arendst/Tasmota #14886: Power ON relays after boot (rules) comes to late to get devices identified FUNC_INIT

stefanbode, 2022-02-17:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
If using for example a DS18B20 and use RELAY1 to power on the device. Additionally use a rule: ON Power1#Boot DO Power on ENDON ON System#Save DO Power off ENDON

The boot order is now quite bad because FUNC_INIT that init the DS18B20 is called BEFORE the power is on. This causes that the init report zero DS18B20 devices. Additionally AFTER setting the relay ON the procedure is NOT called again to scan for devices now present. As a result the use-case does not work.

A bad workaround is to enhance the DS18B20 driver with FUNC_SET_POWER and call the init again on Power change. This fixes the issue, but only for this driver. All other sensors still have the same problem.

My proposed idea to discuss: Doing a FUNC_INIT AFTER the rule #Boot is called.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
08:35:01.000 MQT: hm/status/ESP_32746A/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":66,"Free":445,"Rules":"ON Power1#Boot DO Power on ENDON ON System#Save DO Power off ENDON"}}
08:35:01.211 MQT: hm/status/ESP_32746A/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
08:35:01.463 MQT: hm/status/ESP_32746A/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
08:35:27.041 MQT: hm/status/ESP_32746A/STATUS = {"Status":{"Module":0,"DeviceName":"Fensterbank","FriendlyName":["Fensterbank"],"Topic":"ESP_32746A","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
08:35:27.082 MQT: hm/status/ESP_32746A/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Deep-Sleep Wake","Uptime":"0T00:00:35","StartupUTC":"2022-02-17T07:34:52","Sleep":50,"CfgHolder":4617,"BootCount":25,"BCResetTime":"2022-02-03T14:53:06","SaveCount":96,"SaveAddress":"FA000"}}
08:35:27.115 MQT: hm/status/ESP_32746A/STATUS2 = {"StatusFWR":{"Version":"2022.1.4(tasmota)","BuildDateTime":"2022-02-17T08:17:06","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"385/699"}}
08:35:27.138 MQT: hm/status/ESP_32746A/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["BodeNet_Keller",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00002080","00006000","00004000"]}}
08:35:27.169 MQT: hm/status/ESP_32746A/STATUS4 = {"StatusMEM":{"ProgramSize":634,"Free":368,"Heap":26,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","050013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6,45"}}
08:35:27.210 MQT: hm/status/ESP_32746A/STATUS5 = {"StatusNET":{"Hostname":"ESP-32746A-5226","IPAddress":"192.168.178.149","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.1","DNSServer2":"0.0.0.0","Mac":"A0:20:A6:32:74:6A","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
08:35:27.237 MQT: hm/status/ESP_32746A/STATUS6 = {"StatusMQT":{"MqttHost":"elocm.synology.me","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_32746A","MqttUser":"ESP_32746A","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
08:35:27.263 MQT: hm/status/ESP_32746A/STATUS7 = {"StatusTIM":{"UTC":"2022-02-17T07:35:27","Local":"2022-02-17T08:35:27","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"07:54","Sunset":"18:13"}}
08:35:27.324 MQT: hm/status/ESP_32746A/STATUS10 = {"StatusSNS":{"Time":"2022-02-17T08:35:27","ANALOG":{"A0":107},"DS18B20":{"Id":"031702C3BAFF","Temperature":23.7},"TempUnit":"C"}}
08:35:27.343 MQT: hm/status/ESP_32746A/STATUS11 = {"StatusSTS":{"Time":"2022-02-17T08:35:27","Uptime":"0T00:00:35","UptimeSec":35,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":36,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"BodeNet_Keller","BSSId":"7C:FF:4D:3D:A3:5B","Channel":6,"Mode":"11n","RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-02-17:
@stefanbode 
As suggested on the thread the good solution is to define a GPIO as OUTPUT_HI to power the sensors, any sensors so they are ON before any FUNC_INIT is processed
This has already been discussed many times on Discord (and probably in Discussions too)
Probably that should be documented in the DeepSleep page

stefanbode, 2022-02-17:
I checked a change in xdrv_10_rules.ino:
line: 920ff:
```
        for (uint32_t i = 0; i < TasmotaGlobal.devices_present; i++) {
          uint8_t new_state = (Rules.new_power >> i) &1;
          snprintf_P(json_event, sizeof(json_event), PSTR("{\"Power%d\":{\"Boot\":%d}}"), i +1, new_state);
          RulesProcessEvent(json_event);
          XdrvCall(FUNC_INIT);
          XsnsCall(FUNC_INIT);
        }
```

```
00:00:00.065 Project tasmota - Fensterbank Version 2022.1.4(tasmota)-2_7_4_9(2022-02-17T08:49:46)
00:00:00.067 DSB: Sensors found 0
00:00:00.080 RUL: POWER1#BOOT performs "Power on"
00:00:00.080 SRC: Rule
00:00:00.082 CMD: Grp 0, Cmnd 'POWER', Idx 1, Len 2, Data 'on'
00:00:00.085 RSL: RESULT = {"POWER":"ON"}
00:00:00.086 RSL: POWER = ON
00:00:00.121 DSB: Sensors found 2
00:00:00.181 CFG: Saved to flash at F4, Count 110, Bytes 4096

```

This would also made the deal.

stefanbode, 2022-02-17:
> GPIO as OUTPUT_HI

the `poweronstate 4` will also make it. Most times there is only ONE relay so it will be fine. How to define GPIO HIGH? Compiler option?

sfromis, 2022-02-17:
> How to define GPIO HIGH? Compiler option?

Just the usual gpio config dropdown, `Output Hi`

stefanbode, 2022-02-17:
ok. the `Output Hi` can definitely make it for the deepsleep devices. It is better than the `POWERONSTATE 4`. I will add this to documentation anyhow. Is there a way to call FUNC_INIT with a command?. Then you can add this in a rule, if you have trouble. e.g. a `reinit `command like the `i2scan` command. This anyhow is also a bit useless without the ability to `reinit` afterwards.

sfromis, 2022-02-17:
Do you have a case where a reinit command would be useful?

arendst, 2022-02-17:
Do not do a reinit. It may work for the DS18B20 (but it will add to memory usage as the array is filled again) but it probably fails on other sensors.

The init function is just that; called after power on to setup all kind of memory allocations once.

stefanbode, 2022-02-17:
@sfromis : The use-case behind was to power sensors on/off to avoid e.g. corrosion when there is electrical current flowing. Also to save battery with deepsleep devices. 
But I do understand the concerns of Theo and after looking in some drivers totally agree they are not made to run twice to e.g. add additional sensors. I may work for some but for sure not for all without major investigation. 
rule `power#boot` is running early in the process, but to late for the FUNC_INIT. @arendst is there a chance to reorder this? move power#boot more to the start (assume not easy) or delay FUNC_INIT after the `power#boot` (maybe possible).
For deepsleep only I already updated the documentation to use `OUTPUT HI` or `poweronstate 4` to work with devices that are cut off the battery during deepsleep.

helmar74, 2022-02-17:
@sfromis:
Using a resistor to get low applies on GPIO's D1,D2, D5,D6 and D7. 
D3 and D4 are always high on Deepsleep.
And the "Magic GPIO" D8 is always LOW during deepsleep. 
As described here:
[image]

I used D8 (combined with a BC547 transistor) to bring up the sensor at startup and the sensor are powered off during deepsleep....

arendst, 2022-02-17:
Ah I misunderstood. Allowing rule execution before system#init leads to unpredictable issues as some devices/sensors/memory structs aren't yet initialized. That's why it's called ``system#init`` ;-)

stefanbode, 2022-02-17:
@helmar74 : ok, agree that "OUTPUT HI" is the way to go and poweronstate 4 just creates more problems but no benefits. Will Update documentation.

stefanbode, 2022-02-17:
Regarding the order of events someone (not me) documented this in the deepsleep documentation:

> The following triggers can be used to execute commands upon wake-up or right before entering DeepSleep: - Power1#Boot : is the earliest trigger. But valid only if you have a Relay output defined. - Switch1#Boot : is the next trigger, also occurs very early in the boot process. But valid only if you have Switch input defined. - System#Boot : is occurring later in the Tasmota boot process but is always available. - System#Save : is occuring right before a restart or before entering DeepSleep.

helmar74, 2022-02-17:
poweronstate 4 works for me flawless and is the only poweronstate which is working for my "deepsleep device"

stefanbode, 2022-02-17:
@helmar74 : What about changing configuration on GPIO from RELAY1 --> OUTPUT HI

helmar74, 2022-02-17:
> @helmar74 : What about changing configuration on GPIO from RELAY1 --> OUTPUT HI

Yeah, that did the trick. Changed GPIO D8 to "Output Hi" (from realy 1), disabled rule1.
Works also fine for me, all sensor are properbly initalized after deepsleep.
Have logs running, hopefully I always get the values!!

Thank you very much!!!

stefanbode, 2022-02-17:
@helmar74 : You now should also be able to run the complete cycle with teleperiod 10 to minimize on-time

stefanbode, 2022-02-17:
@arendst : I found a "minimal inversive" change to move the power#boot in front of the FUNC_INIT. 
The problem is, that power#boot is inside RulesEvery50ms(); Even if the relays are set in SetPowerOnState() the rule kicks in to late because of the 50ms window and the init is done. I added just at the end of SetPowerOnState the command to call the RulesEvery50ms. In this case the rule to power on the relay is set before FUNC_INIT and everything is fine. Proposed change:
```
void SetPowerOnState(void)
{
#ifdef ESP8266
  if (MOTOR == TasmotaGlobal.module_type) {
    Settings->poweronstate = POWER_ALL_ON;   // Needs always on else in limbo!
  }

.........

    if (bitRead(TasmotaGlobal.power, i) || (POWER_ALL_OFF_PULSETIME_ON == Settings->poweronstate)) {
      SetPulseTimer(i % MAX_PULSETIMERS, Settings->pulse_timer[i % MAX_PULSETIMERS]);
    }
  }
  TasmotaGlobal.blink_powersave = TasmotaGlobal.power;
  RulesEvery50ms();
}
```

stefanbode, 2022-02-17:
Here the optimized execution:
[image]

stefanbode, 2022-02-17:
> at it's r

I recall my `reinit` request. Too much effort. Specially with the change one comment before

barbudor, 2022-02-17:
> Regarding the order of events someone (not me) documented this in the deepsleep documentation:

Guess I know the guy 😂
[image]

closed 2022-02-17 by stefanbode

helmar74, 2022-02-18:
@stefanbode  : As promised my feedback.
Currently using the following setup:
Deepsleeptime 1800
Teleperiod 10
GPIO Port D8 set as "Output Hi" for powering off sensor during deepsleep.
Here you can see the output of Broker:
```
tele/WemosMiniAkku_1/LWT Online
cmnd/WemosMiniAkku_1/POWER (null)
tele/WemosMiniAkku_1/STATE {"Time":"2022-02-18T08:00:18","Uptime":"0T00:00:11","UptimeSec":11,"Heap":30,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":35,"MqttCount":1,"Wifi":{"AP":2,"SSId":"hs.net_aussen_24GHz","BSSId":"34:31:C4:76:5C:90","Channel":11,"Mode":"11n","RSSI":42,"Signal":-79,"LinkCount":1,"Downtime":"0T00:00:05"}}
tele/WemosMiniAkku_1/SENSOR {"Time":"2022-02-18T08:00:18","ANALOG":{"A0":1024},"DS18B20":{"Id":"000798431FE4","Temperature":6.7},"SR04":{"Distance":29.719},"TempUnit":"C"}
stat/WemosMiniAkku_1/RESULT {"DeepSleep":{"Time":"2022-02-18T08:30:00","Epoch":1645173000}}
tele/WemosMiniAkku_1/LWT Offline
tele/WemosMiniAkku_1/LWT Online
cmnd/WemosMiniAkku_1/POWER (null)
tele/WemosMiniAkku_1/STATE {"Time":"2022-02-18T08:30:06","Uptime":"0T00:00:11","UptimeSec":11,"Heap":30,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":34,"MqttCount":1,"Wifi":{"AP":2,"SSId":"hs.net_aussen_24GHz","BSSId":"34:31:C4:76:5C:90","Channel":11,"Mode":"11n","RSSI":40,"Signal":-80,"LinkCount":1,"Downtime":"0T00:00:05"}}
tele/WemosMiniAkku_1/SENSOR {"Time":"2022-02-18T08:30:06","ANALOG":{"A0":1024},"DS18B20":{"Id":"000798431FE4","Temperature":6.7},"SR04":{"Distance":29.719},"TempUnit":"C"}
stat/WemosMiniAkku_1/RESULT {"DeepSleep":{"Time":"2022-02-18T09:00:00","Epoch":1645174800}}
tele/WemosMiniAkku_1/LWT Offline
tele/WemosMiniAkku_1/LWT Online
cmnd/WemosMiniAkku_1/POWER (null)
tele/WemosMiniAkku_1/STATE {"Time":"2022-02-18T09:00:08","Uptime":"0T00:00:09","UptimeSec":9,"Heap":30,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":35,"MqttCount":1,"Wifi":{"AP":2,"SSId":"hs.net_aussen_24GHz","BSSId":"34:31:C4:76:5C:90","Channel":11,"Mode":"11n","RSSI":40,"Signal":-80,"LinkCount":1,"Downtime":"0T00:00:03"}}
tele/WemosMiniAkku_1/SENSOR {"Time":"2022-02-18T09:00:08","ANALOG":{"A0":1024},"DS18B20":{"Id":"000798431FE4","Temperature":6.7},"SR04":{"Distance":29.193},"TempUnit":"C"}
stat/WemosMiniAkku_1/RESULT {"DeepSleep":{"Time":"2022-02-18T09:30:00","Epoch":1645176600}}
tele/WemosMiniAkku_1/LWT Offline
tele/WemosMiniAkku_1/LWT Online
cmnd/WemosMiniAkku_1/POWER (null)
tele/WemosMiniAkku_1/STATE {"Time":"2022-02-18T09:30:17","Uptime":"0T00:00:13","UptimeSec":13,"Heap":30,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":38,"MqttCount":1,"Wifi":{"AP":2,"SSId":"hs.net_aussen_24GHz","BSSId":"34:31:C4:76:5C:90","Channel":11,"Mode":"11n","RSSI":42,"Signal":-79,"LinkCount":1,"Downtime":"0T00:00:05"}}
tele/WemosMiniAkku_1/SENSOR {"Time":"2022-02-18T09:30:17","ANALOG":{"A0":1024},"DS18B20":{"Id":"000798431FE4","Temperature":6.3},"SR04":{"Distance":29.175},"TempUnit":"C"}
stat/WemosMiniAkku_1/RESULT {"DeepSleep":{"Time":"2022-02-18T10:00:00","Epoch":1645178400}}
tele/WemosMiniAkku_1/LWT Offline
#################### HERE THE ERROR OCCURS ###################
tele/WemosMiniAkku_1/LWT Offline
tele/WemosMiniAkku_1/LWT Online
cmnd/WemosMiniAkku_1/POWER (null)
tele/WemosMiniAkku_1/STATE {"Time":"2022-02-18T10:30:02","Uptime":"0T00:00:12","UptimeSec":12,"Heap":30,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":34,"MqttCount":1,"Wifi":{"AP":2,"SSId":"hs.net_aussen_24GHz","BSSId":"34:31:C4:76:5C:90","Channel":11,"Mode":"11n","RSSI":42,"Signal":-79,"LinkCount":1,"Downtime":"0T00:00:06"}}
tele/WemosMiniAkku_1/SENSOR {"Time":"2022-02-18T10:30:02","ANALOG":{"A0":1024},"DS18B20":{"Id":"000798431FE4","Temperature":6.9},"SR04":{"Distance":29.175},"TempUnit":"C"}
stat/WemosMiniAkku_1/RESULT {"DeepSleep":{"Time":"2022-02-18T11:00:00","Epoch":1645182000}}
tele/WemosMiniAkku_1/LWT Offline
```

What you can see, that the device awakes every 30 minutes and send the topic to the broker.
This worked flawless since yesterday 17:00. But today at 10:00 I didn't received any values. (LWT OFFLINE received twice)
The devices woke up and connected to WIFI (I saw it on the access point).
The values aren't received by the broker only the LWT was sent (Offline) (see output above the line HERE THE ERROR OCCURS)

So it looks like the device went to sleep without sending values to the broker.

The next wakeup at 10:30 (and also at 11:00, not shown in the log output above)  worked flawless again.

Just a thought:
Would it be possible to add something like a acknowledge of the MQTT Broker, that will be sent back to device after broker received the values?
If this acknowledge isn't received by the device it sends teleperiod again... if after 10 cycles (best configurable) the acknowledge isn't still there, the device enters deepsleep

sfromis, 2022-02-18:
MQTT acknowledge would be QoS 1 (delivery at least once) or 2 (exactly once), but [pubsubclient](https://github.com/knolleary/pubsubclient) declares a limitation of:
> It can only publish QoS 0 messages.

Of course, it could be doable for the subscribing end of MQTT (not the broker) to publish a message back to Tasmota, invoking a rule (or whatever) to confirm reception of a message.

helmar74, 2022-02-18:
This wouldn't work as no trigger (e.g LWT ONLINE) is received at error condition.
So my smarthome (openhab) can't trigger a rule....

sfromis, 2022-02-18:
That's not what you asked for, but more or less the opposite. Your backend *could* publish a message back to Tasmota confirming "I got something". This would be acknowledge at the application level, not at the broker level.

helmar74, 2022-02-18:
> s ON and tries to connect and send the message as long as the battery lasts or the MQTT broker was connected. You can get this back with the compiler option regarding timeout. See help. Current functionality will send the device for another cycle into deepsleep and hope the next wakeup that all systems are back working.

Sorry for asking again. Which Compiler Option do you mean?
(DEEPSLEEP_NETWORK_TIMEOUT)?

stefanbode, 2022-02-18:
Yes. Set it to 0 and the device stay awake until able to send the message.

helmar74, 2022-02-18:
OK, but the problem is, that the device connect's to broker.
But only values aren't sent...
So changing it 0 wouldn't help?

helmar74, 2022-02-18:
Same error came up at 14:00.

Device woke up and connected to broker:
```
Feb 18 14:00:03 2022: New connection from 192.168.1.224 on port 1883.
Feb 18 14:00:03 2022: New client connected from 192.168.1.224 as WemosMiniAkku-1v3 (c1, k30, u'DVES_USER').
Feb 18 14:00:48 2022: Client WemosMiniAkku-1v3 has exceeded timeout, disconnecting.
Feb 18 14:00:48 2022: Socket error on client WemosMiniAkku-1v3, disconnecting.
```
The Device is only up 48 seconds, so I think the DEEPSLEEP_NETWORK_TIMEOUT Compiler Option doesn't fix anything in my condition (Currently set to 120 seconds)

... but no data received by broker:
```
tele/WemosMiniAkku_1/STATE {"Time":"2022-02-18T13:30:10","Uptime":"0T00:00:14","UptimeSec":14,"Heap":30,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":33,"MqttCount":1,"Wifi":{"AP":2,"SSId":"hs.net_aussen_24GHz","BSSId":"34:31:C4:76:5C:90","Channel":11,"Mode":"11n","RSSI":42,"Signal":-79,"LinkCount":1,"Downtime":"0T00:00:08"}}
tele/WemosMiniAkku_1/SENSOR {"Time":"2022-02-18T13:30:10","ANALOG":{"A0":1024},"DS18B20":{"Id":"000798431FE4","Temperature":7.1},"SR04":{"Distance":25.912},"TempUnit":"C"}
stat/WemosMiniAkku_1/RESULT {"DeepSleep":{"Time":"2022-02-18T14:00:00","Epoch":1645192800}}
tele/WemosMiniAkku_1/LWT Offline
################## Wakeup at 14:00 ################
tele/WemosMiniAkku_1/LWT Offline
```

I'll try to set teleperiod to 12 in order to receive 2 teleperiod messages...
Let's see

barbudor, 2022-02-18:
Get a serial terminal to log what happens on Tasmota, or use a syslog to get Tasmota logs over Syslog
Without, we don't really know what's Tasmota side of the problem

helmar74, 2022-02-18:
I'll let it run for a day and have an eye on it (changed teleperiod to 12 seconds)
If the error occurs again, I'll setup a syslog server

helmar74, 2022-02-21:
Just a final comment:
I changed to teleperiod 12 on 18th februrary 16:30. Since then I always receives the values on my broker. (evey 1800 secs)
So in summary the change made by @stefanbode in Tasmota 11.0.0.1 that the teleperiod/deepsleep mechanism works like in past versions is working for me. Unfourtnately teleperiod 10 isn't working constantly as the device doesn't always send teleperiod after wakeup reliable. So with my current setting it'll drain battery a bit faster, but I assume that will be no big thing.

Thank you all for supporting especially @stefanbode, @arendst  and @stefanbode 

kind regards 

helmar