arendst/Tasmota #14891: General(18) module in release 11.0 doesn't publish switches messages to  auto discovery

PhilNS, 2022-02-17:
### PROBLEM DESCRIPTION
From Installed Tasmota Release 11.0 on ESP 8266 EX device, it looks like General (18) Module doesn't publish autodiscovery messages about switch so to gets published to HomeAssistant. Basically I get all entities in HA related to device sensors but none about the switch themselves.

I'd be grateful if someone can check where the issue might be and I'm surely available to provide any additional information. Thanks.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)  (Found a similar bug in old issue  #6685
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): ESP8266EX
- [X] Tasmota binary firmware version number used: 11.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: Chrome onto https://tasmota.github.io/install/
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
15:00:59.116 CMD: Backlog Template; Module; GPIO 255
15:00:59.192 MQT: stat/tasmota_3387E1/RESULT = {"NAME":"Generic","GPIO":[160,1,161,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
15:00:59.401 MQT: stat/tasmota_3387E1/RESULT = {"Module":{"18":"Generic"}}
15:00:59.657 MQT: stat/tasmota_3387E1/RESULT = {"GPIO0":{"160":"Switch1"},"GPIO1":{"0":"None"},"GPIO2":{"161":"Switch2"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
15:06:26.589 MQT: stat/tasmota_3387E1/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
15:06:26.801 MQT: stat/tasmota_3387E1/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
15:06:27.051 MQT: stat/tasmota_3387E1/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
15:07:41.639 MQT: stat/tasmota_3387E1/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota-Porta","FriendlyName":["Tasmota Porta"],"Topic":"tasmota_3387E1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
15:07:41.646 MQT: stat/tasmota_3387E1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:09:12","StartupUTC":"2022-02-17T13:58:29","Sleep":50,"CfgHolder":4617,"BootCount":27,"BCResetTime":"2022-02-17T11:29:32","SaveCount":129,"SaveAddress":"F6000"}}
15:07:41.651 MQT: stat/tasmota_3387E1/STATUS2 = {"StatusFWR":{"Version":"11.0.0(tasmota)","BuildDateTime":"2022-02-12T14:13:50","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"357/699"}}
15:07:41.655 MQT: stat/tasmota_3387E1/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["*MYWIFI*",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00028009","2805C80001000600003C5A0A190000000000","00000280","00006000","00004000"]}}
15:07:41.669 MQT: stat/tasmota_3387E1/STATUS4 = {"StatusMEM":{"ProgramSize":620,"Free":380,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14605E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
15:07:41.677 MQT: stat/tasmota_3387E1/STATUS5 = {"StatusNET":{"Hostname":"tasmota-3387E1-2017","IPAddress":"*DEVICE_IP*","Gateway":"*LAN_GATEWAY*","Subnetmask":"255.255.255.0","DNSServer1":"*LAN_DNS*","DNSServer2":"0.0.0.0","Mac":"4C:75:25:33:87:E1","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
15:07:41.683 MQT: stat/tasmota_3387E1/STATUS6 = {"StatusMQT":{"MqttHost":"*MQTTBROKER*","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3387E1","MqttUser":"MQTTUSER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
15:07:41.689 MQT: stat/tasmota_3387E1/STATUS7 = {"StatusTIM":{"UTC":"2022-02-17T14:07:41","Local":"2022-02-17T15:07:41","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+00:00","Sunrise":"07:54","Sunset":"18:13"}}
15:07:41.696 MQT: stat/tasmota_3387E1/STATUS10 = {"StatusSNS":{"Time":"2022-02-17T15:07:41","Switch1":"ON","Switch2":"ON"}}
15:07:41.702 MQT: stat/tasmota_3387E1/STATUS11 = {"StatusSTS":{"Time":"2022-02-17T15:07:41","Uptime":"0T00:09:12","UptimeSec":552,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"MYWIFI","BSSId":"84:3E:92:15:F4:B4","Channel":6,"Mode":"11n","RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

16:08:53.215 CMD: setoption19 0
16:08:53.217 SRC: WebConsole from MQYYBROKER
16:08:53.220 CMD: Grp 0, Cmnd 'SETOPTION', Idx 19, Len 1, Data '0'
16:08:53.226 MQT: stat/tasmota_3387E1/RESULT = {"SetOption19":"OFF"}
16:08:53.639 MQT: homeassistant/device_automation/3387E1_SW_5_HOLD/config =  (retained)
16:08:53.642 MQT: homeassistant/binary_sensor/3387E1_SW_5/config =  (retained)
16:08:53.647 MQT: homeassistant/device_automation/3387E1_SW_6_TOGGLE/config =  (retained)
16:08:53.651 MQT: homeassistant/device_automation/3387E1_SW_6_HOLD/config =  (retained)
16:08:53.655 MQT: homeassistant/binary_sensor/3387E1_SW_6/config =  (retained)
16:08:53.659 MQT: homeassistant/device_automation/3387E1_SW_7_TOGGLE/config =  (retained)
16:08:53.663 MQT: homeassistant/device_automation/3387E1_SW_7_HOLD/config =  (retained)
16:08:53.666 MQT: homeassistant/binary_sensor/3387E1_SW_7/config =  (retained)
16:08:53.670 MQT: homeassistant/device_automation/3387E1_SW_8_TOGGLE/config =  (retained)
16:08:53.674 MQT: homeassistant/device_automation/3387E1_SW_8_HOLD/config =  (retained)
16:08:53.678 MQT: homeassistant/binary_sensor/3387E1_SW_8/config =  (retained)
16:08:53.681 MQT: homeassistant/light/3387E1_LI_1/config =  (retained)
[... 31 lines cut ...]
A) Configure Tasmota 11.0 General (18) module on ESP8266
B) Configure Switch1 on GPO0 and Switch2 on GPO2
C) setoption19 0

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
1) Have Tasmot send autodiscovery message so that Home Assistant identify 11 entities (9 Sensors and 2 Switches) instead currently obtaining only the 9 generic sensors

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

Tasmota messages just afer rebooting the device:
![tasmota-reboot-status](https://user-images.githubusercontent.com/99879088/154511689-735d658f-422d-4d72-80dd-2f39e11394e1.JPG)

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

Triggering the switch attached to the tasmota device , then it looks like triggering some messaging
15:51:23.777 MQT: stat/tasmota_3387E1/RESULT = {"POWER":"ON"}
15:51:23.779 MQT: stat/tasmota_3387E1/POWER = ON
15:51:29.473 MQT: tele/tasmota_3387E1/STATE = {"Time":"2022-02-17T15:51:29","Uptime":"0T00:06:47","UptimeSec":407,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":25,"MqttCount":1,"Wifi":{"AP":1,"SSId":"MYWIFI","BSSId":"84:3E:92:15:F4:B4","Channel":6,"Mode":"11n","RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:05"}}
15:51:29.479 MQT: stat/tasmota_3387E1/RESULT = {"POWER":"OFF"}
15:51:29.483 MQT: stat/tasmota_3387E1/POWER = OFF

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2022-02-17:
Here's what I see in the `tasmota/discovery/.../config` payload when I have `SetOption114` active, as I'd expect to be a very reasonable thing if you want to forward switch actions to HA. Try that.
`"swc":[0,-1,-1,-1,-1,-1,-1,-1],"swn":["Switch1",null,null,null,null,null,null,null],`
This discovery info for a Switch is not included if having `SetOption114 OFF` instead.

What you posted in "additional context" is not at all switch actions being published, but relay actions. These may "happen" to be triggered by switch actions, but that's only one of several ways.

PhilNS, 2022-02-17:
Thanks for the hints. I tried with SetOption114 ON , but it didn't help much in the terms that:
- the switch entities didn't appear in HA
-  the physical triggering of the switch on the device didn't produce anymore a Power ON and OFF on the Tasmota Console

Looking deeper into the Tasmota documentation for your suggestion on SetOption114 I found though a section about SwitchMode setting , and for some reason changing it from Default Toggle: Mode 0, to Mode 1 (ON & OFF) and using Relay instead than Switch made up for the result.

Now using:
- setoption19 off
- setoption114 off
- switchmode 1

A Relay in GPIO0 and a Switch in GPIO2
```lua

22:39:39.576 MQT: stat/tasmota_3387E1/RESULT = {"NAME":"GenericTest","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
22:39:39.806 MQT: stat/tasmota_3387E1/RESULT = {"Module":{"18":"Generic"}}
22:39:40.060 MQT: stat/tasmota_3387E1/RESULT = {"GPIO0":{"224":"Relay1"},"GPIO1":{"0":"None"},"GPIO2":{"161":"Switch2"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```

I gathered in HA:
- the 9 usual sensors (4 disabled) as before
- **1 new Tasmota.Switch** which can be OFF and ON accordingly if contact on GPIO0 is open (OFF) or Grounded (ON).
through these messages:
```lua

22:24:39.783 MQT: tele/tasmota_3387E1/STATE = {"Time":"2022-02-17T22:24:39","Uptime":"0T00:09:10","UptimeSec":550,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"Phil-Wifi2","BSSId":"84:3E:92:15:F4:B4","Channel":6,"Mode":"11n","RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:05"}}
22:24:39.787 MQT: stat/tasmota_3387E1/RESULT = {"POWER":"ON"}
22:24:39.790 MQT: stat/tasmota_3387E1/POWER = ON
22:24:46.893 MQT: tele/tasmota_3387E1/STATE = {"Time":"2022-02-17T22:24:46","Uptime":"0T00:09:17","UptimeSec":557,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Phil-Wifi2","BSSId":"84:3E:92:15:F4:B4","Channel":6,"Mode":"11n","RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:05"}}
22:24:46.899 MQT: stat/tasmota_3387E1/RESULT = {"POWER":"OFF"}
22:24:46.902 MQT: stat/tasmota_3387E1/POWER = OFF
22:25:40.408 MQT: tele/tasmota_3387E1/STATE = {"Time":"2022-02-17T22:25:40","Uptime":"0T00:10:11","UptimeSec":611,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Phil-Wifi2","BSSId":"84:3E:92:15:F4:B4","Channel":6,"Mode":"11n","RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:05"}}
22:25:40.416 MQT: tele/tasmota_3387E1/SENSOR = {"Time":"2022-02-17T22:25:40","Switch2":"ON"}

```

I really believe this is NOT the correct configuration (a Relay which works on GPIO0 as a switch just if I configure also  a Switch on GPIO2 which has no publishing in HA  ?) 

I guess it's a kind of magical electrical coincidence that is though delivering the result that I wished for. 
Would it be stable ? Would it last ? no idea, but I wouldn't trust my life on this device configured as such, but I surely could trust it for the moment to signal some door opening and closure.

Any idea how I could get to some more decent result than this Frankestain configuration ?

sfromis, 2022-02-17:
Well, if you want to stay with not reporting switches, but reporting relays, you could create two virtual relays, each controlled by one switch input.  

Anyway, `SetOption114`, `SwitchMode 15` or using rules are the ways to go if you specifically want switch actions reported (including full flexibility) instead of the indirect path via relays. As I'm not using HA, I do not know what you have to do there, but it may be something like configuring binary sensors.

PhilNS, 2022-02-18:
Indeed maybe I should have stated what was the intended goal from start, which really I think should be the very basic use of the Tasmota firmware on a 8266.
So, the main goal is about having the ESP8266 GPIO2 & GRND pin linked to a Reed switch attached to a door, so that:
- Door is Open(Reed Opened)= ON
- Door is Closed (Reed is Closed) = OFF

From a tutorial I went through everything was plain simple: 
1) Get the ESP8266EX and the USB/TTL adapter: DONE
2) Flash the ESP8266EX with latest (11.0) Tasmota firmware: DONE
3) Attach the ESP8266EX to local WiFI: DONE
4) Configure a Switch on GPIO2 under Generic(18) Template: DONE
5) Configure Tasmota MQTT to Local Broker : DONE

Result:
Find the Tasmota Switch on HA (along with other Tasmota sensor entities) : **FAIL** (No switch under Generic 18 Template)

I get a switch if I use Sonoff Basic Template (even without configuring anything on GPIO0 or 2) but that works only in toggle mode gounding GPIO2.

I'm keeping the Relay on GPIO0 for the moment on first device and door while I'm flashing a second ESP8266 to make some more tests with your indications: Setoption114 ON and Switchmode 15 and different GPIO0 and GPIO2 settings.

I'd be happy to have confirmation though that using the Generic(18) template is the right way to start, just to have a fixed base point before changing the other many options.

Thanks meanwhile for all the hints.

pedrooliveiraJumia, 2022-03-09:
Any news based on this ? 
My switches send the right information to Home Assistant but when I change the status in home assistant it does not reflect in the switch :/

barbudor, 2022-03-09:
> My switches send the right information to Home Assistant but when I change the status in home assistant it does not reflect in the switch :/

I'm not sure to understand what you mean and how it is related to the opened issue.
A Tasmota Switch is an hardware input to the ESP that is suposed to be connected to a flip-switch or a push button. Whenever the status of the input changes because someone flipped the switch or press the button, Tasmota will porcess and eventually publish the state to MQTT.
> when I change the status in home assistant it does not reflect in the switch 

Doesn't means anything regarding the above behavior.

Jason2866, 2022-03-25:
Tasmota does not publish Auto discovery messages when only switches are assigned.
This behavious is as designed.
Easy way (as from @sfromis already suggested) define a dummy relais or use a rule to publish what you need.

closed 2022-03-25 by Jason2866