arendst/Tasmota #14898: full ir support + power sensor not configurable

HMvdZijden, 2022-02-18:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I try to build tasmota with full IR support by activating line #define USE_IR_REMOTE_FULL                     // Support all IR protocols from IRremoteESP8266
After that the compile ends in an error.
Linking .pio/build/tasmota/firmware.elf
/home/gitpod/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/tasmota/src/tasmota.ino.cpp.o:(.text._Z14IrRemoteSendGCPPcjj+0x0): undefined reference to `IRsend::sendGC(unsigned short*, unsigned short)'
/home/gitpod/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/tasmota/src/tasmota.ino.cpp.o: in function `IrRemoteSendGC(char**, unsigned int, unsigned int)':
tasmota.ino.cpp:(.text._Z14IrRemoteSendGCPPcjj+0x7d): undefined reference to `IRsend::sendGC(unsigned short*, unsigned short)'
/home/gitpod/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/tasmota/src/tasmota.ino.cpp.o:(.text._Z23IrRemoteSendRawStandardPPctjj+0x0): undefined reference to `IRsend::sendRaw(unsigned short const*, unsigned short, unsigned short)'
/home/gitpod/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/tasmota/src/tasmota.ino.cpp.o: in function `IrRemoteSendRawStandard(char**, unsigned short, unsigned int, unsigned int)':
tasmota.ino.cpp:(.text._Z23IrRemoteSendRawStandardPPctjj+0x7b): undefined reference to `IRsend::sendRaw(unsigned short const*, unsigned short, unsigned short)'
/home/gitpod/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/tasmota/src/tasmota.ino.cpp.o: in function `IrRemoteSendRawFormatted(char**, unsigned int, unsigned int)':
tasmota.ino.cpp:(.text._Z24IrRemoteSendRawFormattedPPcjj+0x10e): undefined reference to `IRsend::sendRaw(unsigned short const*, unsigned short, unsigned short)'
/home/gitpod/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: tasmota.ino.cpp:(.text._Z24IrRemoteSendRawFormattedPPcjj+0x1eb): undefined reference to `IRsend::sendRaw(unsigned short const*, unsigned short, unsigned short)'
collect2: error: ld returned 1 exit status
*** [.pio/build/tasmota/firmware.elf] Error 1

Building tasmota-ir builds one with full support for ir but does not support power sensors and I need that combination.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____NA
- [x] Tasmota binary firmware version number used: _____latest
  - [ ] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Uncomment  line 836 in my_user_config_h

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
A build output

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-02-18:
Delete the `.pio`, the `.cache` and the `$HOME/.platformio` folders and let PlatformIO redownload fresh and clean environment

Jason2866, 2022-02-18:
It does NOT work this way. You have to define your own env and place it in `platformio_tasmota_cenv.ini`
```
[env:tasmota-HMvdZijden]
build_flags = ${env.build_flags} -DUSE_IR_REMOTE_FULL
```
with `pio run -e tasmota-HMvdZijden` you get your Tasmota version with Full IR support

closed 2022-02-18 by Jason2866

barbudor, 2022-02-18:
@Jason2866 So the instructions are incomplete as they do not refer to a custom env

https://github.com/arendst/Tasmota/blob/202f83b4ff14d8fc7eec79662b8fa82d8fa1a9a6/tasmota/my_user_config.h#L835-L842

We probably should write something here in my_user_config
Do you want me to do it ?

Jason2866, 2022-02-18:
@barbudor  yes, please. I am on Mobile now.

HMvdZijden, 2022-02-18:
I am really sorry, i am such a noob, I really don’t know what to do and why.
Can someone inform me what to do exactly.