arendst/Tasmota #14905: Compile error when Zigbee + Scripting is enabled

Meek-HA, 2022-02-19:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Compiling error when the following is enabled :
#define USE_ZIGBEE
#define USE_SCRIPT   

Error Message is as follows:
xdrv_20_hue.ino:724:35: error: call of overloaded 'EncodeLightId(uint32_t&)' is ambiguous

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2022-02-19:
Hi,

Using latest codebase, defining ZigBee in platformio_override.ini, and defining script and undefining rules in user_config_override.h, it compiles ok.

[image]

Seems that you have a mix of old files with new files. Please, delete everything and start from scratch.

If you need further assistance, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. 

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

TroeperDeTroep, 2022-02-20:
I can confirm this error, I was able to reproduce this, and in discord #lounge, the compile error was confirmed.
To reproduce this, in "user_config_override.h" enter the following lines : 
#undef USE_RULES
#define USE_SCRIPT 
#define USE_ZIGBEE

Command "platformio run -e tasmota" and also "platformio run -e tasmota32" with the newest git release gives the error message : 
xdrv_20_hue.ino:1004:37: error: call of overloaded 'DecodeLightId(uint32_t&)' is ambiguous

Thank you for re-opening this Issue.

ascillato, 2022-02-20:
To use ZigBee, you need to use the ZigBee variant. Just adding USE_ZIGBEE is not enough.

Please, enable scripting in user_config_override.h file and select to compile ZigBee variant in platformio_override.h

s-hadinger, 2022-02-20:
I confirm I can reproduce, let me fix this.

PR #14924 (2022-02-20): Fix Zigbee compilation with Hue emulation
## Description:

**Related issue (if applicable):** fixes #14905

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with Tasmota core ESP32 V.2.0.2.1
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-a [... cut]
closed 2022-02-20 by s-hadinger