arendst/Tasmota #14913: Skip to content arendst / Tasmota Public      Code     Issues 8     Pull requests 7     Discussions     Actions     Projects 3     Security     Insights  jason2866 tasmota specials esp32 ethernet v11 ethernet jason2866 tasmota specials esp32 ethernet v11 ethernet does not work.

smartboart, 2022-02-19:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

[Skip to content](https://github.com/arendst/Tasmota/issues/14912#start-of-content)
arendst /
Tasmota
Public[

](https://github.com/)    Code
    Issues 8
    Pull requests 7
    Discussions
    Actions
    Projects 3
    Security
    Insights

jason2866 tasmota specials esp32 ethernet v11 ethernet does not work #14912
Closed
smartboart opened this issue now · 1 comment
Comments
@smartboart smartboart commented now

jason2866 tasmota specials esp32 ethernet v11 ethernet does not work.
The actual pre compiled Binary file v11.0.0.1 does not Support the ethernet Connectivity. His v10 does it very well...

Did not found a possibilty to open an issue on his github Page...so i do it here..

Binary file see below link..

https://github.com/Jason2866/[Tasmota](https://github.com/arendst/Tasmota)-specials/blob/firmware/firmware/tasmota32/other/tasmota32-ethernet.bin
@close-issue-app close-issue-app bot commented now

This issue has been automatically closed because the PROBLEM REPORT TEMPLATE is missing or incomplete.
Filling the template is required so standard questions don't need to be asked again each time. Our ability to provide assistance is greatly hampered if few minutes are not taken to complete the issue template with the requested information. The details requested potentially affect which options to pursue. The small amount of time you will spend completing the template will also help the volunteers, providing assistance to you, to reduce the time required to help you.
Please, could you be so kind on completing the PROBLEM REPORT TEMPLATE in order to have more information so as to properly help you?
Thank you for taking the time to report, hopefully it can be resolved soon.
Docs for more information.
Discussions for Questions, Feature Requests and Projects.
Chat for more users experience.
Please check the Code of Conduct and the Contributing Guideline and Policy
@close-issue-app close-issue-app bot closed this now
@close-issue-app close-issue-app bot added the template missing/incomplete label now
Remember, contributions to this repository should follow its contributing guidelines and code of conduct.
Assignees
No one assigned
Labels
template missing/incomplete
Projects
None yet
Milestone
No milestone
Linked pull requests

Successfully merging a pull request may close this issue.
None yet
Notifications
Customize

You’re receiving notifications because you authored the thread.
1 participant
@[smartboart](https://github.com/smartboart)[](https://github.com/smartboart)

    © 2022 GitHub, Inc.

    Terms
    Privacy
    Security
    Status
    Docs
    Contact GitHub
    Pricing
    API
    Training
    Blog
    About

close-issue-app bot closed this [now](https://github.com/arendst/Tasmota/issues/14912#issue-1144849513)[](https://github.com/apps/close-issue-app) [close-issue-app](https://github.com/apps/close-issue-app) bot added the template missing/incomplete label [now](https://github.com/arendst/Tasmota/issues/14912#issuecomment-1046105425)[](https://github.com/apps/close-issue-app)[](https://github.com/apps/close-issue-app)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2022-02-20:
Does not work is unspecific. What is not working as expected?
Which build is used (commit)? What settings have you made?
Which board is used (revision)?

Jason2866, 2022-02-20:
If it is a Olimex board, there is a hardware design issue which can hit anytime.
https://github.com/espressif/arduino-esp32/issues/6142

smartboart, 2022-02-20:
It is an olimex  board..aee picture ...and just the ethernet connection does not work..wifi is ok...as soon as i install your precopiled pin v 10 again...it is working...

[image]

smartboart, 2022-02-20:
[image]

arendst, 2022-02-20:
I have the same device. It works great using v11.0.0.1 when ethernet is enabled.

Your issue might be the change in hostname we did from ``xxx_eth`` to ``xxx-eth`` to bring the name in line with specs.

```
00:00:00.002 HDW: ESP32-D0WD 
00:00:00.003 HDW: FoundPSRAM=0 CanUsePSRAM=0
00:00:00.049 UFS: FlashFS mounted with 52 kB free
00:00:00.160 CFG: Loaded from File, Count 105
00:00:00.175 QPC: Count 1
00:00:00.184 ROT: Mode 1
00:00:00.359 CFG: No '*.autoconf' file found
00:00:00.365 BRY: Berry initialized, RAM used=3938 bytes
00:00:00.389 BRY: No 'preinit.be'
00:00:00.393 SRC: Restart
00:00:00.398 Project tasmota - Olimex ESP32-PoE-Iso Version 11.0.0.1(theo)-2_0_2_1(2022-02-20T11:53:50)
00:00:01.760 ETH: Attempting connection...
00:00:01.762 ETH: Connected at 100Mbps Full Duplex
00:00:01.776 ETH: Mac 4C:11:AE:D6:D3:1F, IPAddress 192.168.2.117, Hostname esp32f-eth
00:00:01.937 BRY: No 'autoexec.be'
00:00:02.241 RTC: UTC 2022-02-20T11:12:36, DST 2022-03-27T02:00:00, STD 2022-10-30T03:00:00
12:12:36.220 HTP: Web server active on esp32f-eth with IP address 192.168.2.117
12:12:37.450 MQT: Attempting connection...
12:12:37.464 MQT: Connected
12:12:37.469 MQT: tele/esp32f/LWT = Online (retained)
12:12:37.483 MQT: cmnd/esp32f/POWER = 
12:12:37.484 MQT: Subscribe to cmnd/esp32f/#
12:12:37.488 MQT: Subscribe to cmnd/tasmotas/#
12:12:37.491 MQT: Subscribe to cmnd/DVES_D6D31C_fb/#
12:12:37.499 MQT: tele/esp32f/INFO1 = {"Info1":{"Module":"Olimex ESP32-PoE","Version":"11.0.0.1(theo)","FallbackTopic":"cmnd/DVES_D6D31C_fb/","GroupTopic":"cmnd/tasmotas/"}}
12:12:37.518 MQT: tele/esp32f/INFO2 = {"Info2":{"WebServerMode":"Admin","Ethernet":{"Hostname":"esp32f-eth","IPAddress":"192.168.2.117"}}}
12:12:37.535 MQT: tele/esp32f/INFO3 = {"Info3":{"RestartReason":"Software reset CPU","BootCount":52}}
12:12:37.541 MQT: stat/esp32f/RESULT = {"POWER":"OFF"}
12:12:37.546 MQT: stat/esp32f/POWER = OFF
12:12:37.558 TFS: File 'autoexec.bat' not found
12:12:41.999 MQT: tele/esp32f/STATE = {"Time":"2022-02-20T12:12:41","Uptime":"0T00:00:06","UptimeSec":6,"Heap":170,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":38},"POWER":"OFF"}
12:12:42.024 MQT: tele/esp32f/SENSOR = {"Time":"2022-02-20T12:12:42","ESP32":{"Temperature":53.3},"TempUnit":"C"}
12:12:43.010 QPC: Reset
12:12:45.004 APP: Boot Count 52
12:12:46.349 CFG: Saved, Count 106, Bytes 4096
12:12:48.292 HTP: Console
12:13:03.970 CMD: ethernet
12:13:03.972 SRC: WebConsole from 192.168.2.1
12:13:03.974 CMD: Grp 0, Cmd 'ETHERNET', Idx 1, Len 0, Pld -99, Data ''
12:13:03.985 MQT: stat/esp32f/RESULT = {"Ethernet":"ON"}
12:13:07.486 CMD: wifi
12:13:07.488 SRC: WebConsole from 192.168.2.1
12:13:07.490 CMD: Grp 0, Cmd 'WIFI', Idx 1, Len 0, Pld -99, Data ''
12:13:07.498 MQT: stat/esp32f/RESULT = {"Wifi":"OFF","Mode":"11 "}
12:13:19.602 CMD: template
12:13:19.604 SRC: WebConsole from 192.168.2.1
12:13:19.606 CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 0, Pld -99, Data ''
12:13:19.618 MQT: stat/esp32f/RESULT = {"NAME":"Olimex ESP32-PoE","GPIO":[1,1,1,1,1,1,0,0,5536,1,1,1,1,0,5600,0,0,0,0,5568,0,0,0,0,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
12:13:28.186 CMD: ethaddress
12:13:28.188 SRC: WebConsole from 192.168.2.1
12:13:28.191 CMD: Grp 0, Cmd 'ETHADDRESS', Idx 1, Len 0, Pld -99, Data ''
12:13:28.201 MQT: stat/esp32f/RESULT = {"EthAddress":0}
12:13:33.457 CMD: ethtype
12:13:33.459 SRC: WebConsole from 192.168.2.1
12:13:33.462 CMD: Grp 0, Cmd 'ETHTYPE', Idx 1, Len 0, Pld -99, Data ''
12:13:33.472 MQT: stat/esp32f/RESULT = {"EthType":0}
12:13:39.371 CMD: ethclockmode
12:13:39.373 SRC: WebConsole from 192.168.2.1
12:13:39.376 CMD: Grp 0, Cmd 'ETHCLOCKMODE', Idx 1, Len 0, Pld -99, Data ''
12:13:39.386 MQT: stat/esp32f/RESULT = {"EthClockMode":3}
```

Jason2866, 2022-02-20:
Found the reason. For testing i use latest development Tasmota Arduino Core for the special builds. Something has changed there which broke ethernet.
I just reverted the core to the standard actual used one. With the next commit from Theo here a new build will triggered there and Ethernet is working (again).
Thx for submitting this issue!

smartboart, 2022-02-20:
Welc0me..thanks for providing prebuild bin files for this....makes it easy to Update All my Devices with one click in iobroker... will this Download link Stars the same for future Updates? 
Thanks again.

arendst, 2022-02-20:
I just enabled ethernet in ``tasmota32.bin`` available from the development branch in an hour. No need for "special" binaries regarding ethernet anymore.

smartboart, 2022-02-20:
Ok thanks a lot...i will close this issue...Best regards

closed 2022-02-20 by smartboart