arendst/Tasmota #14929: ESP32: Eth IP address is not shown correctly in GUI and in HASS discovery

altserg, 2022-02-21:
### PROBLEM DESCRIPTION
ESP32 does not show correct Ethernet IP address in GUI and in HASS autodiscovery MQTT messages. The problem exists on Tasmota at least since version 9, and was seen on multiple different HW boards

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Kincony KC868-A8, LiliGo POE board. Issue is device agnostic
- [x] Tasmota binary firmware version number used: 11.0.0.1
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: Platformio built-in
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
09:34:59.111 MQT: stat/GFHVAC/RESULT = {"NAME":"KC868-A8","GPIO":[32,0,1,0,640,608,0,0,0,1,1,1,0,0,5600,0,0,0,0,5568,0,0,0,0,0,0,0,0,1,1,0,0,1,0,0,1],"FLAG":0,"BASE":1}
09:34:59.348 MQT: stat/GFHVAC/RESULT = {"Module":{"0":"KC868-A8"}}
09:34:59.612 MQT: stat/GFHVAC/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"640":"I2C SDA1"},"GPIO5":{"608":"I2C SCL1"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"1312":"DS18x20"},"GPIO15":{"5312":"OpenTherm RX"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"5600":"ETH MDIO"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"5568":"ETH MDC"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"4801":"ADC Button2"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"130":"Button_in3"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
09:33:23.261 CMD: status 0
09:33:23.274 MQT: stat/GFHVAC/STATUS = {"Status":{"Module":0,"DeviceName":"GF HVAC","FriendlyName":["GF_CV_Switch","GF_Reserve_1","GF_Reserve_2","GF_Reserve_3","GF_Reserve_4","GF_HVAC_MAX","GF_HVAC_MEDIUM","GF_HVAC_STANDBY"],"Topic":"GFHVAC","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
09:33:23.327 MQT: stat/GFHVAC/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T11:13:35","StartupUTC":"2022-02-20T21:19:48","Sleep":50,"CfgHolder":3002,"BootCount":16,"BCResetTime":"2022-02-16T22:20:44","SaveCount":35}}
09:33:23.374 MQT: stat/GFHVAC/STATUS2 = {"StatusFWR":{"Version":"11.0.0.1(SRG-K-MIN-BLE-ETH)","BuildDateTime":"2022-02-18T19:36:32","Core":"2_0_2_1","SDK":"v4.4","CpuFrequency":80,"Hardware":"ESP32-D0WDQ6","CR":"480/699"}}
09:33:23.394 MQT: stat/GFHVAC/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Bailar_N",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C008","2805C80001000600003C5A00190000000000","808020A8","00004000","00000000"]}}
09:33:23.431 MQT: stat/GFHVAC/STATUS4 = {"StatusMEM":{"ProgramSize":1300,"Free":1856,"Heap":111,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","00102594","00000001","00000042","00000000","40000200","40000280","00200000","00000008"],"Drivers":"1,2,7,12,28,38,50,52,79,82","Sensors":"2,5,52,62,69,127"}}
09:33:23.477 MQT: stat/GFHVAC/STATUS5 = {"StatusNET":{"Hostname":"ESP32-GF-HVAC","IPAddress":"0.0.0.0","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.1","DNSServer2":"0.0.0.0","Mac":"C8:C9:A3:DC:2E:10","Ethernet":{"Hostname":"ESP32-GF-HVAC-eth","IPAddress":"192.168.2.96","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.1","DNSServer2":"8.8.8.8","Mac":"C8:C9:A3:DC:2E:13"},"Webserver":2,"HTTP_API":0,"WifiConfig":4,"WifiPower":17.0}}
09:33:23.530 MQT: stat/GFHVAC/STATUS6 = {"StatusMQT":{"MqttHost":"smarthome.local","MqttPort":1883,"MqttClientMask":"GF_HVAC","MqttClient":"GF_HVAC","MqttUser":"GFHVAC","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
09:33:23.553 MQT: stat/GFHVAC/STATUS7 = {"StatusTIM":{"UTC":"2022-02-21T08:33:23","Local":"2022-02-21T09:33:23","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":99}}
09:33:23.583 MQT: stat/GFHVAC/STATUS10 = {"StatusSNS":{"Time":"2022-02-21T09:33:23","ESP32":{"Temperature":53.3},"PCF8574-1":{"D0":1,"D1":1,"D2":1,"D3":1,"D4":1,"D5":1,"D6":1,"D7":1},"PCF8574-2":{"D0":1,"D1":1,"D2":1,"D3":1,"D4":1,"D5":1,"D6":1,"D7":1},"TempUnit":"C"}}
09:33:23.618 MQT: stat/GFHVAC/STATUS11 = {"StatusSTS":{"Time":"2022-02-21T09:33:23","Uptime":"0T11:13:35","UptimeSec":40415,"Heap":113,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":39,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":34},"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","POWER7":"OFF","POWER8":"OFF"}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
09:40:07.736 CMD: weblog
09:40:07.746 MQT: stat/GFHVAC/RESULT = {"WebLog":2}
09:40:10.897 CMD: weblog 4
09:40:10.907 MQT: stat/GFHVAC/RESULT = {"WebLog":4}
09:40:11.677 CFG: Saved, Count 36, Bytes 4096
09:40:41.150 SRC: MQTT
09:40:41.154 CMD: Grp 0, Cmd 'POWER', Idx 1, Len 3, Pld 0, Data 'OFF'
09:40:41.164 MQT: stat/GFHVAC/RESULT = {"POWER1":"OFF"}
09:40:41.171 MQT: stat/GFHVAC/POWER1 = OFF
09:40:52.551 M32: Kick off readOneSensor
09:40:52.555 M32: Kick off tele sending
09:41:24.799 CMD: setoption53 0
09:41:24.802 SRC: WebConsole from 192.168.2.112
09:41:24.805 CMD: Grp 0, Cmd 'SETOPTION', Idx 53, Len 1, Pld 0, Data '0'
09:41:24.813 MQT: stat/GFHVAC/RESULT = {"SetOption53":"OFF"}
09:41:25.600 CFG: Saved, Count 37, Bytes 4096
09:41:32.916 CMD: setoption53 1
09:41:32.919 SRC: WebConsole from 192.168.2.112
09:41:32.924 CMD: Grp 0, Cmd 'SETOPTION', Idx 53, Len 1, Pld 1, Data '1'
09:41:32.932 MQT: stat/GFHVAC/RESULT = {"SetOption53":"ON"}
09:41:33.721 CFG: Saved, Count 38, Bytes 4096

```

### TO REPRODUCE
1. Set up the ESP32 device. Attach ethernet cable
2. SetOption53 1
3. WiFi 0 (see first 2 screenshots)
4. Wifi 1 (see last screenshot)

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Correct Eth IP address is displayed in the GUI and in HASS

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]
[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2022-02-21:
You're seeing different issues.

1) IP address 0.0.0.0 means DHCP is providing the correct IP address.
2) IP address 192.168.4.1 is the AP's IP address and not the ethernet IP address. The ethernet IP address won't be shown in the GUI when wifi is ON.

altserg, 2022-02-21:
Hi Theo, 

I disagree.
1. When Wifi is on, then, the IP address in MQTT message and GUI is correct. When Wifi is Off, Eth IP address is incorrect. See the image below showing 2 similar devices with the same FW and one Wifi enabled, another Wifi disabled
[image]
2. My AP has different address, I 192.168.2.XXX network

Jason2866, 2022-02-22:
Getting both (DHCP) IP addresses shown with `status 5`
`
MQT: stat/sonoff-D09CA8/STATUS5 = {"StatusNET":{"Hostname":"sonoff-D09CA8-7336","IPAddress":"192.168.2.179","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.1","DNSServer2":"0.0.0.0","Mac":"94:B9:7E:D0:9C:A8","Ethernet":{"Hostname":"sonoff-D09CA8-7336-eth","IPAddress":"192.168.2.176","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.1","DNSServer2":"0.0.0.0","Mac":"94:B9:7E:D0:9C:AB"},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
`

altserg, 2022-02-23:
With latest git pull I cannot compile the FW anymore. I have cleaned the cache, etc. Is there anything special that I need to do ?

c:/users/stkachen/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\stkachen\AppData\Local\Temp\ccolNs1L.ltrans13.ltrans.o:(.literal+0x49c): undefined reference to `be_load_ccronexpr_class'
c:/users/stkachen/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\stkachen\AppData\Local\Temp\ccolNs1L.ltrans13.ltrans.o: in function `setup()':
<artificial>:(.text+0x3caa): undefined reference to `be_load_ccronexpr_class'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\tasmota32\firmware.elf] Error 1

Jason2866, 2022-02-23:
Nothing special to do. There is no issue, since the automated builds compile all successfully.
Try a complete fresh git clone and delete hidden folder `.platformio`

altserg, 2022-02-23:
still the same error, deleted everything...

altserg, 2022-02-23:
I have found the problem: I use #undef USE_LIGHT, and missing class is guarded by #ifdef USE_LIGHT. The fix is to move one line of code:
diff --git a/lib/libesp32/berry/default/be_modtab.c b/lib/libesp32/berry/default/be_modtab.c
index 684174821..4628e2afb 100644
--- a/lib/libesp32/berry/default/be_modtab.c
+++ b/lib/libesp32/berry/default/be_modtab.c
@@ -208,8 +208,8 @@ BERRY_API void be_load_custom_libs(bvm *vm)
     be_load_md5_lib(vm);
     be_load_serial_lib(vm);
     be_load_ctypes_lib(vm);
-    be_load_ccronexpr_class(vm);
 #ifdef USE_LIGHT
+    be_load_ccronexpr_class(vm);
     be_load_light_state_class(vm);
 #endif
 #ifdef USE_ALEXA_AVS
in /lib/libesp32/berry/default/be_modtab.c b/lib/libesp32/berry/default/be_modtab.c but I do not know if it's right
maybe someone who has more knowledge about the tasmota berry infra can look at it

altserg, 2022-02-23:
With latest fix I confirm that correct IP address is displayed when DHCP is not used
[image]

Correct IP is also shown in MQTT:
[image]

{"ip":"192.168.2.96","dn":"GF HVAC","fn":["GF_CV_Switch","GF_Reserve_1","GF_Reserve_2","GF_Reserve_3","GF_Reserve_4","GF_HVAC_MAX","GF_HVAC_MEDIUM","GF_HVAC_STANDBY"],"hn":"ESP32-GF-HVAC-eth","mac":"C8C9A3DC2E10","md":"KC868-A8","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"11.0.0.2","t":"GFHVAC","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[1,1,1,1,1,1,1,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":0,"lt_st":0,"sho":[0,0,0,0],"ver":1}

Status 5 output shows first IP as 0.0.0.0 but this is probably correct, maybe refers to WiFi

 status 5
16:54:56.262 MQT: stat/GFHVAC/STATUS5 = {"StatusNET":{"Hostname":"ESP32-GF-HVAC","IPAddress":"0.0.0.0","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.1","DNSServer2":"0.0.0.0","Mac":"C8:C9:A3:DC:2E:10","Ethernet":{"Hostname":"ESP32-GF-HVAC-eth","IPAddress":"192.168.2.96","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.1","DNSServer2":"8.8.8.8","Mac":"C8:C9:A3:DC:2E:13"},"Webserver":2,"HTTP_API":0,"WifiConfig":0,"WifiPower":17.0}}

altserg, 2022-02-23:
I will close the issue, but maybe someone can still have a look at the compilation issue

closed 2022-02-23 by altserg

Jason2866, 2022-02-28:
Berry error should be solved with https://github.com/arendst/Tasmota/pull/14993