arendst/Tasmota #14941: v11.0.0 Crashing with smart bulb

ralphrmartin, 2022-02-22:
### PROBLEM DESCRIPTION
I have a smart bulb which worked perfectly with v9 and v10. Installing Tasmota v11, the bulb initially shows a web interface, but crashes some time afterwards (stops responding to the web interface, stops adding info to the the console reports). 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic):LB01-15W-E27-TAS
- [x] Tasmota binary firmware version number used: 11.0.0, 11.0.0.2
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: firmware uploaded from web interface from previous version
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
07:28:04.488 RSL: RESULT = {"NAME":"LB01-15W-E27-TAS","GPIO":[0,0,0,0,416,419,0,0,417,452,418,0,0,0],"FLAG":0,"BASE":18}
07:28:04.697 RSL: RESULT = {"Module":{"0":"LB01-15W-E27-TAS"}}
07:28:04.902 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"416":"PWM1"},"GPIO5":{"419":"PWM4"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"417":"PWM2"},"GPIO13":{"452":"PWM_i5"},"GPIO14":{"418":"PWM3"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
No rules

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
07:29:07.028 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"RalphsNewBulb","FriendlyName":["RalphsNewBulb"],"Topic":"tasmota_D3F022","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
07:29:07.055 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:05:49","StartupUTC":"2022-02-22T07:23:18","Sleep":50,"CfgHolder":4617,"BootCount":41,"BCResetTime":"2021-05-10T11:52:02","SaveCount":973,"SaveAddress":"FA000"}}
07:29:07.089 RSL: STATUS2 = {"StatusFWR":{"Version":"11.0.0.2(tasmota)","BuildDateTime":"2022-02-21T21:48:18","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"359/699"}}
07:29:07.108 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["rrmair",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00208001","0405C80001000600003C5A0A000000000000","08000080","00006400","00004000"]}}
07:29:07.137 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":620,"Free":380,"Heap":26,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
07:29:07.169 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-D3F022-4130","IPAddress":"192.168.1.238","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.102","DNSServer2":"1.1.1.1","Mac":"C4:4F:33:D3:F0:22","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
07:29:07.195 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-02-22T07:29:07","Local":"2022-02-22T07:29:07","StartDST":"2022-03-27T01:00:00","EndDST":"2022-10-30T02:00:00","Timezone":99,"Sunrise":"06:45","Sunset":"17:21"}}
07:29:07.211 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-02-22T07:29:07"}}
07:29:07.220 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-02-22T07:29:07","Uptime":"0T00:05:49","UptimeSec":349,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":100,"Color":"000000FE01","HSBColor":"0,0,0","White":100,"CT":155,"Channel":[0,0,0,100,1],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"rrmair","BSSId":"44:D9:E7:FD:FA:B1","Channel":1,"Mode":"11n","RSSI":52,"Signal":-74,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
Not possible, as the device crashes, and does not respond to console

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Install v11, and wait for some time. Time to crash varies, but seems to be in the range 10 minutes to 1/2 hour

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
No Crash

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
N/A

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Bulb worked perfectly with Tasmota v9 and v10

**(Please, remember to close the issue when the problem has been addressed)**

ralphrmartin, 2022-02-22:
It definitely was crashing with v 11.0.0, both on initial install, and after reinstalling v11.0.0 after reverting to v10.1.0 to check the hardware was ok. I _think_ I installed v11.0.0.2 and it crashed, but after reinstalling v11.0.0.2, it has run for much longer than before without crashing. I will report back later if it crashed

Some time before the crash occurs, there is a line 
`07:33:18.409 APP: Serial logging disabled` 
in the console.

The bulb used is, if I remember correctly, this one:
[https://www.aliexpress.com/item/1005001879808728.html](https://www.aliexpress.com/item/1005001879808728.html)

I installed v11 on some smart sockets, and it ran fine. It is only this bulb which has the issue.

Jason2866, 2022-02-22:
The message `07:33:18.409 APP: Serial logging disabled` is standard and no indicator for something going wrong. Probably the bulb has a hardware issue.

Jason2866, 2022-02-22:
closing since opened in disscusions too. Lets continue in https://github.com/arendst/Tasmota/discussions/14939

closed 2022-02-22 by Jason2866