arendst/Tasmota #14970: Dimmer goes crazy after updating to 11.0

tzagim, 2022-02-24:
### PROBLEM DESCRIPTION
After updating the dimmer from version 10 to version 10.1 and also in version 11. The on and off button of the dimmer remains red (running) and cannot be turned off. Also after a shutdown in the system performs about 10 shutdowns and ignitions before it shuts down.

### REQUESTED INFORMATION
- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Avatto Dimmer
- [X] Tasmota binary firmware version number used: 11
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
{"NAME":"Avatto Dimmer Switch","GPIO":[0,0,0,0,0,0,0,0,0,0,290,0,0,0],"FLAG":0,"BASE":54,"CMND":"TuyaMCU 21,2||TuyaMCU 31,5||DimmerRange 10,30"}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
17:57:25.380 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON","Dimmer":60}
17:57:25.440 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON","Dimmer":55}
17:57:25.499 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON","Dimmer":50}
17:57:25.582 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON","Dimmer":45}
17:57:25.702 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON","Dimmer":40}
17:57:25.718 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON","Dimmer":36}
17:57:25.741 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON","Dimmer":31}
17:57:25.837 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON","Dimmer":26}
17:57:25.860 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON","Dimmer":21}
17:57:25.920 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON","Dimmer":16}
17:57:25.980 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON","Dimmer":12}
17:57:26.043 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON","Dimmer":7}
17:57:26.701 MQT: stat/dimmer_salon/RESULT = {"POWER":"OFF"}
17:57:26.704 MQT: stat/dimmer_salon/POWER = OFF
17:57:26.745 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON"}
17:57:26.750 MQT: stat/dimmer_salon/POWER = ON
17:57:27.362 MQT: stat/dimmer_salon/RESULT = {"POWER":"OFF"}
17:57:27.367 MQT: stat/dimmer_salon/POWER = OFF
17:57:27.405 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON"}
17:57:27.410 MQT: stat/dimmer_salon/POWER = ON
17:57:28.019 MQT: stat/dimmer_salon/RESULT = {"POWER":"OFF"}
17:57:28.022 MQT: stat/dimmer_salon/POWER = OFF
17:57:28.063 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON"}
17:57:28.067 MQT: stat/dimmer_salon/POWER = ON
17:57:28.683 MQT: stat/dimmer_salon/POWER = OFF
17:57:28.723 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON"}
17:57:28.727 MQT: stat/dimmer_salon/POWER = ON

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
Up to version 10.1 it worked fine

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2022-02-24:
Hi, can you share the required information from the template?

Please, share the status 0 output. And please, set weblog to 4 and then share the complete logs when your issue occurs. Thanks

ascillato, 2022-02-24:
This issue is similar to #14398. For Tuya issues we add the label Help Needed due to there is no active development for Tuya drivers in Tasmota at this moment. Sorry. The actual Tuya support is provided "as is" only. Anyway, it is open for anyone interested on fixing its bugs and adding more features to it.

ascillato, 2022-02-24:
> Please, share the `status 0` output. And please, set `weblog` to 4 and then share the complete logs when your issue occurs. Thanks.

What we meant with that is that if you can go to the Tasmota console on that device and type:

```haskell
Weblog 4
Status 0
```

and then copy the output of that and paste here in a comment, please. Thanks.

ascillato2, 2022-02-28:
Please, set `weblog` to 4 and then share the complete logs when your issue occurs and then copy the output of that and paste here. Thanks.

tzagim, 2022-02-28:
```lua
19:32:04.912 CMD: Grp 0, Cmnd 'STATUS', Idx 1, Len 1, Data '0'
19:32:04.918 MQT: stat/dimmer_salon/STATUS = {"Status":{"Module":0,"DeviceName":"Dimmer","FriendlyName":["DIMMER"],"Topic":"dimmer_salon","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
19:32:04.924 MQT: stat/dimmer_salon/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:45","StartupUTC":"2022-02-28T18:31:19","Sleep":50,"CfgHolder":4617,"BootCount":150,"BCResetTime":"2021-07-21T14:56:58","SaveCount":2501,"SaveAddress":"F7000"}}
19:32:04.931 MQT: stat/dimmer_salon/STATUS2 = {"StatusFWR":{"Version":"11.0.0(tasmota)","BuildDateTime":"2022-02-12T14:15:53","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"410/699"}}
19:32:04.938 MQT: stat/dimmer_salon/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXXX","XXXXX"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004000"]}}
19:32:04.954 MQT: stat/dimmer_salon/STATUS4 = {"StatusMEM":{"ProgramSize":622,"Free":380,"Heap":24,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["0000040D","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
19:32:04.961 MQT: stat/dimmer_salon/STATUS5 = {"StatusNET":{"Hostname":"dimmer-salon-9665","IPAddress":"XXX.XXX.XXX.XXX","Gateway":"XXX.XXX.XXX.XXX","Subnetmask":"255.255.255.0","DNSServer1":"XXX.XXX.XXX.XXX","DNSServer2":"0.0.0.0","Mac":"XX:XX:XX:XX:XX:XX","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
19:32:04.968 MQT: stat/dimmer_salon/STATUS6 = {"StatusMQT":{"MqttHost":"XXX.XXX.XXX.XXX","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D19CE2","MqttUser":"XXXXXX","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
19:32:04.978 MQT: stat/dimmer_salon/STATUS7 = {"StatusTIM":{"UTC":"2022-02-28T18:32:04","Local":"2022-02-28T19:32:04","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"07:34","Sunset":"18:31"}}
19:32:04.984 MQT: stat/dimmer_salon/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
19:32:04.995 MQT: stat/dimmer_salon/STATUS10 = {"StatusSNS":{"Time":"2022-02-28T19:32:04","ENERGY":{"TotalStartTime":"2021-07-21T14:56:58","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power": 0}}}
19:32:05.001 MQT: stat/dimmer_salon/STATUS11 = {"StatusSTS":{"Time":"2022-02-28T19:32:04","Uptime":"0T00:00:45","UptimeSec":45,"Heap":24,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":66,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"XXXXXX","BSSId":"XX:XX:XX:XX:XX:XX","Channel":11,"Mode":"11n","RSSI":100,"Signal":-40,"LinkCount":1,"Downtime":"0T00:00:03"}}}
19:32:12.756 MQT: stat/dimmer_salon/STATUS5 = {"StatusNET":{"Hostname":"dimmer-salon-9665","IPAddress":"XXX.XXX.XXX.XXX","Gateway":"XXX.XXX.XXX.XXX","Subnetmask":"255.255.255.0","DNSServer1":"XXX.XXX.XXX.XXX","DNSServer2":"0.0.0.0","Mac":"XX:XX:XX:XX:XX:XX","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
19:32:12.762 MQT: stat/dimmer_salon/STATUS6 = {"StatusMQT":{"MqttHost":"XXX.XXX.XXX.XXX","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D19CE2","MqttUser":"XXXXXX","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
19:32:12.770 MQT: stat/dimmer_salon/STATUS7 = {"StatusTIM":{"UTC":"2022-02-28T18:32:12","Local":"2022-02-28T19:32:12","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"07:34","Sunset":"18:31"}}
19:32:12.777 MQT: stat/dimmer_salon/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
19:32:12.787 MQT: stat/dimmer_salon/STATUS10 = {"StatusSNS":{"Time":"2022-02-28T19:32:12","ENERGY":{"TotalStartTime":"2021-07-21T14:56:58","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power": 0}}}
19:32:12.797 MQT: stat/dimmer_salon/STATUS11 = {"StatusSTS":{"Time":"2022-02-28T19:32:12","Uptime":"0T00:00:53","UptimeSec":53,"Heap":24,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":66,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"XXXXXX","BSSId":"XX:XX:XX:XX:XX:XX","Channel":11,"Mode":"11n","RSSI":100,"Signal":-41,"LinkCount":1,"Downtime":"0T00:00:03"}}}
19:32:13.840 SRC: WebGui from XXX.XXX.XXX.XXX
19:32:13.858 TYA: Send "55aa0006000802020004000000172c"
19:32:13.873 TYA: Send "55aa0006000501010001000d"
19:32:13.898 MQT: stat/dimmer_salon/RESULT = {"POWER":"OFF"}
19:32:13.903 MQT: stat/dimmer_salon/POWER = OFF
19:32:13.974 {"TuyaReceived":{"Data":"55AA03070008020200040000001730","Cmnd":7,"CmndData":"0202000400000017","DpType2Id2":23,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000017"}}}
19:32:13.976 TYA: fnId=21 is set for dpId=2
[... cut ...]
19:32:18.984 {"TuyaReceived":{"Data":"55AA03070008020200040000000821","Cmnd":7,"CmndData":"0202000400000008","DpType2Id2":8,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000008"}}}
19:32:18.987 TYA: fnId=21 is set for dpId=2
19:32:18.989 TYA: RX value 8 from dpId 2 
19:32:20.077 TYA: fnId=21 is set for dpId=2
19:32:20.079 TYA: RX value 0 from dpId 2 
19:32:20.082 {"TuyaReceived":{"Data":"55AA03070008020200040000000B24","Cmnd":7,"CmndData":"020200040000000B","DpType2Id2":11,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000000B"}}}
19:32:20.084 TYA: fnId=21 is set for dpId=2
19:32:20.085 TYA: RX value 11 from dpId 2 
19:32:20.440 WIF: Sending Gratuitous ARP

ascillato, 2022-02-28:
When you said `Current State: On` is that the light is on right? and what are you referring to when you said` MCU State: On`?

> after a shutdown in the system performs about 10 shutdowns and ignitions before it shuts down.

Can you share the log when you perform the shutdown?

tzagim, 2022-02-28:
> When you said `Current State: On` is that the light is ON right? but what are you referring to when you said `MCU State: On`?
> 
> > after a shutdown in the system performs about 10 shutdowns and ignitions before it shuts down.
> 
> Can you share the log when you perform the shutdown?

This is what the log looks like when I perform the shutdown.
He somehow gets by himself ON & OFF commands.

ascillato, 2022-02-28:
It shows `SRC: Switch` so that means that the switch is being pushed or noise is entering the GPIO of the switch.

ascillato, 2022-02-28:
you put in the logs: `MCU State: On`, what are you referring to?

tzagim, 2022-02-28:
It is strange.
1. It's a touch and not a physical switch.
From version 11 the lighting in the switch remains red (on) and can not be switched off at all through the switch.

> > `TYA: RX Relay-1 --> MCU State: On Current State:Off`
> 
> you put in the logs: `MCU State: On`, what are you referring to?

Ahhh, that's how the log came out no I added this.

PS I accidentally clicked on close

ascillato2, 2022-02-28:
Where did you get your template? Seems incomplete.

ascillato2, 2022-02-28:
Can you share the output of `Backlog Template; Module; GPIO 255`

tzagim, 2022-02-28:
I took the template from similar switches in https://templates.blakadder.com

```lua
20:28:06.529 CMD: Backlog Template; Module; GPIO 255
20:28:06.608 MQT: stat/dimmer_salon/RESULT = {"NAME":"Avatto Dimmer Switch","GPIO":[0,0,0,0,0,0,0,0,0,0,290,0,0,0],"FLAG":0,"BASE":54}
20:28:06.819 MQT: stat/dimmer_salon/RESULT = {"Module":{"0":"Avatto Dimmer Switch"}}
20:28:07.025 MQT: stat/dimmer_salon/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"290":"Led3"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

ascillato, 2022-02-28:
> I took the template from similar Dimmers in https://templates.blakadder.com/

which one? I can't find it. Your template is not ok. it has only a GPIO defined as led3.

tzagim, 2022-02-28:
> > I took the template from similar Dimmers in https://templates.blakadder.com/
> 
> which one? I can't find it. Your template is not ok. it has only a GPIO defined as led3.

[Exemple](https://templates.blakadder.com/iswitch_dimmer.html).

I tried to make a change
```lua
20:31:54.372 CMD: Backlog Template; Module; GPIO 255
20:31:54.438 MQT: stat/dimmer_salon/RESULT = {"NAME":"Avatto Dimmer Switch","GPIO":[0,2272,0,2304,0,0,0,0,0,0,290,0,0,0],"FLAG":0,"BASE":54}
20:31:54.650 MQT: stat/dimmer_salon/RESULT = {"Module":{"0":"Avatto Dimmer Switch"}}
20:31:54.905 MQT: stat/dimmer_salon/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"2272":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"2304":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"290":"Led3"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

20:31:07.855 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON","Dimmer":50}
20:31:07.871 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON","Dimmer":45}
20:31:07.897 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON","Dimmer":40}
20:31:07.956 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON","Dimmer":36}
20:31:08.016 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON","Dimmer":31}
20:31:08.076 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON","Dimmer":26}
20:31:08.136 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON","Dimmer":21}
20:31:08.197 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON","Dimmer":16}
20:31:08.256 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON","Dimmer":12}
20:31:08.316 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON","Dimmer":7}
20:31:08.974 MQT: stat/dimmer_salon/RESULT = {"POWER":"OFF"}
20:31:08.979 MQT: stat/dimmer_salon/POWER = OFF
20:31:09.019 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON"}
20:31:09.024 MQT: stat/dimmer_salon/POWER = ON
20:31:09.636 MQT: stat/dimmer_salon/RESULT = {"POWER":"OFF"}
20:31:09.639 MQT: stat/dimmer_salon/POWER = OFF
20:31:09.677 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON"}
20:31:09.683 MQT: stat/dimmer_salon/POWER = ON
20:31:10.297 MQT: stat/dimmer_salon/RESULT = {"POWER":"OFF"}
20:31:10.303 MQT: stat/dimmer_salon/POWER = OFF
20:31:10.337 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON"}
20:31:10.342 MQT: stat/dimmer_salon/POWER = ON
20:31:10.956 MQT: stat/dimmer_salon/RESULT = {"POWER":"OFF"}
20:31:10.961 MQT: stat/dimmer_salon/POWER = OFF
20:31:10.997 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON"}
20:31:11.002 MQT: stat/dimmer_salon/POWER = ON
20:31:11.626 MQT: stat/dimmer_salon/RESULT = {"POWER":"OFF"}
20:31:11.630 MQT: stat/dimmer_salon/POWER = OFF
20:31:11.665 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON"}
20:31:11.668 MQT: stat/dimmer_salon/POWER = ON
20:31:12.286 MQT: stat/dimmer_salon/RESULT = {"POWER":"OFF"}
20:31:12.290 MQT: stat/dimmer_salon/POWER = OFF
20:31:12.385 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON"}
20:31:12.388 MQT: stat/dimmer_salon/POWER = ON
20:31:12.946 MQT: stat/dimmer_salon/RESULT = {"POWER":"OFF"}
20:31:12.950 MQT: stat/dimmer_salon/POWER = OFF
20:31:12.987 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON"}
20:31:12.991 MQT: stat/dimmer_salon/POWER = ON
20:31:13.479 MQT: stat/dimmer_salon/RESULT = {"POWER":"OFF"}
20:31:13.483 MQT: stat/dimmer_salon/POWER = OFF

ascillato, 2022-02-28:
Please, share the same log but with weblog 4

tzagim, 2022-02-28:
> Please, share the same log but with weblog 4

```lua
20:47:11.527 TYA: Send "55aa00060008020200040000001a2f"
20:47:11.541 TYA: Send "55aa0006000501010001000d"
20:47:11.566 MQT: stat/dimmer_salon/RESULT = {"POWER":"OFF"}
20:47:11.571 MQT: stat/dimmer_salon/POWER = OFF
20:47:11.642 {"TuyaReceived":{"Data":"55AA03070008020200040000001A33","Cmnd":7,"CmndData":"020200040000001A","DpType2Id2":26,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000001A"}}}
20:47:11.644 TYA: fnId=21 is set for dpId=2
20:47:11.646 TYA: RX value 26 from dpId 2 
20:47:11.648 {"TuyaReceived":{"Data":"55AA03070005010100010011","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
20:47:11.650 TYA: fnId=11 is set for dpId=1
20:47:11.652 TYA: RX Relay-1 --> MCU State: Off Current State:Off
20:47:11.655 {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
20:47:11.658 TYA: fnId=11 is set for dpId=1
20:47:11.660 TYA: RX Relay-1 --> MCU State: On Current State:Off
20:47:11.661 SRC: Switch
20:47:11.679 TYA: Send "55aa00060008020200040000001a2f"
20:47:11.685 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON"}
20:47:11.689 MQT: stat/dimmer_salon/POWER = ON
20:47:11.694 {"TuyaReceived":{"Data":"55AA03070008020200040000001932","Cmnd":7,"CmndData":"0202000400000019","DpType2Id2":25,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000019"}}}
20:47:11.696 TYA: fnId=21 is set for dpId=2
20:47:11.698 TYA: RX value 25 from dpId 2 
20:47:11.700 {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
20:47:11.702 TYA: fnId=11 is set for dpId=1
20:47:11.704 TYA: RX Relay-1 --> MCU State: On Current State:On
20:47:11.706 {"TuyaReceived":{"Data":"55AA03070008020200040000001831","Cmnd":7,"CmndData":"0202000400000018","DpType2Id2":24,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000018"}}}
20:47:11.708 TYA: fnId=21 is set for dpId=2
20:47:11.710 TYA: RX value 24 from dpId 2 
20:47:11.713 {"TuyaReceived":{"Data":"55AA03070008020200040000001A33","Cmnd":7,"CmndData":"020200040000001A","DpType2Id2":26,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000001A"}}}
20:47:11.714 TYA: fnId=21 is set for dpId=2
20:47:11.716 TYA: RX value 26 from dpId 2 
20:47:13.808 {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
20:47:13.811 TYA: fnId=11 is set for dpId=1
20:47:13.812 TYA: RX Relay-1 --> MCU State: On Current State:On
20:47:13.824 {"TuyaReceived":{"Data":"55AA0307000802020004000000021B","Cmnd":7,"CmndData":"0202000400000002","DpType2Id2":2,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000002"}}}
20:47:13.827 TYA: fnId=21 is set for dpId=2
20:47:13.828 TYA: RX value 2 from dpId 2 
20:47:13.867 {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
20:47:13.870 TYA: fnId=11 is set for dpId=1
20:47:13.872 TYA: RX Relay-1 --> MCU State: On Current State:On
20:47:13.883 {"TuyaReceived":{"Data":"55AA0307000802020004000000011A","Cmnd":7,"CmndData":"0202000400000001","DpType2Id2":1,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000001"}}}
20:47:13.886 TYA: fnId=21 is set for dpId=2
20:47:13.887 TYA: RX value 1 from dpId 2 
20:47:13.928 {"TuyaReceived":{"Data":"55AA03070005010100010011","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
20:47:13.930 TYA: fnId=11 is set for dpId=1
20:47:13.932 TYA: RX Relay-1 --> MCU State: Off Current State:On
20:47:13.934 SRC: Switch
20:47:13.952 TYA: Send "55aa00060008020200040000000b20"
20:47:13.957 MQT: stat/dimmer_salon/RESULT = {"POWER":"OFF"}
20:47:13.962 MQT: stat/dimmer_salon/POWER = OFF
20:47:13.966 {"TuyaReceived":{"Data":"55AA03070008020200040000000019","Cmnd":7,"CmndData":"0202000400000000","DpType2Id2":0,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000000"}}}
20:47:13.969 TYA: fnId=21 is set for dpId=2
20:47:13.970 TYA: RX value 0 from dpId 2 
20:47:13.973 {"TuyaReceived":{"Data":"55AA03070008020200040000000B24","Cmnd":7,"CmndData":"020200040000000B","DpType2Id2":11,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000000B"}}}
20:47:13.975 TYA: fnId=21 is set for dpId=2
20:47:13.977 TYA: RX value 11 from dpId 2 
20:47:13.998 {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
20:47:13.001 TYA: fnId=11 is set for dpId=1
20:47:13.002 TYA: RX Relay-1 --> MCU State: On Current State:Off
20:47:13.004 SRC: Switch
20:47:14.000 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON"}
20:47:14.003 MQT: stat/dimmer_salon/POWER = ON
20:47:14.009 {"TuyaReceived":{"Data":"55AA03070008020200040000000A23","Cmnd":7,"CmndData":"020200040000000A","DpType2Id2":10,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000000A"}}}
20:47:14.011 TYA: fnId=21 is set for dpId=2
20:47:14.013 TYA: RX value 10 from dpId 2 
20:47:14.084 {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
20:47:14.087 TYA: fnId=11 is set for dpId=1
20:47:14.089 TYA: RX Relay-1 --> MCU State: On Current State:On
20:47:14.092 {"TuyaReceived":{"Data":"55AA03070008020200040000000922","Cmnd":7,"CmndData":"0202000400000009","DpType2Id2":9,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000009"}}}
[... 49 lines cut ...]
[... cut ...]
20:47:17.841 {"TuyaReceived":{"Data":"55AA03070008020200040000000019","Cmnd":7,"CmndData":"0202000400000000","DpType2Id2":0,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000000"}}}
20:47:17.844 TYA: fnId=21 is set for dpId=2
20:47:17.845 TYA: RX value 0 from dpId 2 
20:47:17.848 {"TuyaReceived":{"Data":"55AA03070008020200040000000B24","Cmnd":7,"CmndData":"020200040000000B","DpType2Id2":11,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000000B"}}}
20:47:17.850 TYA: fnId=21 is set for dpId=2
20:47:17.852 TYA: RX value 11 from dpId 2

tzagim, 2022-02-28:
And this is a log after downgrade to version 10

```lua
20:55:47.250 CMD: Grp 0, Cmnd 'DIMMER', Idx 1, Len 3, Data '100'
20:55:47.253 SRC: Light
20:55:47.271 TYA: Send "55aa00060008020200040000001e33"
20:55:47.286 TYA: Send "55aa0006000501010001010e"
20:55:47.311 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON","Dimmer":100}
20:55:47.383 {"TuyaReceived":{"Data":"55AA03070008020200040000001E37","Cmnd":7,"CmndData":"020200040000001E","DpType2Id2":30,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000001E"}}}
20:55:47.385 TYA: fnId=21 is set for dpId=2
20:55:47.387 TYA: RX value 30 from dpId 2 
20:55:47.389 {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
20:55:47.392 TYA: fnId=11 is set for dpId=1
20:55:47.393 TYA: RX Relay-1 --> MCU State: On Current State:On
20:55:47.396 {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
20:55:47.399 TYA: fnId=11 is set for dpId=1
20:55:47.400 TYA: RX Relay-1 --> MCU State: On Current State:On
20:55:47.403 {"TuyaReceived":{"Data":"55AA03070008020200040000001E37","Cmnd":7,"CmndData":"020200040000001E","DpType2Id2":30,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000001E"}}}
20:55:47.404 TYA: fnId=21 is set for dpId=2
20:55:47.406 TYA: RX value 30 from dpId 2 
20:55:47.918 CFG: Saved to flash at F7, Count 2573, Bytes 4096
20:55:50.040 TYA: fnId=21 is set for dpId=2
20:55:50.041 TYA: RX value 13 from dpId 2 
20:55:50.043 SRC: Switch
20:55:50.045 CMD: Grp 0, Cmnd 'DIMMER', Idx 3, Len 2, Data '43'
20:55:50.052 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON","Dimmer":43}
20:55:50.082 {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
20:55:50.084 TYA: fnId=11 is set for dpId=1
20:55:50.086 TYA: RX Relay-1 --> MCU State: On Current State:On
20:55:50.097 {"TuyaReceived":{"Data":"55AA03070008020200040000000C25","Cmnd":7,"CmndData":"020200040000000C","DpType2Id2":12,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000000C"}}}
20:55:50.100 TYA: fnId=21 is set for dpId=2
20:55:50.102 TYA: RX value 12 from dpId 2 
20:55:50.104 SRC: Switch
20:55:50.106 CMD: Grp 0, Cmnd 'DIMMER', Idx 3, Len 2, Data '40'
20:55:50.113 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON","Dimmer":40}
20:55:50.141 {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
20:55:50.144 TYA: fnId=11 is set for dpId=1
20:55:50.145 TYA: RX Relay-1 --> MCU State: On Current State:On
20:55:50.157 {"TuyaReceived":{"Data":"55AA03070008020200040000000B24","Cmnd":7,"CmndData":"020200040000000B","DpType2Id2":11,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000000B"}}}
20:55:50.160 TYA: fnId=21 is set for dpId=2
20:55:50.162 TYA: RX value 11 from dpId 2 
20:55:50.163 SRC: Switch
20:55:50.166 CMD: Grp 0, Cmnd 'DIMMER', Idx 3, Len 2, Data '37'
20:55:50.173 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON","Dimmer":37}
20:55:50.202 {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
20:55:50.205 TYA: fnId=11 is set for dpId=1
20:55:50.206 TYA: RX Relay-1 --> MCU State: On Current State:On
20:55:50.217 {"TuyaReceived":{"Data":"55AA03070008020200040000000A23","Cmnd":7,"CmndData":"020200040000000A","DpType2Id2":10,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000000A"}}}
20:55:50.220 TYA: fnId=21 is set for dpId=2
20:55:50.222 TYA: RX value 10 from dpId 2 
20:55:50.224 SRC: Switch
20:55:50.226 CMD: Grp 0, Cmnd 'DIMMER', Idx 3, Len 2, Data '33'
20:55:50.234 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON","Dimmer":33}
20:55:50.262 {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
20:55:50.265 TYA: fnId=11 is set for dpId=1
20:55:50.267 TYA: RX Relay-1 --> MCU State: On Current State:On
20:55:50.276 {"TuyaReceived":{"Data":"55AA03070008020200040000000922","Cmnd":7,"CmndData":"0202000400000009","DpType2Id2":9,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000009"}}}
20:55:50.279 TYA: fnId=21 is set for dpId=2
20:55:50.281 TYA: RX value 9 from dpId 2 
20:55:50.283 SRC: Switch
20:55:50.285 CMD: Grp 0, Cmnd 'DIMMER', Idx 3, Len 2, Data '30'
20:55:50.294 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON","Dimmer":30}
20:55:50.322 {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
20:55:50.325 TYA: fnId=11 is set for dpId=1
20:55:50.326 TYA: RX Relay-1 --> MCU State: On Current State:On
20:55:50.337 {"TuyaReceived":{"Data":"55AA03070008020200040000000821","Cmnd":7,"CmndData":"0202000400000008","DpType2Id2":8,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000008"}}}
20:55:50.340 TYA: fnId=21 is set for dpId=2
20:55:50.342 TYA: RX value 8 from dpId 2 
20:55:50.344 SRC: Switch
20:55:50.346 CMD: Grp 0, Cmnd 'DIMMER', Idx 3, Len 2, Data '27'
20:55:50.353 MQT: stat/dimmer_salon/RESULT = {"POWER":"ON","Dimmer":27}
20:55:50.381 {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
20:55:50.384 TYA: fnId=11 is set for dpId=1
20:55:50.386 TYA: RX Relay-1 --> MCU State: On Current State:On
20:55:50.826 TYA: fnId=21 is set for dpId=2
20:55:50.827 TYA: RX value 0 from dpId 2 
[... cut ...]
20:55:50.923 {"TuyaReceived":{"Data":"55AA03070005010100010011","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
20:55:50.926 TYA: fnId=11 is set for dpId=1
20:55:50.927 TYA: RX Relay-1 --> MCU State: Off Current State:Off
20:55:50.930 {"TuyaReceived":{"Data":"55AA03070008020200040000000019","Cmnd":7,"CmndData":"0202000400000000","DpType2Id2":0,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000000"}}}
20:55:50.933 TYA: fnId=21 is set for dpId=2
20:55:50.935 TYA: RX value 0 from dpId 2

ascillato, 2022-02-28:
Thanks for sharing all your logs. Very much appreciated. I was looking for any issue outside Tuya protocol in order to help you, but with your last logs, it is confirmed that is a Tuya issue.

The actual Tuya Driver is buggy and orphaned. As it is unmaintained, these type of bugs are not going to be solved soon until someone having Tuya hardware, wants to spend time on it and provide a Pull Request. Because of that, we added the label **_Help Needed_**.

This issue is similar to https://github.com/arendst/Tasmota/issues/14398 so a fix to that one, should fix this one too.

tzagim, 2022-02-28:
> Thanks for sharing all your logs. Very much appreciated. I was looking for any issue outside Tuya protocol in order to help you, but with your last logs, it is confirmed that is a Tuya issue.
> 
> The actual Tuya Driver is buggy and orphaned. As it is unmaintained, these type of bugs are not going to be solved soon until someone having Tuya hardware, wants to spend time on it and also provide a Pull Request. Sorry. Because of that, we added the label **_Help Needed_**.
> 
> This issue is similar to #14398 so a fix to that one, should fix this one too.

Thank you very much for your patience and help.
I really hope someone finds the time to take care of this.

closed 2022-03-15 by ascillato2

btsimonh, 2022-03-16:
To add to this.
The central LED on my Athom touch dimmer is controlled by sending a command to the TuyaMCU.
I have two dimmers, and at first they came up with one having the central button lit blue, and the other with a soft red glow.
I complained to Athom that I had two different firmwares...

However, I have now started investigating further.  Both dimmers report the same Tuya firmware (no guarantee it IS identical).
But if you send the correct command, the light changes.
The fact is that the command to set the light is not getting through...  TAS tries to send, but the command is amongst other transactions, and gets discarded?

Now... my problem is that under fairly normal TAS controlled conditions (asking alexa to turn the light on), many times the TuyaMCU stops controlling the actual light (the light is off...).  I have also had the condition where after power on, one dimmer works, and the other does not.
From TAS all looks fine, and the MCU is giving good responses to the requests... it just seems to have shut down it's PWM.
No control from buttons or from TAS.... yet the protocol is still operating.

My GUESS is that the TuyaMCU driver in TAs is not observing the Tuya state machine for the serial protocol correctly, and that in my case this leads to an MCU shutdown of some form.  The BASIC state machine for the protocol is not complex, and well documented [here](https://developer.tuya.com/en/docs/iot/mcu-protocol?id=K9hrdpyujeotg).
My feeling is that we need to be much tighter on the protocol implementation, adhering to timeouts, resends and message order much more strictly if we are to successfully replicate the operation of these devices.

I have disabled the TAS to TuyaMCU link by setting the pins to SeriaTX/RX rather than TuyaTX/RX after the WAF factor completely collapsed today.  My dimmers are in the wall and in use, so I can't develop using them.  I have requested a development unit and help from Athom (@athom-tech - please).

@ascillato2 - if/when I get a dev device, I'll take an initial look.  Some pointers to the originator of the driver may be good if they are happy to chat.

br,
Simon

ascillato, 2022-03-16:
>@ascillato2 - if/when I get a dev device, I'll take an initial look. Some pointers to the originator of the driver may be good if they are happy to chat.

I think a good starting point for this is the issue that is open https://github.com/arendst/Tasmota/issues/14398 which contains a lot of information about Tuya Driver issues.

As the driver is orphaned there is no one that we can refer to. That is why you are more than welcome to take a look into it.

btsimonh, 2022-03-16:
Any sign of a TuyaMCU simulator/emulator?

ascillato, 2022-03-16:
> Any sign of a TuyaMCU simulator/emulator?

Sorry, none that I'm aware of.

btsimonh, 2022-03-16:
noticed you moved it to a discussion, so putting stuff in there now - thank you!