arendst/Tasmota #14975: Russian language is required

fnc12, 2022-02-25:
### PROBLEM DESCRIPTION
Russian language is missing

### REQUESTED INFORMATION
Why Russian language was denied? I am a developer from Kazakhstan, I use Russian.

If you don't know some facts let me tell you:

Russian is a very popular language outside of Russia (Russian has over 258 million total speakers worldwide ([wiki source](https://en.wikipedia.org/wiki/Russian_language)));
Russian Federation has 140 million citizens.
So there are more than 100 million Russian speakers outside of Russia.

Please stop discrimination of people who speaks Russian. We are not about war. We don't like war. We are developers not politics. Denying any language will not stop any war but increase meaningless populism.

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2022-02-25:
Russian is (temporarly) replaced by English because of what happens in the Ukraine.

From your wiki page I understand there are more countries using Russian so you could blame the Russian aggressor or perhaps use the Ukraine languge option (-UK).

fnc12, 2022-02-25:
Ok let me describe you in another way.

1. Assume Germany attacks any country. You don't like, I don't like it also. You replace German language with English. But what about people from Austria and Switzerland? They also speak German.

2. Why don't you replace English language with Arabic cause USA killed Saddam in his country?

What I want to say is that software is not about politics. Developers do software not politics. Bring Russian language back. Thank you.

arendst, 2022-02-25:
Sounds fair.

I'll revert.

arendst, 2022-02-25:
Done.

fnc12, 2022-02-25:
Thanks you. Where are you from? Let's meet and drink some day

arendst, 2022-02-25:
The Netherlands. Tiny country. Big mouth ;-)

ascillato2, 2022-02-25:
Closing this issue as it has been fixed.

closed 2022-02-25 by ascillato2

joba-1, 2022-02-25:
> Ok let me describe you in another way.
>
> 1. Assume Germany attacks any country. You don't like, I don't like it also. You replace German language with English. But what about people from Austria and [...]

I am German, and if Swizerland would attack Kazakhstan, I would be ok with replacing my language with Chinese or whatever and show appreciation for this sign of solidarity, surely not feeling discriminated.

Anyways, the sign is there and you have your language back. So far so good.

jaymemaurice, 2022-02-25:
My wife's father was born in Donbas, married a woman from Kazakhstan who was born there because her father was exiled being an enemy of the state with his religious beliefs. My wife's father's brothers were also jailed for being active Christians. My wife's mother and father moved to the baltics before my wife was born. After the fall of the Soviet Union and independence (to which her parents stood for), they faced discrimination for speaking Russian/being of distant Russian dissent. They eventually all moved to Canada. My wife still thinks in Russian. Despite having a lineage completely screwed up by powers she and her family could not influence, the language was what was used in her formative years and it wasn't really her choice. Her English is fine, but her mother's is not...

DeeKey, 2022-02-26:
More to add - Russian is a mother tongue for many people in Ukraine including president Zelensky. 
Good that the issue was resolved!

AnkBurov, 2022-03-03:
Witch hunt of 21st century.
@arendst you should be ashamed of yourself for this racist behavior when you decided to discriminate the language solely because some nations speak it.

danik-guitar, 2022-03-04:
весь мир такой толерантный, а как что случилось все русскоговорящие вдруг стали одним большим злом.