arendst/Tasmota #14994: IR Send Command On Power On / Reboot Device

jovanotti25, 2022-02-27:
### PROBLEM DESCRIPTION
When i make a restart in tasmota my ir remote send one irsend and turn on my tv  its possible wtich restart do nothing? sorry my bad english 

### REQUESTED INFORMATION
the product is BlitzWolf BW-RC1 IR Controller (56751)
with template 

{"NAME":"BW-RC1","GPIO":[0,0,0,0,56,51,0,0,0,17,8,0,0],"FLAG":0,"BASE":62}

status 0 

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

stat/hubir/STATUS = {"Status":{"Module":0,"DeviceName":"Hub IR","FriendlyName":["Hub IR"],"Topic":"hubir","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
21:13:57.083 MQT: stat/hubir/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota-ir.bin","RestartReason":"Software/System restart","Uptime":"0T00:21:50","StartupUTC":"2022-02-27T19:52:07","Sleep":50,"CfgHolder":4617,"BootCount":26,"BCResetTime":"2022-02-25T16:59:26","SaveCount":48,"SaveAddress":"F6000"}}
21:13:57.089 MQT: stat/hubir/STATUS2 = {"StatusFWR":{"Version":"11.0.0.3(ir)","BuildDateTime":"2022-02-26T16:05:48","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"336/699"}}
21:13:57.094 MQT: stat/hubir/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HouseTAM",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A190000000000","00000080","00006000","00004000"]}}
21:13:57.103 MQT: stat/hubir/STATUS4 = {"StatusMEM":{"ProgramSize":624,"Free":376,"Heap":24,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F5A0582","04000001","00000003","00000000","00000000","00000000","00000000","00000000"],"Drivers":"1,2,4,5,7,9,10,12","Sensors":"1"}}
21:13:57.109 MQT: stat/hubir/STATUS5 = {"StatusNET":{"Hostname":"hubir-3312","IPAddress":"19","Gateway":"192.168.68.1","Subnetmask":"255.","DNSServer1":"212.11","DNSServer2":"62.160","Mac":"2C:F2:84:2C:F0","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
21:13:57.115 MQT: stat/hubir/STATUS6 = {"StatusMQT":{"MqttHost":"192.16118","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0","MqttUser":"","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:13:57.122 MQT: stat/hubir/STATUS7 = {"StatusTIM":{"UTC":"2022-02-27T20:13:57","Local":"2022-02-27T21:13:57","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"07:35","Sunset":"18:29"}}
21:13:57.126 MQT: stat/hubir/STATUS10 = {"StatusSNS":{"Time":"2022-02-27T21:13:57"}}
21:13:57.132 MQT: stat/hubir/STATUS11 = {"StatusSTS":{"Time":"2022-02-27T21:13:57","Uptime":"0T00:21:50","UptimeSec":1310,"Vcc":3.496,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Ho","BSSId":"1C:3B:F3:80:68:EE","Channel":4,"Mode":"11n","RSSI":100,"Signal":-30,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-02-27:
This is missing from your request:
[image]

Please set `Weblog 4` and restart to see why Tasmota is sending the IR command to wake the TV.
It is most likely the command has been sent to Tasmota retained and so is executed again at restart. These logs will help us to be sure.

jovanotti25, 2022-02-27:
this is weblog 4 
 HTP: Main Menu
21:56:33.410 APP: Boot Count 27
21:56:34.305 CFG: Saved to flash at F4, Count 50, Bytes 4096
21:56:34.325 HTP: Console
21:56:34.431 MQT: tele/hubir/STATE = {"Time":"2022-02-27T21:56:34","Uptime":"0T00:00:11","UptimeSec":11,"Vcc":3.497,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"HouseTAM","BSSId":"1C:3B:F3:80:68:EE","Channel":4,"Mode":"11n","RSSI":100,"Signal":-31,"LinkCount":1,"Downtime":"0T00:00:05"}}
21:56:39.420 MQT: tasmota/discovery/2CF432842CF0/config = {"ip":"6667","dn":"Hub IR","fn":["Hub IR",null,null,null,null,null,null,null],"hn":"hubir-3312","mac":"2CF43CF0","md":"BW-RC1","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"11.0.0.3","t":"hubir","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[0,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":0,"lt_st":0,"sho":[0,0,0,0],"ver":1} (retained)
21:56:39.424 MQT: tasmota/discovery/2CF432842CF0/sensors = {"sn":{"Time":"2022-02-27T21:56:39"},"ver":1} (retained)
21:56:48.500 WIF: Checking connection...
21:56:49.350 SRC: MQTT
21:56:49.353 CMD: Grp 0, Cmd 'IRSEND', Idx 1, Len 48, Pld -99, Data '{"Protocol":"NEC","Bits":32,"Data":"0x20DF10EF"}'
21:56:49.473 MQT: stat/hubir/RESULT = {"IRSend":"Done"}
21:57:08.500 WIF: Checking connection...
21:57:25.410 WIF: Sending Gratuitous ARP

barbudor, 2022-02-27:
```haskel
21:56:49.350 SRC: MQTT
21:56:49.353 CMD: Grp 0, Cmd 'IRSEND', Idx 1, Len 48, Pld -99, Data '{"Protocol":"NEC","Bits":32,"Data":"0x20DF10EF"}'
21:56:49.473 MQT: stat/hubir/RESULT = {"IRSend":"Done"}
``` 

Clear like crsytal : something is sending the IRSEND command to your Tasmota which just executed the order.
Now, you have to find what.
As I said earlier, may be you have this message retained in your broker.

jovanotti25, 2022-02-27:
i have send ir in node red , put retain false? 
i put this to make a entity to turn on and off my tv 
[image]

barbudor, 2022-02-27:
` Retain: True` is your error
This makes the command persistant and Tasmota will re-execute it every time it restarts
Set it to ` Retain: false`

barbudor, 2022-02-27:
Please don't forget to close the issue

jovanotti25, 2022-02-27:
I set false but dont resolve make again and send comand :(

barbudor, 2022-02-27:
You first need to clear the retain flag from your broker
Publish a IR command without retained to clear it
Or restart your broker

jovanotti25, 2022-02-27:
Its solved Thanks a lot for your Help

closed 2022-02-27 by jovanotti25

sfromis, 2022-02-27:
To clear retained, you can publish a message, same topic, **with** retain flag but empty payload.