arendst/Tasmota #15009: WebTime breaks >W in scripting

nikito7, 2022-02-28:
### PROBLEM DESCRIPTION

WebTime breaks >W in scripting

I disabled code because main page dont load

```js
;>W
;
;@<b>NTP: </b> %date% %time%
;@<b>Vars: </b> cnt=%0cnt% tper=%0tper% smlj=%0smlj%
;@<b>Vars: </b> wtd=%0wtd% clk=%0clk% old=%0old%
;@<hr>
```

Even a basic code dont work

I suppose that is new Time line:

[image]

Current dev 2022-02-28 23h

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2022-03-02:
Could not reproduce this behavior.
please post a minimal complete script that shows the problem

nikito7, 2022-03-02:
maybe it's sml related

with wget show connection stalled

Any sub pages work

```js
>D

time=""
date=""
cnt=0
clk=0
wtd=0
old=0

>BS

tper=60
smlj=0

=>Sensor53 r

>S

if cnt==30
then
smlj=1
endif

;time=st(tstamp T 2)
;date=st(tstamp T 1)

if cnt<99
then
cnt+=1
endif

; modbus watchdog

clk=sml[3]

if cnt==99
then
wtd+=1
endif

if wtd==1
then
old=sml[3]
endif

if wtd==50
then
wtd=0
if old==clk
then
print modbus error
; 
=>Restart -3
; 
endif
endif

;>W
;
;@<b>NTP: </b> %date% %time%
;@<b>Vars: </b> cnt=%0cnt% tper=%0tper% smlj=%0smlj%
;@<b>Vars: </b> wtd=%0wtd% clk=%0clk% old=%0old%
;@<hr>

; Below registers only work in EB3 !

>M 1

; change to your gpios and mode
; power off is required
;  v  v             v
+1,3,mN1,1,9600,EB3,1,15,r010400010001,r0104006C0007,r010400730007,r0104007A0006,r010400260003,r010400160006,r0104000B0004

1,=h<hr>

; 01

1,01040Cxxxxxxxxxxuu@i0:1,Clock ,h,Clock_H,16
1,01040Cxxxxxxxxxxxxuu@i0:1,Clock ,m,Clock_M,16
1,01040Cxxxxxxxxxxxxxxuu@i0:1,Clock ,s,Clock_S,16

1,=h<hr>

; 6C

1,01040EUUuu@i1:10,Voltage L1 ,V,VolL1,17
1,01040ExxxxxxxxUUuu@i1:10,Voltage L2 ,V,VolL2,17
1,01040ExxxxxxxxxxxxxxxxUUuu@i1:10,Voltage L3 ,V,VolL3,17

1,=h<hr>

1,01040ExxxxxxxxxxxxxxxxxxxxxxxxUUuu@i1:10,Current ,A,Current,17

1,01040ExxxxUUuu@i1:10,Current L1 ,A,CurL1,17
1,01040ExxxxxxxxxxxxUUuu@i1:10,Current L2 ,A,CurL2,17
1,01040ExxxxxxxxxxxxxxxxxxxxUUuu@i1:10,Current L3 ,A,CurL3,17

1,=h<hr>

; 73

1,01041cx8x8x8UUuuUUuu@i2:1,Active Power ,W,ActPow,16

1,01041cUUuuUUuu@i2:1,Active Power L1 ,W,ActPowL1,16
1,01041cx8UUuuUUuu@i2:1,Active Power L2 ,W,ActPowL2,16
1,01041cx8x8UUuuUUuu@i2:1,Active Power L3 ,W,ActPowL3,16

1,=h<hr>

; 7A

1,01040eUUuuUUuu@i3:1,Active Power Export,W,ActPowExp,16

1,01040exxxxxxxxUUuu@i3:1000,Power Factor ,pu,PowFac,19
1,01040exxxxxxxxxxxxUUuu@i3:1000,Power Factor L1 ,pu,PowFacL1,19
1,01040exxxxxxxxxxxxxxxxUUuu@i3:1000,Power Factor L2 ,pu,PowFacL2,19
1,01040exxxxxxxxxxxxxxxxxxxxUUuu@i3:1000,Power Factor L3 ,pu,PowFacL3,19
1,01040exxxxxxxxxxxxxxxxxxxxxxxxUUuu@i3:10,Frequency ,Hz,Frequency,1

1,=h<hr>

; 26

1,01040CUUuuUUuu@i4:1000,Energy T1 Vazio ,kWh,TotEneT1,1
1,01040CxxxxxxxxUUuuUUuu@i4:1000,Energy T2 Ponta ,kWh,TotEneT2,1
1,01040CxxxxxxxxxxxxxxxxUUuuUUuu@i4:1000,Energy T3 Cheias ,kWh,TotEneT3,1

1,=h<hr>

; 16

1,010418UUuuUUuu@i5:1000,Energy Import ,kWh,TotEneImp,17
1,010418xxxxxxxxUUuuUUuu@i5:1000,Energy Export ,kWh,TotEneExp,19

1,=h<hr>

; 0B

1,01040euu@i6:1,Tariff ,,Tariff,16

#

; eof

```

nikito7, 2022-03-02:
Found the issue.

Comment line in >W reboot tasmota.

[image]

gemu2015, 2022-03-02:
comments are not supported in >W, however i did not expect a reboot.
strange, will investigate

gemu2015, 2022-03-02:
fixed, please try

[xdrv_10_scripter.ino.zip](https://github.com/arendst/Tasmota/files/8172361/xdrv_10_scripter.ino.zip)

nikito7, 2022-03-02:
> fixed, please try
> 
> [xdrv_10_scripter.ino.zip](https://github.com/arendst/Tasmota/files/8172361/xdrv_10_scripter.ino.zip)

It work

```js
>W
; fjdjdj
@<b>NTP: </b> %date% %time%
@<b>Vars: </b> cnt=%0cnt% tper=%0tper% smlj=%0smlj%
; sjdjsjs

@<b>Vars: </b> wtd=%0wtd% clk=%0clk% old=%0old%
@<hr>

; dkdjdja
```