arendst/Tasmota #15018: Upgrade from 9.5 to 11 - failed. Not enough space

tb205gti, 2022-03-03:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Ran 9.5 tasmota.bin - updated to 11.0 minimal using file upload in the UI. The Tasmota does not boot properly and ends up in AP mode. I connect to the AP and try the update to the full tasmota.bin.gz via curl (curl -F "u2=@./tasmota-minimal.bin.gz" http://192.168.4.1/u2)

But it fails with Not enough space error.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Sonoff Basic
- [ x] Tasmota binary firmware version number used: 9.5 + 11.0
  - [ x] Pre-compiled
  - [ ] Self-compiled
- [ x] Flashing tools used: web UI + curl
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Update from 9.5 to 11.0 using the webUI and fileupload

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Update successful, starting minimal UI so I can update to the full version.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-03-03:
Your messge is not consistant:
> try the update to the full **tasmota.bin.gz** via curl (curl -F "u2=@./**tasmota-minimal.bin.gz**" http://192.168.4.1/u2)
tasmota or tasmota-minimal ?

Which binary is currently flashed on your device ?
If it is in AP mode, you can still access the information page through http://192.168.1.4/in  or  the update page http://192.168.1.4/up
Avoid using tasmota-minimal during the upgrade, prefereably use `tasmota-lite.bin.gz`

tb205gti, 2022-03-03:
ok, up and running again - it would not update via curl (using the right gzipped tasmota.bin) - but using the webinterface on /up worked and it has the same config as before.

Strange that it would not connect to wifi in minimal, normally I have no issues, and strange that it would not update via curl - giving not 
enough space error.