arendst/Tasmota #15019: Alexa says device is not working properly when using orange or gold as color for rgb strip.

Kugelstift, 2022-03-03:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): ___esp8266__
- [x ] Tasmota binary firmware version number used: ___11.0.0__
  - [ X] Pre-compiled
  - [ ] Self-compiled
- [ X] Flashing tools used: ___webflash__
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Ask alexa to turn "devicename" orange.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Turn strip orange and alexa says ok.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Happens only when using orange , gold etc. 
Alexa via philips emu. Device is working fine but alexa says its not.

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2022-03-03:
Please fill up the required information. Also set `weblog 4` and paste here the logs when the color change fails

Kugelstift, 2022-03-03:
Added everything you should need.

Thank you :)

s-hadinger, 2022-03-03:
This will be fixed in a couple of days. It's an unwanted regression when introducing Berry Alexa support. Stay tuned

s-hadinger, 2022-03-05:
This should be fixed now. I tested with all colors in Alexa app.

Thanks for reporting the issue.

closed 2022-03-05 by s-hadinger

Kugelstift, 2022-03-06:
Thanks for your fast help !

PR #15038 (2022-03-05): Fix Hue emulation for Alexa
## Description:

Uses the new `_hue16` variable to store the high resolution (16 bits) hue set by Alexa and according to Philips Hue API.

**Related issue (if applicable):** fixes #15019 

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with Tasmota core ESP32 V.2.0.2.3
  - [x] [... cut]