arendst/Tasmota #15026: compilation esp32-cam failed when activating home assistant

Enjoy-Combi, 2022-03-03:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Compilation failed when selecting home assistant / esp32-cam

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ X] Device used (e.g., Sonoff Basic): AI THINKER ESP 32 CAM
- [ X] Tasmota binary firmware version number used: 11.0.0
  - [ ] Pre-compiled
  - [ X] Self-compiled
- [ X] Flashing tools used: https://gitpod.io/#https://github.com/benzino77/tasmocompiler
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Use https://gitpod.io/#https://github.com/benzino77/tasmocompiler; select ESP32 CAM & home assistant & version 11.0.0 or development

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Compilation successul

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Here is the error : 
Linking .pio/build/firmware/firmware.elf
/home/gitpod/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /tmp/ccaS7PBT.ltrans13.ltrans.o:(.literal+0x274): undefined reference to `HAssDiscover()'

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2022-03-03:
As designed.

Enjoy-Combi, 2022-03-03:
You mean this is intended. It works without, MQTT is OK

Jason2866, 2022-03-03:
For what reason should Home Assistant be activated?

Enjoy-Combi, 2022-03-03:
I was thinking it was easier to get the data from the BME280 I have attached to the esp32-cam. Finally not needed.

closed 2022-03-03 by Jason2866