arendst/Tasmota #15041: telemetry period to heigh (10)

e-i-k-e, 2022-03-05:
Hello, I have on a Sonoff Pow R2 Tasmota 9.2.0.
This communicates with the ioBroker and for a program I need a telemetry period of <10 (e.g. 5). I didn't manage to configure this value below 10. 
Is there any way to implement this?

closed 2022-03-05 by close-issue-app[bot]

barbudor, 2022-03-06:
Min teleperiod is 10 sec
You need to modify Tasmota code for that
Teleperiod as been limites to 10 to guarantee proper behavior in every case.
Reducing it below may work for you (we don't know your use case) or not. Up to you to discover that.

curzon01, 2022-03-06:
Alternatively without modify Tasmota:
The cmnd `Teleperiod` without an argument triggers the same as teleperiod does each period.
Define a rule with a ruletimer that exec a `Teleperiod` cmnd and triggers itself and one at system startup (the period here is stored in Mem1)

Set period to 2 seconds
```haskell
Mem1 2
```

Define the rule
```haskell
Rule1
ON System#Boot DO Ruletimer1 %Mem1% ENDON
ON Rules#Timer=1 DO Backlog Teleperiod;Ruletimer1 %Mem1% ENDON
```

activate the whole thing
```haskell
Backlog Rule1 4;Rule1 1
```

To change the period simply change Mem1

e-i-k-e, 2022-03-06:
> Alternatively without modify Tasmota: The cmnd `Teleperiod` without an argument triggers the same as teleperiod does each period. Define a rule with a ruletimer that exec a `Teleperiod` cmnd and triggers itself and one at system startup (the period here is stored in Mem1)
> 
> Set period to 5 seconds
> 
> ```haskell
> Mem1 5
> ```
> 
> Define the rule
> 
> ```haskell
> Rule1
> ON System#Boot DO Ruletimer1 %Mem1% ENDON
> ON Rules#Timer=1 DO Backlog Teleperiod;Ruletimer1 %Mem1% ENDON
> ```
> 
> activate the whole thing
> 
> ```haskell
> Backlog Rule1 4;Rule1 1;Ruletimer1 5
> ```
> 
> To change the period simply change Mem1

Thank you for your support!

Since I had never had anything to do with rules, I took over your orders directly.
And it seems to be working! :)

`20:01:59 RUL: RULES#TIMER=1 performs "Backlog Teleperiod;Ruletimer1 6"
20:01:59 MQT: stat/sonoff_powr2/RESULT = {"TelePeriod":10}
20:01:59 MQT: stat/sonoff_powr2/RESULT = {"T1":6,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}`

Does this rule remain permanently active, even after a reboot or do I still have to use something like "savedata 0/1"?

Can it be that no much faster transmissions are possible via WLAN?
At 5sec. I had no regular transmissions.
I'm just testing 6 sec. and then actually get this at about 7 sec. in my database.

curzon01, 2022-03-06:
Rule and Mem stored permanently, no extra save cmnd necessary.

Limit bandwidth might be related to bad WiFi strength and/or slow MQTT server Connection/system