arendst/Tasmota #15070: console log jumps from end to some higher position every 1-2 seconds

joba-1, 2022-03-08:
### PROBLEM DESCRIPTION

By accident I have a web console window open for a long time. It has accumulated >9000 lines.
If I scroll down manually, it always jumps back to a higher position after 1-2s.
Browser is recent firefox.

### REQUESTED INFORMATION

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Sonoff Dual R3
- [ ] Tasmota binary firmware version number used: 11.0.0.3
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: ota
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
18:53:37.713 CMD: Backlog Template; Module; GPIO 255
18:53:37.786 MQT: stat/dualr3/RESULT = {"NAME":"Sonoff Dual R3","GPIO":[32,0,0,0,0,0,0,0,0,576,225,0,0,0,0,0,0,0,0,0,0,7296,7328,224,0,0,0,0,160,161,0,0,0,0,0,0],"FLAG":0,"BASE":1}
18:53:38.024 MQT: stat/dualr3/RESULT = {"Module":{"0":"Sonoff Dual R3"}}
18:53:38.290 MQT: stat/dualr3/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"225":"Relay2"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"7296":"CSE7761 Tx"},"GPIO26":{"7328":"CSE7761 Rx"},"GPIO27":{"224":"Relay1"},"GPIO32":{"160":"Switch1"},"GPIO33":{"161":"Switch2"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
18:53:39.579 MQT: tele/dualr3/STATE = {"Time":"2022-03-08T18:53:39","Uptime":"4T07:17:58","UptimeSec":371878,"Heap":148,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":25,"MqttCount":14,"Berry":{"HeapUsed":3,"Objects":43},"POWER1":"ON","POWER2":"ON","Wifi":{"AP":2,"SSId":"job.fritz.ssid","BSSId":"DC:15:C8:CD:99:F4","Channel":11,"Mode":"11n","RSSI":64,"Signal":-68,"LinkCount":14,"Downtime":"0T00:11:20"}}
18:53:39.607 MQT: tele/dualr3/SENSOR = {"Time":"2022-03-08T18:53:39","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2022-02-02T16:01:12","Total":14.83421,"Yesterday":0.00000,"Today":0.00000,"Period":[0.000,0.000],"Power":[0.000,0.000],"ApparentPower":[0.000,0.000],"ReactivePower":[0.000,0.000],"Factor":[0.00,0.00],"Voltage":0.0,"Current":[0.000,0.000]},"ESP32":{"Temperature":45.6},"TempUnit":"C"}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
18:54:07.832 CMD: status 0
18:54:07.851 MQT: stat/dualr3/STATUS = {"Status":{"Module":0,"DeviceName":"DualR3","FriendlyName":["Gefriertruhe","PowerMoni"],"Topic":"dualr3","ButtonTopic":"0","Power":3,"PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
18:54:07.864 MQT: stat/dualr3/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"4T07:18:26","StartupUTC":"2022-03-04T10:35:41","Sleep":50,"CfgHolder":4617,"BootCount":17,"BCResetTime":"2022-02-02T16:01:12","SaveCount":345}}
18:54:07.881 MQT: stat/dualr3/STATUS2 = {"StatusFWR":{"Version":"11.0.0.3(tasmota)","BuildDateTime":"2022-02-28T16:46:54","Core":"2_0_2_3","SDK":"v4.4-2.0.2-420-g4ff9ceb19a","CpuFrequency":80,"Hardware":"ESP32-D0WD-V3","CR":"432/699"}}
18:54:07.892 MQT: stat/dualr3/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":3,"LogHost":"job4","LogPort":514,"SSId":["job.iot.ssid","job.fritz.ssid"],"TelePeriod":60,"Resolution":"569B98C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004002"]}}
18:54:07.924 MQT: stat/dualr3/STATUS4 = {"StatusMEM":{"ProgramSize":1383,"Free":1856,"Heap":142,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","9FDAC7CF","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","400840D2","20200000","0034482D"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,82,86","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,127"}}
18:54:07.944 MQT: stat/dualr3/STATUS5 = {"StatusNET":{"Hostname":"dualr3","IPAddress":"192.168.1.165","Gateway":"192.168.1.221","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.236","DNSServer2":"0.0.0.0","Mac":"94:B9:7E:68:2B:78","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"0.0.0.0","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00"},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
18:54:07.958 MQT: stat/dualr3/STATUS6 = {"StatusMQT":{"MqttHost":"job4","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_682B78","MqttUser":"DVES_USER","MqttCount":14,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:54:07.977 MQT: stat/dualr3/STATUS7 = {"StatusTIM":{"UTC":"2022-03-08T17:54:07","Local":"2022-03-08T18:54:07","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"07:17","Sunset":"18:43"}}
18:54:07.989 MQT: stat/dualr3/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
18:54:08.012 MQT: stat/dualr3/STATUS10 = {"StatusSNS":{"Time":"2022-03-08T18:54:07","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2022-02-02T16:01:12","Total":14.83421,"Yesterday":0.00000,"Today":0.00000,"Power":[0.000,0.000],"ApparentPower":[0.000,0.000],"ReactivePower":[0.000,0.000],"Factor":[0.00,0.00],"Voltage":0.0,"Current":[0.000,0.000]},"ESP32":{"Temperature":46.1},"TempUnit":"C"}}
18:54:08.030 MQT: stat/dualr3/STATUS11 = {"StatusSTS":{"Time":"2022-03-08T18:54:08","Uptime":"4T07:18:27","UptimeSec":371907,"Heap":144,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":14,"Berry":{"HeapUsed":3,"Objects":43},"POWER1":"ON","POWER2":"ON","Wifi":{"AP":2,"SSId":"job.fritz.ssid","BSSId":"DC:15:C8:CD:99:F4","Channel":11,"Mode":"11n","RSSI":66,"Signal":-67,"LinkCount":14,"Downtime":"0T00:11:20"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
see pic below

```
### TO REPRODUCE
_Steps to reproduce the behavior:_
wait long enough with web console open

### EXPECTED BEHAVIOUR
view port does not jump by itself

### SCREENSHOTS
[image]

### ADDITIONAL CONTEXT
not the most important thing in the world, but hey...

ascillato, 2022-03-08:
Hi, I can't reproduce it Chrome. Can you test it in Chrome as seems to be an issue of Firefox itself?

sfromis, 2022-03-08:
I've seen something similar with Chrome, but only after waiting a long time with the browser log view not scrolled to the bottom.

Jason2866, 2022-03-08:
Yes, jumps back with a big log history. Using Chrome with tablet.

pkkrusty, 2022-03-08:
Have had this happen too.

arendst, 2022-03-09:
You did it again reaching an edge case! I never anticipated a german would leave his console open for so long ;-)

In file ``HTTP_SCRIPT_CONSOLE.h`` change ``scrollTop`` from 99999 to 9999999 and see what happens.

arendst, 2022-03-09:
I'll fix it now as it needs to be changed for Berry too. Hold on.

joba-1, 2022-03-09:
> You did it again reaching an edge case! I never anticipated a german would leave his console open for so long ;-)

Ok, now I would like to hear the story of why germans are not supposed to do that. I feel discriminated :D

arendst, 2022-03-09:
Kidding.

Pls test and report back in a few days ;-)

ascillato2, 2022-03-11:
Closing as this issue has been fixed. Thanks.

closed 2022-03-11 by ascillato2

joba-1, 2022-03-16:
no more jumps - all fine