arendst/Tasmota #15075: ESP32: Can not Reset to Factory (Device Recovery)

UBWH, 2022-03-09:
### PROBLEM DESCRIPTION
The 2 hardware Device Recovery methods here (https://tasmota.github.io/docs/Device-Recovery/) do not seem to work with this (pretty generic) ESP32 Development board:
https://templates.blakadder.com/lilygo_T-Internet-POE.html

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): [lilygo_T-Internet-POE.html](https://templates.blakadder.com/lilygo_T-Internet-POE.html)
- [x] Tasmota binary firmware version number used: http://ota.tasmota.com/tasmota32/tasmota32.bin  (Latest dev. 11.0.0.3)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA Upgrade
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
Backlog Template; Module; GPIO 255
01:37:13.297 RSL: RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
01:37:13.531 RSL: RESULT = {"Module":{"1":"ESP32-DevKit"}}
01:37:13.797 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
No Rules
```
- [x] Provide the output of this command: `Status 0`:
```lua
01:39:43.366 CMD: status 0
01:39:43.375 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_54B470","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
01:39:43.413 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:09:01","StartupUTC":"2022-03-09T00:30:42","Sleep":50,"CfgHolder":4617,"BootCount":13,"BCResetTime":"2022-03-09T01:12:21","SaveCount":21}}
01:39:43.440 RSL: STATUS2 = {"StatusFWR":{"Version":"11.0.0.3(tasmota)","BuildDateTime":"2022-03-08T22:21:06","Core":"2_0_2_3","SDK":"v4.4-2.0.2-476-gbdacda661c","CpuFrequency":80,"Hardware":"ESP32-D0WDQ6","CR":"386/699"}}
01:39:43.467 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["UniFi",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A190000000000","00000080","00006000","00004000"]}}
01:39:43.517 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1385,"Free":1856,"Heap":152,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","9FDAC7CF","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","400840D2","20200000","0034482D"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,82,86","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,127"}}
01:39:43.567 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-54B470-5232","IPAddress":"10.1.1.20","Gateway":"10.1.1.1","Subnetmask":"255.255.255.0","DNSServer1":"10.1.1.1","DNSServer2":"0.0.0.0","Mac":"EC:94:CB:54:B4:70","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"0.0.0.0","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00"},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
01:39:43.604 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_54B470","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
01:39:43.637 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-03-09T00:39:43","Local":"2022-03-09T01:39:43","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"07:15","Sunset":"18:45"}}
01:39:43.666 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-03-09T01:39:43","ESP32":{"Temperature":53.3},"TempUnit":"C"}}
01:39:43.679 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-03-09T01:39:43","Uptime":"0T00:09:01","UptimeSec":541,"Heap":154,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":43},"Wifi":{"AP":1,"SSId":"UniFi","BSSId":"80:2A:A8:D7:00:AC","Channel":6,"Mode":"11n","RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  n/a
```

### TO REPRODUCE
Load V10.0.3
Reset to defaults usung GUI (works fine)
Connect to AP and enter SSID/Password of AP1
Device reboots, connects to WiFi. 
Can open web page and view normal Tasmota GUI.
Note: Even though the hardware I am using has a LAN interface ... it is NOT configured in the template, and not being used.

RECOVERY METHOD 1
* 7-power cycle method. 6 times power on 5 sec. 7th time leave power on.
```
00:00:00.003 HDW: ESP32-D0WDQ6 
00:00:00.089 UFS: FlashFS mounted with 308 kB free
00:00:00.100 CFG: Loaded from NVS, Count 25
00:00:00.104 FRC: Some settings have been reset (2)                    <--------------  is the (2) correct??
00:00:00.382 BRY: Berry initialized, RAM used=4153 bytes
00:00:00.416 Project tasmota - Tasmota Version 11.0.0.3(tasmota)-2_0_2_3(2022-03-08T22:21:06)
```

RECOVERY METHOD 2
* Press Button 40+ sec
The BOT button on this board shorts GPIO00 to GND. This was pressed for 40+ sec, then device power cycled
Device reconnected to AP1 - meaning the settings were not reset to 'factory'

### EXPECTED BEHAVIOUR
Settings erased and device reboot into Initial Config (WiFi - STA) mode.

### SCREENSHOTS
N/A

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2022-03-09:
Power Recovery is not working with all devices. It depends on the hardware design. 
For Button reset in Tasmota the button needs to be defined

UBWH, 2022-03-09:
Jason thanks.

1) Button to be defined: Thanks. Will read-up on how that is done. I assumed it was GPIO00.

2) 7 power cycle reset:
I have looked at the code, and added some AddLog() messages to the console on boot up.

in file: settings.ino
function: UpdateQuickPowerCycle()

```
#ifdef ESP32
....
 AddLog(LOG_LEVEL_INFO, PSTR("ROB QPC: Counter %u QPC_COUNT:%u"),counter,QPC_COUNT);
...
#endif
```

If I:
* powercycle 4 times, I see ROB QPC: Counter 4 QPC_COUNT:7
* powercycle 5 times, I see ROB QPC: Counter 5 QPC_COUNT:7
* powercycle 6 times, It kinda resets and all I see of interest is: FRC: Some settings have been reset (2) 

So:
* Tasmota32 is correctly counting the fast powercycles 
* But on the 7th power up - for some reason - this code is not executing???
```
 if (QPC_COUNT == (counter)) {  // 7 power cycles in a row
      SettingsErase(3);          // Quickly reset all settings including QuickPowerCycle flag
      EspRestart();
...
```