arendst/Tasmota #15084: Power status after power cycle is wrong

joba-1, 2022-03-09:
### PROBLEM DESCRIPTION
If I power cycle my dual r3 with both channels ON the web interface and logs show both channels OFF afterwards.
Also power readings (voltage, etc.) are zero
But in reality they are ON, as they should always be according my settings.
If I switch power to ON on the web console, text toggles from OFF to ON, power readings are displayed and real power stays ON.

### REQUESTED INFORMATION

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [*] Device used (e.g., Sonoff Basic): Sonoff Dual R3
- [*] Tasmota binary firmware version number used: 11.0.0.3 from right now
  - [*] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
12:13:40.571 CMD: Backlog Template; Module; GPIO 255
12:13:40.643 MQT: stat/dualr3/RESULT = {"NAME":"Sonoff Dual R3","GPIO":[32,0,0,0,0,0,0,0,0,576,225,0,0,0,0,0,0,0,0,0,0,7296,7328,224,0,0,0,0,160,161,0,0,0,0,0,0],"FLAG":0,"BASE":1}
12:13:40.887 MQT: stat/dualr3/RESULT = {"Module":{"0":"Sonoff Dual R3"}}
12:13:41.115 MQT: stat/dualr3/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"225":"Relay2"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"7296":"CSE7761 Tx"},"GPIO26":{"7328":"CSE7761 Rx"},"GPIO27":{"224":"Relay1"},"GPIO32":{"160":"Switch1"},"GPIO33":{"161":"Switch2"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
12:14:09.747 CMD: Status 0
12:14:09.763 MQT: stat/dualr3/STATUS = {"Status":{"Module":0,"DeviceName":"DualR3","FriendlyName":["Gefriertruhe","PowerMoni"],"Topic":"dualr3","ButtonTopic":"0","Power":0,"PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
12:14:09.807 MQT: stat/dualr3/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:09:01","StartupUTC":"2022-03-09T11:05:08","Sleep":50,"CfgHolder":4617,"BootCount":20,"BCResetTime":"2022-02-02T16:01:12","SaveCount":369}}
12:14:09.849 MQT: stat/dualr3/STATUS2 = {"StatusFWR":{"Version":"11.0.0.3(tasmota)","BuildDateTime":"2022-03-09T09:34:33","Core":"2_0_2_3","SDK":"v4.4-2.0.2-476-gbdacda661c","CpuFrequency":80,"Hardware":"ESP32-D0WD-V3","CR":"432/699"}}
12:14:09.868 MQT: stat/dualr3/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":3,"LogHost":"job4","LogPort":514,"SSId":["job.iot.ssid","job.fritz.ssid"],"TelePeriod":60,"Resolution":"569B98C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004002"]}}
12:14:09.919 MQT: stat/dualr3/STATUS4 = {"StatusMEM":{"ProgramSize":1385,"Free":1856,"Heap":146,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","9FDAC7CF","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","400840D2","20200000","0034482D"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,82,86","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,127"}}
12:14:10.001 MQT: stat/dualr3/STATUS5 = {"StatusNET":{"Hostname":"dualr3","IPAddress":"192.168.1.165","Gateway":"192.168.1.221","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.236","DNSServer2":"0.0.0.0","Mac":"94:B9:7E:68:2B:78","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"0.0.0.0","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00"},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
12:14:10.043 MQT: stat/dualr3/STATUS6 = {"StatusMQT":{"MqttHost":"job4","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_682B78","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
12:14:10.064 MQT: stat/dualr3/STATUS7 = {"StatusTIM":{"UTC":"2022-03-09T11:14:10","Local":"2022-03-09T12:14:10","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"07:15","Sunset":"18:45"}}
12:14:10.085 MQT: stat/dualr3/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
12:14:10.117 MQT: stat/dualr3/STATUS10 = {"StatusSNS":{"Time":"2022-03-09T12:14:10","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2022-02-02T16:01:12","Total":14.83421,"Yesterday":0.00000,"Today":0.00000,"Power":[0.000,0.000],"ApparentPower":[0.000,0.000],"ReactivePower":[0.000,0.000],"Factor":[0.00,0.00],"Voltage":0.0,"Current":[0.000,0.000]},"ESP32":{"Temperature":47.8},"TempUnit":"C"}}
12:14:10.165 MQT: stat/dualr3/STATUS11 = {"StatusSTS":{"Time":"2022-03-09T12:14:10","Uptime":"0T00:09:02","UptimeSec":542,"Heap":144,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":43},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":2,"SSId":"job.fritz.ssid","BSSId":"DC:15:C8:CD:99:F4","Channel":7,"Mode":"11n","RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
12:25:26.671 MQT: tele/dualr3/INFO3 = {"Info3":{"RestartReason":"Vbat power on reset","BootCount":22}}
12:25:26.689 MQT: stat/dualr3/RESULT = {"POWER1":"OFF"}
12:25:26.693 MQT: stat/dualr3/POWER1 = OFF
12:25:26.697 MQT: stat/dualr3/RESULT = {"POWER2":"OFF"}
12:25:26.701 MQT: stat/dualr3/POWER2 = OFF
12:25:26.708 C61: Tx A526
12:25:26.712 TFS: File 'autoexec.bat' not found
12:25:26.714 C61: Rx 234A8E39
12:25:26.716 C61: Tx A523
12:25:26.720 C61: Rx 22F223
12:25:26.725 C61: Tx A524
12:25:26.729 C61: Rx 0020BF57
12:25:26.731 C61: Tx A52C
12:25:26.735 C61: Rx FFFD65903D
12:25:26.737 C61: Tx A525
12:25:26.740 C61: Rx 00041918
12:25:26.742 C61: Tx A52D
12:25:26.746 C61: Rx FFFFA56426
[... 1218 lines cut ...]
12:25:44.972 C61: Tx A52C
12:25:44.976 C61: Rx FFEAC299EA
12:25:44.978 C61: Tx A525
12:25:44.982 C61: Rx 00029B98
12:25:44.986 C61: Tx A52D
12:25:44.989 C61: Rx FFFFE3B09C
12:25:44.994 C61: F8946, U2303912, I32858/0, P1391975/0
12:25:45.151 C61: Tx A526
12:25:45.155 C61: Rx 232C964F
12:25:45.159 C61: Tx A523
12:25:45.163 C61: Rx 22F223
12:25:45.166 C61: Tx A524
12:25:45.170 C61: Rx 00E079DD
12:25:45.173 C61: Tx A52C

```

### TO REPRODUCE
Maybe set power state on boot as always on)

### EXPECTED BEHAVIOUR
Displayed state matches real state

### ADDITIONAL CONTEXT
having a run today...

ascillato2, 2022-03-09:
Hi,

The `PowerOnState` parameter is `1` that is **turn power(s) ON after power up**.
The WebUI is not updated if you powercycle your device. You need to refresh the browser.

joba-1, 2022-03-09:
I (re)started the browser after the device was up and refreshed several times before I believed there is sth strange. The OFF can also be seen in the logs, not just the browser.

ascillato, 2022-03-09:
Do you have any rule or berry script in your device?

ascillato, 2022-03-09:
Do you have weblog 2. Can you do the powercycle but defining weblog to 4? So as to see if there isn't any retained MQTT message.

joba-1, 2022-03-09:
no.

just noticed these two tele messages from log above (power off, switch on).
Don’t know what that means.

```
12:25:33.006 MQT: tele/dualr3/STATE = {"Time":"2022-03-09T12:25:32","Uptime":"0T00:00:13","UptimeSec":13,"Heap":151,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":378,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":43},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":2,"SSId":"job.fritz.ssid","BSSId":"DC:15:C8:CD:99:F4","Channel":7,"Mode":"11n","RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:06"}}
12:25:33.073 MQT: tele/dualr3/SENSOR = {"Time":"2022-03-09T12:25:33","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2022-02-02T16:01:12","Total":14.83421,"Yesterday":0.00000,"Today":0.00000,"Period":[0.000,0.000],"Power":[0.000,0.000],"ApparentPower":[0.000,0.000],"ReactivePower":[0.000,0.000],"Factor":[0.00,0.00],"Voltage":0.0,"Current":[0.000,0.000]},"ESP32":{"Temperature":52.8},"TempUnit":"C"}
```

joba-1, 2022-03-09:
the log is level 4 (reason for the ton of c61 messages)

I filled in status 0, then read I should use level 4 for logs, that‘s why it is different (and to test you of course :) )

joba-1, 2022-03-09:
> SWITCH ON means

ah, understood. Nothing connected though.

ascillato, 2022-03-09:
> ah, understood. Nothing connected though.

Which SWITCHMODE do you have?

ascillato, 2022-03-09:
> ah, understood. Nothing connected though.

So, maybe it needs to set `switchmode1 2` and `switchmode2 2` (inverted follow)

joba-1, 2022-03-09:
```
19:40:41.631 CMD: switchmode
19:40:41.635 SRC: WebConsole from 192.168.1.199
19:40:41.641 CMD: Grp 0, Cmd 'SWITCHMODE', Idx 1, Len 0, Pld -99, Data ''
19:40:41.652 MQT: stat/dualr3/RESULT = {"SwitchMode1":0}

```

ascillato, 2022-03-09:
So, maybe it needs to set `switchmode1 2` and `switchmode2 2` (inverted follow)

ascillato, 2022-03-09:
After that, please, can you do a power cycle and copy the log from the beginning up to:
```haskell
12:25:26.671 MQT: tele/dualr3/INFO3 = {"Info3":{"RestartReason":"Vbat power on reset","BootCount":22}}
12:25:26.689 MQT: stat/dualr3/RESULT = {"POWER1":"OFF"}
12:25:26.693 MQT: stat/dualr3/POWER1 = OFF
12:25:26.697 MQT: stat/dualr3/RESULT = {"POWER2":"OFF"}
12:25:26.701 MQT: stat/dualr3/POWER2 = OFF
```
?

joba-1, 2022-03-09:
ok, I guess log level 2 is fine for this?

ascillato, 2022-03-09:
No, please set weblog to 4

joba-1, 2022-03-09:
ok, until then, just for completeness: just did a normal restart and that did not help. Switch is now also displayed as OFF in the sensor message.

joba-1, 2022-03-09:
Hm, doesnt seem the weblog catches the power cycle:

I filtered out the 3000+ C61 lines...
```
19:50:45.481 CMD: weblog 4
19:50:45.485 SRC: WebConsole from 192.168.1.199
19:50:45.491 CMD: Grp 0, Cmd 'WEBLOG', Idx 1, Len 1, Pld 4, Data '4'
19:50:45.555 MQT: stat/dualr3/RESULT = {"WebLog":4}
19:50:47.895 CFG: Saved, Count 383, Bytes 4096
19:51:10.983 WIF: Checking connection...
19:51:53.613 MQT: tasmota/discovery/94B97E682B78/config = {"ip":"192.168.1.165","dn":"DualR3","fn":["Gefriertruhe","PowerMoni",null,null,null,null,null,null],"hn":"dualr3","mac":"94B97E682B78","md":"Sonoff Dual R3","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"11.0.0.3","t":"dualr3","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[1,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":0,"lt_st":0,"sho":[0,0,0,0],"ver":1} (retained)
19:51:53.664 MQT: tasmota/discovery/94B97E682B78/sensors = {"sn":{"Time":"2022-03-09T19:51:53","Switch1":"OFF","Switch2":"OFF","ENERGY":{"TotalStartTime":"2022-02-02T16:01:12","Total":14.83421,"Yesterday":0.00000,"Today":0.00000,"Power":[0.000,0.000],"ApparentPower":[0.000,0.000],"ReactivePower":[0.000,0.000],"Factor":[0.00,0.00],"Voltage":0.0,"Current":[0.000,0.000]},"ESP32":{"Temperature":56.7},"TempUnit":"C"},"ver":1} (retained)
```

joba-1, 2022-03-09:
P.S. the switchmode 2 actually made the relay follow the power OFF state. The fridge was off when I arrived.

ascillato, 2022-03-09:
> Hm, doesnt seem the weblog catches the power cycle:
> I filtered out the 3000+ C61 lines...

What I want to see is the full log when the device starts, up to when it say that the relay is OFF.

> P.S. the switchmode 2 actually made the relay follow the power OFF state. The fridge was off when I arrived.

If you are not going to use the inputs, it is better that you undefine the SWITCH GPIOs or disable the inputs by `SO114 1`

joba-1, 2022-03-09:
This is what syslog has to say about that power cycle:

```
2022-03-09T19:51:40.413763+01:00 dualr3 ESP-HDW: ESP32-D0WD-V3
2022-03-09T19:51:40.413763+01:00 dualr3 ESP-HDW: FoundPSRAM=0 CanUsePSRAM=0
2022-03-09T19:51:40.413763+01:00 dualr3 ESP-UFS: FlashFS mounted with 304 kB free
2022-03-09T19:51:40.413763+01:00 dualr3 ESP-QPC: Count 1
2022-03-09T19:51:40.413763+01:00 dualr3 ESP-ROT: Mode 1
2022-03-09T19:51:40.413763+01:00 dualr3 ESP-BRY: GC from 1479 to 1479 bytes, objects freed 0/1 (in 1 ms) - slots from 6/46 to 6/46
2022-03-09T19:51:40.413808+01:00 dualr3 ESP-BRY: GC from 3789 to 3589 bytes, objects freed 4/23 (in 0 ms) - slots from 35/61 to 27/61
2022-03-09T19:51:40.413808+01:00 dualr3 ESP-CFG: No '*.autoconf' file found
2022-03-09T19:51:40.422653+01:00 dualr3 ESP-BRY: GC from 5274 to 4153 bytes, objects freed 6/41 (in 1 ms) - slots from 60/76 to 38/76
2022-03-09T19:51:40.422653+01:00 dualr3 ESP-BRY: Berry initialized, RAM used=4153 bytes
2022-03-09T19:51:40.422653+01:00 dualr3 ESP-BRY: No 'preinit.be'
2022-03-09T19:51:40.428544+01:00 dualr3 ESP-SRC: Restart
2022-03-09T19:51:40.428544+01:00 dualr3 ESP-Project tasmota - DualR3 Version 11.0.0.3(tasmota)-2_0_2_3(2022-03-09T09:34:33)
2022-03-09T19:51:40.437234+01:00 dualr3 ESP-ETH: No ETH MDC and/or ETH MDIO GPIO defined
2022-03-09T19:51:40.437234+01:00 dualr3 ESP-BRY: No 'autoexec.be'
2022-03-09T19:51:40.437234+01:00 dualr3 ESP-WIF: Checking connection...
2022-03-09T19:51:40.447835+01:00 dualr3 ESP-WIF: Attempting connection...
2022-03-09T19:51:40.447835+01:00 dualr3 ESP-WIF: Connecting to AP2 job.fritz.ssid Channel 11 BSSId DC:15:C8:CD:99:F4 in mode 11n as dualr3...
2022-03-09T19:51:40.447835+01:00 dualr3 ESP-WIF: Checking connection...
2022-03-09T19:51:40.455802+01:00 dualr3 ESP-WIF: Attempting connection...
2022-03-09T19:51:40.455802+01:00 dualr3 ESP-WIF: Checking connection...
2022-03-09T19:51:40.455802+01:00 dualr3 ESP-WIF: Attempting connection...
2022-03-09T19:51:40.463534+01:00 dualr3 ESP-WIF: Checking connection...
2022-03-09T19:51:40.463534+01:00 dualr3 ESP-WIF: Attempting connection...
2022-03-09T19:51:40.471728+01:00 dualr3 ESP-WIF: Checking connection...
2022-03-09T19:51:40.471728+01:00 dualr3 ESP-WIF: Attempting connection...
2022-03-09T19:51:40.618500+01:00 dualr3 ESP-QPC: Reset
2022-03-09T19:51:40.618500+01:00 dualr3 ESP-WIF: Checking connection...
2022-03-09T19:51:40.618500+01:00 dualr3 ESP-WIF: Connected
2022-03-09T19:51:40.624565+01:00 dualr3 ESP-HTP: Web server active on dualr3 with IP address 192.168.1.165
2022-03-09T19:51:41.447575+01:00 dualr3 ESP-NTP: Sync time...
2022-03-09T19:51:41.493659+01:00 dualr3 ESP-RTC: UTC 2022-03-09T18:51:40, DST 2022-03-27T02:00:00, STD 2022-10-30T03:00:00
2022-03-09T19:51:41.497008+01:00 dualr3 ESP-RTC: Synced by NTP
2022-03-09T19:51:41.500614+01:00 dualr3 ESP-NRG: Energy reset by invalid data
2022-03-09T19:51:41.654625+01:00 dualr3 ESP-APP: Boot Count 24
2022-03-09T19:51:42.431149+01:00 dualr3 ESP-MQT: Attempting connection...
2022-03-09T19:51:42.431149+01:00 dualr3 ESP-MQT: Connected
2022-03-09T19:51:42.433356+01:00 dualr3 ESP-MQT: tele/dualr3/LWT = Online (retained)
2022-03-09T19:51:42.440099+01:00 dualr3 ESP-MQT: cmnd/dualr3/POWER =
2022-03-09T19:51:42.440099+01:00 dualr3 ESP-MQT: Subscribe to cmnd/dualr3/#
2022-03-09T19:51:42.442823+01:00 dualr3 ESP-MQT: Subscribe to cmnd/tasmotas/#
2022-03-09T19:51:42.446489+01:00 dualr3 ESP-MQT: Subscribe to cmnd/DVES_682B78_fb/#
2022-03-09T19:51:42.451425+01:00 dualr3 ESP-MQT: Unsubscribe from homeassistant/status
2022-03-09T19:51:42.519143+01:00 dualr3 ESP-MQT: tele/dualr3/INFO1 = {"Info1":{"Module":"Sonoff Dual R3","Version":"11.0.0.3(tasmota)","FallbackTopic":"cmnd/DVES_682B78_fb/","GroupTopic":"cmnd/tasmotas/"}}
2022-03-09T19:51:42.523400+01:00 dualr3 ESP-MQT: tele/dualr3/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"dualr3","IPAddress":"192.168.1.165"}}
2022-03-09T19:51:42.571942+01:00 dualr3 ESP-MQT: tele/dualr3/INFO3 = {"Info3":{"RestartReason":"Vbat power on reset","BootCount":25}}
2022-03-09T19:51:42.576075+01:00 dualr3 ESP-MQT: stat/dualr3/RESULT = {"POWER1":"OFF"}
2022-03-09T19:51:42.576606+01:00 dualr3 ESP-MQT: stat/dualr3/POWER1 = OFF
2022-03-09T19:51:42.587989+01:00 dualr3 ESP-MQT: stat/dualr3/RESULT = {"POWER2":"OFF"}
2022-03-09T19:51:42.595274+01:00 dualr3 ESP-MQT: stat/dualr3/POWER2 = OFF

```

joba-1, 2022-03-09:
ah - but it is level 2...

ascillato, 2022-03-09:
It is more complete if you can catch the one from the WEBUI console.

joba-1, 2022-03-09:
How do I do that? I mean during power up there is no ip and no webserver? And the tons of C61 messages probably flood the buffers tasmota might have to retain early boot messages. But I'll try again. Maybe I just wasn't lucky enough.

ascillato, 2022-03-09:
Try with weblog 3 instead

joba-1, 2022-03-09:
weblog had no other info. Syslog with level 4 now, so you have sth to read while I go for weblog 3 (2 stories between me and that switch, you realize this is more sports than I had the whole day :D )

```
2022-03-09T20:17:06+01:00 ax1 dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.165 94:b9:7e:68:2b:78 dualr3
2022-03-09T20:17:07.929920+01:00 dualr3 ESP-HDW: ESP32-D0WD-V3
2022-03-09T20:17:07.929920+01:00 dualr3 ESP-HDW: FoundPSRAM=0 CanUsePSRAM=0
2022-03-09T20:17:07.939529+01:00 dualr3 ESP-UFS: FlashFS mounted with 304 kB free
2022-03-09T20:17:07.939529+01:00 dualr3 ESP-QPC: Count 1
2022-03-09T20:17:07.939529+01:00 dualr3 ESP-ROT: Mode 1
2022-03-09T20:17:07.939529+01:00 dualr3 ESP-BRY: GC from 1479 to 1479 bytes, objects freed 0/1 (in 0 ms) - slots from 6/46 to 6/46
2022-03-09T20:17:07.939621+01:00 dualr3 ESP-BRY: GC from 3789 to 3589 bytes, objects freed 4/23 (in 1 ms) - slots from 35/61 to 27/61
2022-03-09T20:17:07.939621+01:00 dualr3 ESP-CFG: No '*.autoconf' file found
2022-03-09T20:17:07.948838+01:00 dualr3 ESP-BRY: GC from 5274 to 4153 bytes, objects freed 6/41 (in 2 ms) - slots from 60/76 to 38/76
2022-03-09T20:17:07.957415+01:00 dualr3 ESP-BRY: Berry initialized, RAM used=4153 bytes
2022-03-09T20:17:07.957415+01:00 dualr3 ESP-BRY: No 'preinit.be'
2022-03-09T20:17:07.957415+01:00 dualr3 ESP-SRC: Restart
2022-03-09T20:17:07.957415+01:00 dualr3 ESP-Project tasmota - DualR3 Version 11.0.0.3(tasmota)-2_0_2_3(2022-03-09T09:34:33)
2022-03-09T20:17:07.957484+01:00 dualr3 ESP-ETH: No ETH MDC and/or ETH MDIO GPIO defined
2022-03-09T20:17:07.966818+01:00 dualr3 ESP-BRY: No 'autoexec.be'
2022-03-09T20:17:07.966818+01:00 dualr3 ESP-WIF: Checking connection...
2022-03-09T20:17:07.967158+01:00 dualr3 ESP-WIF: Attempting connection...
2022-03-09T20:17:07.975494+01:00 dualr3 ESP-WIF: Connecting to AP2 job.fritz.ssid Channel 11 BSSId DC:15:C8:CD:99:F4 in mode 11n as dualr3...
2022-03-09T20:17:07.975494+01:00 dualr3 ESP-WIF: Checking connection...
2022-03-09T20:17:07.975494+01:00 dualr3 ESP-WIF: Attempting connection...
2022-03-09T20:17:07.984507+01:00 dualr3 ESP-WIF: Checking connection...
2022-03-09T20:17:07.984507+01:00 dualr3 ESP-WIF: Attempting connection...
2022-03-09T20:17:07.984507+01:00 dualr3 ESP-WIF: Checking connection...
2022-03-09T20:17:07.993519+01:00 dualr3 ESP-WIF: Attempting connection...
2022-03-09T20:17:07.998439+01:00 dualr3 ESP-WIF: Checking connection...
2022-03-09T20:17:08.005060+01:00 dualr3 ESP-WIF: Attempting connection...
2022-03-09T20:17:08.156760+01:00 dualr3 ESP-WIF: Checking connection...
2022-03-09T20:17:08.156760+01:00 dualr3 ESP-WIF: Connected
2022-03-09T20:17:08.156760+01:00 dualr3 ESP-QPC: Reset
2022-03-09T20:17:08.549468+01:00 dualr3 ESP-NTP: Sync time...
2022-03-09T20:17:08.599134+01:00 dualr3 ESP-RTC: UTC 2022-03-09T19:17:08, DST 2022-03-27T02:00:00, STD 2022-10-30T03:00:00
2022-03-09T20:17:08.599134+01:00 dualr3 ESP-RTC: Synced by NTP
2022-03-09T20:17:08.604974+01:00 dualr3 ESP-HTP: Web server active on dualr3 with IP address 192.168.1.165
2022-03-09T20:17:09.088421+01:00 dualr3 ESP-NRG: Energy reset by invalid data
2022-03-09T20:17:09.292377+01:00 dualr3 ESP-APP: Boot Count 26
2022-03-09T20:17:10.507128+01:00 dualr3 ESP-MQT: Attempting connection...
2022-03-09T20:17:10.510298+01:00 dualr3 ESP-MQT: Connected
2022-03-09T20:17:10.513586+01:00 dualr3 ESP-MQT: tele/dualr3/LWT = Online (retained)
2022-03-09T20:17:10.516994+01:00 dualr3 ESP-MQT: cmnd/dualr3/POWER =
2022-03-09T20:17:10.519325+01:00 dualr3 ESP-MQT: Subscribe to cmnd/dualr3/#
2022-03-09T20:17:10.585308+01:00 dualr3 ESP-MQT: Subscribe to cmnd/tasmotas/#
2022-03-09T20:17:10.595827+01:00 dualr3 ESP-MQT: Subscribe to cmnd/DVES_682B78_fb/#
2022-03-09T20:17:10.602003+01:00 dualr3 ESP-MQT: Unsubscribe from homeassistant/status
2022-03-09T20:17:10.608486+01:00 dualr3 ESP-MQT: tele/dualr3/INFO1 = {"Info1":{"Module":"Sonoff Dual R3","Version":"11.0.0.3(tasmota)","FallbackTopic":"cmnd/DVES_682B78_fb/","GroupTopic":"cmnd/tasmotas/"}}
2022-03-09T20:17:10.613555+01:00 dualr3 ESP-MQT: tele/dualr3/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"dualr3","IPAddress":"192.168.1.165"}}
2022-03-09T20:17:10.614513+01:00 dualr3 ESP-MQT: tele/dualr3/INFO3 = {"Info3":{"RestartReason":"Vbat power on reset","BootCount":27}}
2022-03-09T20:17:10.626033+01:00 dualr3 ESP-MQT: stat/dualr3/RESULT = {"POWER1":"OFF"}
2022-03-09T20:17:10.631382+01:00 dualr3 ESP-MQT: stat/dualr3/POWER1 = OFF
2022-03-09T20:17:10.639266+01:00 dualr3 ESP-MQT: stat/dualr3/RESULT = {"POWER2":"OFF"}
2022-03-09T20:17:10.645841+01:00 dualr3 ESP-MQT: stat/dualr3/POWER2 = OFF
2022-03-09T20:17:10.651176+01:00 dualr3 ESP-TFS: File 'autoexec.bat' not found

```

joba-1, 2022-03-09:
```
00:00:00.002-210/49 HDW: ESP32-D0WD-V3 
00:00:00.003-210/49 HDW: FoundPSRAM=0 CanUsePSRAM=0
00:00:00.077-208/49 UFS: FlashFS mounted with 304 kB free
00:00:00.180 CFG: Loaded from File, Count 391
00:00:00.196 QPC: Count 1
00:00:00.205 ROT: Mode 1
00:00:00.473 CFG: No '*.autoconf' file found
00:00:00.479 BRY: Berry initialized, RAM used=4153 bytes
00:00:00.503 BRY: No 'preinit.be'
00:00:00.507 SRC: Restart
00:00:00.514 Project tasmota - DualR3 Version 11.0.0.3(tasmota)-2_0_2_3(2022-03-09T09:34:33)
00:00:00.515 ETH: No ETH MDC and/or ETH MDIO GPIO defined
00:00:00.733 BRY: No 'autoexec.be'
00:00:00.022 WIF: Attempting connection...
00:00:01.317 WIF: Connecting to AP2 job.fritz.ssid Channel 11 BSSId DC:15:C8:CD:99:F4 in mode 11n as dualr3...
00:00:04.036 WIF: Attempting connection...
00:00:04.985 WIF: Attempting connection...
00:00:05.558 WIF: Attempting connection...
00:00:06.797 QPC: Reset
00:00:06.985 WIF: Attempting connection...
00:00:07.787 NRG: Energy reset by invalid data
00:00:07.985 WIF: Connected
00:00:08.027 C61: CSE7761 found
00:00:08.252 HTP: Web server active on dualr3 with IP address 192.168.1.165
00:00:08.787 APP: Boot Count 27
00:00:08.985 RTC: UTC 2022-03-09T19:28:17, DST 2022-03-27T02:00:00, STD 2022-10-30T03:00:00
20:28:17.000 RTC: Synced by NTP
20:28:18.252 MQT: Attempting connection...
20:28:18.306 MQT: Connected
20:28:18.315 MQT: tele/dualr3/LWT = Online (retained)
20:28:18.322 MQT: cmnd/dualr3/POWER = 
20:28:18.324 MQT: Subscribe to cmnd/dualr3/#
20:28:18.334 MQT: Subscribe to cmnd/tasmotas/#
20:28:18.341 MQT: Subscribe to cmnd/DVES_682B78_fb/#
20:28:18.347 MQT: Unsubscribe from homeassistant/status
20:28:18.365 MQT: tele/dualr3/INFO1 = {"Info1":{"Module":"Sonoff Dual R3","Version":"11.0.0.3(tasmota)","FallbackTopic":"cmnd/DVES_682B78_fb/","GroupTopic":"cmnd/tasmotas/"}}
20:28:18.384 MQT: tele/dualr3/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"dualr3","IPAddress":"192.168.1.165"}}
20:28:18.399 MQT: tele/dualr3/INFO3 = {"Info3":{"RestartReason":"Vbat power on reset","BootCount":28}}
20:28:18.404 MQT: stat/dualr3/RESULT = {"POWER1":"OFF"}
20:28:18.407 MQT: stat/dualr3/POWER1 = OFF
20:28:18.410 MQT: stat/dualr3/RESULT = {"POWER2":"OFF"}
20:28:18.423 MQT: stat/dualr3/POWER2 = OFF
20:28:18.436 TFS: File 'autoexec.bat' not found
20:28:18.740 SRC: MQTT
20:28:18.743 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 0, Pld -99, Data ''
20:28:18.758 MQT: stat/dualr3/STATUS = {"Status":{"Module":0,"DeviceName":"DualR3","FriendlyName":["Gefriertruhe","PowerMoni"],"Topic":"dualr3","ButtonTopic":"0","Power":0,"PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[2,2,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
20:28:19.154 CFG: Saved, Count 392, Bytes 4096
20:28:23.786 MQT: tele/dualr3/STATE = {"Time":"2022-03-09T20:28:23","Uptime":"0T00:00:16","UptimeSec":16,"Heap":151,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":32,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":43},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":2,"SSId":"job.fritz.ssid","BSSId":"DC:15:C8:CD:99:F4","Channel":11,"Mode":"11n","RSSI":60,"Signal":-70,"LinkCount":1,"Downtime":"0T00:00:09"}}
20:28:23.854 MQT: tele/dualr3/SENSOR = {"Time":"2022-03-09T20:28:23","ENERGY":{"TotalStartTime":"2022-02-02T16:01:12","Total":14.83421,"Yesterday":0.00000,"Today":0.00000,"Period":[0.000,0.000],"Power":[0.000,0.000],"ApparentPower":[0.000,0.000],"ReactivePower":[0.000,0.000],"Factor":[0.00,0.00],"Voltage":0.0,"Current":[0.000,0.000]},"ESP32":{"Temperature":53.3},"TempUnit":"C"}
```

ascillato, 2022-03-09:
Please, try `SO114 1` in the console so as to detach the switches from the relays and try again please.

joba-1, 2022-03-09:
I already kicked them out of the config. Is that enough?

```
20:35:57.081 CMD: status 0
20:35:57.084 SRC: WebConsole from 192.168.1.199
20:35:57.087 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
20:35:57.098 MQT: stat/dualr3/STATUS = {"Status":{"Module":0,"DeviceName":"DualR3","FriendlyName":["Gefriertruhe","PowerMoni"],"Topic":"dualr3","ButtonTopic":"0","Power":0,"PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[2,2,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
20:35:57.140 MQT: stat/dualr3/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:07:50","StartupUTC":"2022-03-09T19:28:07","Sleep":50,"CfgHolder":4617,"BootCount":27,"BCResetTime":"2022-02-02T16:01:12","SaveCount":392}}
20:35:57.184 MQT: stat/dualr3/STATUS2 = {"StatusFWR":{"Version":"11.0.0.3(tasmota)","BuildDateTime":"2022-03-09T09:34:33","Core":"2_0_2_3","SDK":"v4.4-2.0.2-476-gbdacda661c","CpuFrequency":80,"Hardware":"ESP32-D0WD-V3","CR":"432/699"}}
20:35:57.207 MQT: stat/dualr3/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":3,"MqttLog":0,"SysLog":2,"LogHost":"job4","LogPort":514,"SSId":["job.iot.ssid","job.fritz.ssid"],"TelePeriod":60,"Resolution":"569B98C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004002"]}}
20:35:57.263 MQT: stat/dualr3/STATUS4 = {"StatusMEM":{"ProgramSize":1385,"Free":1856,"Heap":144,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","9FDAC7CF","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","400840D2","20200000","0034482D"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,82,86","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,127"}}
20:35:57.319 MQT: stat/dualr3/STATUS5 = {"StatusNET":{"Hostname":"dualr3","IPAddress":"192.168.1.165","Gateway":"192.168.1.221","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.236","DNSServer2":"0.0.0.0","Mac":"94:B9:7E:68:2B:78","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"0.0.0.0","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00"},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
20:35:57.362 MQT: stat/dualr3/STATUS6 = {"StatusMQT":{"MqttHost":"job4","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_682B78","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
20:35:57.386 MQT: stat/dualr3/STATUS7 = {"StatusTIM":{"UTC":"2022-03-09T19:35:57","Local":"2022-03-09T20:35:57","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"07:15","Sunset":"18:45"}}
20:35:57.409 MQT: stat/dualr3/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
20:35:57.454 MQT: stat/dualr3/STATUS10 = {"StatusSNS":{"Time":"2022-03-09T20:35:57","ENERGY":{"TotalStartTime":"2022-02-02T16:01:12","Total":14.83421,"Yesterday":0.00000,"Today":0.00000,"Power":[0.000,0.000],"ApparentPower":[0.000,0.000],"ReactivePower":[0.000,0.000],"Factor":[0.00,0.00],"Voltage":0.0,"Current":[0.000,0.000]},"ESP32":{"Temperature":48.9},"TempUnit":"C"}}
20:35:57.503 MQT: stat/dualr3/STATUS11 = {"StatusSTS":{"Time":"2022-03-09T20:35:57","Uptime":"0T00:07:50","UptimeSec":470,"Heap":146,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":43},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":2,"SSId":"job.fritz.ssid","BSSId":"DC:15:C8:CD:99:F4","Channel":11,"Mode":"11n","RSSI":60,"Signal":-70,"LinkCount":1,"Downtime":"0T00:00:09"}}}
```

joba-1, 2022-03-09:
```
20:38:16.386 CMD: Backlog Template; Module; GPIO 255:
20:38:16.389 SRC: WebConsole from 192.168.1.199
20:38:16.392 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 27, Pld -99, Data 'Template; Module; GPIO 255:'
20:38:16.461 SRC: Backlog
20:38:16.465 CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 0, Pld -99, Data ''
20:38:16.480 MQT: stat/dualr3/RESULT = {"NAME":"Sonoff Dual R3","GPIO":[32,0,0,0,0,0,0,0,0,576,225,0,0,0,0,0,0,0,0,0,0,7296,7328,224,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}
20:38:16.704 SRC: Backlog
20:38:16.708 CMD: Grp 0, Cmd 'MODULE', Idx 1, Len 0, Pld -99, Data ''
20:38:16.723 MQT: stat/dualr3/RESULT = {"Module":{"0":"Sonoff Dual R3"}}
20:38:16.953 SRC: Backlog
20:38:16.957 CMD: Grp 0, Cmd 'GPIO', Idx 1, Len 4, Pld 255, Data '255:'
20:38:16.983 MQT: stat/dualr3/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"225":"Relay2"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"7296":"CSE7761 Tx"},"GPIO26":{"7328":"CSE7761 Rx"},"GPIO27":{"224":"Relay1"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
20:38:23.817 MQT: tele/dualr3/STATE = {"Time":"2022-03-09T20:38:23","Uptime":"0T00:10:16","UptimeSec":616,"Heap":149,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":43},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":2,"SSId":"job.fritz.ssid","BSSId":"DC:15:C8:CD:99:F4","Channel":11,"Mode":"11n","RSSI":60,"Signal":-70,"LinkCount":1,"Downtime":"0T00:00:09"}}
20:38:23.855 MQT: tele/dualr3/SENSOR = {"Time":"2022-03-09T20:38:23","ENERGY":{"TotalStartTime":"2022-02-02T16:01:12","Total":14.83421,"Yesterday":0.00000,"Today":0.00000,"Period":[0.000,0.000],"Power":[0.000,0.000],"ApparentPower":[0.000,0.000],"ReactivePower":[0.000,0.000],"Factor":[0.00,0.00],"Voltage":0.0,"Current":[0.000,0.000]},"ESP32":{"Temperature":48.9},"TempUnit":"C"}
```

ascillato, 2022-03-09:
Ok, thanks for all the logs. I think we cover the most common scenarios for that problem. Let me do some tests with an ESP32 and I will come back to you.

joba-1, 2022-03-09:
ok, then here is the final log from the power cycle with additional so114 1
```

00:00:00.003-210/49 HDW: ESP32-D0WD-V3 
00:00:00.004-210/49 HDW: FoundPSRAM=0 CanUsePSRAM=0
00:00:00.017-208/49 UFS: FlashFS mounted with 304 kB free
00:00:00.053 CFG: Loaded from File, Count 393
00:00:00.069 QPC: Count 1
00:00:00.078 ROT: Mode 1
00:00:00.169 CFG: No '*.autoconf' file found
00:00:00.175 BRY: Berry initialized, RAM used=4153 bytes
00:00:00.188 BRY: No 'preinit.be'
00:00:00.192 SRC: Restart
00:00:00.199 Project tasmota - DualR3 Version 11.0.0.3(tasmota)-2_0_2_3(2022-03-09T09:34:33)
00:00:00.200 ETH: No ETH MDC and/or ETH MDIO GPIO defined
00:00:00.265 BRY: No 'autoexec.be'
00:00:00.479 WIF: Attempting connection...
00:00:00.973 WIF: Connecting to AP2 job.fritz.ssid Channel 11 BSSId DC:15:C8:CD:99:F4 in mode 11n as dualr3...
00:00:01.163 WIF: Attempting connection...
00:00:01.113 WIF: Attempting connection...
00:00:03.114 WIF: Attempting connection...
00:00:04.115 WIF: Attempting connection...
00:00:05.116 WIF: Connected
00:00:05.381 HTP: Web server active on dualr3 with IP address 192.168.1.165
00:00:05.482 QPC: Reset
00:00:05.558 C61: CSE7761 found
00:00:06.112 RTC: UTC 2022-03-09T19:42:55, DST 2022-03-27T02:00:00, STD 2022-10-30T03:00:00
20:42:55.001 RTC: Synced by NTP
20:42:55.369 NRG: Energy reset by invalid data
20:42:55.814 MQT: Attempting connection...
20:42:56.007 MQT: Connected
20:42:56.014 MQT: tele/dualr3/LWT = Online (retained)
20:42:56.018 MQT: cmnd/dualr3/POWER = 
20:42:56.020 MQT: Subscribe to cmnd/dualr3/#
20:42:56.026 MQT: Subscribe to cmnd/tasmotas/#
20:42:56.029 MQT: Subscribe to cmnd/DVES_682B78_fb/#
20:42:56.033 MQT: Unsubscribe from homeassistant/status
20:42:56.043 MQT: tele/dualr3/INFO1 = {"Info1":{"Module":"Sonoff Dual R3","Version":"11.0.0.3(tasmota)","FallbackTopic":"cmnd/DVES_682B78_fb/","GroupTopic":"cmnd/tasmotas/"}}
20:42:56.067 MQT: tele/dualr3/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"dualr3","IPAddress":"192.168.1.165"}}
20:42:56.072 MQT: tele/dualr3/INFO3 = {"Info3":{"RestartReason":"Vbat power on reset","BootCount":28}}
20:42:56.088 MQT: stat/dualr3/RESULT = {"POWER1":"OFF"}
20:42:56.091 MQT: stat/dualr3/POWER1 = OFF
20:42:56.094 MQT: stat/dualr3/RESULT = {"POWER2":"OFF"}
20:42:56.097 MQT: stat/dualr3/POWER2 = OFF
20:42:56.104 TFS: File 'autoexec.bat' not found
20:42:56.317 SRC: MQTT
20:42:56.320 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 0, Pld -99, Data ''
20:42:56.350 MQT: stat/dualr3/STATUS = {"Status":{"Module":0,"DeviceName":"DualR3","FriendlyName":["Gefriertruhe","PowerMoni"],"Topic":"dualr3","ButtonTopic":"0","Power":0,"PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[2,2,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
20:42:56.462 APP: Boot Count 28
20:42:57.660 CFG: Saved, Count 394, Bytes 4096
20:43:01.473 MQT: tele/dualr3/STATE = {"Time":"2022-03-09T20:43:01","Uptime":"0T00:00:14","UptimeSec":14,"Heap":150,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":41,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":43},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":2,"SSId":"job.fritz.ssid","BSSId":"DC:15:C8:CD:99:F4","Channel":11,"Mode":"11n","RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:07"}}
20:43:01.525 MQT: tele/dualr3/SENSOR = {"Time":"2022-03-09T20:43:01","ENERGY":{"TotalStartTime":"2022-02-02T16:01:12","Total":14.83421,"Yesterday":0.00000,"Today":0.00000,"Period":[0.000,0.000],"Power":[0.000,0.000],"ApparentPower":[0.000,0.000],"ReactivePower":[0.000,0.000],"Factor":[0.00,0.00],"Voltage":0.0,"Current":[0.000,0.000]},"ESP32":{"Temperature":48.3},"TempUnit":"C"}
```

joba-1, 2022-03-09:
No problem, take your time.
The important thing for me is, that the relay powers are on after a power cycle, and that works fine.

ascillato, 2022-03-09:
Hi,

I couldn't reproduce it.

PowerOnState is made to work ONLY when the device is Powered Up (not in a restart).

I run several tests and `PowerOnState 1` turns all my relays to ON and also shows at boot time the `{"POWER1":"ON"}` message. But again, ONLY at power up, not in a restart.

Please, can you confirm that you see the issue at power up or only at restart?

joba-1, 2022-03-09:
All of the above tests were power cycles, not just restarts. Strange.

But I also did restarts and real power was always on (like I need it) and power in logs and on web ui always off.

Since you cannot reproduce it, I'll try to build my own version with some more logs. But with loow prio...

ascillato, 2022-03-10:
It is very strange. I don't have a Sonoff Dual R3 to test, but I have applied the template in an ESP32 and on power up, it reports the Power ON in the console and it turns the output to ON.

Can you start from scratch? like updating to latest from the development branch and doing a reset 1 in the console?

jbakuwel, 2022-03-11:
For what it's worth ... no V (and presumably also A) readings after turning the relays on on the DualR3 here. Tried with stable and dev branches (currently on 11.0.0.3), some results. Power cycled multiple times.

ascillato, 2022-03-11:
Hi @jbakuwel 

Your issue is different that this one. Please, open a new one with the troubleshooting template fully completed. Thanks.

joba-1, 2022-03-15:
just updated my dualr3 and it works just fine now (except for the c61 0V readings). Thanks!

closed 2022-03-15 by joba-1