arendst/Tasmota #15095: Template Shelly Plus 1 PM incorrect

cybermaus, 2022-03-10:
### PROBLEM DESCRIPTION
Using template for Shelly Plus 1 PM causes the switch input to not drive the relay.
looking at the console putput, it seem GPIO4 is missing from the template.

Manually configuring the Shelly as per [this webpage](https://templates.blakadder.com/shelly_plus_1PM.html) however does make it work correctly: 
`{"NAME":"Shelly Plus 1PM","GPIO":[0,0,0,0,192,2720,0,0,0,0,0,0,0,0,2656,0,0,0,0,2624,0,0,224,0,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}`

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X ] Device used (e.g., Sonoff Basic): Shelly Plus 1 PM
- [X ] Tasmota binary firmware version number used: _____
  - [X ] Pre-compiled : V11.0.0 (downloaded from OTA
  - [X ] Self-compiled : V11.0.3 (compiled from yesterday's clone)
- [X ] Flashing tools used: prebuild with cable, compiled using web-upgrade
- [X ] Provide the output of command: `Backlog Template; Module; GPIO 255`:

Working manual config:
```13:32:35.678 CMD: Backlog Template; Module; GPIO 255
13:32:35.748 MQT: stat/tasmota_79B2B0/RESULT = {"NAME":"Shelly Plus 1PM","GPIO":[0,0,0,0,0,2720,0,0,0,0,0,0,0,0,2656,0,0,0,0,2624,0,160,224,0,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}
13:32:35.986 MQT: stat/tasmota_79B2B0/RESULT = {"Module":{"1":"ESP32-DevKit"}}
13:32:36.246 MQT: stat/tasmota_79B2B0/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"192":"Switch_n1"},"GPIO5":{"2720":"BL0937 CF"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"2656":"HLWBL CF1"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"2624":"HLWBL SEL_i"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"224":"Relay1"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```

Non working autoconfig of module (no responce to switch)
```13:36:08.905 CMD: Backlog Template; Module; GPIO 255
13:36:08.993 MQT: stat/tasmota_79B2B0/RESULT = {"NAME":"Shelly Plus 1PM","GPIO":[0,0,0,0,0,2720,0,0,0,0,0,0,0,0,2656,0,0,0,0,2624,0,160,224,0,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}
13:36:09.195 MQT: stat/tasmota_79B2B0/RESULT = {"Module":{"0":"Shelly Plus 1PM"}}
13:36:09.464 MQT: stat/tasmota_79B2B0/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"2720":"BL0937 CF"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"2656":"HLWBL CF1"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"2624":"HLWBL SEL_i"},"GPIO24":{"0":"None"},"GPIO25":{"160":"Switch1"},"GPIO26":{"224":"Relay1"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```

```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Program a Shelly Plus 1 PM with autoconfig of template, and try to use the switch

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Depending on the switchmode, it should drive the relay or send MQTT message (switchmode 15)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I guess I will work with manual template for now, but it is a bit of a pity.
pending inclusion into the product, some hint as to which file to edit to fix the template in my self-compiled build would be nice.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2022-03-10:
It seems you have not activated the Template in case one.

Jason2866, 2022-03-10:
I deleted my comment because i realized it was wrong.

sfromis, 2022-03-10:
Checked a Shelly Plus 1PM configured with the template, and as expected the gpios include: `"GPIO4":{"192":"Switch_n1"}`

The template content you posted in both working and non-working config is bad:
`{"NAME":"Shelly Plus 1PM","GPIO":[0,0,0,0,0,2720,0,0,0,0,0,0,0,0,2656,0,0,0,0,2624,0,160,224,0,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}`
Missing gpio4. Looks like you've broken the template. Try reconfiguring with the correct template from:
https://templates.blakadder.com/shelly_plus_1PM.html

cybermaus, 2022-03-11:
Thanks

if you check the link I provided in the report, you'll find I mention the exact same page template from @blakadder as you suggest, and I confirm that one is working as intended.

But it is not working if I simply install the precompiled OTA, select auto-configure.
It does find he correct device, and applies *most* of said template, but not the GPIO4.

Also, you say "Missing gpio4"
However, note that while indeed the 1st line is missing CPIO4 but the 3rd line does show the GPIO4.
And both lines are provided by the requested debug info: `Backlog Template; Module; GPIO 255`
```
13:36:08.993 MQT: stat/tasmota_79B2B0/RESULT = {"NAME":"Shelly Plus 1PM","GPIO":[0,0,0,0,0,2720,0,0,0,0,0,0,0,0,2656,0,0,0,0,2624,0,160,224,0,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}
13:36:09.195 MQT: stat/tasmota_79B2B0/RESULT = {"Module":{"0":"Shelly Plus 1PM"}}
13:32:36.246 MQT: stat/tasmota_79B2B0/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"192":"Switch_n1"},"GPIO5":{"2720":"BL0937 CF"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"2656":"HLWBL CF1"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"2624":"HLWBL SEL_i"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"224":"Relay1"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```

You suggest "Looks like you've broken the template"
Ok, but then educate me, how can I even break it. All I did/do is a full precompiled OTA flash, full reset, select "auto-config" module, and confirm the correctly found "Shelly Plus 1 PM" So at that point, I am not yet trying to do any manual setting.

As mentioned, I am able to get it to work, by using blakadders config manually, so I am not stuck or anything.
But should it not simply be working from auto-config?

Jason2866, 2022-03-11:
Fixed. Thx @sfromis

closed 2022-03-11 by Jason2866

cybermaus, 2022-03-11:
works now, thanks