arendst/Tasmota #15134: VEML60 reporting wrong value

ThomasT500, 2022-03-14:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
VEML60 reporting wrong UV Power value 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): _____
- [x ] Tasmota binary firmware version number used: _____
  - [x ] Pre-compiled
  - [x] Self-compiled
- [x ] Flashing tools used: Tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
ESP32 with Tasmota:
Program Version	11.0.0.2(allsensors)
Build Date & Time	2022-02-21T22:07:30

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
 Rule1: on tele-VEML6070#UvIndexPower do publish domoticz/in {"idx":61,"nvalue":0,"svalue":"%value%"} endon

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
None

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
None

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Tried new flash and new configuration, same result

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Publish this value in the right dataformat ór - better - publish it inW/m2 to get a more readable output for sending to e.g. Domoticz
and in this way also is published e.g. by “Buitenradar” and others

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Tried by myself with NO succes:  
Rule1: on tele-VEML6070#UvIndexPower do publish domoticz/in {"idx":61,"nvalue":0,"svalue":"%value%"*1000} endon

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-03-14:
You don't explain why the value is incorrect
The UV Index is 0.98 which is totally coherent with a UV Power of 0.025 W/m2

sfromis, 2022-03-14:
> Rule1: on tele-VEML6070#UvIndexPower do publish domoticz/in {"idx":61,"nvalue":0,"svalue":"%value%"*1000} endon

You can't have expression mixed with the text of a command, but instead first calculate the value in one rule, and then refer it in another.