arendst/Tasmota #15140: can't play rtsp stream esp32-cam

jenniferlee1818, 2022-03-14:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Can't play the rtsp stream.  Log repeats rtsp created, rtsp stopped.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32-CAM https://smile.amazon.com/gp/product/B094NCWSDX
- [x] Tasmota binary firmware version number used: 11.0.0
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: ESP-Flasher-Windows-x64
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
CMD: Grp 0, Cmnd 'TEMPLATE', Idx 1, Len 0, Data ''
19:52:00.020 RSL: RESULT = {"NAME":"ESP32-CAM","GPIO":[4992,1,672,1,416,5088,1,1,1,6720,736,704,1,1,5089,5090,0,5091,5184,5152,0,5120,5024,5056,0,0,0,0,4928,576,5094,5095,5092,0,0,5093],"FLAG":0,"BASE":2}

19:52:10.492 CMD: Grp 0, Cmnd 'MODULE', Idx 1, Len 0, Data ''
19:52:10.505 RSL: RESULT = {"Module":{"0":"ESP32-CAM"}}

CMD: Grp 0, Cmnd 'GPIO', Idx 1, Len 3, Data '255'
19:52:50.610 RSL: RESULT = {"GPIO0":{"4992":"CAM_XCLK"},"GPIO1":{"0":"None"},"GPIO2":{"672":"SPI MISO1"},"GPIO3":{"0":"None"},"GPIO4":{"416":"PWM1"},"GPIO5":{"5088":"CAM_DATA1"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"6720":"SDCard CS"},"GPIO14":{"736":"SPI CLK1"},"GPIO15":{"704":"SPI MOSI1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"5089":"CAM_DATA2"},"GPIO19":{"5090":"CAM_DATA3"},"GPIO20":{"0":"None"},"GPIO21":{"5091":"CAM_DATA4"},"GPIO22":{"5184":"CAM_PCLK"},"GPIO23":{"5152":"CAM_HREF"},"GPIO24":{"0":"None"},"GPIO25":{"5120":"CAM_VSYNC"},"GPIO26":{"5024":"CAM_SIOD"},"GPIO27":{"5056":"CAM_SIOC"},"GPIO32":{"4928":"CAM_PWDN"},"GPIO33":{"576":"LedLink_i"},"GPIO34":{"5094":"CAM_DATA7"},"GPIO35":{"5095":"CAM_DATA8"},"GPIO36":{"5092":"CAM_DATA5"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"5093":"CAM_DATA6"}}
19:52:50.708

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
Grp 0, Cmnd 'RULE', Idx 1, Len 0, Data ''
19:56:58.861 RSL: RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":34,"Free":477,"Rules":"on system#boot do meshbroker endon"}}

19:57:01.050 CMD: Grp 0, Cmnd 'RULE', Idx 2, Len 0, Data ''
19:57:01.053 RSL: RESULT = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":30,"Free":481,"Rules":"ON System#Boot DO WcInit ENDON"}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
19:57:53.304 CMD: Grp 0, Cmnd 'STATUS', Idx 1, Len 1, Data '0'
19:57:53.308 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Hello Cam","FriendlyName":["Hello Cam"],"Topic":"hello-cam","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
19:57:53.348 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:01:57","StartupUTC":"2022-03-14T18:55:56","Sleep":50,"CfgHolder":4680,"BootCount":35,"BCResetTime":"2022-03-14T00:09:29","SaveCount":57}}
19:57:53.376 RSL: STATUS2 = {"StatusFWR":{"Version":"11.0.0(webcam)","BuildDateTime":"2022-03-13T17:01:14","Core":"2_0_2_2","SDK":"v4.4-3-g6afb23d90a","CpuFrequency":240,"Hardware":"ESP32-D0WDQ6","CR":"371/699"}}
19:57:53.401 RSL: STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["BATTLECAT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A190000000000","00000080","00006000","00004000"]}}
19:57:53.430 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1199,"Free":1856,"Heap":125,"StackLowMark":2,"PsrMax":4092,"PsrFree":3289,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":80,"FlashMode":3,"Features":["00000809","0F9005CA","00040001","00000082","00000000","40000981","80000080","00200000","00000208"],"Drivers":"1,2,4,7,8,9,10,12,24,27,29,38,50,52,57,81","Sensors":"2,6,127"}}
19:57:53.471 RSL: STATUS5 = {"StatusNET":{"Hostname":"hello-cam-3508","IPAddress":"10.0.0.35","Gateway":"10.0.0.1","Subnetmask":"255.255.255.0","DNSServer1":"10.0.0.1","DNSServer2":"0.0.0.0","Mac":"40:91:51:AB:6D:B4","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
19:57:53.498 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.5","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AB6DB4","MqttUser":"mqttadmin","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
19:57:53.514 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-03-14T18:57:53","Local":"2022-03-14T19:57:53","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"07:05","Sunset":"18:52"}}
19:57:53.542 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-03-14T19:57:53","ESP32":{"Temperature":72.8},"TempUnit":"C"}}
19:57:53.546 BRY: GC from 6528 to 3344 bytes, objects freed 11/31 (in 1 ms)
19:57:53.562 BRY: GC from 3344 to 3243 bytes, objects freed 1/31 (in 0 ms)
19:57:53.566 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-03-14T19:57:53","Uptime":"0T00:01:57","UptimeSec":117,"Heap":125,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":32,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":31},"POWER":"OFF","Dimmer":10,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"BATTLECAT","BSSId":"6C:CD:D6:04:A0:EE","Channel":3,"Mode":"11n","RSSI":100,"Signal":-36,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
Creating TSP streamer
Created streamer width=480, height=320
Creating RTSP session
20:00:16.921 CAM: RTSP stream created
20:00:16.926 SRC: MQTT
20:00:16.928 CMD: Grp 0, Cmnd 'STATUS', Idx 1, Len 2, Data '11'
20:00:16.931 BRY: GC from 3875 to 3243 bytes, objects freed 2/31 (in 1 ms)
20:00:16.950 MQT: hello-cam/stat/STATUS11 = {"StatusSTS":{"Time":"2022-03-14T20:00:16","Uptime":"0T00:00:20","UptimeSec":20,"Heap":119,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":32,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":31},"POWER":"OFF","Dimmer":10,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"BATTLECAT","BSSId":"6C:CD:D6:04:A0:EE","Channel":3,"Mode":"11n","RSSI":100,"Signal":-36,"LinkCount":1,"Downtime":"0T00:00:04"}}}
[... 127 lines cut ...]
Creating RTSP session
20:04:32.431 CAM: RTSP stream created
RTSP received OPTIONS
client has closed the socket
client closed socket, exiting
closing TCP socket
closing UDP socket
closing UDP socket
20:04:39.753 CAM: RTSP stopped

```

### TO REPRODUCE
power up the esp32-cam

### EXPECTED BEHAVIOUR
play the rtsp stream

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2022-03-14:
Since the persistent symptom is `client has closed the socket`, it looks like something is going wrong there. What client program are you using, and what messages do you get there?

jenniferlee1818, 2022-03-15:
@sfromis yes, you're right.  I'm not sure what exactly was going wrong there, but after a million dropped packets and a million reboots, it works now.  And it works every time.  This isn't a problem with Tasmota.

closed 2022-03-15 by jenniferlee1818