arendst/Tasmota #15144: Upgrade from 5.14 to 6.7.1 does not work

jayjupdhig, 2022-03-15:
https://tasmota.github.io/docs/Upgrading/#gz-binaries

Get a "buffer miscompare" error when i load http://ota.tasmota.com/tasmota/release-6.7.1/sonoff.bin to a Sonoff with:

[image]

HERE:

[image]

THEN...

[image]

(WTF...)

What you write here seems to be wrong!!

[image]

Why that!?

Thx for feedback,
Jan

closed 2022-03-15 by close-issue-app[bot]

jayjupdhig, 2022-03-15:
...

ascillato, 2022-03-15:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you.

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

jayjupdhig, 2022-03-15:
Your link on https://tasmota.github.io/docs/Upgrading/#gz-binaries seems to be wrong... on the chat somebody told me i need minimal- or (???) basic- of Tasmota 6.7.1.bin file...

But which one now - "minimal" or "basic" ?? 

[image]

jayjupdhig, 2022-03-15:
After installing 6.7.1-basic, both of my devices are destroyed... only to recover with RS-232 which makes no sense for a price for 5-7 dollar...

jayjupdhig, 2022-03-15:
It was just a reset to factory default - so i didin't put them into the trash. (Sry. i was a lil bit stressed - due to many technical problem in the same moment)