arendst/Tasmota #15167: Tasmota Web Installer with NodeMCU v2 (Amica) hangs

alexwala, 2022-03-18:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Trying to flash a NodeMCUv2 board (amica, cp2102) via tasmota web installer.
Hangs on 'Preparing Installation'.
Using Tasmotizer by checking 'Self-resetting device (NodeMCU, Wemos)' works fine.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): NodeMCU v2 (Amica)
- [ ] Tasmota binary firmware version number used: v11.0.0
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: Tasmota Web Installer, Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Try flash NodeMCU v2 (Amica) via Tasmota Web Installer

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Flashing is performed.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2022-03-18:
Does a other NodeMCU or WemosD1 mini works with WebInstaller?

alexwala, 2022-03-18:
Actually having only one: NodeMCU v2 (Amica)