arendst/Tasmota #15211: BLE sensors are not reported in new Tasmota native discovery, and not visible in HASS GUI

altserg, 2022-03-23:
### PROBLEM DESCRIPTION
BLE sensors are not reported in Tasmota native discovery. They are not visible in HASS GUI. The behavior is the same regardless of MI32Option6 setting

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Kincony KC868-A8
- [ ] Tasmota binary firmware version number used: 11.0.0.4
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: platformio
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
Backlog Template; Module; GPIO 255
12:57:37.916 MQT: stat/FFHeating/RESULT = {"NAME":"KC868-A8","GPIO":[32,0,1,0,640,608,0,0,0,1,1,1,0,0,5600,0,0,0,0,5568,0,0,0,0,0,0,0,0,1,1,0,0,1,0,0,1],"FLAG":0,"BASE":1}
12:57:38.151 MQT: stat/FFHeating/RESULT = {"Module":{"0":"KC868-A8"}}
12:57:38.419 MQT: stat/FFHeating/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"640":"I2C SDA1"},"GPIO5":{"608":"I2C SCL1"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"5600":"ETH MDIO"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"5568":"ETH MDC"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
 Rules not used

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

status 0
12:58:29.525 MQT: stat/FFHeating/STATUS = {"Status":{"Module":0,"DeviceName":"FF Heating","FriendlyName":["GSM Repeater 220V","Roof 220V","FF_G6_VeronicaBedroom2","FF_G5_VeronicaBedroom1","FF_G3_WorkRoom1","FF_G4_WorkRoom2","FF_G2_MainBathroom","FF_G1_MainBedroom1"],"Topic":"FFHeating","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
12:58:29.580 MQT: stat/FFHeating/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T02:44:13","StartupUTC":"2022-03-23T09:14:16","Sleep":50,"CfgHolder":3002,"BootCount":61,"BCResetTime":"2021-12-23T22:34:50","SaveCount":493}}
12:58:29.624 MQT: stat/FFHeating/STATUS2 = {"StatusFWR":{"Version":"11.0.0.4(SRG-K-MIN-BLE-ETH)","BuildDateTime":"2022-03-23T10:08:02","Core":"2_0_3","SDK":"v.4.4.0","CpuFrequency":80,"Hardware":"ESP32-D0WDQ6","CR":"536/699"}}
12:58:29.645 MQT: stat/FFHeating/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Bailar_N",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["0002C008","2805C80001000600003C5A0A000000000000","80000288","00004000","00000002"]}}
12:58:29.684 MQT: stat/FFHeating/STATUS4 = {"StatusMEM":{"ProgramSize":1300,"Free":1856,"Heap":95,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","08100584","00000001","00000042","00000000","40000200","40000080","00200000","00000008"],"Drivers":"1,2,7,10,12,28,38,50,52,56,79,82","Sensors":"2,5,52,62,127"}}
12:58:29.733 MQT: stat/FFHeating/STATUS5 = {"StatusNET":{"Hostname":"ESP32-FFHeating","IPAddress":"0.0.0.0","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.1","DNSServer2":"0.0.0.0","Mac":"4C:EB:D6:49:0C:B4","Ethernet":{"Hostname":"ESP32-FFHeating-eth","IPAddress":"192.168.2.95","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.1","DNSServer2":"8.8.8.8","Mac":"4C:EB:D6:49:0C:B7"},"Webserver":2,"HTTP_API":0,"WifiConfig":4,"WifiPower":17.0}}
12:58:29.783 MQT: stat/FFHeating/STATUS6 = {"StatusMQT":{"MqttHost":"smarthome.local","MqttPort":1883,"MqttClientMask":"ESP32-FF-Heating","MqttClient":"ESP32-FF-Heating","MqttUser":"FFHeating","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
12:58:29.813 MQT: stat/FFHeating/STATUS7 = {"StatusTIM":{"UTC":"2022-03-23T11:58:29","Local":"2022-03-23T12:58:29","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":99}}
12:58:29.841 MQT: stat/FFHeating/STATUS10 = {"StatusSNS":{"Time":"2022-03-23T12:58:29","ESP32":{"Temperature":53.3},"PCF8574-1":{"D0":1,"D1":1,"D2":1,"D3":1,"D4":1,"D5":1,"D6":1,"D7":1},"PCF8574-2":{"D0":1,"D1":1,"D2":1,"D3":1,"D4":1,"D5":1,"D6":1,"D7":1},"TempUnit":"C"}}
12:58:29.876 MQT: stat/FFHeating/STATUS11 = {"StatusSTS":{"Time":"2022-03-23T12:58:29","Uptime":"0T02:44:13","UptimeSec":9853,"Heap":97,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":38},"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","POWER7":"OFF","POWER8":"OFF"}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
 Kick off readOneSensor
13:04:17.611 M32: Hit tele time, restarted but not finished last - lost from slot 1
13:04:18.640 MQT: tele/tasmota_ble/AndrewBedroom = {"Time":"2022-03-23T13:04:18","alias":"AndrewBedroom","mac":"a4c13803bc36","RSSI":-77}
13:04:19.628 MQT: tele/tasmota_ble/MainBedroom = {"Time":"2022-03-23T13:04:19","alias":"MainBedroom","mac":"a4c138064916","Temperature":22.4,"Humidity":42.5,"DewPoint":9.0,"Battery":62,"RSSI":-66,"TempUnit":"C"}
13:04:19.643 M32: readOneSensor - already active reading 0
13:04:19.734 BLE: failed to connect to device
13:04:20.626 M32: Operation failed -11 for A4C13803BC36
13:04:20.639 MQT: tele/FFHeating/BLE = {"BLEOperation":{"opid":"70","stat":"-11","state":"FAILCONNECT","MAC":"A4C13803BC36","u":"84213760","svc":"ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6","notifychar":"ebe0ccc1-7a0a-4b0c-8a1a-6ff2997da3a6"}}
13:04:20.661 MQT: tele/tasmota_ble/WorkRoom = {"Time":"2022-03-23T13:04:20","alias":"WorkRoom","mac":"a4c138969b08","Temperature":22.4,"Humidity":41.3,"DewPoint":8.6,"RSSI":-65,"TempUnit":"C"}
13:04:21.607 MQT: tele/tasmota_ble/VeronicaBedroom = {"Time":"2022-03-23T13:04:21","alias":"VeronicaBedroom","mac":"a4c1389224d1","Temperature":22.5,"Humidity":44.6,"DewPoint":9.8,"Battery":53,"RSSI":-64,"TempUnit":"C"}
13:04:23.587 BRY: GC from 4960 to 3571 bytes, objects freed 5/38 (in 1 ms) - slots from 41/122 to 39/122

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Add any (Xiaomi) BLE temperature/humidity sensor to Tasmota. The sensor data are correctly reported using both "mi32option6 0"  and "mi32option6 1". However, the sensor is not reported via Tasmota native discovery to homeassistant. Legacy Home Assistant discovery via MQTT integration works fine. 
There is a thread about it in hass community with possible workaround: https://community.home-assistant.io/t/tasmota-xaomi-ble-temp-sensors/311314/4
However, I think that the behavior of legacy HASS discovery and new Tasmota native discovery should be the same

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Native Tasmota discovery should report BLE sensors, and they should be visible in HASS. The same as legacy Home Assistant discovery

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]
[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-03-23:
@altserg I'm looking into this but I am not sure that this is even possible.
The problem is that normally tasmota sends a single discovery message at the start with all its sensors in it.
With dynamically added sensors like BLE, or Zigbee, I'm not sure that the Tasmota plugin in HA would support receiving multiple sensor discovery messages.

You can try the attached [tasmota32-bluetooth-test-discovery1.zip](https://github.com/arendst/Tasmota/files/8337083/tasmota32-bluetooth-test-discovery1.zip)
 which sends discovery mesage as below.
Let me know how it works

```haskell
2022-03-23 23:31:57: tasmota/discovery/A4CF1204BFD0/sensors {"sn":{"ATC92d36d":{"alias":"chbre_coline","mac":"a4c13892d36d","Temperature":18.4,"Humidity":51.2,"DewPoint":8.1,"Btn":1,"Battery":84,"RSSI":-95},"TempUnit":"C"},"ver":1}
```

altserg, 2022-03-24:
Can you send me changed source files ? I'm compiling my own ESP32 builds

altserg, 2022-03-24:
I have used my spare ESP32 module, I see no MQTT messages for BLE sensors, see below. All BLE keys are OK
[image]
[image]

znanev, 2022-03-24:
@altserg Hi, I think you're experiencing the same issue as I described some time ago:

https://github.com/arendst/Tasmota/issues/14857

The stale bot closed the issue and I didn't want to nag the developers any more, as I found a work around that works fine for me, even though it prevents me automatically updating the firmware, as I need to recompile with specific flags enabed / disabled:

https://github.com/arendst/Tasmota/issues/14857#issuecomment-1044852323

altserg, 2022-03-24:
> @altserg Hi, I think you're experiencing the same issue as I described some time ago:
> 
> #14857
> 
> The stale bot closed the issue and I didn't want to nag the developers any more, as I found a work around that works fine for me, even though it prevents me automatically updating the firmware, as I need to recompile with specific flags enabed / disabled:
> 
> [#14857 (comment)](https://github.com/arendst/Tasmota/issues/14857#issuecomment-1044852323)

I also see the same behavior. But the issue is different as I do not want to use legacy HASS discovery but native Tasmota discovery. Legacy HASS discovery is not supported anymore.

znanev, 2022-03-24:
@altserg I totally agree with you. I was just coming from a use case, where I had already a working Home Assistant setup with a Tasmotized ESP32 used as BLE bridge to extend the range of remote and outside Xiaomi sensors. I was surprised that the HASS discovery was dropped with no equivalent support with native Tasmota discovery, that's why my immediate goal was to have my working setup back again, although with deprecated functionality.

So I'm actually in the same boat as you - I'd like to have the dynamically discovered BLE sensors autoconfigured in HA using the native Tasmota discovery mechanism. Let's hope this is doable after all.

barbudor, 2022-03-24:
I have 7 Xiaomi T&H sensors and I get discovery publications with the above firmware
```
21:17:00.681 MQT: tele/esp32dev3/STATE = {"Time":"2022-03-24T21:17:00+01:00","Uptime":"0T22:00:11","UptimeSec":79211,"Heap":86,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":57,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":43},"Wifi":{"AP":1,"SSId":"BNETWLAN","BSSId":"88:C3:97:F2:25:EC","Channel":1,"Mode":"11n","RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:06"}}
21:17:00.806 MQT: tele/esp32dev3/SENSOR = {"Time":"2022-03-24T21:17:00+01:00","ANALOG":{"A1":4081,"A2":4023},"ESP32":{"Temperature":60.0},"TempUnit":"C"} (retained)
21:17:00.834 MQT: tele/esp32dev3/BLE = {"Time":"2022-03-24T21:17:00+01:00","BLEDevices":{"total":7,"A4C138C09AD3":{"i":0,"r":-79,"a":"chbre_manon"},"F8042EBE0971":{"i":1,"r":-70},"A4C138352C48":{"i":2,"r":-71,"a":"salon"},"A4C138BE32D8":{"i":3,"r":-71,"a":"chbre_jmv"},"A4C13860B9F8":{"i":4,"r":-68,"a":"salle_bain"},"A4C1382BDB8E":{"i":5,"r":-86,"a":"chbre_lucas"},"A4C138C7C998":{"i":6,"r":-95,"a":"garage"}}}
21:17:00.876 MQT: tele/esp32dev3/BLE = {"Time":"2022-03-24T21:17:00+01:00","BLE":{"scans":3944,"adverts":556251,"devices":7,"resets":0}}
21:21:49.668 MQT: tasmota/discovery/A4CF1204BFD0/sensors = {"sn":{"ATCc09ad3":{"alias":"chbre_manon","mac":"a4c138c09ad3","Temperature":23.5,"Humidity":41.2,"DewPoint":9.5,"Btn":1,"Battery":86,"RSSI":-85},"TempUnit":"C"},"ver":1} (retained)
21:21:49.690 MQT: tele/tasmota_ble/chbre_manon = {"Time":"2022-03-24T21:21:49+01:00","alias":"chbre_manon","mac":"a4c138c09ad3","Temperature":23.5,"Humidity":41.2,"DewPoint":9.5,"Btn":1,"Battery":86,"RSSI":-85,"TempUnit":"C"} (retained)
21:21:50.651 MQT: tasmota/discovery/A4CF1204BFD0/sensors = {"sn":{"ATC352c48":{"alias":"salon","mac":"a4c138352c48","Temperature":18.3,"Humidity":48.5,"DewPoint":7.3,"Btn":1,"Battery":45,"RSSI":-72},"TempUnit":"C"},"ver":1} (retained)
21:21:50.674 MQT: tele/tasmota_ble/salon = {"Time":"2022-03-24T21:21:50+01:00","alias":"salon","mac":"a4c138352c48","Temperature":18.3,"Humidity":48.5,"DewPoint":7.3,"Btn":1,"Battery":45,"RSSI":-72,"TempUnit":"C"} (retained)
21:21:51.701 MQT: tasmota/discovery/A4CF1204BFD0/sensors = {"sn":{"ATCbe32d8":{"alias":"chbre_jmv","mac":"a4c138be32d8","Temperature":19.4,"Humidity":48.7,"DewPoint":8.3,"Btn":1,"Battery":55,"RSSI":-62},"TempUnit":"C"},"ver":1} (retained)
21:21:51.719 MQT: tele/tasmota_ble/chbre_jmv = {"Time":"2022-03-24T21:21:51+01:00","alias":"chbre_jmv","mac":"a4c138be32d8","Temperature":19.4,"Humidity":48.7,"DewPoint":8.3,"Btn":1,"Battery":55,"RSSI":-62,"TempUnit":"C"} (retained)
21:21:52.689 MQT: tasmota/discovery/A4CF1204BFD0/sensors = {"sn":{"ATC60b9f8":{"alias":"salle_bain","mac":"a4c13860b9f8","Temperature":19.5,"Humidity":53.1,"DewPoint":9.6,"Btn":1,"Battery":54,"RSSI":-80},"TempUnit":"C"},"ver":1} (retained)
21:21:52.718 MQT: tele/tasmota_ble/salle_bain = {"Time":"2022-03-24T21:21:52+01:00","alias":"salle_bain","mac":"a4c13860b9f8","Temperature":19.5,"Humidity":53.1,"DewPoint":9.6,"Btn":1,"Battery":54,"RSSI":-80,"TempUnit":"C"} (retained)
21:21:53.667 MQT: tasmota/discovery/A4CF1204BFD0/sensors = {"sn":{"ATC2bdb8e":{"alias":"chbre_lucas","mac":"a4c1382bdb8e","Temperature":20.3,"Humidity":44.9,"DewPoint":7.9,"Btn":1,"Battery":78,"RSSI":-94},"TempUnit":"C"},"ver":1} (retained)
21:21:53.695 MQT: tele/tasmota_ble/chbre_lucas = {"Time":"2022-03-24T21:21:53+01:00","alias":"chbre_lucas","mac":"a4c1382bdb8e","Temperature":20.3,"Humidity":44.9,"DewPoint":7.9,"Btn":1,"Battery":78,"RSSI":-94,"TempUnit":"C"} (retained)
21:21:54.653 MQT: tasmota/discovery/A4CF1204BFD0/sensors = {"sn":{"ATCc7c998":{"alias":"garage","mac":"a4c138c7c998","Temperature":14.3,"Humidity":66.0,"DewPoint":8.1,"Btn":1,"Battery":4,"RSSI":-96},"TempUnit":"C"},"ver":1} (retained)
21:21:54.673 MQT: tele/tasmota_ble/garage = {"Time":"2022-03-24T21:21:54+01:00","alias":"garage","mac":"a4c138c7c998","Temperature":14.3,"Humidity":66.0,"DewPoint":8.1,"Btn":1,"Battery":4,"RSSI":-96,"TempUnit":"C"} (retained)
21:22:00.675 MQT: tele/esp32dev3/STATE = {"Time":"2022-03-24T21:22:00+01:00","Uptime":"0T22:05:11","UptimeSec":79511,"Heap":85,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":52,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":43},"Wifi":{"AP":1,"SSId":"BNETWLAN","BSSId":"88:C3:97:F2:25:EC","Channel":1,"Mode":"11n","RSSI":100,"Signal":-49,"LinkCount":1,"Downtime":"0T00:00:06"}}
21:22:00.795 MQT: tele/esp32dev3/SENSOR = {"Time":"2022-03-24T21:22:00+01:00","ANALOG":{"A1":3752,"A2":4013},"ESP32":{"Temperature":60.6},"TempUnit":"C"} (retained)
21:22:00.823 MQT: tele/esp32dev3/BLE = {"Time":"2022-03-24T21:22:00+01:00","BLEDevices":{"total":7,"A4C138C09AD3":{"i":0,"r":-87,"a":"chbre_manon"},"F8042EBE0971":{"i":1,"r":-70},"A4C138352C48":{"i":2,"r":-73,"a":"salon"},"A4C138BE32D8":{"i":3,"r":-71,"a":"chbre_jmv"},"A4C13860B9F8":{"i":4,"r":-71,"a":"salle_bain"},"A4C1382BDB8E":{"i":5,"r":-94,"a":"chbre_lucas"},"A4C138C7C998":{"i":6,"r":-96,"a":"garage"}}}
21:22:00.870 MQT: tele/esp32dev3/BLE = {"Time":"2022-03-24T21:22:00+01:00","BLE":{"scans":3959,"adverts":557779,"devices":7,"resets":0}}
```

barbudor, 2022-03-24:
So apprently you are using `MI32Option6 0` while I was using `Mi32Option6 1`
With `Mi32Option6 0` I don't get any discovery messages anymore.
I'll have a look

barbudor, 2022-03-24:
Hi @emontnemery 
I trying to solve this issue to publish proper tasmota discovery for dynamically detected BLE sensors.
Could you please advise what would be the expected behavior from Tasmota regarding drivers that comes up at any time ?

altserg, 2022-03-25:
I tried with MI32Ooption6 1, and indeed, I can see one sensor, that is detected in HASS. See below.
However, I see that there is only one MQTT discovery message related to the latest updated sensor. If there are multiple BLE sensors, you only can see the last updated
[image]
[image]

barbudor, 2022-03-25:
Thanks @altserg
That's something I'd like to discuss with someone from HA because having to manage that in the embedded device seems overkill.

Jason2866, 2022-03-25:
Moving to discussions since it is a feature request