arendst/Tasmota #15212: Sporadic 0-values via MQTT and on the WebUI (SML)

Pe-MaKer, 2022-03-23:
### PROBLEM DESCRIPTION
When reading out my electricity meter, sporadic zero values occur for the values:
1.8.0 / Total_in,
1.8.1 / HT_Total_in and
1.8.2 / NT_Total_in.

The SML messages always contain the correct values, but the web interface and MQTT transmit these incorrect 0 values.
I have no idea where these incorrect values occur, because everything arrives error-free via the serial interface.

The electricity meter transmits an unusually large number of values and needs about 10 seconds to transmit all values. If required, I can provide an extended log with all messages.

The developer of the reader hardware @esplesekopf is also hoping to find a solution to the problem, because he can reproduce the error with his hardware and his electricity meter.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP8285
- [x] Tasmota binary firmware version number used: Tasmota 10.1.0.1 ([Link](https://github.com/esplesekopf/Tasmota))
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: ESP8266Flasher
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
17:38:10.859 CMD: Backlog Template; Module; GPIO 255
17:38:10.862 SRC: WebConsole from 10.0.0.68
17:38:10.864 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 26, Data 'Template; Module; GPIO 255'
17:38:10.953 SRC: Backlog
17:38:10.955 CMD: Grp 0, Cmnd 'TEMPLATE', Idx 1, Len 0, Data ''
17:38:10.964 MQT: stat/stromzaehler_8F80AE/RESULT = {"NAME":"ESPLesekopf","GPIO":[1,1,320,1,1,1,1,1,1,1,1,256,1,1],"FLAG":0,"BASE":18}
17:38:11.159 SRC: Backlog
17:38:11.161 CMD: Grp 0, Cmnd 'MODULE', Idx 1, Len 0, Data ''
17:38:11.168 MQT: stat/stromzaehler_8F80AE/RESULT = {"Module":{"0":"ESPLesekopf"}}
17:38:11.364 SRC: Backlog
17:38:11.366 CMD: Grp 0, Cmnd 'GPIO', Idx 1, Len 3, Data '255'

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
17:58:30.978 CMD: Status 0
17:58:30.988 MQT: stat/stromzaehler_8F80AE/STATUS = {"Status":{"Module":0,"DeviceName":"Stromzähler","FriendlyName":["Stromzähler"],"Topic":"stromzaehler_8F80AE","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
17:58:30.999 MQT: stat/stromzaehler_8F80AE/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://homeassistant.fritz.box:9541/data/firmwares/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T20:05:51","StartupUTC":"2022-03-22T20:52:39","Sleep":50,"CfgHolder":4617,"BootCount":88,"BCResetTime":"2022-03-21T09:45:14","SaveCount":203,"SaveAddress":"F9000"}}
17:58:31.003 MQT: stat/stromzaehler_8F80AE/STATUS2 = {"StatusFWR":{"Version":"10.1.0.1(tasmota)","BuildDateTime":"2022.03.20 17:44:02","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"364/699"}}
17:58:31.014 MQT: stat/stromzaehler_8F80AE/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["PMK-Net",""],"TelePeriod":1,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004000"]}}
17:58:31.031 MQT: stat/stromzaehler_8F80AE/STATUS4 = {"StatusMEM":{"ProgramSize":650,"Free":352,"Heap":10,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14325E","FlashFrequency":40,"FlashMode":255,"Features":["00000407","87DAC787","043E8001","000000CF","010013C0","C000F989","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6,53"}}
17:58:31.042 MQT: stat/stromzaehler_8F80AE/STATUS5 = {"StatusNET":{"Hostname":"stromzaehler-8F80AE-0174","IPAddress":"10.0.0.139","Gateway":"10.0.0.1","Subnetmask":"255.0.0.0","DNSServer1":"10.0.0.1","DNSServer2":"0.0.0.0","Mac":"9C:9C:1F:8F:80:AE","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
17:58:31.051 MQT: stat/stromzaehler_8F80AE/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.4","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_8F80AE","MqttUser":"DVES_USER","MqttCount":7,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
17:58:31.065 MQT: stat/stromzaehler_8F80AE/STATUS7 = {"StatusTIM":{"UTC":"2022-03-23T16:58:31","Local":"2022-03-23T17:58:31","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"06:46","Sunset":"19:06"}}
17:58:31.081 MQT: stat/stromzaehler_8F80AE/STATUS10 = {"StatusSNS":{"Time":"2022-03-23T17:58:31","":{"Meter_number":20100458,"Serial_number":0,"time-stamp":"17:58:09","Total_in":12.403,"HT_Total_in":6.023,"NT_Total_in":6.379,"kW":0.457,"react_pwr":0.228,"V_L1":225.10,"V_L2":225.30,"V_L3":225.30,"I_L1":0.65,"I_L2":1.15,"I_L3":1.16,"Freq":49.97}}}
17:58:31.093 MQT: stat/stromzaehler_8F80AE/STATUS11 = {"StatusSTS":{"Time":"2022-03-23T17:58:31","Uptime":"0T20:05:52","UptimeSec":72352,"Heap":10,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":7,"POWER":"ON","Wifi":{"AP":1,"SSId":"PMK-Net","BSSId":"74:42:7F:86:1C:9D","Channel":6,"Mode":"11n","RSSI":66,"Signal":-67,"LinkCount":5,"Downtime":"0T07:34:45"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
18:01:39.814 MQT: tele/stromzaehler_8F80AE/SENSOR = {"Time":"2022-03-23T18:01:39","":{"Meter_number":20100458,"Serial_number":0,"time-stamp":"18:01:25","Total_in":12.403,"HT_Total_in":6.024,"NT_Total_in":6.379,"kW":0.451,"react_pwr":0.227,"V_L1":225.50,"V_L2":225.70,"V_L3":226.10,"I_L1":0.66,"I_L2":1.14,"I_L3":1.13,"Freq":50.05}}
18:01:40.799 MQT: tele/stromzaehler_8F80AE/STATE = {"Time":"2022-03-23T18:01:40","Uptime":"0T20:09:01","UptimeSec":72541,"Heap":11,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":7,"POWER":"ON","Wifi":{"AP":1,"SSId":"PMK-Net","BSSId":"74:42:7F:86:1C:9D","Channel":6,"Mode":"11n","RSSI":66,"Signal":-67,"LinkCount":5,"Downtime":"0T07:34:45"}}
18:01:40.815 MQT: tele/stromzaehler_8F80AE/SENSOR = {"Time":"2022-03-23T18:01:40","":{"Meter_number":20100458,"Serial_number":0,"time-stamp":"18:01:25","Total_in":12.403,"HT_Total_in":6.024,"NT_Total_in":6.379,"kW":0.451,"react_pwr":0.227,"V_L1":225.50,"V_L2":225.70,"V_L3":226.10,"I_L1":0.66,"I_L2":1.14,"I_L3":1.13,"Freq":50.05}}
18:01:41.801 MQT: tele/stromzaehler_8F80AE/STATE = {"Time":"2022-03-23T18:01:41","Uptime":"0T20:09:02","UptimeSec":72542,"Heap":11,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":7,"POWER":"ON","Wifi":{"AP":1,"SSId":"PMK-Net","BSSId":"74:42:7F:86:1C:9D","Channel":6,"Mode":"11n","RSSI":66,"Signal":-67,"LinkCount":5,"Downtime":"0T07:34:45"}}
[... 109 lines cut ...]
1,1.8.2(@1000,NT,KWh,NT_Total_in,3  
1,=h<hr/>
1,1.7.0(@1,Power,kW,kW,3              ;momentane Wirkleistung P+
1,4.7.0(@1,Reactive_Power,kvar,react_pwr,3        ;momentane Blindleistung Q-
1,32.7(@1,Spannung_L1,V,V_L1,2        ;momentaner Spannungswert L1
1,52.7(@1,Spannung_L2,V,V_L2,2        ;momentaner Spannungswert L2     
1,72.7(@1,Spannung_L3,V,V_L3,2        ;momentaner Spannungswert L3
1,31.7(@1,Strom_L1,A,I_L1,2           ;momentaner Stromwert L1 
1,51.7(@1,Strom_L2,A,I_L2,2           ;momentaner Stromwert L2 
1,71.7(@1,Strom_L3,A,I_L3,2           ;momentaner Stromwert L3  
1,14.7(@1,Frequenz,Hz,Freq,2          ;momentane Frequenz
#
```

### EXPECTED BEHAVIOUR
The fields must not contain any incorrect 0-values, as the following processing would sum up the changed power consumption. 

### SCREENSHOTS
[image]

### ADDITIONAL CONTEXT
Sensor53 d1:
[sensor53 d1.txt](https://github.com/arendst/Tasmota/files/8335552/sensor53.d1.txt)

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2022-03-23:
I notice that you have `TelePeriod 1`, way below the usual minimum of 10 seconds. While scripting does not appear to have this check, I'm wondering if such a fast TelePeriod may cause issues.

Pe-MaKer, 2022-03-24:
> I notice that you have `TelePeriod 1`, way below the usual minimum of 10 seconds. While scripting does not appear to have this check, I'm wondering if such a fast TelePeriod may cause issues.> 

Thank you for taking the time to look at my problem.
I have already tried the TelePeriod with 10 and also 60 seconds and got the same result.
Every few times, the shown value drops to "0.000".
What surprises me most about the whole problem: All other values that I determine in my script, voltage, frequency, etc., are constantly transmitted correctly after the first reading. Only the elementary values for "total positive active energy" always have these drops to "0.000".

barbudor, 2022-03-24:
Could it just be possible that you actually have transmission problems ? Noise ?
I see that the script is changing baudrate a each frame between 300 and 9600 baud.
This could be a good reason to have garbage data in the buffer.

I don't know if there is options in SML to dump the data received on serial.

gemu2015, 2022-03-25:
your meter emits a lot of data per data set. may be the decoder is too slow to keep track and therefore loses serial data. try to increase the IRQ buffer TMSBSIZ to e.g. 1024 bytes
#define TMSBSIZ 1024

Pe-MaKer, 2022-03-25:
> ust be possible that you actually have transmission problems ? Noise ? I see that the script is changing baudrate a each frame between 300 and 9600 baud. This could be a good reason to have garbage data in the buffer.
> 
> I don't know if there is options in SML to dump the data received on
Of course, transmission problems can never be ruled out - especially because it is an optical transmission method.
But the SML messages in the console are all error-free.
Switching between 300 and 9600 baud is necessary because the electricity meter has to be triggered with a pulse first. After that, the meter sends a very comprehensive dataset with 9600 baud.

Pe-MaKer, 2022-03-25:
> Could it just be possible that you actually have transmission problems ? Noise ? I see that the script is changing baudrate a each frame between 300 and 9600 baud. This could be a good reason to [...]

Of course, transmission problems can never be ruled out - especially because it is an optical transmission method.
But the SML messages in the console are all error-free.
Switching between 300 and 9600 baud is necessary because the electricity meter has to be triggered with a pulse first. After that, the meter sends a very comprehensive dataset with 9600 baud.

Pe-MaKer, 2022-03-25:
> your meter emits a lot of data per data set. may be the decoder is too slow to keep track and therefore loses serial data. try to increase the IRQ buffer TMSBSIZ to e.g. 1024 bytes #define TMSBSIZ 1024

Yes, you're right @gemu2015  - the meter is just throwing a lot of data around. It seems that Siemens has listed almost the entire OBIS code list of IEC 62056-61 here. German over-engineering. ;)
Thanks for the tip with the buffer. I'll also pass it on to @esplesekopf , who provides pre-built firmware for his users.

Jason2866, 2022-03-25:
@Pe-MaKer Please close if the suggested solution does fix the issue. Thx

Pe-MaKer, 2022-03-26:
Sadly, increasing TMSBSIZ does not fix the issue.
As mentioned above, the buffer already has 2048 byte.

gemu2015, 2022-03-26:
could you please count the bytes of one burst of your meter ?
it is extremely large probably more than 2k
if the burst is larger enlarge TMSBSIZ further.

keep SML_BSIZ as low as possible. it has to fit the longest line of your meter.

probably enlarge the poll  interval to may be 10 seconds

;Restart sequence after 100x100ms    
case 100  
; 10000ms later \> restart sequence    
scnt=0

esplesekopf, 2022-03-27:
Hello to all,
I would also like to comment briefly on this topic.

The hint with SML_BSIZ was important, because here I had an error while copying. There should be 60 and not 1060. I have then worked with different TMSBSIZ buffers. Once you look at the amount that the smartmeter outputs, you have to work with 6144. 

Now I have inserted here two outputs of sensor53 d1. As you can see the amount of data is extreme. 

I noticed that the OBIS numbers like 1.8.0 or 1.8.1 or even 2.8.1 appear several times (1.8.1 and 1.8.1&12, 1.8.1*11 etc.) I assume that these data really come from the meter. Interestingly, the problems with the "temporary" 0 values occur only with THESE OBIS numbers - interestingly excluded is 1.8.0 - this is always displayed correctly. 
If I include such OBIS numbers in the script, which are present only once, no problems occur here. Also an OBIS number at the end of the record is displayed correctly. I did this to test the buffer length.

Can it be that there are possible problems with the parsing of the OBIS values, if these occur several times?

`16:01:46.893 /SAT63511B12849000120`
`16:01:47.350 : F.F(00000000)`
`16:01:47.370 : 0.0.0(846100587)`
`16:01:47.404 : 0.0.1(G23-511-B.12/849-000120)`
`16:01:47.431 : C.70.0()`
`16:01:47.514 : 1.128.0(0.000*kWh)`
`16:01:47.516 : 0.1.0(12)`
`16:01:47.518 : 0.1.2&12(07-02-01 00:01)`
`16:01:47.528 : 0.1.2*11(07-02-01 00:00)`
`16:01:47.560 : 0.1.2&10(07-03-30 09:36)`
`16:01:47.587 : 0.1.2&09(07-03-30 09:14)`
`16:01:47.622 : 0.1.2&08(07-01-31 23:09)`
`16:01:47.697 : 0.1.2&07(07-01-31 23:07)`
`16:01:47.699 : 0.1.2&06(08-09-08 12:55)`
`16:01:47.706 : 0.1.2&05(08-09-05 20:16)`
`16:01:47.738 : 0.1.2&04(08-09-05 20:13)`
`16:01:47.766 : 0.1.2&03(07-01-03 10:12)`
`16:01:47.797 : 0.1.2&02(05-01-01 00:04)`
`16:01:47.827 : 0.1.2&01(05-01-01 00:00)`
`16:01:47.858 : 0.1.2*00(00-00-00 00:00)`
`16:01:47.888 : 0.1.2*00(00-00-00 00:00)`
`16:01:47.918 : 0.1.2*00(00-00-00 00:00)`
`16:01:47.937 : 1.2.0(54.639*kW)`
`16:01:47.957 : 2.2.0(37.920*kW)`
`16:01:47.981 : 1.4.0(06  0.080*kW)`
`16:01:48.008 : 2.4.0(06  0.000*kW)`
`16:01:48.046 : 1.6.0(0.986*kW)(05-01-01 01:00)`
`16:01:48.085 : 1.6.0&12(0.000)(05-01-01 00:00)`
`16:01:48.124 : 1.6.0*11(0.000)(05-01-01 00:00)`
`16:01:48.164 : 1.6.0&10(0.000)(05-01-01 00:00)`
`16:01:48.204 : 1.6.0&09(0.000)(05-01-01 00:00)`
`16:01:48.245 : 1.6.0&08(0.000)(05-01-01 00:00)`
`16:01:48.285 : 1.6.0&07(13.263)(07-01-31 23:00)`
`16:01:48.324 : 1.6.0&06(41.376)(08-09-08 12:15)`
`16:01:48.364 : 1.6.0&05(0.000)(05-01-01 00:00)`
`16:01:48.404 : 1.6.0&04(0.000)(05-01-01 00:00)`
`16:01:48.774 : 1.6.0&03(0.000)(05-01-01 00:00)`
`16:01:48.776 : 1.6.0&02(0.000)(05-01-01 00:00)`
`16:01:48.778 : 1.6.0&01(0.000)(05-01-01 00:00)`
`16:01:48.780 : 1.6.0*00(0.000)(00-00-00 00:00)`
`16:01:48.781 : 1.6.0*00(0.000)(00-00-00 00:00)`
`16:01:48.783 : 1.6.0*00(0.000)(00-00-00 00:00)`
`16:01:48.785 : 2.6.0(0.049*kW)(05-01-01 00:30)`
`16:01:48.787 : 2.6.0&12(0.000)(05-01-01 00:00)`
`16:01:48.788 : 2.6.0*11(0.000)(05-01-01 00:00)`
`16:01:48.803 : 2.6.0&10(33.987)(07-03-30 09:30)`
`16:01:48.844 : 2.6.0&09(0.000)(05-01-01 00:00)`
`16:01:48.884 : 2.6.0&08(0.334)(07-01-31 23:09)`
`16:01:48.924 : 2.6.0&07(0.000)(05-01-01 00:00)`
`16:01:48.963 : 2.6.0&06(3.598)(08-09-08 12:45)`
`16:01:49.003 : 2.6.0&05(0.000)(05-01-01 00:00)`
`16:01:49.043 : 2.6.0&04(0.001)(07-01-05 11:15)`
`16:01:49.083 : 2.6.0&03(0.000)(05-01-01 00:00)`
`16:01:49.124 : 2.6.0&02(0.000)(05-01-01 00:00)`
`16:01:49.164 : 2.6.0&01(0.000)(05-01-01 00:00)`
`16:01:49.204 : 2.6.0*00(0.000)(00-00-00 00:00)`
`16:01:49.243 : 2.6.0*00(0.000)(00-00-00 00:00)`
`16:01:49.282 : 2.6.0*00(0.000)(00-00-00 00:00)`
[... 661 lines cut ...]
`16:02:15.980 : C.7.1(0)(05-01-01 00:00)`
`16:02:16.010 : C.7.2(2)(05-01-01 00:00)`
`16:02:16.041 : C.7.3(2)(05-01-01 00:00)`
`16:02:16.052 : C.C.1(0)`
`16:02:16.071 : C.1.8.1(0.0*kW)`
`16:02:16.090 : C.1.8.2(0.0*kW)`
`16:02:16.111 : C.1.8.3(0.0*kW)`
`16:02:16.130 : C.1.8.4(0.0*kW)`
`16:02:16.153 : C.1.8.5(0.0*kW)`
`16:02:16.171 : C.1.8.6(0.0*kW)`
`16:02:16.191 : C.2.8.1(0.0*kW)`
`16:02:16.211 : C.2.8.2(0.0*kW)`
`16:02:16.230 : C.2.8.3(0.0*kW)`
`16:02:16.252 : C.2.8.4(0.0*kW)`
`16:02:16.270 : C.2.8.5(0.0*kW)`
`16:02:16.292 : C.2.8.6(0.0*kW)`
`16:02:16.303 : C.70.2()`
`16:02:16.321 : C.71.1(1)`
`16:02:16.346 : C.71.2(120*A)`
`16:02:16.363 : C.71.3(1)`
`16:02:16.382 : C.71.4(3)`
`16:02:16.401 : C.71.5(2)`
`16:02:16.451 : C.60.4.1(00010 001 0000 0CD6)(07-02-01 00:01)`
`16:02:16.501 : C.60.4.2(00010 001 0000 0002)(07-02-01 00:02)`
`16:02:16.552 : C.60.4.3(00010 001 0000 00FF)(07-02-01 00:02)`

gemu2015, 2022-03-27:
6144 bytes at 9600 baud takes about 6 seconds!
this meter is crazy
you must increase poll time to 10 seconds as stated above

this is the clash that creates the zero values
2.8.1(1.227*kWh)
C.2.8.1(0.0*kW)

the pattern matching detects both values

you have to switch from pattern matching to line matching.

#define SML_OBIS_LINE

esplesekopf, 2022-03-28:
@gemu2015 
Thanks a lot!

`#define SML_OBIS_LINE`

 has solved the problem. This is what my first tests show.

Will test some more now and then give feedback.

Is there any problem or any concern to leave `#define SML_OBIS_LINE` as default?Could there be a problem with this on other meters?

ascillato2, 2022-03-29:
> Thanks a lot! #define SML_OBIS_LINE has solved the problem.

Great to know that your issue is fixed.

closed 2022-03-29 by ascillato2

gemu2015, 2022-03-29:
@esplesekopf 
you may set SML_OBIS_LINE as default. i don't think there will be problems.

Pe-MaKer, 2022-03-29:
> #define SML_OBIS_LINE

Thanks for this hint @gemu2015 .
Now I have consistent sensor readings.

Are you considering the possibility of defining the type of PatternMatching within the script in the future? This would eliminate the need to build a separate firmware for this particular electricity meter.