arendst/Tasmota #15261: Support for SonOff POW R3

blackjack75, 2022-03-30:
I bought some SonOff POW R3 assuming they'd work just like R2 with the same template (it's supposedly the same but supporting 25A).

Turns out the POW R2 template *nearly* works but there's a small difference: On and OFF on the switch are inverted ! All the rest seems identical (energy monitoring etc seems to work just fine). 

I can solve it at home by just creating my own template based on POW R2 and just changing the relay to relay_i.

I am afraid I don't have the skills to build a proper PR without risking to mess up things but for someone who knows the project it should be pretty straightforward to clone the R2 template and make a R3  with the inverted relay.

closed 2022-03-30 by close-issue-app[bot]

barbudor, 2022-03-30:
There's no need for a PR.
It's a new device. 
So scroll down the left side of https://templates.blakadder.com/ until you find the "ADD NEW TEMPLATE" button
Thanks for your contribution

barbudor, 2022-03-30:
In fact the template is already there : https://templates.blakadder.com/sonoff_POWR3.html

blackjack75, 2022-03-30:
Thanks a lot ! Somehow I hadn't realized it was so recent (and somehow had not found this exist template when googling either).