arendst/Tasmota #15271: Thermostat mode: Stops controlling if no WiFi despite local sensor

PjotrekSE, 2022-04-02:
### PROBLEM DESCRIPTION
Compiled 11.0.0 with thermostat enabled. Parametrized for freezer, works perfectly as long as connected to WIFi, stops controlling when no WiFi connection possible, even though a local sensor is used. Previously used version (6.7.1) id not have this problem.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): _____sonoff basic
- [ ] Tasmota binary firmware version number used: ____11.0.0_
  - [ ] Pre-compiled
  - [ x] Self-compiled
- [ x] Flashing tools used: _____tasmotizer-1.2
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
11:49:08.841 CMD: Backlog Template; Module; GPIO 255
11:49:08.908 MQT: stat/freezer1/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
11:49:09.118 MQT: stat/freezer1/RESULT = {"Module":{"1":"Sonoff Basic"}}
11:49:09.374 MQT: stat/freezer1/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"1312":"DS18x20"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
11:50:55.741 CMD: Backlog Rule1; Rule2; Rule3
11:50:55.811 MQT: stat/freezer1/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":440,"Free":71,"Rules":"ON System#Boot DO backlog sensorinputset1 1; climatemodeset1 1; tempformatset1 0; controllermodeset1 0; tempfrostprotectset1 -25; temptargetset1 -18; timepicycleset1 15; propbandset1 1; timeresetset1 1200; tempantiwindupresetset1 0.8; temphystset1 0.3; timemaxactionset1 10; timeminactionset1 3; temprupdeltinset1 0.4; timeallowrampupset1 300; timerampupcycleset1 30; timerampupmaxset1 960; thermostatmodeset1 1; perflevelautotune1 1; ENDON"}}
11:50:56.028 MQT: stat/freezer1/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
11:50:56.278 MQT: stat/freezer1/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
11:51:07.422 MQT: tele/freezer1/STATE = {"Time":"2022-04-02T11:51:07","Epoch":1648893067,"Uptime":"3T09:51:10","UptimeSec":294670,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"solvik-bod","BSSId":"14:D6:4D:29:25:54","Channel":13,"Mode":"11n","RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:03"}}
11:51:07.433 MQT: tele/freezer1/SENSOR = {"Time":"2022-04-02T11:51:07","Epoch":1648893067,"DS18B20":{"Id":"03165038F8FF","Temperature":-18.1},"Thermostat0":{"ThermostatModeSet":1,"TempTargetSet":-18.00,"CtrDutyCycleRead":0},"TempUnit":"C"}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
11:51:46.665 CMD: Status 0
11:51:46.672 MQT: stat/freezer1/STATUS = {"Status":{"Module":1,"DeviceName":"freezer1","FriendlyName":["freezer1"],"Topic":"freezer1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
11:51:46.683 MQT: stat/freezer1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Hardware Watchdog","Uptime":"3T09:51:49","StartupUTC":"2022-03-29T23:59:57","Sleep":50,"CfgHolder":4712,"BootCount":12,"BCResetTime":"2022-02-27T11:00:03","SaveCount":3978,"SaveAddress":"FA000"}}
11:51:46.692 MQT: stat/freezer1/STATUS2 = {"StatusFWR":{"Version":"11.0.0(tasmota)","BuildDateTime":"2022-02-16T16:08:21","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"366/699"}}
11:51:46.702 MQT: stat/freezer1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":1,"SysLog":2,"LogHost":"10.91.50.81","LogPort":514,"SSId":["solvik-bod","solvik1"],"TelePeriod":60,"Resolution":"558180D0","SetOption":["00008009","2805C80001000600003C5A0A190000000000","000000A8","00006000","00004000"]}}
11:51:46.716 MQT: stat/freezer1/STATUS4 = {"StatusMEM":{"ProgramSize":560,"Free":440,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F000782","04368001","00000043","00000000","C000F981","00000404","00001000","00000000"],"Drivers":"1,2,4,6,9,10,16,18,19,20,21,22,24,27,29,30,35,37,39,45","Sensors":"1,2,4,5"}}
11:51:46.726 MQT: stat/freezer1/STATUS5 = {"StatusNET":{"Hostname":"freezer1-3662","IPAddress":"10.91.50.100","Gateway":"10.91.50.1","Subnetmask":"255.255.255.0","DNSServer1":"10.91.50.1","DNSServer2":"10.91.50.1","Mac":"68:C6:3A:8A:6E:4E","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
11:51:46.734 MQT: stat/freezer1/STATUS6 = {"StatusMQT":{"MqttHost":"10.91.50.81","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_8A6E4E","MqttUser":"soliot","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
11:51:46.744 MQT: stat/freezer1/STATUS7 = {"StatusTIM":{"UTC":"2022-04-02T09:51:46","Local":"2022-04-02T11:51:46","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":99,"Sunrise":"06:13","Sunset":"19:33"}}
11:51:46.756 MQT: stat/freezer1/STATUS10 = {"StatusSNS":{"Time":"2022-04-02T11:51:46","Epoch":1648893106,"DS18B20":{"Id":"03165038F8FF","Temperature":-18.0},"Thermostat0":{"ThermostatModeSet":1,"TempTargetSet":-18.00,"CtrDutyCycleRead":0},"TempUnit":"C"}}
11:51:46.768 MQT: stat/freezer1/STATUS11 = {"StatusSTS":{"Time":"2022-04-02T11:51:46","Epoch":1648893106,"Uptime":"3T09:51:49","UptimeSec":294709,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"solvik-bod","BSSId":"14:D6:4D:29:25:54","Channel":13,"Mode":"11n","RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
Cannot as when the problem occurs, there is no wifi

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Turn off wifi router

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Thermostat stops controlling, leaves freezer off, temperature rises 1 C/hour

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
Not possible, se above

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-04-02:
Is Tasmota rebooting in Wifimanager mode with AP ? What is you WIfiConfig mode ?

PjotrekSE, 2022-04-02:
> 
WifiConfig 4
But it should *NOT* stop controlling the freezer when WiFi is not available!

barbudor, 2022-04-02:
I don't see anything in the Thermostat code that is relevant to Wifi loss
So it might be something else, more general to Tasmota.

Some work has been done recently around the dependency regarding access to NTP server so it could be worth to upgrade to latest dev version.
It could be usefull also if you could get logs from the serial console but that requires disconnecting the Basic from mains voltage.

sfromis, 2022-04-02:
Your pasted status info shows an uptime of nearly 3½ days. Did the issue occur during that time?

Point being that without Wifi, it seems that Tasmota may (not sure when) decide to restart (even with `WifiConfig 4`), and when unable to get a real time (from NTP server) after that, Tasmota would not work well.

ascillato2, 2022-04-05:
Hi, any news on this?

PjotrekSE, 2022-04-05:
Well, for a thermostat controlling a freezer it is absolutely necessary
that it works in all possible situations, as temperature rises by 1°C/hour,
and if above ~ -8°C food for thousands of € is wasted.
Not working at loss of WiFi is NOT ,acceptable.
I can imagine other situations where it is even more critical.

PjotrekSE, 2022-04-05:
...that it has been up for 3 1/2 days is irrelevant. If it is up 200 days
and then fails when noone is about and can intervene, it is just as
catastrophal

ascillato, 2022-04-05:
Please, can you do some tests so as to gather some logs?

With your device working, please, turn off your Wi-Fi Router for a minute or so. Then turn your router on again and go to the Tasmota console and copy and paste the logs. The last lines of the logs are stored in Tasmota RAM, so if the device is not restarting, you should be able to check it. Please, set WEBLOG to 4, before doing this test. Thanks.

sfromis, 2022-04-05:
For a device where a failure is critical, I'd insist on having a backend process to raise an alarm if the controller is somehow not operational. Simplest step would be to react to the device going LWT offline, or no longer publishing MQTT messages. Of course also check the reported sensor temperature readings.

PjotrekSE, 2022-04-05:
Of course I have. But that does not help if noone is nearby. The device
must not fail, at least not in such a silly way - just quitting for an
irrelevant reason where it could just go on working, AS the older version
did.

ascillato, 2022-04-05:
>Of course I have. But that does not help if noone is nearby. The device
must not fail, at least not in such a silly way - just quitting for an
irrelevant reason where it could just go on working, AS the older version
did.

In order to have some insight of what is really happening in your device, please, can you do some tests so as to gather some logs?

Please, set WEBLOG to 4, before doing this test.

With your device working, please, turn off your Wi-Fi Router for a minute or so. Then turn your router on again and go to the Tasmota console and copy and paste the logs. The last lines of the logs are stored in Tasmota RAM, so if the device is not restarting, you should be able to check it.

Thanks.

ascillato2, 2022-04-08:
Please, when you have some time, do the quick test to gather some logs so as to have information of what may be happening in your device. Remember to set weblog to 4. Thanks.

PjotrekSE, 2022-04-10:
Test logs.  Topology: MQTT broker (mb) ethernet connected to main router, freezer house connected from main router via radio link (rl), freezer connected to access point (ap) in freezer house.
ap turned off while rl on and mb on133946-134022, rl turned off while ap and mb on 134653-134756, mb off whilw ap on and rl on 135131_135234.
To be conclusive on function, the unit should be noncommunicado for two hours or so, otherwise it will be difficult to judge if there is regulation or not. But then the logs will be less useful as they would show very little relevant data remaining.

13:34:07.162 SRC: Backlog
13:34:07.164 CMD: Grp 0, Cmnd 'SENSORINPUTSET', Idx 1, Len 1, Data '1'
13:34:07.171 MQT: stat/freezer1/RESULT = {"SensorInputSet1":1}
13:34:07.411 SRC: Backlog
13:34:07.414 CMD: Grp 0, Cmnd 'CLIMATEMODESET', Idx 1, Len 1, Data '1'
13:34:07.419 MQT: stat/freezer1/RESULT = {"ClimateModeSet1":1}
13:34:07.661 SRC: Backlog
13:34:07.663 CMD: Grp 0, Cmnd 'TEMPFORMATSET', Idx 1, Len 1, Data '0'
13:34:07.669 MQT: stat/freezer1/RESULT = {"TempFormatSet1":0}
13:34:07.912 SRC: Backlog
13:34:07.914 CMD: Grp 0, Cmnd 'CONTROLLERMODESET', Idx 1, Len 1, Data '0'
13:34:07.920 MQT: stat/freezer1/RESULT = {"ControllerModeSet1":0}
13:34:08.162 SRC: Backlog
13:34:08.164 CMD: Grp 0, Cmnd 'TEMPFROSTPROTECTSET', Idx 1, Len 3, Data '-25'
13:34:08.170 MQT: stat/freezer1/RESULT = {"TempFrostProtectSet1":-25.0}
13:34:08.412 SRC: Backlog
13:34:08.415 CMD: Grp 0, Cmnd 'TEMPTARGETSET', Idx 1, Len 3, Data '-18'
13:34:08.421 MQT: stat/freezer1/RESULT = {"TempTargetSet1":-18.0}
13:34:08.618 SRC: Backlog
13:34:08.621 CMD: Grp 0, Cmnd 'TIMEPICYCLESET', Idx 1, Len 2, Data '15'
13:34:08.627 MQT: stat/freezer1/RESULT = {"TimePiCycleSet1":15}
13:34:08.868 SRC: Backlog
13:34:08.870 CMD: Grp 0, Cmnd 'PROPBANDSET', Idx 1, Len 1, Data '1'
13:34:08.877 MQT: stat/freezer1/RESULT = {"PropBandSet1":1}
13:34:09.118 SRC: Backlog
13:34:09.120 CMD: Grp 0, Cmnd 'TIMERESETSET', Idx 1, Len 4, Data '1200'
13:34:09.127 MQT: stat/freezer1/RESULT = {"TimeResetSet1":1200}
13:34:09.368 SRC: Backlog
13:34:09.371 CMD: Grp 0, Cmnd 'TEMPANTIWINDUPRESETSET', Idx 1, Len 3, Data '0.8'
13:34:09.378 MQT: stat/freezer1/RESULT = {"TempAntiWindupResetSet1":0.8}
13:34:09.619 SRC: Backlog
13:34:09.622 CMD: Grp 0, Cmnd 'TEMPHYSTSET', Idx 1, Len 3, Data '0.3'
13:34:09.629 MQT: stat/freezer1/RESULT = {"TempHystSet1":0.3}
13:34:09.869 SRC: Backlog
13:34:09.871 CMD: Grp 0, Cmnd 'TIMEMAXACTIONSET', Idx 1, Len 2, Data '10'
13:34:09.878 MQT: stat/freezer1/RESULT = {"TimeMaxActionSet1":10}
13:34:10.119 SRC: Backlog
13:34:10.121 CMD: Grp 0, Cmnd 'TIMEMINACTIONSET', Idx 1, Len 1, Data '3'
13:34:10.128 MQT: stat/freezer1/RESULT = {"TimeMinActionSet1":3}
13:34:10.369 SRC: Backlog
13:34:10.372 CMD: Grp 0, Cmnd 'TEMPRUPDELTINSET', Idx 1, Len 3, Data '0.4'
13:34:10.379 MQT: stat/freezer1/RESULT = {"TempRupDeltInSet1":0.4}
13:34:10.493 QPC: Reset
13:34:10.590 SRC: Backlog
13:34:10.592 CMD: Grp 0, Cmnd 'TIMEALLOWRAMPUPSET', Idx 1, Len 3, Data '300'
13:34:10.600 MQT: stat/freezer1/RESULT = {"TimeAllowRampupSet1":300}
13:34:10.840 SRC: Backlog
13:34:10.843 CMD: Grp 0, Cmnd 'TIMERAMPUPCYCLESET', Idx 1, Len 2, Data '30'
13:34:10.849 MQT: stat/freezer1/RESULT = {"TimeRampupCycleSet1":30}
13:34:11.090 SRC: Backlog
13:34:11.093 CMD: Grp 0, Cmnd 'TIMERAMPUPMAXSET', Idx 1, Len 3, Data '960'
13:34:11.099 MQT: stat/freezer1/RESULT = {"TimeRampupMaxSet1":960}
13:34:11.340 SRC: Backlog
13:34:11.342 CMD: Grp 0, Cmnd 'THERMOSTATMODESET', Idx 1, Len 1, Data '1'
13:34:11.349 MQT: stat/freezer1/RESULT = {"ThermostatModeSet1":1}
13:34:11.471 MQT: tele/freezer1/STATE = {"Time":"2022-04-10T13:34:11","Epoch":1649590451,"Uptime":"0T00:00:09","UptimeSec":9,"Heap":30,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"solvik-bod","BSSId":"14:D6:4D:29:25:54","Channel":11,"Mode":"11n","RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:03"}}
13:34:11.503 MQT: tele/freezer1/SENSOR = {"Time":"2022-04-10T13:34:11","Epoch":1649590451,"DS18B20":{"Id":"03165038F8FF","Temperature":-18.2},"Thermostat0":{"ThermostatModeSet":1,"TempTargetSet":-18.00,"CtrDutyCycleRead":0},"TempUnit":"C"}
13:34:11.573 SRC: Backlog
13:34:11.576 CMD: Grp 0, Cmnd 'PERFLEVELAUTOTUNE', Idx 1, Len 1, Data '1'
13:34:11.582 MQT: stat/freezer1/RESULT = {"PerfLevelAutotune1":1}
13:34:12.425 APP: Boot Count 13
13:34:12.623 CFG: Saved to flash at F9, Count 4987, Bytes 4096
13:34:16.945 HTP: Main Menu
13:34:24.084 HTP: Console
13:34:25.799 WIF: Checking connection...
13:34:45.776 WIF: Checking connection...
[... 207 lines cut ...]
13:52:03.285 MQT: Connect failed to 10.91.50.81:1883, rc -2. Retry in 30 sec
13:52:04.264 WIF: Sending Gratuitous ARP
13:52:11.310 RSL: STATE = {"Time":"2022-04-10T13:52:11","Epoch":1649591531,"Uptime":"0T00:18:09","UptimeSec":1089,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":3,"POWER":"OFF","Wifi":{"AP":1,"SSId":"solvik-bod","BSSId":"14:D6:4D:29:25:54","Channel":11,"Mode":"11n","RSSI":100,"Signal":-50,"LinkCount":2,"Downtime":"0T00:01:16"}}
13:52:11.321 RSL: SENSOR = {"Time":"2022-04-10T13:52:11","Epoch":1649591531,"DS18B20":{"Id":"03165038F8FF","Temperature":-18.3},"Thermostat0":{"ThermostatModeSet":1,"TempTargetSet":-18.00,"CtrDutyCycleRead":27},"TempUnit":"C"}
13:52:23.011 WIF: Checking connection...
13:52:34.344 MQT: Attempting connection...
13:52:34.362 MQT: Connected
13:52:34.366 MQT: tele/freezer1/LWT = Online (retained)
[... cut ...]
13:54:11.314 MQT: tele/freezer1/STATE = {"Time":"2022-04-10T13:54:11","Epoch":1649591651,"Uptime":"0T00:20:09","UptimeSec":1209,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":4,"POWER":"OFF","Wifi":{"AP":1,"SSId":"solvik-bod","BSSId":"14:D6:4D:29:25:54","Channel":11,"Mode":"11n","RSSI":100,"Signal":-50,"LinkCount":2,"Downtime":"0T00:01:16"}}
13:54:11.325 MQT: tele/freezer1/SENSOR = {"Time":"2022-04-10T13:54:11","Epoch":1649591651,"DS18B20":{"Id":"03165038F8FF","Temperature":-18.3},"Thermostat0":{"ThermostatModeSet":1,"TempTargetSet":-18.00,"CtrDutyCycleRead":27},"TempUnit":"C"}
13:54:23.283 WIF: Checking connection...
13:54:43.303 WIF: Checking connection...

ascillato, 2022-05-05:
bump. not addresses yet.

serifis84, 2022-05-31:
Hello there, same issue with my setup, didn't realize it was not working whithout been connected to WiFi. Fresh compiled and install firmware, let it without WiFi connection but serially connected to view the console logging, Thermostatmodeset 1 but no action. After connected to WiFi everything works fine as every other thermostat i have in several installations. I supose thermostat driver does not get temp value from the connected DS18B20

arendst, 2022-06-12:
#14394 might solve your issue too

Jason2866, 2022-06-20:
Please try latest release v12.0.2

PjotrekSE, 2022-06-23:
Sorry, wilco as soon as all guests left. In Scandinavia, midsummer is just as big a celebration as Xmas.

mattgit, 2022-06-28:
> 

> Please try latest release v12.0.2

still not working for me

closed 2022-07-28 by github-actions[bot]

circuitmike, 2022-08-20:
Another update: I was able to work around the problem by creating rules that invoke the `teleperiod` command on boot and on wifi disconnection:
```
on power1#boot do backlog power1 on; teleperiod 10 endon on wifi#disconnected do teleperiod 10 endon
```
This is an ugly hack, though, and I think this issue should be reopened.