arendst/Tasmota #15273: Tasmota uses wrong BSSid when switching between AP1 and AP2 on start

tolland, 2022-04-02:
### PROBLEM DESCRIPTION
When migrating tasmota devices between 2 wifi networks, tasmota uses the BSSid from the old device, and fails to connect to the new one.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Basic
- [ ] Tasmota binary firmware version number used:11.0.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Web gui
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
14:42:38.961 CMD: Backlog Template; Module; GPIO 255
14:42:38.979 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
14:42:39.208 RSL: RESULT = {"Module":{"1":"Sonoff Basic"}}
14:42:39.462 RSL: RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
14:43:36.376 CMD: Status 0
14:43:36.382 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Sonoff-003","FriendlyName":["Sonoff-003"],"Topic":"tasmota_7D0E09","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
14:43:36.389 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T00:17:05","StartupUTC":"2022-04-02T13:26:31","Sleep":50,"CfgHolder":4617,"BootCount":28,"BCResetTime":"2021-09-29T11:47:13","SaveCount":685,"SaveAddress":"F9000"}}
14:43:36.394 RSL: STATUS2 = {"StatusFWR":{"Version":"11.0.0(tasmota)","BuildDateTime":"2022-02-12T14:13:50","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"435/699"}}
14:43:36.399 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IOT","SKY4172"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000081","00006000","00004000"]}}
14:43:36.411 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":620,"Free":380,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
14:43:36.418 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-7D0E09-3593","IPAddress":"192.168.90.103","Gateway":"192.168.90.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.90.1","DNSServer2":"0.0.0.0","Mac":"60:01:94:7D:0E:09","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
14:43:36.421 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.102","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7D0E09","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
14:43:36.428 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-04-02T13:43:36","Local":"2022-04-02T14:43:36","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"06:25","Sunset":"19:21"}}
14:43:36.433 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-04-02T14:43:36"}}
14:43:36.438 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-04-02T14:43:36","Uptime":"0T00:17:05","UptimeSec":1025,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":2,"SSId":"SKY4172","BSSId":"8C:3B:AD:30:8A:71","Channel":11,"Mode":"11n","RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:18"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Add 2 wifi networks. Fails to connect to the new one, reporting the same BSSid for the 2nd network as the first, which is incorrect

### EXPECTED BEHAVIOUR
Would expect tasmota to detect the correct BSSid for the 2nd network

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Logs of error here:
https://pastebin.com/L3vHekE4

**(Please, remember to close the issue when the problem has been addressed)**

tolland, 2022-04-02:
I noticed that if I did the work around, which was to shutdown the old AP, and reboot the tasmota that it is logging a similar issue, but the other way. i.e.

```
00:00:00.065 Project tasmota - Sonoff-003 Version 11.0.0(tasmota)-2_7_4_9(2022-02-12T14:13:50)
00:00:00.554 WIF: Connecting to AP2 IOT Channel 11 BSSId 8C:3B:AD:30:8A:71 in mode 11n as tasmota-7D0E09-3593...
00:00:06.921 QPC: Reset
00:00:09.378 WIF: Connect failed as AP cannot be reached
00:00:09.715 WIF: Connecting to AP1 SKY4172 in mode 11n as tasmota-7D0E09-3593...
00:00:14.005 WIF: Connected
```

it's trying to use the AP1 BSSid for the AP2:

```
# nmcli dev wifi
IN-USE  BSSID              SSID                         MODE   CHAN  RATE        SIGNAL  BARS  SECURITY
*       8C:3B:AD:30:8A:70  SKY9271                      Infra  36    540 Mbit/s  69      ▂▄▆_  WPA2
        C2:C9:E3:10:C3:DB  IOT                          Infra  10    130 Mbit/s  42      ▂▄__  WPA2
```

barbudor, 2022-04-02:
What are your settings for SetOption56 and 57 ?

What happens if you ask tasmota to switch to the other AP using the AP command ?
Like whne on AP1, type "AP2" in the console or the opposite

tolland, 2022-04-02:
> What are your settings for SetOption56 and 57 ?

```
16:20:51.124 CMD: SetOption56
16:20:51.129 RSL: RESULT = {"SetOption56":"OFF"}
16:20:58.285 CMD: SetOption57
16:20:58.289 RSL: RESULT = {"SetOption57":"ON"}
```

> What happens if you ask tasmota to switch to the other AP using the AP command ?
> Like whne on AP1, type "AP2" in the console or the opposite

trying to switch to the other AP doesn't appear to do anything, with or without a reboot. (I am not familiar with the console commands, so I am probably doing something wrong...)

If I do this

```
16:29:14.807 CMD: AP
16:29:14.811 RSL: RESULT = {"Ap":{"1":"SKY4172"}}
16:29:18.916 CMD: AP2
16:29:18.920 RSL: RESULT = {"Ap":{"1":"SKY4172"}}
```

it stays on AP1, and after a reboot, it connects to AP1 again

barbudor, 2022-04-02:
Sorry the command has a space between "AP" and the number
Also what is your WifiConfig 

I'm not sure to understand in which case you have the problem
I now have 2 AP, my standard AP : BNETWLAN with BSSid 88:C3:97:F2:25:EC and another one BNETTEST wiht BSSid EC:88:8F:DC:5E:84
Both are set on Channel 1
If I'm switching from "AP 1" to "AP 2" or vice-versa I get the proper message:
```
00:00:00.515 WIF: Connecting to AP1 BNETWLAN Channel 1 BSSId 88:C3:97:F2:25:EC in mode 11n as nodemcu...
or
00:00:00.515 WIF: Connecting to AP2 BNETTEST Channel 1 BSSId EC:88:8F:DC:5E:84 in mode 11n as nodemcu...
```

Now being on AP2, and using `WifiConfig 4` (retry the other AP without rebooting), switching off the AP2:
```
AP
17:54:47.803 CMD: AP
17:54:47.804 SRC: Serial
17:54:47.806 CMD: Grp 0, Cmd 'AP', Idx 1, Len 0, Pld -99, Data ''
17:54:47.813 MQT: stat/nodemcu/AP = {"Ap":{"2":"BNETTEST"}}
WifiConfig
17:54:50.831 CMD: WifiConfig
17:54:50.833 SRC: Serial
17:54:50.834 CMD: Grp 0, Cmd 'WIFICONFIG', Idx 1, Len 0, Pld -99, Data ''
17:54:50.841 MQT: stat/nodemcu/WIFICONFIG = {"WifiConfig":{"4":"Retry"}}
17:55:09.516 WIF: Attempting connection...
17:55:09.853 WIF: Connecting to AP2 BNETTEST Channel 1 BSSId 88:C3:97:F2:25:EC in mode 11n as nodemcu...
17:55:12.285 WIF: Attempting connection...
17:55:13.287 WIF: Attempting connection...
17:55:14.288 WIF: Attempting connection...
17:55:15.290 WIF: Attempting connection...
17:55:16.290 WIF: Attempting connection...
17:55:17.295 WIF: Attempting connection...
17:55:18.294 WIF: Attempting connection...
17:55:19.297 WIF: Connect failed as AP cannot be reached
17:55:20.101 CFG: Saved to flash at F5, Count 40, Bytes 4096
17:55:20.286 WIF: Connect failed as AP cannot be reached
17:55:20.622 WIF: Connecting to AP1 BNETWLAN in mode 11n as nodemcu...
17:55:22.860 CFG: Saved to flash at F4, Count 41, Bytes 4096
17:55:23.094 WIF: Attempting connection...
17:55:24.504 WIF: Connected
17:55:24.754 MQT: Attempting connection...
17:55:24.777 MQT: Connected
17:55:24.780 MQT: tele/nodemcu/LWT = Online (retained)
17:55:24.782 MQT: cmnd/nodemcu/POWER = 
17:55:24.784 MQT: Subscribe to cmnd/nodemcu/#
17:55:24.786 MQT: Subscribe to cmnd/alldevices/#
17:55:24.787 MQT: Subscribe to cmnd/18FE34A6D683_fb/#
17:55:25.315 CFG: Saved to flash at FB, Count 42, Bytes 4096
```

Indeed I see at 17:55:09.853 an attempt to connect to AP2 with wrong BSSId
Than after a few seconds it correctly connects to AP1

So although there is a attempt to reconnect to the shotdown AP, it successfully connect back to AP1
Everything seems logical (except that log with bad BSSid)

tolland, 2022-04-02:
> So although there is a attempt to reconnect to the shotdown AP, it successfully connect back to AP1 Everything seems logical (except that log with bad BSSid)

Looking at the code here:
https://github.com/arendst/Tasmota/blob/964f0bd62d9a490e5d8141c68e27418620015dfe/tasmota/support_wifi.ino#L219

The BSSId log message seems to be generated on line 226, and used by WIFI.begin on line 219, so it's not just a log message, it's actually using the wrong BSSID. (i.e. if you turned the AP back on, it would fail to connect...)

> (except that log with bad BSSid)

I think this is the key. 

There is some niche case in the logic where it's failing to copy the new bssid over the `last_bssid` around here somewhere:
https://github.com/arendst/Tasmota/blob/964f0bd62d9a490e5d8141c68e27418620015dfe/tasmota/support_wifi.ino#L287

(Anyway, this is a bit painful to do over the web gui, so I'll try and replicate with serial console and more debugging...)

barbudor, 2022-04-02:
Indeed something strange:
```
19:34:57.461 WIF: Network (re)scan started...
19:34:59.463 WIF: Network 0, AP1, SSId BNETWLAN, Channel 1, BSSId 88:C3:97:F2:25:EC, RSSI -40, Encryption 1
19:34:59.464 WIF: Network 1, AP2, SSId BNETTEST, Channel 1, BSSId EC:88:8F:DC:5E:84, RSSI -23, Encryption 1
19:34:59.838 WIF: Connecting to AP2 BNETTEST Channel 1 BSSId EC:88:8F:DC:5E:84 in mode 11n as nodemcu...
```
As it finds AP2 to have a better signal, its trying AP2 but wrong BSSId
Need more logs

tolland, 2022-04-02:
I built and uploaded off the current development branch, and was able to replicate the problem at the beginning a couple of times. However after adding some logging in there, it stopped happening.

I think it requires a specific order of changes to trigger. i.e. changing the SSID in the gui of the current AP causes it to restart and then connect again with the new SSID and old BSSID. But i could reliably replicate it, so I'll close the ticket unless I get time to further investigate and find something reproducible.

Thanks for your time!

closed 2022-04-02 by tolland

didim99, 2025-01-28:
For those who have similar issue, using command `AP 1` works perfectly for me without need to power off second AP.

For maintainers, @barbudor, I'm get into similar situation as @tolland.

In short terms: if you change BSSID of your AP that configured in Tasmota, device no longer connects to that AP automatically until you manually select this AP via console command.

I have 2 APs configured in Tasmota (originally 13.1.0 updated to 14.4.1 via OTA but seems to work same way). The signal quality of AP2 (Ch 6, RSSI 72) was better than AP1 (Ch 11, RSSI 55) and previously my device always (almost) connected to AP2. Some time ago I replaced AP1's hardware with a new one, and configured it similar to old (same SSID and password, but obviously BSSID is changed). Now signal quality of AP1 (Ch 7, RSSI 90) is significantly better than AP2, but my device still connects to AP2. In log I can see similar situation as TS: Tasmota tries to connect to AP1 with old BSSID and old channel 11, then fails, then connects to AP2. It seems to BSSID and channel number is cached somewhere in device configuration.

I solve my issue by manually switching AP via console command `AP 1`, then device was successfully connected to new AP with new BSSID (and connects automatically after reboot). But this behaviour really weird and seems to be a bug.

barbudor, 2025-01-31:
@didim99 I think `reset 3` should reset Tamsota's memory of BSSId and allow to rescan properly