arendst/Tasmota #15275: Berry tasmota.resp_cmnd_str() response garbadge

curzon01, 2022-04-02:
### PROBLEM DESCRIPTION
Berry `tasmota.resp_cmnd_str()` response garbadge for command name (key in response json) when also using `tasmota.cmd()` within own command:
```lua
16:13:09.663 MQT: stat/tasmota/GA_po_IO_Oxilife/RESULT = {"��?@�?":"OK"}
```
see 
### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32 D1 (ESP32-D0WDQ6)
- [x] Tasmota binary firmware version number used: 11.0.0.5
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: VSC
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
16:15:37.703 MQT: stat/tasmota/GA_po_IO_Oxilife/RESULT = {"NAME":"Oxilife","GPIO":[1,1,544,1,1,1,0,1,224,225,226,227,228,0,7008,6976,0,1,1312,1,0,1,1,1,0,0,0,0,1,1,1,0,1,0,0,1],"FLAG":0,"BASE":1}
16:15:37.912 MQT: stat/tasmota/GA_po_IO_Oxilife/RESULT = {"Module":{"0":"Oxilife"}}
16:15:38.172 MQT: stat/tasmota/GA_po_IO_Oxilife/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"544":"LedLink"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"225":"Relay2"},"GPIO14":{"226":"Relay3"},"GPIO15":{"227":"Relay4"},"GPIO16":{"228":"Relay5"},"GPIO17":{"0":"None"},"GPIO18":{"7008":"NeoPool Rx"},"GPIO19":{"6976":"NeoPool Tx"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"1312":"DS18x20"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
no rules used

- [x] Provide the output of this command: `Status 0`:
```lua
16:17:18.209 MQT: stat/tasmota/GA_po_IO_Oxilife/STATUS = {"Status":{"Module":0,"DeviceName":"PHTasmota47","FriendlyName":["","","","",""],"Topic":"GA_po_IO_Oxilife","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":1,"PowerRetain":1,"InfoRetain":0,"StateRetain":0}}
16:17:18.323 MQT: stat/tasmota/GA_po_IO_Oxilife/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"neopool","OtaUrl":"http://tasmota.int.prsolution.eu/local/tasmota32-neopool.bin","RestartReason":"Software reset CPU","Uptime":"0T02:19:54","StartupUTC":"2022-04-02T11:57:24","Sleep":50,"CfgHolder":4617,"BootCount":211,"BCResetTime":"2021-11-21T16:51:03","SaveCount":1301}}
16:17:18.423 MQT: stat/tasmota/GA_po_IO_Oxilife/STATUS2 = {"StatusFWR":{"Version":"11.0.0.5(tasmota)","BuildDateTime":"2022-04-02T13:41:49","Core":"2_0_3","SDK":"v.4.4.0","CpuFrequency":80,"Hardware":"ESP32-D0WDQ6","CR":"625/699"}}
16:17:18.514 MQT: stat/tasmota/GA_po_IO_Oxilife/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":2,"LogHost":"syslog.int.prsolution.eu","LogPort":514,"SSId":["\\_=_/","<*_*>"],"TelePeriod":60,"Resolution":"558780C0","SetOption":["00208229","2805C8000100060000005A0A000000000000","004000CD","00206000","00004000"]}}
16:17:18.632 MQT: stat/tasmota/GA_po_IO_Oxilife/STATUS4 = {"StatusMEM":{"ProgramSize":1360,"Free":1856,"Heap":121,"StackLowMark":1,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","9FAA07CF","0011B1A1","B7F7BFCF","15DABBC0","20360D47","400840D2","28200000","0434482D"],"Drivers":"1,2,3,4,5,8,9,10,11,13,14,16,17,20,21,24,27,29,34,38,50,52,56,59,62,82,86","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,35,37,39,40,42,43,45,47,51,52,55,56,58,59,64,66,67,74,83,85,92,95,127"}}
16:17:18.747 MQT: stat/tasmota/GA_po_IO_Oxilife/STATUS5 = {"StatusNET":{"Hostname":"phtasmota47","IPAddress":"172.18.10.167","Gateway":"172.18.10.4","Subnetmask":"255.255.255.0","DNSServer1":"172.18.10.10","DNSServer2":"172.18.10.4","Mac":"3C:61:05:14:04:00","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"172.18.10.1","Subnetmask":"255.255.255.0","DNSServer1":"172.18.10.10","DNSServer2":"172.18.10.4","Mac":"00:00:00:00:00:00"},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
16:17:18.847 MQT: stat/tasmota/GA_po_IO_Oxilife/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.int.prsolution.eu","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_140400","MqttUser":"tasmota","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
16:17:18.942 MQT: stat/tasmota/GA_po_IO_Oxilife/STATUS7 = {"StatusTIM":{"UTC":"2022-04-02T14:17:18","Local":"2022-04-02T16:17:18","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":99,"Sunrise":"06:58","Sunset":"19:56"}}
16:17:19.097 MQT: stat/tasmota/GA_po_IO_Oxilife/STATUS10 = {"StatusSNS":{"Time":"2022-04-02T16:17:19+02:00","DS18B20-1":{"Id":"0213139D39AA","Temperature":8.4},"DS18B20-2":{"Id":"01131B69B529","Temperature":5.5},"DS18B20-3":{"Id":"01131B6C4959","Temperature":10.9},"NeoPool":{"Time":"2022-04-02T16:17:14","Type":"Oxilife","Modules":{"pH":1,"Redox":1,"Hydrolysis":1,"Chlorine":0,"Conductivity":0,"Ionization":0},"Temperature":10.2,"Power":{"Version":"V3.57","NodeID":"05DB 2D32 3130 5750 4313 281A","5V":5.017,"12V":14.578,"24-30V":33.960,"4-20mA":0.020},"pH":{"Data":7.3,"Min":7.0,"Max":7.5,"State":0,"Pump":0,"FL1":0,"Tank":1},"Redox":{"Data":732,"Setpoint":750},"Hydrolysis":{"Data": 0,"Unit":"%","Runtime":{"Total":"29T06:28:12","Part":"29T06:27:55","Pol1":"14T16:15:38","Pol2":"14T14:12:34","Changes":280},"State":"OFF","Cover":0,"Boost":0,"Low":0},"Filtration":{"State":0,"Mode":0},"Light":0,"Relay":{"State":[0,0,0,0,0,0,0],"Acid":0}},"ESP32":{"Temperature":53.3},"TempUnit":"C"}}
16:17:19.237 MQT: stat/tasmota/GA_po_IO_Oxilife/STATUS11 = {"StatusSTS":{"Time":"2022-04-02T16:17:19+02:00","Uptime":"0T02:19:55","UptimeSec":8395,"Heap":124,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":81,"MqttCount":1,"Berry":{"HeapUsed":18,"Objects":322},"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","Wifi":{"AP":2,"SSId":"<*_*>","BSSId":"9C:C7:A6:46:A8:A3","Channel":8,"Mode":"11n","RSSI":100,"Signal":-28,"LinkCount":1,"Downtime":"0T00:00:08"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
16:20:00.538 CMD: TestCmnd
16:20:00.545 CMD: Grp 0, Cmd 'TESTCMND', Idx 1, Len 0, Pld -99, Data ''
16:20:00.553 SRC: Berry
16:20:00.556 CMD: Grp 0, Cmd 'DELAY', Idx 1, Len 1, Pld 1, Data '1'
16:20:00.571 MQT: stat/tasmota/GA_po_IO_Oxilife/RESULT = {"Delay":2}
16:20:00.657 MQT: stat/tasmota/GA_po_IO_Oxilife/RESULT = {"��?@�?":"OK"}
16:20:00.842 BRY: GC from 27264 to 17142 bytes, objects freed 242/290 (in 4 ms) - slots from 680/697 to 255/487
16:20:02.002 BRY: GC from 27466 to 19315 bytes, objects freed 178/307 (in 4 ms) - slots from 585/607 to 270/455
```

### TO REPRODUCE
Define berry cmd in Berry Scripting console:

```berry
def TestCmnd()
  tasmota.cmd("Delay 1")
  tasmota.resp_cmnd_str("OK")
end
tasmota.add_cmd('TestCmnd', TestCmnd)
```

Test command on Console
```lua
16:13:09.558 CMD: TestCmnd
16:13:09.582 MQT: stat/tasmota/RESULT = {"Delay":2}
16:13:09.663 MQT: stat/tasmota/RESULT = {"��?@�?":"OK"}
```

It does not happen, if ` tasmota.cmd("Delay 1")` is removed from TestCmnd()

### EXPECTED BEHAVIOUR
Name of user defined command should be saved for `tasmota.resp_cmnd_str()` when also using `tasmota.cmd()`

sfromis, 2022-04-03:
May be too much of a race condition to call one command from another. I'd push it using 
`tasmota.set_timer(0, tasmota.cmd('Delay 1')`

Note: It is nonsensical to use `Delay` via `tasmota.cmd`, as the timing shows it does not work, which does not surprise me due to `Delay` being "too unreliable" anyway.

curzon01, 2022-04-03:
I notice it was a stupid example to use "Delay" :upside_down_face: - replace `DELAY` with a command or command series whose result you need to construct the `tasmota.resp_cmnd_str()` return.

The problem occurs with every Tasmota command and I need the result of a command series as return code for tasmota.resp_cmnd_str() so `tasmota.set_timer()`, which is a workaround, does not work here.

I can also simply build the return value with `tasmota.resp_cmnd()`, that's not the point. I just wanted to report that using `tasmota.cmd()` within a user command destroys the key of `tasmota.resp_cmnd_str()`.

sfromis, 2022-04-03:
Yes a race condition. Not surprising given that Tasmota in no way tries to be threadsafe.

curzon01, 2022-04-03:
so should we close? it's really not an issue for me

s-hadinger, 2022-04-03:
Tasmota commande are not reentrant so you can't call a command from within a command

Jason2866, 2022-04-03:
Closing since answered. The behaviour can not be changed

closed 2022-04-03 by Jason2866