arendst/Tasmota #15294: WPA2/3 Enterprise (802.1x) Support in Tasmota? (Based on ESPHome support?)

victorhooi, 2022-04-04:
### PROBLEM DESCRIPTION
Tasmota currently does not support WPA Enterprise (802.1x) wireless encryption, which makes it hard to setup Tasmota devices with such networks.

ESPHome added similar support a few years ago - is there something we can learn from their support?

https://github.com/esphome/feature-requests/issues/35
https://github.com/esphome/esphome/pull/1080
https://github.com/esphome/esphome-docs/pull/633

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Shelly Dimmer 2, Sonoff Pow R2
- [x] Tasmota binary firmware version number used: 11.0.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
23:25:48.993 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
23:25:49.242 RSL: RESULT = {"Module":{"43":"Sonoff Pow R2"}}
23:25:49.459 RSL: RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"3072":"CSE7766 Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3104":"CSE7766 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
23:26:13.032 RSL: STATUS = {"Status":{"Module":43,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_B1DC57","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
23:26:13.036 RSL: STATUS1 = {"StatusPRM":{"Baudrate":4800,"SerialConfig":"8E1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"4T13:53:18","StartupUTC":"2022-03-31T08:32:55","Sleep":50,"CfgHolder":4617,"BootCount":18,"BCResetTime":"2021-01-16T22:32:01","SaveCount":568,"SaveAddress":"F4000"}}
23:26:13.040 RSL: STATUS2 = {"StatusFWR":{"Version":"10.1.0(tasmota)","BuildDateTime":"2021-12-08T14:47:33","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"350/699"}}
23:26:13.044 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MIMO",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000680003C5A0A000000000000","00000080","00006000","00004000"]}}
23:26:13.053 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":616,"Free":384,"Heap":26,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
23:26:13.058 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-B1DC57-7255","IPAddress":"10.5.1.242","Gateway":"10.5.1.1","Subnetmask":"255.255.255.0","DNSServer1":"10.5.1.1","DNSServer2":"0.0.0.0","Mac":"84:F3:EB:B1:DC:57","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
23:26:13.061 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B1DC57","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
23:26:13.067 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-04-04T22:26:13","Local":"2022-04-04T23:26:13","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"06:21","Sunset":"19:24"}}
23:26:13.070 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
23:26:13.077 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-04-04T23:26:13","ENERGY":{"TotalStartTime":"2021-01-16T22:32:01","Total":440.516,"Yesterday":2.582,"Today":3.968,"Power":304,"ApparentPower":351,"ReactivePower":175,"Factor":0.87,"Voltage":243,"Current":1.443}}}
23:26:13.083 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-04-04T23:26:13","Uptime":"4T13:53:18","UptimeSec":395598,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"MIMO","BSSId":"00:4E:35:13:E2:80","Channel":11,"Mode":"11n","RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
N/A

```

### TO REPRODUCE
Attempt to connect a Tasmota device to a WPA Enterprise network.

### EXPECTED BEHAVIOUR
Tasmota should be able to connect to an Enterprise wifi network.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

So there are very old Github issues dating back years, on WPA Enterprise support (e.g. https://github.com/arendst/Tasmota/issues/3777 and https://github.com/arendst/Tasmota/issues/745). At the time, they appeared to indicate that the limitation was an upstream one.

However, I noticed that ESPHome actually added WPA Enterprise support a couple of years ago:

https://github.com/esphome/feature-requests/issues/35
https://github.com/esphome/esphome/pull/1080
https://github.com/esphome/esphome-docs/pull/633

Does this give any clues as to how Tasmota might be able to add WPA Enterprise support as well?

Are there technical issues that their approach wouldn't work here?

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2022-04-04:
Tasmota supports WPA3 for ESP32 devices the same as ESPHome does. PR that added that: https://github.com/arendst/Tasmota/pull/13526

There is no support for WPA3 in ESP8266 devices due to hardware restrictions. Sorry.

victorhooi, 2022-04-05:
@ascillato Thanks for replying!

However, I think there's a mixup - I'm referring to WPA2/WPA3 **Enterprise** specifically - i.e. using 802.1x authentication on wifi.

If we only support WPA2 Enterprise - that is fine - but it doesn't seem like that support is in Tasmota yet?

ascillato, 2022-04-05:
Exactly, that is not supported in ESP8266 due to the SDK. In ESP32, the SDK and the Arduino core supports EAP but that part is not  implemented in Tasmota. Anyway, if implemented, it would work only in ESP32 devices and not the ones you listed.

Moving this to **Ideas** due to it is a **Feature request** for supporting WPA2/WPA3 EAP in ESP32. Thanks.