arendst/Tasmota #15313: Compilation 11.0.0.6 failed for ESP32 due to wrong core used?

helmar74, 2022-04-08:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I just want to give a hint. I tried to compile dev version 11.0.0.6 for esp32 but it failed with error:
```
Executing task in folder Tasmota-development_v11006: C:\Users\user\.platformio\penv\Scripts\platformio.exe run <

Processing tasmota32 (framework: arduino; platform: https://github.com/tasmota/platform-espressif32/releases/download/v2.0.3rc1/platform-espressif32-2.0.3.zip; board: esp32_4M)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Platform Manager: Installing https://github.com/tasmota/platform-espressif32/releases/download/v2.0.3rc1/platform-espressif32-2.0.3.zip
Unpacking  [#################################---]   91%  00:00:00
Unpacking...
Error: Traceback (most recent call last):
  File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\commands\platform.py", line 322, in init_platform
    return PlatformFactory.new(name)
  File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\platform\factory.py", line 54, in new
    raise UnknownPlatform(pkg_or_spec)
platformio.platform.exception.UnknownPlatform: Unknown development platform 'https://github.com/tasmota/platform-espressif32/releases/download/v2.0.3rc1/platform-espressif32-2.0.3.zip'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 38, in unpack
    return fu.unpack(dst, with_progress=with_progress)
  File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\package\unpack.py", line 179, in unpack
    self._archiver.extract_item(item, dest_dir)
  File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\package\unpack.py", line 48, in extract_item
    self._afo.extract(item, dest_dir)
  File "C:\Users\user\.platformio\python3\lib\zipfile.py", line 1616, in extract
    return self._extract_member(member, path, pwd)
  File "C:\Users\user\.platformio\python3\lib\zipfile.py", line 1687, in _extract_member
    open(targetpath, "wb") as target:
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\user\\.platformio\\.cache\\tmp\\pkg-installing-mgstdoib\\__MACOSX\\platform-espressif32-2.0.3rc1\\examples\\espidf-aws-iot\\components\\esp-aws-iot\\aws-iot-device-sdk-embedded-C\\samples\\linux\\subscribe_publish_library_sample\\._subscribe_publish_library_sample.c'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\__main__.py", line 121, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "C:\Users\user\.platformio\penv\lib\site-packages\click\core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\user\.platformio\penv\lib\site-packages\click\core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "C:\Users\user\.platformio\penv\lib\site-packages\click\core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\user\.platformio\penv\lib\site-packages\click\core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\user\.platformio\penv\lib\site-packages\click\core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\user\.platformio\penv\lib\site-packages\click\decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\commands\run\command.py", line 134, in cli
    process_env(
  File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\commands\run\command.py", line 176, in process_env
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
  File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\commands\run\processor.py", line 67, in process
    result = init_platform(self.options["platform"]).run(
  File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\commands\platform.py", line 325, in init_platform
    _platform_install([name], skip_default_package=skip_default_package)
  File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\commands\platform.py", line 212, in _platform_install
    pkg = pm.install(
  File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\package\manager\platform.py", line 51, in install
    pkg = super(PlatformPackageManager, self).install(
  File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 48, in install
    pkg = self._install(
  File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 97, in _install
    pkg = self.install_from_url(spec.url, spec, silent=silent)
  File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 139, in install_from_url
    self.unpack(dl_path, tmp_dir)
[... 58 lines cut ...]
```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-04-08:
No problem building  with 2.0.3rc1 on a clean and fresh install of PlatformIO in VSCode
Try to clean up your platformio environment as it easily get corrupted.

```
Processing tasmota32 (framework: arduino; platform: https://github.com/tasmota/platform-espressif32/releases/download/v2.0.3rc1/platform-espressif32-2.0.3.zip; 
board: esp32_4M)
----------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
FILESYSTEM  littlefs
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32_4M.html
PLATFORM: Espressif 32 (2.0.3) > Espressif Generic ESP32 4M Flash, Tasmota 1856k Code/OTA, 320k FS
HARDWARE: ESP32 80MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 2.0.3
 - tool-esptoolpy 1.30300.220323 (3.3.0)
 - tool-mklittlefs 1.203.210628 (2.3)
 - tool-mkspiffs 2.230.0 (2.30)
 - toolchain-xtensa-esp32 8.4.0+2021r2-patch3
Converting tasmota.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Found 111 compatible libraries
Scanning dependencies...

.......

Archiving .pio\build\tasmota32\libFrameworkArduino.a
Linking .pio\build\tasmota32\firmware.elf
Retrieving maximum program size .pio\build\tasmota32\firmware.elf
Checking size .pio\build\tasmota32\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  19.2% (used 62880 bytes from 327680 bytes)
Flash: [=======   ]  73.9% (used 1403753 bytes from 1900544 bytes)
Building .pio\build\tasmota32\firmware.bin
esptool.py v3.3
Creating esp32 image...
Merged 25 ELF sections
Successfully created esp32 image.
esp32_create_combined_bin([".pio\build\tasmota32\firmware.bin"], [".pio\build\tasmota32\firmware.elf"])
    Offset | File
 -  0x1000 | C:\Users\jeanm\.platformio\packages\framework-arduinoespressif32\tools\sdk\esp32\bin\bootloader_dout_40m.bin
 -  0x8000 | C:\_perso\repos\Tasmota\.pio\build\tasmota32\partitions.bin
 -  0xe000 | C:\Users\jeanm\.platformio\packages\framework-arduinoespressif32\tools\partitions\boot_app0.bin
 - 0x10000 | C:\_perso\repos\Tasmota\.pio\build\tasmota32/firmware.bin
esptool.py v3.3
Wrote 0x168260 bytes to file C:\_perso\repos\Tasmota\.pio\build\tasmota32/firmware.factory.bin, ready to flash to offset 0x0
bin_map_copy([".pio\build\tasmota32\firmware.bin"], [".pio\build\tasmota32\firmware.elf"])
map_gzip([".pio\build\tasmota32\firmware.bin"], [".pio\build\tasmota32\firmware.elf"])
================================================================ [SUCCESS] Took 308.34 seconds ================================================================

Environment    Status    Duration
-------------  --------  ------------
tasmota32      SUCCESS   00:05:08.345
================================================================= 1 succeeded in 00:05:08.345 ================================================================= 
```

helmar74, 2022-04-08:
I tried to clean up my PlatformIO/VSC environment, that doesn't work either:
```
r Terminalprozess "C:\Users\helmar\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'clean'" wurde mit folgendem Exitcode beendet: 1.

Das Terminal wird von Aufgaben wiederverwendet, drücken Sie zum Schließen eine beliebige Taste.

> Executing task in folder Tasmota-development_v11006: C:\Users\helmar\.platformio\penv\Scripts\platformio.exe run --target clean <

Processing tasmota32 (framework: arduino; platform: https://github.com/tasmota/platform-espressif32/releases/download/v2.0.3rc1/platform-espressif32-2.0.3.zip; board: esp32_4M)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Platform Manager: Installing https://github.com/tasmota/platform-espressif32/releases/download/v2.0.3rc1/platform-espressif32-2.0.3.zip
Unpacking  [#################################---]   91%
Unpacking...
Error: Traceback (most recent call last):
  File "C:\Users\helmar\.platformio\penv\lib\site-packages\platformio\commands\platform.py", line 322, in init_platform
    return PlatformFactory.new(name)
  File "C:\Users\helmar\.platformio\penv\lib\site-packages\platformio\platform\factory.py", line 54, in new
    raise UnknownPlatform(pkg_or_spec)
platformio.platform.exception.UnknownPlatform: Unknown development platform 'https://github.com/tasmota/platform-espressif32/releases/download/v2.0.3rc1/platform-espressif32-2.0.3.zip'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\helmar\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 38, in unpack
    return fu.unpack(dst, with_progress=with_progress)
  File "C:\Users\helmar\.platformio\penv\lib\site-packages\platformio\package\unpack.py", line 179, in unpack
    self._archiver.extract_item(item, dest_dir)
  File "C:\Users\helmar\.platformio\penv\lib\site-packages\platformio\package\unpack.py", line 48, in extract_item
    self._afo.extract(item, dest_dir)
  File "C:\Users\helmar\.platformio\python3\lib\zipfile.py", line 1616, in extract
    return self._extract_member(member, path, pwd)
  File "C:\Users\helmar\.platformio\python3\lib\zipfile.py", line 1687, in _extract_member
    open(targetpath, "wb") as target:
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\helmar\\.platformio\\.cache\\tmp\\pkg-installing-e9af_35y\\__MACOSX\\platform-espressif32-2.0.3rc1\\examples\\espidf-aws-iot\\components\\esp-aws-iot\\aws-iot-device-sdk-embedded-C\\samples\\linux\\subscribe_publish_library_sample\\._subscribe_publish_library_sample.c'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\helmar\.platformio\penv\lib\site-packages\platformio\__main__.py", line 121, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "C:\Users\helmar\.platformio\penv\lib\site-packages\click\core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\helmar\.platformio\penv\lib\site-packages\click\core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "C:\Users\helmar\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "C:\Users\helmar\.platformio\penv\lib\site-packages\click\core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\helmar\.platformio\penv\lib\site-packages\click\core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\helmar\.platformio\penv\lib\site-packages\click\core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\helmar\.platformio\penv\lib\site-packages\click\decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "C:\Users\helmar\.platformio\penv\lib\site-packages\platformio\commands\run\command.py", line 134, in cli
    process_env(
  File "C:\Users\helmar\.platformio\penv\lib\site-packages\platformio\commands\run\command.py", line 176, in process_env
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
  File "C:\Users\helmar\.platformio\penv\lib\site-packages\platformio\commands\run\processor.py", line 67, in process
    result = init_platform(self.options["platform"]).run(
  File "C:\Users\helmar\.platformio\penv\lib\site-packages\platformio\commands\platform.py", line 325, in init_platform
    _platform_install([name], skip_default_package=skip_default_package)
  File "C:\Users\helmar\.platformio\penv\lib\site-packages\platformio\commands\platform.py", line 212, in _platform_install
    pkg = pm.install(
  File "C:\Users\helmar\.platformio\penv\lib\site-packages\platformio\package\manager\platform.py", line 51, in install
    pkg = super(PlatformPackageManager, self).install(
  File "C:\Users\helmar\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 48, in install
    pkg = self._install(
  File "C:\Users\helmar\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 97, in _install
    pkg = self.install_from_url(spec.url, spec, silent=silent)
[... 10 lines cut ...]
  File "C:\Users\helmar\.platformio\python3\lib\zipfile.py", line 1687, in _extract_member
[... cut ...]
* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

Der Terminalprozess "C:\Users\helmar\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'clean'" wurde mit folgendem Exitcode beendet: 1.

Das Terminal wird von Aufgaben wiederverwendet, drücken Sie zum Schließen eine beliebige Taste.
```

With Core 2.0.3 it works...

helmar74, 2022-04-08:
Just a stupid question wouldn't core 2.0.3 be better than rc1 version?

barbudor, 2022-04-08:
There might be a mistagging as AFAIK 2.0.3 is not yet out
But I easily get confused
Also sometime, @Jason2866 revert to previous core or at least revert some commits because it creates problems with Tasmota

Still I don't know why you can compile from the default
Are you using PIO command-line ? I was using in VSCode. Not sure what different it makes
I can make a try in Gitpod where it use the command line to see if there's a difference