arendst/Tasmota #15321: Compile Error Tasmota 11.0.0.6 on ESP32 when USE_SRIPT Enabled

helmar74, 2022-04-08:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

In my self compiled firmware I use Scripts.
So i defined in user_config_override.h:
```
#undef USE_RULES 
#define USE_SCRIPT
```
With this options I get the following compile error un Tasmota 11.0.0.6:
```
exe: error: ld returned 1 exit status
*** [.pio\build\tasmota32\firmware.elf] Error 1
================================================= [FAILED] Took 83.22 seconds =================================================

Environment    Status    Duration
-------------  --------  ------------
tasmota32      FAILED    00:01:23.216
```

This Options were working until Tasmota 11.0.0. If I set
```
//#define USE_SCRIPT
```
I can compile, but then of course Scripts aren't working.

So how do I get Scripts running on tasmota32?

Just for your information: For ESP8266
```
#undef USE_RULES 
#define USE_SCRIPT
```

is working

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): NodeMCU ESP32
- [X] Tasmota binary firmware version number used: 11.0.0.6
  - [ ] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used: VSC with PlatformIO on Windows
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2022-04-08:
Hi,

It is fixed now.

[image]

Please, download latest Tasmota codebase and try again.

Thanks for reporting.

closed 2022-04-08 by ascillato2

helmar74, 2022-04-09:
Hi, yes now it works like charme.
Thank you for fixing it so quickly!