arendst/Tasmota #15363: AFter Update to 11.1.0 all Files on SD CARD are renamed

helmar74, 2022-04-13:
### PROBLEM DESCRIPTION
Hi,

I updated my Infodisply to Tasmota 11.1.0. After the upgrade all lot of my files where renamed like this:

[image]

With Version 11.0.0 it looked like this:
[image]

Maybe I have to set something in user_config_override.h?

Anyone an idea what causes the issue?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): ESP32 NodeMCU
- [X] Tasmota binary firmware version number used: 11.1.0
  - [ ] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used: OTA / VSC with PlatformIO
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-04-13:
Is it the same SDCard ? You didn't reformatted it since you upgraded ?

The files are not renamed but it is showing the FAT 8.3 filenames instead of the usual FAT32 long filenames.
FAT32 always has 2 entries for each file : 1 in the old 8.3 mode compatible with old "DOS" and the long filename.

@s-hadinger @Jason2866 
I can't reproduce on the internal flash filesystem and I don't have (yet) a board with SDcard.
May be something a recent change in the library handling SDcard ?

helmar74, 2022-04-13:
> Is it the same SDCard ? You didn't reformatted it since you upgraded ?
No the SDCard hasn't been formated

> The files are not renamed but it is showing the FAT 8.3 filenames instead of the usual FAT32 long filenames. FAT32 always has 2 entries for each file : 1 in the old 8.3 mode compatible with old [...]

Jason2866, 2022-04-13:
Can not reproduce
[image]

helmar74, 2022-04-14:
@Jason2866 Precompiled or selfcompiled?

I'm using the following options in user_config_override.h
```
#define USE_UFILESYS
#define USE_SDCARD
```

Can that cause trouble?

helmar74, 2022-04-14:
I commented put #define USE_UFILESYS

still the same issue..

Reverted back to 11.0.0, that works flawless

Here's my user_config_override.h : 
```
/*
  user_config_override.h - user configuration overrides my_user_config.h for Tasmota

  Copyright (C) 2020  Theo Arends

  This program is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/

#ifndef _USER_CONFIG_OVERRIDE_H_
#define _USER_CONFIG_OVERRIDE_H_

// force the compiler to show a warning to confirm that this file is included
#warning **** user_config_override.h: Using Settings from this File ****

/*****************************************************************************************************\
 * USAGE:
 *   To modify the stock configuration without changing the my_user_config.h file:
 *   (1) copy this file to "user_config_override.h" (It will be ignored by Git)
 *   (2) define your own settings below
 *
 ******************************************************************************************************
 * ATTENTION:
 *   - Changes to SECTION1 PARAMETER defines will only override flash settings if you change define CFG_HOLDER.
 *   - Expect compiler warnings when no ifdef/undef/endif sequence is used.
 *   - You still need to update my_user_config.h for major define USE_MQTT_TLS.
 *   - All parameters can be persistent changed online using commands via MQTT, WebConsole or Serial.
\*****************************************************************************************************/

// -- Master parameter control --------------------
#undef  CFG_HOLDER
#define CFG_HOLDER        4617                   // [Reset 1] Change this value to load SECTION1 configuration parameters to flash

// -- Setup your own Wifi settings  ---------------
#undef  STA_SSID1
#define STA_SSID1         "XXXXXXXXXXXX"             // [Ssid1] Wifi SSID

#undef  STA_PASS1
#define STA_PASS1         "XXXXXXXXXXXXXX"     // [Password1] Wifi password

// -- Setup your own MQTT settings  ---------------
#undef  MQTT_HOST
#define MQTT_HOST         "192.168.1.xxx" // [MqttHost]

#undef  MQTT_PORT
#define MQTT_PORT         1883                   // [MqttPort] MQTT port (10123 on CloudMQTT)

// You might even pass some parameters from the command line ----------------------------
// Ie:  export PLATFORMIO_BUILD_FLAGS='-DUSE_CONFIG_OVERRIDE -DMY_IP="192.168.1.99" -DMY_GW="192.168.1.1" -DMY_DNS="192.168.1.1"'

/*
#ifdef MY_IP
[... 293 lines cut ...]
[library]
shared_libdeps_dir      = lib
; *** Library disable / enable for variant Tasmota(32). Disable reduces compile time
; *** !!! Disabling needed libs will generate compile errors !!!
; *** The resulting firmware will NOT be different if you leave all libs enabled
; *** Disabling by putting a ";" in front of the lib name
; *** If you dont know what it is all about, do not change
lib_extra_dirs           =
; *** Only disabled for Tasmota minimal and Tasmota light. For all other variants needed!
                           lib/lib_basic
; **** I2C devices. Most sensors. Disable only if you dont have ANY I2C device enabled
                           lib/lib_i2c
; *** Displays. Disable if you dont have any Display activated
                           lib/lib_display
; *** Bear SSL and base64. Disable if you dont have SSL or TLS activated
                           lib/lib_ssl
; *** Audio needs a lot of time to compile. Mostly not used functions. Recommended to disable
                           lib/lib_audio
; *** RF 433 stuff (not RF Bridge). Recommended to disable
                           lib/lib_rf
; *** Mostly not used functions. Recommended to disable
                           lib/lib_div
```

I need SCRIPTS, SDCARD, Display RA8876 Display Support. I also compile ST7789 Display as I want to use this firmware for a other device also.

helmar74, 2022-04-14:
Sorry for asking, where is the .platformio folder?
I'm using VSC with platformio addon...

helmar74, 2022-04-14:
> No, looks like using a framework without long file names support. We had a short time such a version. Delete folder `.platformio` and try again.

Yeah that did the trick. I deleted .platformio folder in C:\Users\USER
After that platformio addon was installed newly on VSC. After recompiling the files look like this:
[image]

Many thanks for all your support!

closed 2022-04-14 by helmar74