arendst/Tasmota #15372: Tasmota devices are unavailable during 30 seconds every minute when they cannot reach NTP servers to update the time

Heisenberg2980, 2022-04-15:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When a Tasmota device cannot reach the NTP servers to update the time, it becomes unavailable during 30 seconds every minute (i.e. device is not available 50% of the time).
The way I found this issue is by installing Pihole to resolve DNS queries within my network. As soon as I enabled Pihole and changed the router settings to use Pihole as DNS server, all tasmota lights started to have this behaviour.
Please note I am aware Pihole should not be blocking the NTP servers, but that is a different issue I will raise in the Pihole repository.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP8266 light bulb
- [x] Tasmota binary firmware version number used: 11.1.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
00:04:04.164 MQT: stat/tasmota_E02CA3/RESULT = {"NAME":"HP-BR40-12W","GPIO":[0,0,0,0,2912,417,0,0,416,0,2944,0,0,0],"FLAG":0,"BASE":27}
00:04:04.376 MQT: stat/tasmota_E02CA3/RESULT = {"Module":{"0":"HP-BR40-12W"}}
00:04:04.632 MQT: stat/tasmota_E02CA3/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"2912":"SM16716 CLK"},"GPIO5":{"417":"PWM2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"416":"PWM1"},"GPIO13":{"0":"None"},"GPIO14":{"2944":"SM16716 DAT"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
00:05:58.899 MQT: stat/tasmota_E02CA3/STATUS = {"Status":{"Module":0,"DeviceName":"Cloakroom Bulb","FriendlyName":["Cloakroom Bulb"],"Topic":"tasmota_E02CA3","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
00:05:58.934 MQT: stat/tasmota_E02CA3/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:03:19","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":333,"BCResetTime":"2020-08-05T21:03:19","SaveCount":6282,"SaveAddress":"FB000"}}
00:05:58.965 MQT: stat/tasmota_E02CA3/STATUS2 = {"StatusFWR":{"Version":"11.1.0(tasmota)","BuildDateTime":"2022-04-13T06:40:42","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"437/699"}}
00:05:58.987 MQT: stat/tasmota_E02CA3/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["2ND-FLOOR-2G",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0002A009","2805C80001000600003C5A0A000000000000","00000200","00006000","00004000"]}}
00:05:59.002 MQT: stat/tasmota_E02CA3/STATUS4 = {"StatusMEM":{"ProgramSize":626,"Free":376,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","04000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,56,62","Sensors":"1,2,3,4,5,6"}}
00:05:59.039 MQT: stat/tasmota_E02CA3/STATUS5 = {"StatusNET":{"Hostname":"tasmota-E02CA3-3235","IPAddress":"192.168.0.178","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.0.1","DNSServer2":"0.0.0.0","Mac":"BC:DD:C2:E0:2C:A3","Webserver":2,"HTTP_API":1,"WifiConfig":5,"WifiPower":17.0}}
00:05:59.062 MQT: stat/tasmota_E02CA3/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.31","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E02CA3","MqttUser":"heisenberg","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":60,"SOCKET_TIMEOUT":15}}
00:05:59.087 MQT: stat/tasmota_E02CA3/STATUS7 = {"StatusTIM":{"UTC":"1970-01-01T00:05:59","Local":"1970-01-01T00:05:59","StartDST":"1970-01-01T00:00:00","EndDST":"1970-01-01T00:00:00","Timezone":"+00:00","Sunrise":"20:13","Sunset":"05:47"}}
00:05:59.106 MQT: stat/tasmota_E02CA3/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:05:59"}}
00:05:59.119 MQT: stat/tasmota_E02CA3/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:05:59","Uptime":"0T00:03:19","UptimeSec":199,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Dimmer":100,"Color":"0,0,0,0,255","HSBColor":"0,0,0","White":100,"CT":500,"Channel":[0,0,0,0,100],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"2ND-FLOOR-2G","BSSId":"20:0D:B0:58:41:FA","Channel":6,"Mode":"11n","RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:07:12.437 NTP: Sync time...
00:07:40.421 NTP: Unable to resolve IP address
00:07:40.430 SRC: WebGui from 192.168.0.5
00:07:40.441 MQT: tele/tasmota_E02CA3/STATE = {"Time":"1970-01-01T00:07:40","Uptime":"0T00:04:06","UptimeSec":246,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":2851,"MqttCount":1,"POWER":"OFF","Dimmer":100,"Color":"0,0,0,0,255","HSBColor":"0,0,0","White":100,"CT":500,"Channel":[0,0,0,0,100],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"2ND-FLOOR-2G","BSSId":"20:0D:B0:58:41:FA","Channel":6,"Mode":"11n","RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:04"}}
00:07:40.481 MQT: stat/tasmota_E02CA3/RESULT = {"POWER":"OFF"}
00:07:40.485 MQT: stat/tasmota_E02CA3/POWER = OFF

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Block the NTP servers in your network and then restart the Tasmota light. 

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Tasmota devices should be available all the time regardless of their time being updated or not.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2022-04-15:
This a well-known thing outside the direct control of Tasmota. If you want proper operation, make sure to have a real time source (NTP, GPS or RTC chip), or maybe use the Time command to set a fake time.

Heisenberg2980, 2022-04-15:
Thanks for the quick response, I will open a PR to add this info to the FAQ

closed 2022-04-15 by Heisenberg2980

Heisenberg2980, 2022-04-15:
PR created: https://github.com/tasmota/docs/pull/969

ascillato, 2022-04-15:
Problem already reported in the opened https://github.com/arendst/Tasmota/issues/14394

ascillato, 2022-04-15:
As it is a known bug that needs to be corrected and not a normal behavior, it is better to continue the report and the work as an issue and not as a part of the documentation.