arendst/Tasmota #15374: Negative Active Power using BL0942 Energy Monitor Chip since Tasmota Version 11.1.

sergios100, 2022-04-15:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic):LSPA8
- [x] Tasmota binary firmware version number used: 11.1.0.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: ota
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
00:00:00.001 HDW: ESP8285
00:00:00.046 CFG: Loaded from flash at FB, Count 47
00:00:00.266 QPC: Reset
00:00:00.277 Project tasmota - Tasmota Version 11.1.0.1(tasmota)-2_7_4_9(2022-04-14T21:36:33)
00:00:00.281 SNS: Hardware Serial
00:00:00.886 WIF: Connecting to AP1 Apple Network Channel 11 BSSId 24:F5:A2:23:E7:E0 in mode 11n as tasmota-A852C2-4802...
00:00:02.771 WIF: Connected
00:00:03.024 HTP: Web server active on tasmota-A852C2-4802 with IP address 192.168.1.35
18:47:02.036 RSL: INFO1 = {"Info1":{"Module":"Generic","Version":"11.1.0.1(tasmota)","FallbackTopic":"cmnd/DVES_A852C2_fb/","GroupTopic":"cmnd/tasmotas/"}}
18:47:02.038 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-A852C2-4802","IPAddress":"192.168.1.35"}}
18:47:02.040 RSL: INFO3 = {"Info3":{"RestartReason":"Software/System restart","BootCount":13}}
18:47:02.041 RSL: RESULT = {"POWER":"ON"}
18:47:02.043 RSL: POWER = ON
18:47:06.421 RSL: STATE = {"Time":"2022-04-15T18:47:06","Uptime":"0T00:00:10","UptimeSec":10,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"Apple Network","BSSId":"24:F5:A2:23:E7:E0","Channel":11,"Mode":"11n","RSSI":40,"Signal":-80,"LinkCount":1,"Downtime":"0T00:00:04"}}
18:47:06.430 RSL: SENSOR = {"Time":"2022-04-15T18:47:06","ENERGY":{"TotalStartTime":"2022-01-18T19:13:55","Total":0.002,"Yesterday":0.000,"Today":0.002,"Period": 0,"Power":-0,"ApparentPower":75,"ReactivePower":75,"Factor":-0.00,"Voltage":229,"Current":0.327}}
18:52:06.420 RSL: STATE = {"Time":"2022-04-15T18:52:06","Uptime":"0T00:05:10","UptimeSec":310,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"Apple Network","BSSId":"24:F5:A2:23:E7:E0","Channel":11,"Mode":"11n","RSSI":46,"Signal":-77,"LinkCount":1,"Downtime":"0T00:00:04"}}
18:52:06.428 RSL: SENSOR = {"Time":"2022-04-15T18:52:06","ENERGY":{"TotalStartTime":"2022-01-18T19:13:55","Total":0.002,"Yesterday":0.000,"Today":0.002,"Period": 0,"Power":-0,"ApparentPower":75,"ReactivePower":75,"Factor":-0.00,"Voltage":229,"Current":0.326}}
18:53:58.923 CMD: Backlog Template; Module; GPIO 255
18:53:58.982 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
18:53:59.187 RSL: RESULT = {"Module":{"0":"Generic"}}
18:53:59.444 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"3200":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"7520":"BL0942 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"32":"Button1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"224":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
18:54:13.687 CMD: Backlog Template; Module; GPIO 255
18:54:13.751 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
18:54:13.958 RSL: RESULT = {"Module":{"0":"Generic"}}
18:54:14.214 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"3200":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"7520":"BL0942 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"32":"Button1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"224":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
18:56:48.313 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_A852C2","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
18:56:48.318 RSL: STATUS1 = {"StatusPRM":{"Baudrate":4800,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota-minimal.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:09:52","StartupUTC":"2022-04-15T16:46:56","Sleep":50,"CfgHolder":4617,"BootCount":13,"BCResetTime":"2022-01-18T19:13:55","SaveCount":49,"SaveAddress":"FA000"}}
18:56:48.325 RSL: STATUS2 = {"StatusFWR":{"Version":"11.1.0.1(tasmota)","BuildDateTime":"2022-04-14T21:36:33","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"364/699"}}
18:56:48.330 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Apple Network",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004000"]}}
18:56:48.341 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":626,"Free":376,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","04000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,56,62","Sensors":"1,2,3,4,5,6"}}
18:56:48.346 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-A852C2-4802","IPAddress":"192.168.1.35","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"C4:4F:33:A8:52:C2","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
18:56:48.350 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A852C2","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:56:48.355 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-04-15T16:56:48","Local":"2022-04-15T18:56:48","StartDST":"2022-03-27T01:00:00","EndDST":"2022-10-30T01:00:00","Timezone":99,"Sunrise":"06:35","Sunset":"20:13"}}
18:56:48.358 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
18:56:48.366 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-04-15T18:56:48","ENERGY":{"TotalStartTime":"2022-01-18T19:13:55","Total":0.002,"Yesterday":0.000,"Today":0.002,"Power":-0,"ApparentPower":75,"ReactivePower":75,"Factor":-0.00,"Voltage":229,"Current":0.327}}}
18:56:48.372 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-04-15T18:56:48","Uptime":"0T00:09:52","UptimeSec":592,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"Apple Network","BSSId":"24:F5:A2:23:E7:E0","Channel":11,"Mode":"11n","RSSI":44,"Signal":-78,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

Issue occurs immediately

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Download Firmware Version 11.1. or newer. Was working on 11.0.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

I expect the consumed Power not to be negative, as it was before Version 11.1

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

Attached is a 70W incandescent light bulb.
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2022-04-16:
Caused by this https://github.com/arendst/Tasmota/pull/15265

Will re-evaluate as I see some issues in there,

barbudor, 2022-04-16:
It may be possible that different hardware designers have been using power monitoing chips in one direction or the other. The previous abs() was then masking this.
May be introducing yet another SO for changing the sign ?

ascillato2, 2022-04-16:
Closing as the issue has been fixed by Theo. Thanks.

sergios100, 2022-04-21:
Thanks everyone for the quick help!
One Question: the issue seems to be fixed, but where can I get the fixed release?
I tried this here: [http://ota.tasmota.com/tasmota/tasmota.bin.gz](url) but my issue still persists.

Thanks!

sfromis, 2022-04-21:
Fix should be included in current development binary, the URL you link. Better check the "Build Date & Time" on your device, like on the Tasmota "Information" page, to verify which one you have flashed.

sergios100, 2022-04-21:
Hm, ok I think than we still have the issue:
[image]

[image]

once again `Status 0:`
```lua
16:22:21.796 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_A852CB","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
16:22:21.801 RSL: STATUS1 = {"StatusPRM":{"Baudrate":4800,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota-minimal.bin.gz","RestartReason":"Software/System restart","Uptime":"0T06:13:40","StartupUTC":"2022-04-21T08:08:41","Sleep":50,"CfgHolder":4617,"BootCount":36,"BCResetTime":"2022-01-18T19:13:55","SaveCount":192,"SaveAddress":"F4000"}}
16:22:21.806 RSL: STATUS2 = {"StatusFWR":{"Version":"11.1.0.1(tasmota)","BuildDateTime":"2022-04-20T21:05:51","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"364/699"}}
16:22:21.810 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Apple Network",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004000"]}}
16:22:21.820 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":626,"Free":376,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","04000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
16:22:21.825 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-A852CB-4811","IPAddress":"192.168.1.5","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"C4:4F:33:A8:52:CB","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
16:22:21.828 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A852CB","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
16:22:21.833 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-04-21T14:22:21","Local":"2022-04-21T16:22:21","StartDST":"2022-03-27T01:00:00","EndDST":"2022-10-30T01:00:00","Timezone":99,"Sunrise":"06:24","Sunset":"20:22"}}
16:22:21.836 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
16:22:21.845 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-04-21T16:22:21","ENERGY":{"TotalStartTime":"2022-01-18T19:13:55","Total":-0.000,"Yesterday":0.000,"Today":-0.000,"Power":-0,"ApparentPower":71,"ReactivePower":71,"Factor":-0.00,"Voltage":233,"Current":0.306}}}
16:22:21.850 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-04-21T16:22:21","Uptime":"0T06:13:40","UptimeSec":22420,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"Apple Network","BSSId":"24:F5:A2:23:E7:E0","Channel":11,"Mode":"11n","RSSI":76,"Signal":-62,"LinkCount":3,"Downtime":"0T00:00:12"}}}
```
and output of command: `Backlog Template; Module; GPIO 255`:
```lua
16:25:51.137 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
16:25:51.344 RSL: RESULT = {"Module":{"0":"Generic"}}
16:25:51.558 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"3200":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"7520":"BL0942 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"32":"Button1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"224":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```

sergios100, 2022-04-21:
Hi Theo!
Yes now it works after I did send the command "PowerSet":

[image]

Thank you very much!

closed 2022-04-21 by ascillato2

Le-Duc0069, 2024-05-24:
> Chào Theo! Vâng, bây giờ nó hoạt động sau khi tôi đã gửi lệnh "PowerSet":
> 
> [image]
> 
> Cảm ơn rất nhiều!

How do you use powerset to make it a positive value? I did it but it didn't work

g6094199, 2024-06-05:
i see this on 14.1.0 too on a tuya device

[image]