arendst/Tasmota #15391: Compile errors with #undef USE_AUTOCONF #undef USE_BERRY.

mikep1998, 2022-04-17:
### PROBLEM DESCRIPTION
Compile errors with 
#undef USE_AUTOCONF 
#undef USE_BERRY.

ERROR:
Linking error 
in function `re_pattern_split_run':
<artificial>:(.text+0x3cc4): undefined reference to `berry_malloc'

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32_____
- [x] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Compile with 
#undef USE_AUTOCONF
#undef USE_BERRY.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2022-04-18:
Hi,

I can't reproduce the error. Sorry.

I have just compiled `tasmota32` adding `#undef USE_AUTOCONF` and `#undef USE_BERRY` to **_user_config_override.h_**, and it compiled ok without errors.

Please, try again with latest Tasmota codebase. Also it would be a good practice to delete the `.cache` folder.

Thanks.

closed 2022-04-18 by ascillato2

mikep1998, 2022-04-18:
Please compile for tasmota32-lvgl with  #undef USE_AUTOCONF
#undef USE_BERRY.

sfromis, 2022-04-18:
The "preselected" tasmota32-xxxxxx builds are not meant to amend (add/subtract features), but only to use as is. To make your own mix, you should start with just "tasmota" and make your selections.

mikep1998, 2022-04-18:
tasmota32-lvgl is the simplest to get to the error.

sfromis, 2022-04-18:
The error is not with Tasmota, but with you doing something which is not meant to be supported.

mikep1998, 2022-04-18:
LVGL was supported without berry now it is giving errors

sfromis, 2022-04-18:
This is not a question of what is what is supported with LVGL, but about that you cannot take the tasmota32-lvgl build and add/subtract features, within the standard build setup. Instead you have to start from just "tasmota32", and then add/substract LVGL, Berry and so on, to create your own mix.

sfromis, 2022-04-18:
IIRC, the LVGL build has been expanded to include OpenHASP for an enhanced way of doing display stuff, and that implementation needs Berry too.

mikep1998, 2022-04-18:
Ok thanks for the info. I will upload my configuration built on tasmota32 that causes the error

sfromis, 2022-04-18:
To avoid the requirement for Berry, then you may have to `#undef USE_LVGL_OPENHASP`

mikep1998, 2022-04-18:
Thanks for the info. I will try that. 
Question shouldn't USE_BERRY be a higher priority then some of the other #defines that seem to require Berry or just enable Berry?

mikep1998, 2022-04-18:
#undef USE_LVGL_OPENHASP makes no difference. It does not appear to be defined in the first place.

sfromis, 2022-04-18:
There is no "should" here, there is no 100% cross-connection of features and prerequisites. It is not implausible that you may have to selectively disable further features, if you insist on the non-standard config of not having Berry included. By now, more and more newer Tasmota features depend on having Berry present.

mikep1998, 2022-04-18:
Ok I have got it down to building for tasmota32 with

#define USE_LVGL
#define USE_DISPLAY

#undef USE_AUTOCONF
#undef USE_BERRY

fails.

ascillato, 2022-04-18:
`USE_LVGL` and `USE_DISPLAY` requires `USE_BERRY` to work. Sorry, they can't be disabled.

Why you want to disable Berry? ESP32 has enough RAM and FLASH. Why standard `tasmota32-lvgl` is not enough that you need to compile your own version?

mikep1998, 2022-04-18:
Ok, thanks for your support. I will work this out my self.