arendst/Tasmota #15395: Exception 29 on Webinterface Access after connecting Waveshare 4.2 ePaper display

qroac, 2022-04-18:
### PROBLEM DESCRIPTION
After connecting and configuring a Waveshare 4.2 ePaper display to a Wemos D1 mini as described [here ](https://github.com/arendst/Tasmota/discussions/11850#discussioncomment-718091), I can no longer use the web interface.
Each attempt to open the interface ends up in a restart with exception code 29.
A webconsole that has been opened before is still functional and shows the output of the module.

Disconnecting the display from the module makes no difference.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): __Generic (18) on Wemos D1 mini ESP8266EX__
- [x] Tasmota binary firmware version number used: __11.1.0__
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: __Tasmotizer__
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
14:59:42.800 CMD: Backlog Template; Module; GPIO 255
14:59:42.802 SRC: WebConsole from 192.168.0.27
14:59:42.805 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 26, Pld -99, Data 'Template; Module; GPIO 255'
14:59:43.100 SRC: Backlog
14:59:43.103 CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 0, Pld -99, Data ''
14:59:43.107 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
14:59:43.315 SRC: Backlog
14:59:43.318 CMD: Grp 0, Cmd 'MODULE', Idx 1, Len 0, Pld -99, Data ''
14:59:43.321 RSL: RESULT = {"Module":{"18":"Generic"}}
14:59:43.564 SRC: Backlog
14:59:43.567 CMD: Grp 0, Cmd 'GPIO', Idx 1, Len 3, Pld 255, Data '255'
14:59:43.576 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"864":"SSPI MOSI"},"GPIO13":{"896":"SSPI SCLK"},"GPIO14":{"6016":"EPaper42 CS"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
  No rules used

```
- [x] Provide the output of this command: `Status 0`:
```lua
14:58:42.141 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
14:58:42.146 RSL: STATUS = {"Status":{"Module":18,"DeviceName":"Luftsensor Wohnzimmer","FriendlyName":["Luftsensor Wohnzimmer"],"Topic":"tasmota_3A2E84","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
14:58:42.176 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Exception","Uptime":"0T00:08:09","StartupUTC":"2022-04-18T13:50:33","Sleep":50,"CfgHolder":4617,"BootCount":13,"BCResetTime":"2022-04-18T14:17:59","SaveCount":20,"SaveAddress":"3FB000"}}
14:58:42.207 RSL: STATUS2 = {"StatusFWR":{"Version":"11.1.0(TasmoCompiler-esp82664M)","BuildDateTime":"2022.04.18 12:16:25","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"469/699"}}
14:58:42.227 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["fuermann-iot",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A190000000000","00000080","00006000","00004000"]}}
14:58:42.262 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":818,"Free":1228,"Heap":4,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640D8","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FA0458F","04261FA1","75F607CF","5B4C13C4","D836FDC1","082D60BC","003E9000","0400E8E0"],"Drivers":"1,2,3,4,6,8,9,10,13,18,19,22,24,26,27,29,30,35,37,38,45,50,56,59,62","Sensors":"1,2,3,4,5,6,7,8,9,14,15,17,18,19,20,21,26,27,31,38,39,42,44,46,47,55,56,58,59,63,65,66,72,73,74,88,89,91,92,93"}}
14:58:42.302 RSL: STATUS5 = {"StatusNET":{"Hostname":"air-livingroom","IPAddress":"192.168.252.127","Gateway":"192.168.252.1","Subnetmask":"255.255.252.0","DNSServer1":"192.168.252.1","DNSServer2":"0.0.0.0","Mac":"24:A1:60:3A:2E:84","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
14:58:42.325 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3A2E84","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
14:58:42.346 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-04-18T13:58:42","Local":"2022-04-18T14:58:42","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"05:19","Sunset":"19:09"}}
14:58:42.364 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-04-18T14:58:42"}}
14:58:42.370 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-04-18T14:58:42","Uptime":"0T00:08:09","UptimeSec":489,"Heap":4,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"XXX","BSSId":"XXX","Channel":8,"Mode":"11n","RSSI":100,"Signal":-33,"LinkCount":1,"Downtime":"0T00:00:04"}}}
14:58:42.407 RSL: STATUS12 = {"StatusSTK":{"Exception":29,"Reason":"Exception","EPC":["4027cee8","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["40281a37","402565f5","40256931","402569b6","40256a90","4023d228","4026ddd0","402c0baf","402321ac","4023d26a","402c0baf","4023d338","402c0baf","4023d376","4026ddd0","40240514","402b3947","402c15c6","402c0baf","402078bf","402c0baf","40241c8c","40241e3c","4024c0f2","402568b2","40256e81","4026e424","402321ac","402757b6","40275814","401000e1"]}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:00:00.001 HDW: ESP8266EX
00:00:00.013 UFS: FlashFS mounted with 1992 kB free
00:00:00.067 CFG: Loaded from File, Anzahl 15
00:00:00.072 QPC: Count 1
00:00:00.078 SPI: Software using GPIO13(CLK) and GPIO12(MOSI)
00:00:00.080 ROT: Mode 1
00:00:08.522 DSP: E-Paper 4.2
00:00:08.530 SRC: Restart
00:00:08.532 Projekt tasmota - Luftsensor Wohnzimmer Version 11.1.0(TasmoCompiler-esp82664M)-2_7_4_9(2022.04.18 12:16:25)
[... 89 lines cut ...]
```

2. Flash it to the ESP and configure it as Generic (18)

3. Open two web interfaces, one for configuration and one with web console

4. Solder the BS jumper of the display from 0 to 1 to make it 3-wire SPI and connect the display as shown [here](https://github.com/arendst/Tasmota/discussions/11850#discussioncomment-718091)

5. Configure D5,D6,D7 according to the image (D5 as ePaper 42 cs)

6. Hit the save button, make sure to have the console still open

The ESP restarts, the display initializes and the console shows a E-Paper 4.2.
From now on, each time you try to open the webinterface, you see the initialization sequence of the display and a restart with Exception Code 29 in the console.

### EXPECTED BEHAVIOUR
The configuration of a display doesn't affect the web interface and yield a restart with exception each time the webinterface is called.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2022-04-19:
be aware that this display needs 15k RAM for screen buffer. This is probably too much for an esp8266 if you enable other options that consume memory.
i think it crashes because of low heap memory.
you should use an esp32 with this display, it has much more RAM available

ascillato, 2022-04-19:
Just for testing, the exception occurs if you disable the following?
```cpp
#define USE_EXPRESSION
#define SUPPORT_IF_STATEMENT
#define USE_INFLUXDB
```

qroac, 2022-04-20:
Thanks for the hint on the memory limitation @gemu2015 

I tried the suggestion of @ascillato, and yes, without expressions, if-statements and influxdb, it worked.

I think I can manage without expressions and ifs, but as environment sensor I still wanted to report measurements to influxdb.

So I first un-checked the display-support in TasmoCompiler, which previously added support for a huge range of displays with I2C and SPI and manually enabled USE_SPI, USE_DISPLAY and USE_DISPLAY_EPAPER_42
I also removed USE_AWATCH and SHOW_SPLASH.

And now the memory consumption seems to match my esp8266 :)

Thank you for your advice.

closed 2022-04-20 by qroac