arendst/Tasmota #15405: DeepSleepTime isn't working, Exception:28, Reason:Blocked Loop

mitch-geht-ab, 2022-04-19:
### PROBLEM DESCRIPTION
After setting DeepSleepTime 14400 (4h) Wemos reboots every (more or less) 150 sec. with Exception:28 and Reason:Blocked Loop.
Sensor (SR04) is working. As I can observe until now, with DeepSleepTime of 1h or 2h is working fine. After disable it with mqtt to 0 and set it via Console to 4h I see this strange behaviour.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Basic Generic 18
- [x] Tasmota binary firmware version number used: 11.1.0(sensors)  2_7_4_9/2.2.2-dev(38a443e)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
20:35:08.553 MQT: tele/zisterne/SENSOR = {"Time":"2022-04-19T20:35:08","SR04":{"Distance":36.193}}
20:35:10.395 WIF: Sending Gratuitous ARP
20:35:12.966 WIF: Checking connection...
20:35:18.398 MQT: tele/zisterne/STATE = {"Time":"2022-04-19T20:35:18","Uptime":"0T00:01:10","UptimeSec":70,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"waifai-iot","BSSId":"82:8A:20:54:7C:A5","Channel":11,"Mode":"11n","RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:04"}}
20:35:18.423 MQT: tele/zisterne/SENSOR = {"Time":"2022-04-19T20:35:18","SR04":{"Distance":36.193}}
20:35:28.428 MQT: tele/zisterne/STATE = {"Time":"2022-04-19T20:35:28","Uptime":"0T00:01:20","UptimeSec":80,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"waifai-iot","BSSId":"82:8A:20:54:7C:A5","Channel":11,"Mode":"11n","RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:04"}}
20:35:28.452 MQT: tele/zisterne/SENSOR = {"Time":"2022-04-19T20:35:28","SR04":{"Distance":36.175}}
20:35:32.972 WIF: Checking connection...
20:35:38.397 MQT: tele/zisterne/STATE = {"Time":"2022-04-19T20:35:38","Uptime":"0T00:01:30","UptimeSec":90,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"waifai-iot","BSSId":"82:8A:20:54:7C:A5","Channel":11,"Mode":"11n","RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:04"}}
20:35:38.421 MQT: tele/zisterne/SENSOR = {"Time":"2022-04-19T20:35:38","SR04":{"Distance":36.175}}
20:35:44.211 HTP: Configuration
20:35:45.661 HTP: Main Menu
20:35:47.735 HTP: Information
20:35:48.392 MQT: tele/zisterne/STATE = {"Time":"2022-04-19T20:35:48","Uptime":"0T00:01:40","UptimeSec":100,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":30,"MqttCount":1,"Wifi":{"AP":1,"SSId":"waifai-iot","BSSId":"82:8A:20:54:7C:A5","Channel":11,"Mode":"11n","RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:04"}}
20:35:48.417 MQT: tele/zisterne/SENSOR = {"Time":"2022-04-19T20:35:48","SR04":{"Distance":36.228}}
20:35:52.926 WIF: Checking connection...
20:35:58.408 MQT: tele/zisterne/STATE = {"Time":"2022-04-19T20:35:58","Uptime":"0T00:01:50","UptimeSec":110,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"waifai-iot","BSSId":"82:8A:20:54:7C:A5","Channel":11,"Mode":"11n","RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:04"}}
20:35:58.433 MQT: tele/zisterne/SENSOR = {"Time":"2022-04-19T20:35:58","SR04":{"Distance":36.211}}
20:36:08.385 MQT: tele/zisterne/STATE = {"Time":"2022-04-19T20:36:08","Uptime":"0T00:02:00","UptimeSec":120,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"waifai-iot","BSSId":"82:8A:20:54:7C:A5","Channel":11,"Mode":"11n","RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:04"}}
20:36:08.412 MQT: tele/zisterne/SENSOR = {"Time":"2022-04-19T20:36:08","SR04":{"Distance":36.246}}
20:36:10.413 WIF: Sending Gratuitous ARP
20:36:12.936 WIF: Checking connection...
20:36:18.410 MQT: tele/zisterne/STATE = {"Time":"2022-04-19T20:36:18","Uptime":"0T00:02:10","UptimeSec":130,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"waifai-iot","BSSId":"82:8A:20:54:7C:A5","Channel":11,"Mode":"11n","RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:04"}}
20:36:18.436 MQT: tele/zisterne/SENSOR = {"Time":"2022-04-19T20:36:18","SR04":{"Distance":36.211}}
20:36:28.409 MQT: tele/zisterne/STATE = {"Time":"2022-04-19T20:36:28","Uptime":"0T00:02:20","UptimeSec":140,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"waifai-iot","BSSId":"82:8A:20:54:7C:A5","Channel":11,"Mode":"11n","RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:04"}}
20:36:28.436 MQT: tele/zisterne/SENSOR = {"Time":"2022-04-19T20:36:28","SR04":{"Distance":36.246}}
20:36:29.016 HTP: Main Menu
20:36:30.609 HTP: Console
20:36:32.689 CMD: Backlog Template; Module; GPIO 255
20:36:32.691 SRC: WebConsole from 192.168.2.5
20:36:32.694 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 26, Pld -99, Data 'Template; Module; GPIO 255'
20:36:32.766 SRC: Backlog
20:36:32.769 CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 0, Pld -99, Data ''
20:36:32.776 MQT: stat/zisterne/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
20:36:32.933 WIF: Checking connection...
20:36:32.982 SRC: Backlog
20:36:32.985 CMD: Grp 0, Cmd 'MODULE', Idx 1, Len 0, Pld -99, Data ''
20:36:32.989 MQT: stat/zisterne/RESULT = {"Module":{"18":"Generic"}}
20:36:33.233 SRC: Backlog
20:36:33.236 CMD: Grp 0, Cmd 'GPIO', Idx 1, Len 3, Pld 255, Data '255'
20:36:33.246 MQT: stat/zisterne/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"1888":"SR04 Ech/RX"},"GPIO13":{"1856":"SR04 Tri/TX"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
20:36:38.394 MQT: tele/zisterne/STATE = {"Time":"2022-04-19T20:36:38","Uptime":"0T00:02:30","UptimeSec":150,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":44,"MqttCount":1,"Wifi":{"AP":1,"SSId":"waifai-iot","BSSId":"82:8A:20:54:7C:A5","Channel":11,"Mode":"11n","RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:04"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
[... 15 lines cut ...]

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

- Set deepsleep to 4h via console

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

wakeup from deepsleep after (more or less) 4h, read sensor, go to deepsleep

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2022-04-21:
I will take a look on this. There is no code change if it is 1h or 4h. Anyhow, thanks for bringing it up

mitch-geht-ab, 2022-04-22:
Have tested it with 1h, 2h, 3h... working like a charm.
As far as I switch to 4h I got this strange behaviour.
As you can see in my splunk logs for mqtt msg. and the console command to set DeepSleepTime.
Set it at around 12:06... expected next wakeup at 16 but had some flapping until 12:30 and than again at 15:36 til 16:16.

```
2022-04-22 16:16:04.429 [vent.ItemStateChangedEvent] - zisterne_deepsleep changed from {"DeepSleep":{"Time":"2022-04-23T00:00:00","Epoch":1650672000}} to {"DeepSleep":{"Time":"2022-04-22T20:00:00","Epoch":1650657600}}
2022-04-22 16:13:20.406 [vent.ItemStateChangedEvent] - zisterne_deepsleep changed from {"DeepSleep":{"Time":"2022-04-22T20:00:00","Epoch":1650657600}} to {"DeepSleep":{"Time":"2022-04-23T00:00:00","Epoch":1650672000}}
2022-04-22 16:10:41.224 [vent.ItemStateChangedEvent] - zisterne_deepsleep changed from {"DeepSleep":{"Time":"2022-04-23T00:00:00","Epoch":1650672000}} to {"DeepSleep":{"Time":"2022-04-22T20:00:00","Epoch":1650657600}}
2022-04-22 16:07:57.621 [vent.ItemStateChangedEvent] - zisterne_deepsleep changed from {"DeepSleep":{"Time":"2022-04-22T20:00:00","Epoch":1650657600}} to {"DeepSleep":{"Time":"2022-04-23T00:00:00","Epoch":1650672000}}
2022-04-22 16:05:16.278 [vent.ItemStateChangedEvent] - zisterne_deepsleep changed from {"DeepSleep":{"Time":"2022-04-23T00:00:00","Epoch":1650672000}} to {"DeepSleep":{"Time":"2022-04-22T20:00:00","Epoch":1650657600}}
2022-04-22 16:02:34.841 [vent.ItemStateChangedEvent] - zisterne_deepsleep changed from {"DeepSleep":{"Time":"2022-04-22T20:00:00","Epoch":1650657600}} to {"DeepSleep":{"Time":"2022-04-23T00:00:00","Epoch":1650672000}}	
2022-04-22 15:59:53.435 [vent.ItemStateChangedEvent] - zisterne_deepsleep changed from {"DeepSleep":{"Time":"2022-04-22T16:00:00","Epoch":1650643200}} to {"DeepSleep":{"Time":"2022-04-22T20:00:00","Epoch":1650657600}}
2022-04-22 15:39:02.146 [vent.ItemStateChangedEvent] - zisterne_deepsleep changed from {"DeepSleep":{"Time":"2022-04-22T20:00:00","Epoch":1650657600}} to {"DeepSleep":{"Time":"2022-04-22T16:00:00","Epoch":1650643200}}
2022-04-22 15:36:20.789 [vent.ItemStateChangedEvent] - zisterne_deepsleep changed from {"DeepSleep":{"Time":"2022-04-22T16:00:00","Epoch":1650643200}} to {"DeepSleep":{"Time":"2022-04-22T20:00:00","Epoch":1650657600}}
2022-04-22 12:33:06.576 [vent.ItemStateChangedEvent] - zisterne_deepsleep changed from {"DeepSleep":{"Time":"2022-04-22T20:00:00","Epoch":1650657600}} to {"DeepSleep":{"Time":"2022-04-22T16:00:00","Epoch":1650643200}}
2022-04-22 12:30:25.185 [vent.ItemStateChangedEvent] - zisterne_deepsleep changed from {"DeepSleep":{"Time":"2022-04-22T16:00:00","Epoch":1650643200}} to {"DeepSleep":{"Time":"2022-04-22T20:00:00","Epoch":1650657600}}
2022-04-22 12:27:45.741 [vent.ItemStateChangedEvent] - zisterne_deepsleep changed from {"DeepSleep":{"Time":"2022-04-22T20:00:00","Epoch":1650657600}} to {"DeepSleep":{"Time":"2022-04-22T16:00:00","Epoch":1650643200}}
2022-04-22 12:25:02.384 [vent.ItemStateChangedEvent] - zisterne_deepsleep changed from {"DeepSleep":{"Time":"2022-04-22T16:00:00","Epoch":1650643200}} to {"DeepSleep":{"Time":"2022-04-22T20:00:00","Epoch":1650657600}}
2022-04-22 12:22:23.279 [vent.ItemStateChangedEvent] - zisterne_deepsleep changed from {"DeepSleep":{"Time":"2022-04-22T20:00:00","Epoch":1650657600}} to {"DeepSleep":{"Time":"2022-04-22T16:00:00","Epoch":1650643200}}
2022-04-22 12:19:41.574 [vent.ItemStateChangedEvent] - zisterne_deepsleep changed from {"DeepSleep":{"Time":"2022-04-22T16:00:00","Epoch":1650643200}} to {"DeepSleep":{"Time":"2022-04-22T20:00:00","Epoch":1650657600}}
2022-04-22 12:16:58.172 [vent.ItemStateChangedEvent] - zisterne_deepsleep changed from {"DeepSleep":{"Time":"2022-04-22T20:00:00","Epoch":1650657600}} to {"DeepSleep":{"Time":"2022-04-22T16:00:00","Epoch":1650643200}}
2022-04-22 12:14:16.753 [vent.ItemStateChangedEvent] - zisterne_deepsleep changed from {"DeepSleep":{"Time":"2022-04-22T16:00:00","Epoch":1650643200}} to {"DeepSleep":{"Time":"2022-04-22T20:00:00","Epoch":1650657600}}
2022-04-22 12:11:35.393 [vent.ItemStateChangedEvent] - zisterne_deepsleep changed from {"DeepSleep":{"Time":"2022-04-22T20:00:00","Epoch":1650657600}} to {"DeepSleep":{"Time":"2022-04-22T16:00:00","Epoch":1650643200}}
2022-04-22 12:08:53.984 [vent.ItemStateChangedEvent] - zisterne_deepsleep changed from {"DeepSleep":{"Time":"2022-04-22T16:00:00","Epoch":1650643200}} to {"DeepSleep":{"Time":"2022-04-22T20:00:00","Epoch":1650657600}}
2022-04-22 12:06:34.555 [vent.ItemStateChangedEvent] - zisterne_deepsleep changed from {"DeepSleepTime":14400} to {"DeepSleep":{"Time":"2022-04-22T16:00:00","Epoch":1650643200}}
2022-04-22 12:06:34.243 [vent.ItemStateChangedEvent] - zisterne_deepsleep changed from {"DeepSleepTime":0} to {"DeepSleepTime":14400}
```

Some Logs from console

```
...
12:06:16.528 HTP: Main Menu
12:06:22.385 MQT: tele/zisterne/STATE = {"Time":"2022-04-22T12:06:22","Uptime":"0T00:03:40","UptimeSec":220,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"waifai-iot","BSSId":"82:8A:20:54:7C:A5","Channel":11,"Mode":"11n","RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:04"}}
12:06:22.412 MQT: tele/zisterne/SENSOR = {"Time":"2022-04-22T12:06:22","SR04":{"Distance":38.386}}
12:06:26.514 WIF: Checking connection...
[... cut ...]
12:06:34.296 CFG: Saved to flash at F5, Count 71, Bytes 4096
12:06:34.408 DSL: Reset wrong settings wakeup: 0, slip 10560
12:06:34.415 MQT: stat/zisterne/RESULT = {"DeepSleep":{"Time":"2022-04-22T16:00:00","Epoch":1650643200}}
```

with setting to 14400 you can follow it on splunk logs (see above)

How do I switch DeepSleepTime?

- Set retain mqtt cmnd/zisterne/DeepSleepTime to 0
- Wait for reaching next wakeup
- delete retain mqtt DeepSleepTime msg
- set new DeepSleepTime via tasmota console

Is there anything wrong?

mitch-geht-ab, 2022-04-28:
Even with 3h DeepDleep there are some blackouts:

```
2022-04-27 21:04:45.532	{"DeepSleep":{"Time":"2022-04-28T00:00:00","Epoch":1651104000}}
2022-04-27 18:00:18.956	{"DeepSleep":{"Time":"2022-04-27T21:00:00","Epoch":1651093200}}
2022-04-27 17:51:09.834	{"DeepSleep":{"Time":"2022-04-27T18:00:00","Epoch":1651082400}}
2022-04-27 17:48:26.434	{"DeepSleep":{"Time":"2022-04-27T21:00:00","Epoch":1651093200}}
2022-04-27 15:14:32.367	{"DeepSleep":{"Time":"2022-04-27T18:00:00","Epoch":1651082400}}
2022-04-27 12:01:45.433	{"DeepSleep":{"Time":"2022-04-27T15:00:00","Epoch":1651071600}}
2022-04-27 09:00:12.228	{"DeepSleep":{"Time":"2022-04-27T12:00:00","Epoch":1651060800}}
2022-04-27 08:49:31.731	{"DeepSleep":{"Time":"2022-04-27T09:00:00","Epoch":1651050000}}
2022-04-27 08:46:50.301	{"DeepSleep":{"Time":"2022-04-27T12:00:00","Epoch":1651060800}}
2022-04-27 06:00:00.623	{"DeepSleep":{"Time":"2022-04-27T09:00:00","Epoch":1651050000}}
2022-04-27 05:57:32.976	{"DeepSleep":{"Time":"2022-04-27T06:00:00","Epoch":1651039200}}
2022-04-27 05:54:54.229	{"DeepSleep":{"Time":"2022-04-27T09:00:00","Epoch":1651050000}}
2022-04-27 03:00:13.202	{"DeepSleep":{"Time":"2022-04-27T06:00:00","Epoch":1651039200}}
2022-04-27 02:53:13.365	{"DeepSleep":{"Time":"2022-04-27T03:00:00","Epoch":1651028400}}
2022-04-27 02:50:32.248	{"DeepSleep":{"Time":"2022-04-27T06:00:00","Epoch":1651039200}}
```

I guess (but I'm very sure) that exception:28 is the reason, even if I'm not recording the whole day the console.

BR
mitch

stefanbode, 2022-05-04:
I can confirm, that I have the same behavior here. In the old version there was a 71min limit and a small wakeup. This worked fine. Now in arduino core the deepsleeptime is a int32 and a longer sleep should be possible. But as you said it looks strange....

mitch-geht-ab, 2022-05-04:
@stefanbode 
> ... In the old version there was a 71min limit and a small wakeup. This worked fine. Now in arduino core the  ...
can you tell me when this change was done, in which version?

As far as I can see in my notes from starting with tasmota+wemos, the earliest version was 8.5.1 or 8.3.1

stefanbode, 2022-05-04:
@barbudor , @ascillato2 : can you help me to minimize the casts is this expression?
`uint64_t deepsleepticker = (uint64_t) ((uint64_t)100 * (uint64_t)RtcSettings.deepsleep_slip * ((uint64_t)RtcSettings.nextwakeup - (uint64_t)LocalTime()));`

these are more or less all uint32. I know that the following is NOT enough:
`uint64_t deepsleepticker = (uint64_t) (100 * RtcSettings.deepsleep_slip * (RtcSettings.nextwakeup - LocalTime()));`

stefanbode, 2022-05-04:
The Exeption 28 seems to be a second bug. Dig into this now.

stefanbode, 2022-05-04:
@arendst : Hi Theo, I can confirm that the device executes the command ESP.deepSleep(). Then after some time, there is this exception 28 and the device is not waked up upfront. Looks like this happens in deepsleep, but this makes absolutely no sense.

stefanbode, 2022-05-04:
Thats the stacktrace:

```
Decoding stack results
0x4025e16e: pgm_read_byte_inlined at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/sys/xtensa/sys/pgmspace.h line 72
0x4025e5e6: memchr_P at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/sys/xtensa/string_pgmspace.c line 351
0x4025e830: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 180
0x4025e830: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 180
0x4025e8fa: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x4025e8fa: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x4025e8fa: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x4025e8fa: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x4025e830: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 180
0x4025f35c: _printf_i at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c line 246
0x4025e8fa: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x4025e8fa: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x4025eae4: _svfprintf_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 531
0x4025e8fa: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x4025ed2f: _svfprintf_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 667
0x4025e0dc: vsnprintf at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/vsnprintf.c line 42
0x4025e099: _vsnprintf_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/vsnprintf.c line 73
0x4025e8fa: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x4025e099: _vsnprintf_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/vsnprintf.c line 73
0x4025e0dc: vsnprintf at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/vsnprintf.c line 42
0x4025e170: pgm_read_byte_inlined at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/sys/xtensa/sys/pgmspace.h line 72
0x4025e5e6: memchr_P at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/sys/xtensa/string_pgmspace.c line 351
0x4025e830: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 180
0x4025e830: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 180
0x4025e8fa: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x4025e8fa: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x4025e8fa: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x4025e8fa: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x4025e830: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 180
0x4025f35c: _printf_i at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c line 246
0x4025e8fa: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x4025eae4: _svfprintf_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 531
0x4025f35c: _printf_i at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c line 246
0x4025e8fa: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x4025e8fa: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x4025e8fa: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x4025e8fa: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x4025e830: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 180
0x4025f35c: _printf_i at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c line 246
0x4025e8fa: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x4025e8fa: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x4025eae4: _svfprintf_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 531
0x4025e099: _vsnprintf_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/vsnprintf.c line 73
0x4025e8fa: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x4025e099: _vsnprintf_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/vsnprintf.c line 73
0x4025e0dc: vsnprintf at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/vsnprintf.c line 42
0x4025e0dc: vsnprintf at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/vsnprintf.c line 42
0x4025e150: snprintf_P at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/sys/xtensa/stdio_pgmspace.c line 33
0x4025e0dc: vsnprintf at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/vsnprintf.c line 42
```

arendst, 2022-05-04:
@stefanbode Hi Stepfan. Do you get a Blocked Loop with this exception or just the exception. If just the exception that's weird as it happens in the xtensa code not the arduino core.

stefanbode, 2022-05-04:
I go back to the old method of having a very small wakeup every hour and go sleep again if there is time. This was working and it looks like there is something special with the uint64_t range of deepsleep.

ascillato2, 2022-05-04:
Fixed by https://github.com/arendst/Tasmota/pull/15536.

closed 2022-05-04 by ascillato2