arendst/Tasmota #15435: Disable ON/OFF in image

dmasar, 2022-04-23:
### PROBLEM DESCRIPTION
Hi,

I would like to ask wheter there is posibility disable ON/OFF switch functionality at all.
I mean to build image with it disabled.

Motivation:
I have tasmota devices used only for power metering. no need to turn off them.
And I need to be sure nobody turn off them. Wheter it is because of mistake, home automation errors, configuration error, or by intention. I just need "powered on socket with power metering".
I have searched my_user_config.h, but didnt found nothing related.

more context:
as example am using device with Template {"NAME":"Gosund SP111 V","GPIO":[57,255,56,255,132,134,0,0,131,17,0,21,0],"FLAG":0,"BASE":45};

Any hint appreciated.

thanks.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [+ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ +] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ +] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [+ ] Self-compiled
  tag: v11.1.0
- [ +] Flashing tools used: esptool.py
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
09:45:37.774 MQT: dom/garage/socket1-gosund75-socket/stat/RESULT = {"NAME":"Gosund SP111 V","GPIO":[321,1,320,1,2656,2720,0,0,2624,32,0,224,0,0],"FLAG":0,"BASE":45}
09:45:37.997 MQT: dom/garage/socket1-gosund75-socket/stat/RESULT = {"Module":{"0":"Gosund SP111 V"}}
09:45:38.253 MQT: dom/garage/socket1-gosund75-socket/stat/RESULT = {"GPIO0":{"321":"Led_i2"},"GPIO1":{"0":"None"},"GPIO2":{"320":"Led_i1"},"GPIO3":{"0":"None"},"GPIO4":{"2656":"HLWBL CF1"},"GPIO5":{"2720":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"32":"Button1"},"GPIO14":{"0":"None"},"GPIO15":{"224":"Relay1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-04-23:
This is not an issue and you should have open a discussion instead.

Check doc for [PowerOnState](https://tasmota.github.io/docs/Commands/#poweronstate)
Default value for PowerOnState can be hardcoded in your `user_config_override.h` with
```
#define APP_POWERON_STATE  POWER_ALL_ALWAYS_ON
```

You can also replace the `Relay` GPIO by `OUTPUT_HI` which will always drive the relay ON

dmasar, 2022-04-23:
Thank you this is exactly what I need.

closed 2022-04-23 by dmasar