arendst/Tasmota #15438: scd30 sensor not recognized when Power Cycle of ESP8266 (Wemos)

solarslurpi, 2022-04-23:
### PROBLEM DESCRIPTION
scd30 sensor with ESP826 Wemos is not detected on Power cycle.  The scd30 sensor is detected and readings are available on `restart 1`

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): __Wemos D1 (clone?) EDP286 _ Adafruit scd30 sensor___
- [ ] Tasmota binary firmware version number used: _____
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ x] Flashing tools used: __https://tasmota.github.io/install/___
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
14:45:05.628 CMD: Backlog Template; Module; GPIO 255
14:45:05.705 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
14:45:05.913 RSL: RESULT = {"Module":{"0":"Generic"}}
14:45:06.168 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"640":"I2C SDA"},"GPIO5":{"608":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"4704":"ADC Input"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
ame":["Tasmota"],"Topic":"tasmota_25EEA5","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
14:46:01.494 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota-minimal.bin.gz","RestartReason":"External System","Uptime":"1T22:12:05","StartupUTC":"2022-04-21T15:33:56","Sleep":50,"CfgHolder":4617,"BootCount":22,"BCResetTime":"2022-04-20T23:07:20","SaveCount":45,"SaveAddress":"F5000"}}
14:46:01.499 RSL: STATUS2 = {"StatusFWR":{"Version":"11.1.0.1(sensors)","BuildDateTime":"2022-04-19T20:29:35","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"369/699"}}
14:46:01.502 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["happyday",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008109","2805C80001000600003C5A0A190000000000","00000080","00006000","00004000"]}}
14:46:01.517 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":682,"Free":320,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000000","04000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34,62","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74"}}
14:46:01.522 RSL: STATUS5 = {"StatusNET":{"Hostname":"snifferbuddy","IPAddress":"192.168.86.41","Gateway":"192.168.86.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.86.1","DNSServer2":"0.0.0.0","Mac":"BC:DD:C2:25:EE:A5","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
14:46:01.526 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_25EEA5","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
14:46:01.531 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-04-23T13:46:01","Local":"2022-04-23T14:46:01","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"05:43","Sunset":"19:52"}}
14:46:01.576 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-04-23T14:46:01","ANALOG":{"A0":15},"SCD30":{"CarbonDioxide":614,"eCO2":583,"Temperature":71.9,"Humidity":56.3,"DewPoint":55.5},"TempUnit":"F"}}
14:46:01.581 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-04-23T14:46:01","Uptime":"1T22:12:05","UptimeSec":166325,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"happyday","BSSId":"7C:2E:BD:91:8C:F7","Channel":1,"Mode":"11n","RSSI":72,"Signal":-64,"LinkCount":3,"Downtime":"0T00:00:10"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Power cycle an ESP826 (wemos - not sure that matters).

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I expected to see CO2 readings.

This is what I did to fix it for me:
```
void Scd30Detect(void)
{
#ifdef ESP32
  if (!I2cSetDevice(SCD30_ADDRESS)) { return; }
#endif
  scd30.begin();
```
ie.: Not running I2cSetDevice() if the chipset is 8266.  This is not a robust way.  However, I take it the scd30 needs more time to get back to Tasmota than Tasmota is giving it...I don't want to introduce a delay.  But I do not have enough context in the code to fix timing of loading i2c drivers.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2022-04-23:
I've also seen SCD30 not being detected for power-up boot with ESP8266, but only after a restart. Maybe the sensor is slow to boot till ready on the I2C bus. Saw no similar trouble with ESP32.

I'm concerned that the "fix" suggested here is not good, as I suspect that it will work badly in case of the driver being included in a build where no SCD30 sensor is present. Has this been tested?

arnowelzel, 2022-08-13:
I also experience this issue. When power cycle a D1 mini connected to a SCD30 I have to reset the device once to get it recognize the SCD30.

arendst, 2022-08-14:
From the datasheet it needs 2 seconds to power up. The current code doesn't allow this.

I'll order one and rewrite the setup code once received.

arendst, 2022-08-14:
@anowelzel could you pls try to change
```
bool Xsns42(byte function)
{
  if (!I2cEnabled(XI2C_29)) { return false; }

  bool result = false;

  if (FUNC_INIT == function) {
    Scd30Detect();
  }
  else if (scd30Found) {
```

into
```
bool scd30InitOnce = false;

bool Xsns42(byte function)
{
  if (!I2cEnabled(XI2C_29)) { return false; }

  bool result = false;

/*
  if (FUNC_INIT == function) {
    Scd30Detect();
  }
*/
  if (!scd30InitOnce && (FUNC_EVERY_SECOND == function) && (TasmotaGlobal.uptime > 2)) {
    scd30InitOnce = true;
    Scd30Detect();
  }
  else if (scd30Found) {
```

and report back.

arendst, 2022-08-14:
@sfromis if you can find the time could pls try the above code change to see if it fixes the power on SCD30 detection?

arnowelzel, 2022-08-14:
> @arnowelzel could you pls try to change
> 
[...]
> into
>
> ```
> bool scd30InitOnce = false;
>
> bool Xsns42(byte function)
> {
>   if (!I2cEnabled(XI2C_29)) { return false; }
>
>   bool result = false;
>
> /*
>   if (FUNC_INIT == function) {
>     Scd30Detect();
>   } [...]

Just tried that change and looks good so far. Even after a power cycle the SCD30 was detected properly and everything is fine:

[image]

The code makes sense as well - wait until the global uptime is more than 2 seconds before trying to detect the SCD30.

Thanks for the quick fix!

arnowelzel, 2022-08-14:
Just for completeness - after I let some fresh air in, the readings also got down:

[image]

sfromis, 2022-08-14:
I can also see it working after a power cycle restart.
`00:00:01.925 I2C: SCD30 found at 0x61`

arendst, 2022-08-15:
Thx. Integrated just now.

closed 2022-08-15 by arendst

DrUseless, 2022-09-15:
Has this issue been fixed in the Tasmota-Sensors.bin (12.1.1) file. If I load the latest release of the bin file I still have the issue of the SCD30 not being detected after a power cycle. Only when I restart from the web interface it detects the SCD30. I'm a bit of a newbie so I apologies if this has been addressed somewhere else.

arendst, 2022-09-15:
Should have been fixed in 12.1.1

If not try latest dev release (available in an hour) which has an additional waiting second before initializing

DrUseless, 2022-09-15:
Thank I loaded the new Dev file . Stupid question. Think I might be missing something. With the Dev Release I'm still experiencing the power cycle issue. Do I need to build my own bin file where I #define USE_SCD30? Currently I'm just flashing my D1 Mini with the Tasmota-sensor.bin file and then just assigning the I2C ports. Doing it this way gives me the issue with the power cycle . Like I said Newbie here 🤦🏻‍♂️

arendst, 2022-09-15:
As I thought. The Wemos D1 is flakey at power on.

I just included a wait for a second before init I2C. Than it works.

I will need time to find a solution.

arendst, 2022-09-15:
OK.

In latest dev there is a new command called ``SetOption46 0..255`` allowing you to stall initialization for 0..255 * 10 milliseconds to let stabilize the local Wemos power.

You might want to experiment with values like ``SO46 10`` os ``SO46 20`` for a 100mSec or 200mSec delay.

If you own the Adafruit SCD30  you might also want to power the device from 5V as it draws a lot of power when measuring (notice the fainting power led because the wemos LDO just cannot provide enough power for the device.

YMMV.

DrUseless, 2022-09-17:
Apologies for the late response. So I tried the suggestion you said with the new Dev firmware Tasmota-Sensor.bin file and SetOption46 0-255. I still can't get the scd30 to work on power up. No matter the value I set.  I will either move to a different MCU or perhaps try running a rule for the device to restart after a Power cycle. Thanks

DrUseless, 2022-09-19:
Okay so I tried another D1 mini and the SetOption46 200 does work. Think there might have been an issue with the one I initially tested on. So every time I power cycle the SCD30 comes online . Thanks for the assistance in getting this too function. Much Appreciated

kdelios, 2023-11-09:
I have exactly the same behavior with esp32 (esp32 wroom-32 development board). Tested at 13.2.0, 13.2.0.2 and older 13.1.0 with no success. Each (100%) power recycle leads to unrecognized sensor with empty sensor details etc. Sensor is a genuine Sensirion SCD30, supplied with 5V.
Only after restart is OK

kdelios, 2023-11-09:
Thanks I saw the possible solution. Didn't tried yet, (I'm out of office) and I have no physical access to it. 
I am panning to do it tomorrow morning and I will report back.

sfromis, 2023-11-09:
Did a small test hooking up a SCD30, and can confirm that it was not found on first boot after power up. When restarting, it was found, but I notice that other I2C sensors on same board were found less than 0.3 seconds after boot, while the SCD30 was only found nearly 2.7 seconds later. So there is already a delay for SCD30.

When using `SetOption46 255` (the max of 2.55 secs), the whole I2C initialization was delayed by that time, and SCD30 was found at first boot after power up. It may be feasible with a shorter delay than 255, if you want to make boot a bit faster.

kdelios, 2023-11-10:
SO46 seems to work.
In any case, as a failsafe action I am detecting tasmota restart season as "Vbat power on reset" at node red and publish a restart 1 mqtt message after 5 sec.

B0rax, 2024-04-29:
I think this issue should be opened again. I am facing the same issue with an ESP32 C3 and the SCD30 sensor. Even SO46 255 does not fix the issue. Only a software restart allows the sensor to be recognized.

sfromis, 2024-04-29:
With ESP32-C3, Tasmota 13.4.1.2 dev build and `SetOption46 255`, right after power on:
`"SCD30":{"CarbonDioxide":769,"eCO2":798,"Temperature":26.54,"Humidity":38.42,"DewPoint":11.20,"HeatIndex":26.48},"TempUnit":"C"}`

B0rax, 2024-05-01:
I am running a pre compiled binary from the web flasher: Tasmota 13.4.1.2 (5ecd45e-tasmota32).
The board I am using is an ESP32 c3 super mini.
I2C is connected to gpio 3 (SDA) and 4 (SCL).

It is definitely not recognized on every normal boot.

jeremyherbert, 2024-09-18:
I am running 14.2.0 with an SCD30 on this board: https://wiki.luatos.org/chips/esp32c3/board.html

I2CScan detects the device, and I see the following detection on startup:

```
00:00:00.199 CMD: Using USB CDC
00:00:00.200 HDW: ESP32-C3 v0.3 
00:00:00.213 UFS: FlashFS mounted with 304 kB free
00:00:00.223 CFG: Loaded from File, Count 38
00:00:00.225 SER: Set to 8N1 115200 bit/s
00:00:00.226 SER: HWCDC supports 115200 bit/s only
00:00:00.230 QPC: Count 1
00:00:02.553 I2C: Bus1 using GPIO06(SCL) and GPIO07(SDA)
00:00:02.587 BRY: Berry initialized, RAM used 4244 bytes
00:00:02.598 Project tasmota - Tasmota-SCD30-B Version 14.2.0(release-tasmota32)-3_0_4(2024-08-14T12:40:03)
00:00:04.001 WIF: Connecting to AP1 turtle_IoT Channel 1 BSSId 72:A7:41:BB:6F:19 in mode HT40 as tasmota-79B5A8-5544...
00:00:05.698 I2C: SCD30 found at 0x61
00:00:06.094 WIF: Connected
00:00:06.301 HTP: Web server active on tasmota-79B5A8-5544 with IP address 192.168.1.238
08:27:07.301 MQT: Attempting connection...
08:27:07.320 MQT: Connected
08:27:07.324 MQT: tele/tasmota_79B5A8/LWT = Online (retained)
08:27:07.326 MQT: cmnd/tasmota_79B5A8/POWER = 
08:27:07.331 MQT: tele/tasmota_79B5A8/INFO1 = {"Info1":{"Module":"ESP32C3","Version":"14.2.0(release-tasmota32)","FallbackTopic":"cmnd/DVES_79B5A8_fb/","GroupTopic":"cmnd/tasmotas/"}}
08:27:07.334 MQT: tele/tasmota_79B5A8/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-79B5A8-5544","IPAddress":"192.168.1.238","IP6Global":"","IP6Local":"fe80::6255:f9ff:fe79:b5a8%st1"}}
08:27:07.337 MQT: tele/tasmota_79B5A8/INFO3 = {"Info3":{"RestartReason":"Software reset digital core","BootCount":27}}
08:27:08.406 MQT: stat/tasmota_79B5A8/STATUS11 = {"StatusSTS":{"Time":"2024-09-18T08:27:08","Uptime":"0T00:00:06","UptimeSec":6,"Heap":189,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":4,"Objects":45},"Wifi":{"AP":1,"SSId":"turtle_IoT","BSSId":"72:A7:41:BB:6F:19","Channel":1,"Mode":"HT40","RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:04"}}}
08:27:08.452 MQT: stat/tasmota_79B5A8/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32c3.bin","RestartReason":"Software reset digital core","Uptime":"0T00:00:06","StartupUTC":"2024-09-18T07:27:02","Sleep":50,"CfgHolder":4617,"BootCount":26,"BCResetTime":"2024-09-18T05:02:45","SaveCount":38}}
08:27:09.711 QPC: Reset
08:27:11.665 MQT: tele/tasmota_79B5A8/STATE = {"Time":"2024-09-18T08:27:11","Uptime":"0T00:00:09","UptimeSec":9,"Heap":189,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":4,"Objects":45},"Wifi":{"AP":1,"SSId":"turtle_IoT","BSSId":"72:A7:41:BB:6F:19","Channel":1,"Mode":"HT40","RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:04"}}
```

But it rarely goes through with the reading of the device and instead I get the amber light stuck on/off.

I have two identical setups, and both exhibit the same behaviour. I have tried `SetOption46 255` but it does not make a difference.

sfromis, 2024-09-18:
The SCD30 driver was already changed to delay detection after boot, instead of needing `SetOption46` delay for power-on boot (not restart boot), and this was about the sensor being detected or not, with your log indeed showing that it was detected.

Testing a SCD30 with another ESP32-C3 board, I saw no trouble.

Maybe your board is not able to supply the needed power, the SCD30 may need a spike of 75 mA during measurement, which is a lot more than most sensors.

jeremyherbert, 2024-09-18:
I have just tried adding 4.7k hardware pullups - no change
Added a lot more decoupling on the 3.3V side - no change
Changed power VIN for sensor to 5V from USB (supply has at least 5V2A behind it) - seems to sort of improve it, it seems that on first power on the SCD30 doesn't show up, after running `restart 1` it seems to work most of the time. 

I will leave the devices running overnight. Though interestingly one of the two units is much worse than the other.

jeremyherbert, 2024-09-18:
OK, sadly no this did not fix this issue.

jeremyherbert, 2024-09-19:
So I read the SCD30 datasheet and it recommends running the I2C interface at 50kHz clock, not 100kHz. Lo and behold, I recompiled tasmota32c3 with a 50kHz SCL frequency and now all of my SCD30 devices are working after a software reset with `Restart 1`. I can still get some to not read or detect the sensor properly on reboot, but after a reset they are 100% working, whereas before they were working less than 50% of the time after a software reset.

The datasheet says the absolute maximum clock rate is 100kHz but also states that you should run at 50kHz. I guess it is possible if different ESP32C3 boards have their base clock frequency slightly high/low this could cause some to work better or worse than others (which is what I observed). Or it may be true that different SCD30 units have a better ability to work closer to 100kHz. Ultimately, the sensor itself is running the I2C slave in software via a STM32 MCU, so it will be quite sensitive to this timing being too fast I imagine.

I think this issue maybe needs to be reopened? There definitely seems to be something here, though I don't think tasmota has any support for changing the I2C clock speed at runtime.

arendst, 2024-09-19:
Thx for testing.

From your comments I can deduct two actions:
- Start recognizing the STM32 controlled device later than shortly after power on. I noticed the same with M5Unit scales, also using STM32. There I delay recognizing by two seconds after power on fixing the detection of the device.
- Dynamically change I2C busspeed from default 110k to 50k while processing SCD30 functions. This needs to be implemented but I see no reason why it could be a problem. I forsee lowering the clock for SCD30 functions and at function exit raising the clock again to last clock speed.

I will implement both options in due time as I'm currently working on I2C implementation changes anyway.

jeremyherbert, 2024-09-19:
Both of these sound great, thank you. I also am going to try to do some more testing at 100kHz to see if I can actually observe the problem occurring with the logic analyser, because sometimes it doesn't come up at physical power on. 

Maybe it is a case of needing a soft reboot if it is stuck in this state.

arendst, 2024-09-19:
At power on I merely suspect the slow start of the STM32. As said, I noticed the same on the M5Unit scales where the STM32 needed almost 900msec to respond to first I2C command. This was too late for Tasmota's initial I2C detection leaving the device undetected.

arendst, 2024-09-19:
Ah, I see the SCD30 driver already has the power-on delay (as suggested earlier in this threat).
```
  // https://github.com/arendst/Tasmota/issues/15438 and datasheet (The boot-up time is < 2 s.)
/*
  if (FUNC_INIT == function) {
    Scd30Detect();
  }
*/
  if (!Scd30.init_once && (FUNC_EVERY_SECOND == function) && (TasmotaGlobal.uptime > 3)) {
    Scd30.init_once = true;
    Scd30Detect();
  }
```

I also see it uses a patched library with specific ESP8266 code:
```
#ifdef ESP8266
    this->pWire->setClockStretchLimit(200000);
#endif
```
So I'll focus on setting the SCD30 clockspeed to 50k

arendst, 2024-09-19:
Try latest dev branch. This loweres I2C clock for SCD30 only from 100k to 50k.

At least on my ESP8266 it still works fine.

jeremyherbert, 2024-09-20:
Thanks. Works well on my latest version.

I have narrowed down the reboot problem further I think, it seems that the the sensor gets stuck reporting "no data" sometimes. Since the device is supposed to sample at 2Hz and there is a "no data" counter in the code, I think the sensor should probably be soft-reset or similar if the counter reaches 5, since that will correspond to 2.5 failed measurements.

Triox85, 2025-02-13:
Hey, 
I read through the entire thread and was initially pleased that the topic was addressed here and hopefully resolved. The current situation for me is that I connect the ESP8266 to voltage, wait for it to start up, see whether the BME280 and the SHT30 as well as the SCD30 are displayed.
BME280 and SHT30 are always displayed, the SCD30 only after a restart via the software buttons.
Then the SCD30 is found, but the data is not sent via MQTT (iObrocker). Increase the idea over time in xsns_42_scd30.ino line 355
 ` if (!Scd30.init_once && (FUNC_EVERY_SECOND == function) && (TasmotaGlobal.uptime > 5)) `
Going from 3 to 5 didn't bring any improvement either.

Also swapping the I2C bus, creating 2x I2C BUS (1 and 2), hanging the device alone on the bus did not bring any improvement.

[Tasmota 14.4.1 (tasmota) von Theo Arends]

Start via voltage

`00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: aus Flash geladen von F8, Anzahl 52
00:00:00.056 QPC: Count 1
00:00:02.562 I2C: Bus1 using GPIO14(SCL) and GPIO12(SDA)
00:00:02.563 I2C: Bus2 using GPIO05(SCL) and GPIO04(SDA)
00:00:02.569 Projekt tasmota - Messgeraet2 Version 14.4.1(tasmota)-2_7_8(2025.02.12 10:02:27)
00:00:02.580 I2C: BMP280 found at 0x76 (bus2)
00:00:02.614 I2C: SHT3X found at 0x44 (bus2)
00:00:04.001 WIF: verbinden mit AP1 WLAN-Kam Channel 1 BSSId FC:EC:DA:E9:3F:E6 in Modus 11n als Messgeraet2...
00:00:05.564 WIF: verbunden
00:00:05.819 HTP: Webserver aktiv Messgeraet2 mit IP-Adresse 192.168.179.88
18:30:48.488 MQT: Verbindungsversuch…
18:30:48.511 MQT: verbunden
18:30:48.514 MQT: tele/tasmota_14B1BF/LWT = Online (beibehalten)
18:30:48.517 MQT: cmnd/tasmota_14B1BF/POWER = 
18:30:48.523 MQT: tele/tasmota_14B1BF/INFO1 = {"Info1":{"Module":"Generic","Version":"14.4.1(tasmota)","FallbackTopic":"cmnd/Messgeraet2_fb/","GroupTopic":"cmnd/tasmotas/"}}
18:30:48.532 MQT: tele/tasmota_14B1BF/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"Messgeraet2","IPAddress":"192.168.179.88"}}
18:30:48.543 MQT: tele/tasmota_14B1BF/INFO3 = {"Info3":{"RestartReason":"External System","BootCount":36}}
18:30:50.271 QPC: Reset
18:30:52.230 MQT: tele/tasmota_14B1BF/STATE = {"Time":"2025-02-13T18:30:52","Uptime":"0T00:00:09","UptimeSec":9,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"WLAN-Kam","BSSId":"FC:EC:DA:E9:3F:E6","Channel":1,"Mode":"11n","RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:04"}}
18:30:52.253 MQT: domoticz/in = {"idx":2,"nvalue":0,"svalue":"23.8;42.4;1","Battery":100,"RSSI":5}
18:30:52.261 MQT: domoticz/in = {"idx":3,"nvalue":0,"svalue":"28.0;0.0;0;1017.1;5","Battery":100,"RSSI":5}
18:30:52.270 MQT: domoticz/in = {"idx":2,"nvalue":0,"svalue":"26.9;36.7;2","Battery":100,"RSSI":5}
18:30:52.279 MQT: tele/tasmota_14B1BF/SENSOR = {"Time":"2025-02-13T18:30:52","AM2301":{"Temperature":23.8,"Humidity":42.4,"DewPoint":10.2,"HeatIndex":23.3},"BMP280":{"Temperature":28.0,"Pressure":1017.1},"SHT3X":{"Temperature":26.9,"Humidity":36.7,"DewPoint":10.8,"HeatIndex":26.6},"PressureUnit":"hPa","TempUnit":"C"}`

Start via Reset

`00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: aus Flash geladen von F7, Anzahl 53
00:00:00.055 QPC: Count 1
00:00:02.562 I2C: Bus1 using GPIO14(SCL) and GPIO12(SDA)
00:00:02.563 I2C: Bus2 using GPIO05(SCL) and GPIO04(SDA)
00:00:02.568 Projekt tasmota - Messgeraet2 Version 14.4.1(tasmota)-2_7_8(2025.02.12 10:02:27)
00:00:02.580 I2C: BMP280 found at 0x76 (bus2)
00:00:02.614 I2C: SHT3X found at 0x44 (bus2)
00:00:04.000 WIF: verbinden mit AP1 WLAN-Kam Channel 1 BSSId FC:EC:DA:E9:3F:E6 in Modus 11n als Messgeraet2...
00:00:06.566 WIF: verbunden
00:00:06.819 HTP: Webserver aktiv Messgeraet2 mit IP-Adresse 192.168.179.88
18:32:52.040 I2C: SCD30 found at 0x61
18:32:52.681 MQT: Verbindungsversuch…
18:32:53.010 MQT: verbunden
18:32:53.014 MQT: tele/tasmota_14B1BF/LWT = Online (beibehalten)
18:32:53.017 MQT: cmnd/tasmota_14B1BF/POWER = 
18:32:53.023 MQT: tele/tasmota_14B1BF/INFO1 = {"Info1":{"Module":"Generic","Version":"14.4.1(tasmota)","FallbackTopic":"cmnd/Messgeraet2_fb/","GroupTopic":"cmnd/tasmotas/"}}
18:32:53.032 MQT: tele/tasmota_14B1BF/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"Messgeraet2","IPAddress":"192.168.179.88"}}
18:32:53.043 MQT: tele/tasmota_14B1BF/INFO3 = {"Info3":{"RestartReason":"Software/System restart","BootCount":37}}
18:32:54.267 QPC: Reset
18:32:57.219 MQT: tele/tasmota_14B1BF/STATE = {"Time":"2025-02-13T18:32:57","Uptime":"0T00:00:11","UptimeSec":11,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"WLAN-Kam","BSSId":"FC:EC:DA:E9:3F:E6","Channel":1,"Mode":"11n","RSSI":56,"Signal":-72,"LinkCount":1,"Downtime":"0T00:00:05"}}
18:32:57.242 MQT: domoticz/in = {"idx":2,"nvalue":0,"svalue":"23.6;40.2;1","Battery":100,"RSSI":5}
18:32:57.249 MQT: domoticz/in = {"idx":3,"nvalue":0,"svalue":"25.7;0.0;0;1017.1;5","Battery":100,"RSSI":5}
18:32:57.258 MQT: domoticz/in = {"idx":2,"nvalue":0,"svalue":"25.1;39.8;2","Battery":100,"RSSI":5}
18:32:57.267 MQT: tele/tasmota_14B1BF/SENSOR = {"Time":"2025-02-13T18:32:57","AM2301":{"Temperature":23.6,"Humidity":40.2,"DewPoint":9.3,"HeatIndex":23.1},"BMP280":{"Temperature":25.7,"Pressure":1017.1},"SHT3X":{"Temperature":25.1,"Humidity":39.8,"DewPoint":10.4,"HeatIndex":24.7},"PressureUnit":"hPa","TempUnit":"C"}`