arendst/Tasmota #15505: [Telegram] tmsend fails to fail any messages unless SetOption132 1 is used

juliogonzalez, 2022-05-01:
### PROBLEM DESCRIPTION
`tmsend` fails to fail any messages unless `SetOption132 1` is used.

As discussed at https://github.com/arendst/Tasmota/discussions/15499, maybe the GoDaddy certificate at tasmota is not the right one anymore.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Basic
- [x] Tasmota binary firmware version number used: 11.1.0.1 
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: Web browser
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

19:21:04.973 CMD: Backlog Template; Module; GPIO 255
19:21:04.995 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
19:21:05.207 RSL: RESULT = {"Module":{"1":"Sonoff Basic"}}
19:21:05.461 RSL: RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

19:21:27.769 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota-WIFI-Madrid","FriendlyName":["Tasmota WIFI Madrid"],"Topic":"tasmota1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
19:21:27.800 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:17:53","StartupUTC":"2022-05-01T18:03:34","Sleep":50,"CfgHolder":4617,"BootCount":27,"BCResetTime":"2022-04-30T23:11:19","SaveCount":67,"SaveAddress":"FA000"}}
19:21:27.827 RSL: STATUS2 = {"StatusFWR":{"Version":"11.1.0.1(tasmota)","BuildDateTime":"2022-05-01T17:58:27","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"567/699"}}
19:21:27.853 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":2,"LogHost":"","LogPort":514,"SSId":["Republica",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C80001000600003C5A00190000000000","00000080","00206000","00004000"]}}
19:21:27.883 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":562,"Free":440,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8D8007C7","04000001","000000C7","000000C0","00000000","0000C080","00000000","04000020"],"Drivers":"1,2,3,4,8,9,10,20,38,40,62","Sensors":"1,2,3,5,6"}}
19:21:27.913 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota1-2092","IPAddress":"192.168.1.18","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.2","DNSServer2":"192.168.1.3","Mac":"84:F3:EB:3E:08:2C","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
19:21:27.939 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-05-01T18:21:27","Local":"2022-05-01T19:21:27","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00"}}
19:21:27.963 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-05-01T19:21:27"}}
19:21:27.987 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-05-01T19:21:27","Uptime":"0T00:17:53","UptimeSec":1073,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"MySSId","BSSId":"XX:XX:XX:XX:XX:XX","Channel":11,"Mode":"11n","RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

19:23:48.297 CMD: tmsend A test
19:23:48.962 RSL: RESULT = {"TmSend":"Failed"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
1. Generate a build with telegram enabled
2. Configure `tmchatid`, `tmtoken` and `tmstate`
3. Try to send a message with `tmsend`, it will fail
4. Run `SetOption132 1`
5. Try to send a message with `tmsend`, it will work

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
`tmsend` should work out of the box, without the need of adjusting the value of `SetOption132`

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Somehow, I think the certificate provided at https://github.com/arendst/Tasmota/blob/development/tasmota/tasmota_ca.ino#L178 is not right anymore.

The text mentions the certificate will expire on 20220523, but according to https://www.sslshopper.com/ssl-checker.html#hostname=api.telegram.org, the certificate for ` Go Daddy Secure Certificate Authority - G2` will expire on 20310503. That certificate seems to be https://certs.godaddy.com/repository/gdig2.crt.pem.

I tried to prepare a PR on my own, but sadly had some problems as you can see at https://github.com/arendst/Tasmota/discussions/15499#discussioncomment-2670009 (I admit I am not experience enough to do such changes :-))

**(Please, remember to close the issue when the problem has been addressed)**

nikvostro, 2022-05-02:
The same for me. Probably stopped working yesterday. Two devices (esp32 and esp8266) with different tokens fails to send messages using tmsend.

FerFemNemBem, 2022-05-04:
Same here.
Have 5 devices that worked since a long time. All of a sudden "tmsend" fails.
```
14:32:12.953 CMD: tmsend test
14:32:12.955 SRC: WebConsole from 192.168.178.77
14:32:12.958 CMD: Grp 0, Cmnd 'TMSEND', Idx 1, Len 4, Data 'test'
14:32:12.961 TGM: sendMessage
14:32:13.590 MQT: steckdose/unten/bad/waschmaschine/RESULT = {"TmSend":"Failed"} 
```

arendst, 2022-05-04:
BTW is there any reason to favour the certificate over the fingerprint?

FerFemNemBem, 2022-05-04:
Can confirm, that `SetOption132 1` solves the issue!
I have in my self compiled tasmota set:
```
#define USE_TELEGRAM
#define USE_TELEGRAM_FINGERPRINT "\xB2\x72\x47\xA6\x69\x8C\x3C\x69\xF9\x58\x6C\xF3\x60\x02\xFB\x83\xFA\x8B\x1F\x23"
```
This worked until - hmmm last week(?) without problems.
 
Is there anything I have to uncomment too?
If not I have to enhance my "USER_BACKLOG" in future builds with `SetOption132 1` to get it work defaultly.

arendst, 2022-05-04:
No uncomments needed. ``SO132 1`` and (default enabled) fingerprint is all you need.

closed 2022-06-03 by github-actions[bot]

FerFemNemBem, 2022-06-16:
Would it make sense to:
`#define MQTT_TLS_FINGERPRINT   true`
as default inside "my_user_config.h"?
So one don't have to `SO132 1` to get the Telegram stuff working.
I don't have the knowledge about side effects - just thinking...

FerFemNemBem, 2022-06-17:
I'm very often flash new Shelly devices with a self compiled tasmota with `#define USE_TELEGRAM`.
This does not work anymore out of the box. 
Maybe set `#define MQTT_TLS_FINGERPRINT true` implicitly if `#define USE_TELEGRAM` is set in standard tasmota?
For now I have adapted my `user_config_override.h` files but maybe others will run into this too and its not obvious that `#define MQTT_TLS_FINGERPRINT true` is mandatory in order to get telegram working.