arendst/Tasmota #15527: Request for telemetry command

ssjophus, 2022-05-03:
### PROBLEM DESCRIPTION
There needs to be a way to poll sensors and return status on request (in addition to or instead of regular intervals), and not just data already in the buffer. Sometimes the sensor data is to old using the minimum 10 sec telemetry period EX; If I need to bring in data from multiple devices (into Node Red) and compare them. 10 seconds is a lifetime when trying to get accurate data. there needs to be a way to send a poll request and have the device send the data at the time of the poll. If all devices see the same poll request, all devices (subscribed) will send there data from the same time period. thus would be accurate to each other. I could have one sensor measuring pressure, and one measuring mADC, the poll would bring in the mADC at the time of a specific pressure.
My personal usage for this is for instrument calibration in environments that are hostile to sensative test equipment, ie wet locations. It is much safer to position my expensive equipment in a safe place and read data with low cost (easily replacable) readers. This fix would only apply to sensors and serial devices as digital inputs already update as soon as a change in state is detected.
I have searched through threads and found a few authors with similar issues but all the ways to circumvent the problem involve a hard rule around a specific value, instead of a value at a called moment. The command: cmnd/topic/Status (10) only relays data that's in the buffer, ie old data, it doesn't update the buffer first.
Thank you for your concideration,

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2022-05-03:
Hi,

The command `teleperiod` without arguments will give you as response the teleperiod time and then the full teleperiod sensor data.

closed 2022-05-03 by ascillato2

sfromis, 2022-05-04:
> The command: cmnd/topic/Status (10) only relays data that's in the buffer, ie old data

This is not correct. Repeated `Status 10` does provide "reasonably current" data.