arendst/Tasmota #15533: Cannot connect to Tasmota device using Hostname

sityware, 2022-05-04:
### PROBLEM DESCRIPTION
Unable to access Tasmota devices using Hostname

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Gosund UP111
- [x] Tasmota binary firmware version number used: 11.1.0 - 2_7_4_9/2.2.2-dev(38a443e)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
11:43:47.785 RSL: RESULT = {"NAME":"Gosund UP111","GPIO":[0,320,0,32,2720,2656,0,0,2624,576,224,0,0,0],"FLAG":0,"BASE":18}
11:43:47.992 RSL: RESULT = {"Module":{"0":"Gosund UP111"}}
11:43:48.250 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"320":"Led_i1"},"GPIO2":{"0":"None"},"GPIO3":{"32":"Button1"},"GPIO4":{"2720":"BL0937 CF"},"GPIO5":{"2656":"HLWBL CF1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"224":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
11:42:51.674 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Light_LivingRoom","FriendlyName":["Light_LivingRoom"],"Topic":"tasmota_6858C2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
11:42:51.701 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:08:29","StartupUTC":"2022-05-04T10:34:22","Sleep":50,"CfgHolder":4617,"BootCount":18,"BCResetTime":"2022-04-19T21:54:27","SaveCount":60,"SaveAddress":"F5000"}}
11:42:51.735 RSL: STATUS2 = {"StatusFWR":{"Version":"11.1.0(tasmota)","BuildDateTime":"2022-04-13T06:40:42","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"380/699"}}
11:42:51.754 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Europa2","Europa"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","000000A0","00006000","00004000"]}}
11:42:51.785 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":626,"Free":376,"Heap":27,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","04000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,56,62","Sensors":"1,2,3,4,5,6"}}
11:42:51.816 RSL: STATUS5 = {"StatusNET":{"Hostname":"lightlivingroom","IPAddress":"192.168.1.32","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"70:03:9F:68:58:C2","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
11:42:51.839 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_6858C2","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
11:42:51.860 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-05-04T10:42:51","Local":"2022-05-04T11:42:51","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"05:24","Sunset":"20:09"}}
11:42:51.877 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
11:42:51.895 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-05-04T11:42:51","ENERGY":{"TotalStartTime":"2022-04-19T21:54:27","Total":0.049,"Yesterday":0.000,"Today":0.000,"Power": 0,"ApparentPower": 0,"ReactivePower": 0,"Factor":0.00,"Voltage": 0,"Current":0.000}}}
11:42:51.915 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-05-04T11:42:51","Uptime":"0T00:08:29","UptimeSec":509,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":2,"SSId":"Europa","BSSId":"62:38:E0:06:0B:DC","Channel":8,"Mode":"11n","RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  

```

### TO REPRODUCE
Set Wi-Fi Config
Set Wi-Fi Hostname (used all lowercase, meeting standards)
My Router is OpenWRT and confirmed other Wi-Fi devices can be accessed via hostname, e.g a Raspberry Pi.
Router and Tasmota device confirmed within the same subnet 255.255.255.0

Access Console and enable mDNS
SetOption55 1

```lua
AP2 SSId (RSSI) | wifinet (58%, -71 dBm) 11n
lightlivingroom
70:04:9C:62:58:C2
ip: 192.168.1.32
gateway: 192.168.1.1
subnet: 255.255.255.0
dns1: 192.168.1.1
dns2: 0.0.0.0

```

### EXPECTED BEHAVIOUR
Able to access Tasmota device via hostname from web browser.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2022-05-04:
You should not mix up expectations for mDNS and what dnsmasq can provide. In my setup, with dnsmasq on my router, and no mDNS whatsoever, dnsmasq will resolve host name to the IP, with no "extra" config done.

Note that precompiled binaries do *not* include mDNS, meaning that `SetOption55` has no effect in this case.

sityware, 2022-05-04:
so it should work 

> You should not mix up expectations for mDNS and what dnsmasq can provide. In my setup, with dnsmasq on my router, and no mDNS whatsoever, dnsmasq will resolve host name to the IP, with no "extra" [...]

I'll remove the mDNS reference to avoid confusion. Thanks
Might be worth someone else testing with OpenWRT.

barbudor, 2022-05-04:
Are you using DHCP ?
I do not expect the hostname to be registered by your OpenWRT DNS if it does not come from a DHCP request?

sfromis, 2022-05-04:
For me, using dnsmasq (on my Google Wifi setup) to provide both DNS and DHCP works nicely to have DNS resolution of local host names, without me even having worried about the details....

sityware, 2022-05-04:
> Are you using DHCP ? I do not expect the hostname to be registered by your OpenWRT DNS if it does not come from a DHCP request?

Yes, I have notice the difference between my Pi and S10 request and Tasmota.
When trying to access via web I receive 'This site can’t be reached'

These are the first lines exclusive to Tasmota device.
dnsmasq-dhcp: DHCPDISCOVER(br-lan) 70:03:9f:68:58:c2
dnsmasq-dhcp: DHCPOFFER(br-lan) 192.168.1.122 70:03:9f:68:58:c2

sityware, 2022-05-04:
Resolved.
OpenWRT has an option under DHCP and DNS > Gerneral Settings

Local domain : lan
Local domain suffix appended to DHCP names and hosts file entries

if I suffix lan on the end http://lightlivingroom.lan/ 
it works now. Maybe if anyone else has the query it could be something like this.

closed 2022-05-04 by sityware