arendst/Tasmota #15547: Homeassistant IP update not detected by tasmota's

helderem, 2022-05-05:
My tasmota devices are setup with MQTT using a hostname ("homeassistant"). The IP address of my MQTT broker has changed because of a router change. Tasmota devices are still connecting to the old IP address, only when I change the value to "homeassistant.local" the change is being picked up by the device.

I've also tried to change the value to "homeAssistant" but that is not picked up as a change, also changing the value to an IP address and then back to the old "homeassistant" value doesn't solve the issue.

Broken state:
[image]

Fixed:
[image]

Is this an caching issue in tasmota or is there a issue with the DNS of my internal network?

closed 2022-05-05 by close-issue-app[bot]

helderem, 2022-05-05:
DNS server has been updated automatically when the tasmota's connected to the new router:
16:13:25.912 CMD: IpAddress4
16:13:25.919 MQT: stat/sonoff_light_hallwaydownstairs/RESULT = {"IPAddress4":"192.168.1.1"}

My old router had the 192.168.178.x range. By the way, I just found out that my opentherm gateway has the same issue. So it might also be an ESP8266 issue.

ascillato2, 2022-05-05:
https://github.com/arendst/Tasmota/discussions/15548