arendst/Tasmota #15550: IRRECEIVE / IRHVAC issue with recognizing the protocols

hristo-atanasov, 2022-05-05:
### PROBLEM DESCRIPTION
Inconsistent recognition of IR remotes from ACs. The issue might be from a long time ago, but it still exists. On very old Tasmota 8.1 ircustom IRHVAC ELECTRA protocol/provider is always recognized perfectly, but on 11.1 and also development branch it is not and showed as UNKNOWN most of the time - like 1 recognized out of 10 times. Sending with IRHVAC is OK, but not the receiving. I tried with IRrecvDumpV2, the tool from IRRemoteESP8266 - the library always recognise it. I guess this might be some timing issue when Tasmota is receiving IR codes. When Tasmota recognize the code it shows 104bits, when it doesn't it shows 106bits. A few times it was recognized as KELVINATOR vendor/protocol. I opened an issue on IRremoteESP8266, but they told me, that if IRrecvDumpV2 always recognize it, the problem might be in the implementation in Tasmota.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): NodeMCU V3
- [x] Tasmota binary firmware version number used: 11.1 IR and development branch enabling irfull
  - [x] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: VS Code and Tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
16:39:48.888 CMD: Backlog Template; Module; GPIO 255
16:39:48.963 MQT: stat/testMultisensor/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
16:39:49.175 MQT: stat/testMultisensor/RESULT = {"Module":{"18":"Generic"}}
16:39:49.430 MQT: stat/testMultisensor/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"1088":"IRrecv"},"GPIO3":{"0":"None"},"GPIO4":{"1056":"IRsend"},"GPIO5":{"160":"Switch1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"608":"I2C SCL"},"GPIO14":{"640":"I2C SDA"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
16:39:50.377 MQT: stat/testMultisensor/RESULT = {"POWER":"OFF"}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
16:41:47.694 CMD: Status 0
16:41:47.702 MQT: stat/testMultisensor/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"testMultisensor","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
16:41:47.736 MQT: stat/testMultisensor/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T00:05:37","StartupUTC":"2022-05-05T15:36:10","Sleep":50,"CfgHolder":4617,"BootCount":15,"BCResetTime":"2022-04-13T13:11:38","SaveCount":14886,"SaveAddress":"FD000"}}
16:41:47.768 MQT: stat/testMultisensor/STATUS2 = {"StatusFWR":{"Version":"11.1.0(TasmoCompiler-esp8266generic)","BuildDateTime":"2022-04-14T18:39:35","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"364/699"}}
16:41:47.792 MQT: stat/testMultisensor/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["gh0s7",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A190000000000","00000080","00006000","00004000"]}}
16:41:47.828 MQT: stat/testMultisensor/STATUS4 = {"StatusMEM":{"ProgramSize":787,"Free":1260,"Heap":10,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0FCA4586","04260001","600207C3","180C0000","D032F981","0009208C","00201000","04000040"],"Drivers":"1,2,4,5,6,8,9,10,12,18,19,22,24,26,27,29,30,35,37,38,45,50,56,62","Sensors":"1,2,4,5,6,7,8,9,14,26,27,38,39,46,47,55,58,59,63,65,72,73,74,88"}}
16:41:47.867 MQT: stat/testMultisensor/STATUS5 = {"StatusNET":{"Hostname":"testMultisensor-7102","IPAddress":"192.168.3.129","Gateway":"192.168.1.1","Subnetmask":"255.255.252.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"FC:F5:C4:94:7B:BE","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
16:41:47.894 MQT: stat/testMultisensor/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.122","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_947BBE","MqttUser":"ghost","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
16:41:47.918 MQT: stat/testMultisensor/STATUS7 = {"StatusTIM":{"UTC":"2022-05-05T15:41:47","Local":"2022-05-05T16:41:47","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"05:22","Sunset":"20:10"}}
16:41:47.939 MQT: stat/testMultisensor/STATUS10 = {"StatusSNS":{"Time":"2022-05-05T16:41:47","Switch1":"ON","BME280":{"Temperature":25.5,"Humidity":27.2,"DewPoint":5.2,"Pressure":1015.6},"PressureUnit":"hPa","TempUnit":"C"}}
16:41:47.956 MQT: stat/testMultisensor/STATUS11 = {"StatusSTS":{"Time":"2022-05-05T16:41:47","Uptime":"0T00:05:37","UptimeSec":337,"Heap":8,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"ghost","BSSId":"F2:9F:C2:6E:75:EF","Channel":1,"Mode":"11n","RSSI":64,"Signal":-68,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
Always recognize the right vendor/protocol as the IRrecvDumpV2 tool.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
I think there is a difference between how Tasmota and IRremoteESP8266 are handling the received IR codes.

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2022-05-05:
Can you check the value of the tolerance `SetOption44` ?

It used to be hard-coded at `25` but was then moved to `SetOption44` which was reused from the deprecated `TUYA_VOLTAGE_ID`. It's possible that there is a left-over in this option.

hristo-atanasov, 2022-05-05:
I tried setting it from 6 all the way up to 50 .. Going from 30 and up it gets even worse .. :)

s-hadinger, 2022-05-05:
Do you by chance have 2 IR devices? It will be difficult for me to test since I don't have the remote.

hristo-atanasov, 2022-05-05:
I have 4 ACs, with 3 different brands, but 3 of them use same remotes. Only one uses different remote/vendor.

s-hadinger, 2022-05-06:
Sorry, I meant 2 IR Tasmota devices. I would be interested in using one to emulate the remote.

hristo-atanasov, 2022-05-06:
I have more than 20 PCs. (I'm assembling them for friends). I already made such tests, but I'll test it again, because I can't remember what was the result. :)

closed 2022-06-05 by github-actions[bot]

Ghvinerias, 2022-11-24:
I am Having the same issue, does anybody have any update?