arendst/Tasmota #15555: shutter tilt discovery for Home Assistant not working

tsrubar, 2022-05-05:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Hi, shutter tilt discovery for Home Assistant not working, even if ShutterTiltConfig1 is set. Using native integration in HA, deleted device from integration, restarted device itself, it reappeared by discovery, but tilt controls not. Sensor entity of tilt is visible.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ V ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ V ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ V ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ V ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ V ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ V ] Device used (e.g., Sonoff Basic): Sonoff Dual R3 v2
- [ V ] Tasmota binary firmware version number used: Tasmota 11.1.0.1 - bluetooth
  - [ V ] Pre-compiled
  - [ X ] Self-compiled
- [ V ] Flashing tools used: Tasmotizer
- [ V ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
21:11:56.478 CMD: Backlog Template; Module; GPIO 255
21:11:56.534 MQT: stat/tasmota_0CFC20/RESULT = {"NAME":"Sonoff Dual R3 v2","GPIO":[32,0,0,0,0,0,0,0,0,576,225,0,0,0,0,0,0,0,0,0,0,3200,8128,224,0,0,0,0,34,33,0,0,0,0,0,0],"FLAG":0,"BASE":1}
21:11:56.736 MQT: stat/tasmota_0CFC20/RESULT = {"Module":{"0":"Sonoff Dual R3 v2"}}
21:11:56.000 MQT: stat/tasmota_0CFC20/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"225":"Relay2"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"3200":"Serial Tx"},"GPIO26":{"8128":"BL0939 Rx"},"GPIO27":{"224":"Relay1"},"GPIO32":{"34":"Button3"},"GPIO33":{"33":"Button2"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ V ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
21:12:41.356 CMD: Backlog Rule1; Rule2; Rule3
21:12:41.411 MQT: stat/tasmota_0CFC20/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":230,"Free":281,"Rules":"ON Button2#State=3 DO Power2 2 ENDON   ON Button2#State=2 DO Backlog ShutterStop; ShutterTiltChange1 -30 ENDON   ON Button3#State=3 DO Power1 2 ENDON   ON Button3#State=2 DO Backlog ShutterStop; ShutterTiltChange1 30 ENDON Rule1 1"}}
21:12:41.641 MQT: stat/tasmota_0CFC20/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
21:12:41.892 MQT: stat/tasmota_0CFC20/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ V ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
21:13:30.808 CMD: Status 0
21:13:30.821 MQT: stat/tasmota_0CFC20/STATUS = {"Status":{"Module":0,"DeviceName":"Zaluzie-spalna","FriendlyName":["Tasmota",""],"Topic":"tasmota_0CFC20","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"","SwitchMode":[5,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1,"InfoRetain":0,"StateRetain":0}}
21:13:30.882 MQT: stat/tasmota_0CFC20/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32-bluetooth.bin","RestartReason":"Software reset CPU","Uptime":"0T04:24:33","StartupUTC":"2022-05-05T14:48:57","Sleep":50,"CfgHolder":4617,"BootCount":101,"BCResetTime":"2022-01-14T11:06:27","SaveCount":342}}
21:13:30.942 MQT: stat/tasmota_0CFC20/STATUS2 = {"StatusFWR":{"Version":"11.1.0.1(bluetooth)","BuildDateTime":"2022-04-13T17:24:59","Core":"2_0_3","SDK":"v.4.4.0","CpuFrequency":80,"Hardware":"ESP32-D0WD-V3","CR":"403/699"}}
21:13:30.978 MQT: stat/tasmota_0CFC20/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Loki",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008029","0A05C80001000600003C5A0A000000000000","40000080","00006000","00004000"]}}
21:13:31.032 MQT: stat/tasmota_0CFC20/STATUS4 = {"StatusMEM":{"ProgramSize":1513,"Free":1856,"Heap":105,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC7C7","00148001","000000CF","010013C0","C0000981","00004080","00200000","0400082C"],"Drivers":"1,2,3,4,5,7,8,9,10,12,16,20,21,24,26,27,29,35,38,50,52,56,59,62,79,85","Sensors":"1,2,3,5,6,52,62,127"}}
21:13:31.096 MQT: stat/tasmota_0CFC20/STATUS5 = {"StatusNET":{"Hostname":"tasmota-0CFC20-7200","IPAddress":"192.168.83.201","Gateway":"192.168.83.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.83.1","DNSServer2":"0.0.0.0","Mac":"E0:E2:E6:0C:FC:20","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
21:13:31.146 MQT: stat/tasmota_0CFC20/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.83.1","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0CFC20","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:13:31.192 MQT: stat/tasmota_0CFC20/STATUS7 = {"StatusTIM":{"UTC":"2022-05-05T19:13:31","Local":"2022-05-05T21:13:31","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":99,"Sunrise":"06:22","Sunset":"21:10"}}
21:13:31.226 MQT: stat/tasmota_0CFC20/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
21:13:31.272 MQT: stat/tasmota_0CFC20/STATUS10 = {"StatusSNS":{"Time":"2022-05-05T21:13:31","ENERGY":{"TotalStartTime":"2022-01-14T11:06:27","Total":0.241,"Yesterday":0.001,"Today":0.007,"Power":[ 0, 0],"ApparentPower":[ 0, 0],"ReactivePower":[ 0, 0],"Factor":[0.00,0.00],"Voltage": 0,"Current":[0.000,0.000],"BL09XX":{"Temperature":31.7}},"ESP32":{"Temperature":49.4},"Shutter1":{"Position":0,"Direction":0,"Target":0,"Tilt":0},"TempUnit":"C"}}
21:13:31.339 MQT: stat/tasmota_0CFC20/STATUS11 = {"StatusSTS":{"Time":"2022-05-05T21:13:31","Uptime":"0T04:24:34","UptimeSec":15874,"Heap":104,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":25,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":40},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Loki","BSSId":"D2:21:F9:17:D9:7D","Channel":1,"Mode":"11n","RSSI":56,"Signal":-72,"LinkCount":1,"Downtime":"0T00:00:03"}}}
21:13:31.396 MQT: stat/tasmota_0CFC20/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":534,"Close":510,"50perc":50,"Delay":3,"Opt":"0000","Calib":[240,433,649,880,1000],"Mode":"1"}}}

```
- [ X ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
N/A

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Enable ShutterTiltConfig1 <a> <b> <c> <d> <e>
Let device be discovered by native Tasmota integration in HA

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Tilt controls should be discovered when ShutterTiltConfig defined as stated in xdrv_12_home_assistant.ino, commit 17066e404c2b15076116d2115280ccea14ca2359

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
N/A

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
N/A

**(Please, remember to close the issue when the problem has been addressed)**

tsrubar, 2022-05-06:
My wild guess is, that [here ](https://github.com/arendst/Tasmota/blob/ebadc63abcf30b22dc062d4e712d981928d222e1/tasmota/xdrv_12_home_assistant.ino#L1053):
`      if (Settings->shutter_tilt_config[3][i] != Settings->shutter_tilt_config[4][i]) {
`
should be [i+1] instead of both [i], since 'i' counter starts at 0, if just one 'for' cycle run in case of one defined shutter and therefore discovery condition is always false if comparing both values multiplied by 'i'.  (if A[i] != B[i] will always be false while i=0)
Sorry if i have mistaken, not a programmer, just a sysadmin using general logic here.

barbudor, 2022-05-06:
HI @tsrubar 
I'm not using Shutters myself but I've looked at the code you are refering and I don't agree
`shutter_tilt_config[3][i]` and `shutter_tilt_config[4][i]` are respectively `openposition` and `closeposition` values for the `ShutterTiltConfig<i>` for the shutter #i
So it wouldn't make sense to compare the openposition of shutter i with close position of shutter i+1

What this code is doing is that it decide if Tilt has been configured.
If Tilt hasn't been configured, open and close position are the same (likely 0) and no Tilt is used
If Tilt has been configured, open and close are necesseraly different and then the Tilt is announced to HA.

Did you used ShutterTiltConfig ?

I hope my explanation makes sense to you.

barbudor, 2022-05-06:
Note that you haven't really filled the section "Steps to reproduce"
We are expecting here the full set of commands that you have used here from a blank Tasmota config to your config.

tsrubar, 2022-05-06:
Hi, first, thank you for your reply. 
As I stated sooner, it was just my guess, I thought that if statement works as evaluation logic if shuttertiltconfig is set and therefore compares shuttertiltconfig 0 0 0 0 0 with shuttertiltconfig <a> <b> <c> <d> <e>.
Yes, I have shuttertiltconfig set, as stated in description of this issue, my values are these and everything works just fine, except HA discovery:
`stat/tasmota_0CFC20/RESULT = {"ShutterTiltConfig1":"SHT:0 -90 0 24 0 -90"}`

Steps to reproduce are as stated already in description, where is also vidible that shuttertiltconfig was used:

- Enable ShutterTiltConfig1
- Let device be discovered by native Tasmota integration in HA

barbudor, 2022-05-06:
Not sure to follow
Don't forget that the index in C starts a 0 so `ShutterTiltConfig1` goes in `shutter_tilt_config[0]`
So `i` is counting from `0` to `MAX-1` and then any message or display shows `i+1` from `1` to `MAX`

But you are refering to code that is depending on legacy HA discovery (SetOption19 1) as per https://github.com/arendst/Tasmota/blob/ebadc63abcf30b22dc062d4e712d981928d222e1/tasmota/xdrv_12_home_assistant.ino#L1034
where `Settings->flag.hass_discovery` is SetOption19 so this code is not executed

As far as I can see, there is no Tasmota Discovery message for `SetOption19 0`
I can't tell if it is missing or if it is because the Tasmota integration plugin do not support it.
It is one of the difficulties to synchronize the development around HA as we have no contacts with the HA team but happy to integrate any PR they would submit.
May be you should ask to the people in charge of the Tasmota plugin if they support announcement of Shutters, and if so, either they submit a PR to Tasmota or they tell us exctly what they expect and we could amend that?

tsrubar, 2022-05-06:
Oh, I didnt notice that it is config for old mqtt discovery at all. Thanks for pointing it out and sorry for false alarm.
I am struggling with this for two weeks now, tried to find out if problem is, that discovery message from tasmota does not contain data about tilt, or it does, but HA tasmota integration wont ingest them. This was point beyond my programming knowledge, so maybe if you could point me in right direction, what to look for in my discovery messages, I could possibly find out on my own. Shutters get recognized in HA Tasmota integration well, apart from tilt controls. I will try to look into its code, maybe i will find something. Tried to ask this in various ways, without opening an issue there, but never got an answer from anybody :(

Discovery messages:
config:
`{"ip":"192.168.83.201","dn":"Zaluzie-spalna","fn":["Tasmota","",null,null,null,null,null,null],"hn":"tasmota-0CFC20-7200","mac":"E0E2E60CFC20","md":"Sonoff Dual R3 v2","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"11.1.0.1","t":"tasmota_0CFC20","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[3,3,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":0,"lt_st":0,"sho":[0,0,0,0],"ver":1}`

sensors:
`{"sn":{"Time":"2022-05-06T11:23:10","ENERGY":{"TotalStartTime":"2022-01-14T11:06:27","Total":0.241,"Yesterday":0.001,"Today":0.007,"Power":[ 0, 0],"ApparentPower":[ 0, 0],"ReactivePower":[ 0, 0],"Factor":[0.00,0.00],"Voltage": 0,"Current":[0.000,0.000],"BL09XX":{"Temperature":31.3}},"ESP32":{"Temperature":47.8},"Shutter1":{"Position":0,"Direction":0,"Target":0,"Tilt":0},"TempUnit":"C"},"ver":1}`

tsrubar, 2022-05-06:
My questions are heavily ignored on HA discord and nowhere on the internet is answer to be found. I mean, am I only one in the whole world willing to use native integration to discover tilt controls for venetian shutter? After two weeks of unsuccessfull attempts to make it work I feel really frustrated now. Especially when I see in statistics that half of HA installations use cover devices. If I do something wrong, I just want to find out what it is, since setting shuttertiltconfig should be enough, but clearly not in my case :(

Jason2866, 2022-05-14:
Have you tried latest development version with [this](https://github.com/arendst/Tasmota/pull/15612) change ?

tsrubar, 2022-05-14:
Not yet, as I would need to test with yet to be released Tasmota 12 in my r2 sonoff shutters which are mounted in walls and dont have test piece to be easily reflashed if something goes wrong. That issue in Home Assistant was opened by me. I checked code changes and it wasnt just hassle with few variables, so will need for Tasmota 12 release anyway. When it comes out, I will test it asap

closed 2022-05-14 by ascillato2

barbudor, 2022-05-14:
Why do you consider it needs Tasmota 12 ? You can flash the development builds. It doesn't make a change.

tsrubar, 2022-05-14:
I've seen it somewhere yesterday, when was reading changes. These sonoffs have esp32 and i am not really sure right now if development builds are usable for them. If so, I could try it.

ascillato, 2022-05-14:
At http://ota.tasmota.com/tasmota32/ you have all the official Tasmota development versions for ESP32.

tsrubar, 2022-05-14:
Just did successful ota to 11.1.0.3.
[Here](https://github.com/home-assistant/core/pull/71789) in proposed change is included in Tasmota v 12.0.0, and daily dev builds starting from 2022-05-14, so I am going to try find right development HA build and test it

ascillato, 2022-05-14:
Tasmota v12 does not exists yet. That is a typo in that issue. All changes that are discussed in this issue and in the one that you have linked are already implemented in Tasmota 11.1.0.3 which can be downloaded from http://ota.tasmota.com/tasmota32/

arendst, 2022-05-14:
No typo. It's the official released discussed with the Hass driver people emontnemery. For now 11.1.0.3 is fine

tsrubar, 2022-05-14:
Just recreated HA container with latest dev image (Home Assistant Core 2022.6.0.dev20220514), deleted device, restarted device, it reappeared in HA with Tassmota 11.1.0.3, but no joy yet :(
[image]

ascillato, 2022-05-14:
@tsrubar Please, report that in https://github.com/home-assistant/core/pull/71789 since from Tasmota side, you have the correct version.

tsrubar, 2022-05-14:
Reported and included discovery messages