arendst/Tasmota #15623: TYWE3S adding an external relay doesn't work

Smandurlo, 2022-05-14:
### PROBLEM DESCRIPTION

I have an ERC-310 dimmer which works ok with TUYA MCU. Inside there is a TYWE3S module. I added an external relay on a free pin and selected relay on the tasmota menu on that pin.

Oddly, it doesn't work. I tried every free good pin (4 5 12 13 14). They all work as switches (input) and as LEDs. So, they work as input and output as LED, but not as relay.

Using the LED configuration, I can drive the external relay, but it "floats" while booting (it "blinks" 1 time like the onboard led) and while restarting/upgrading firmware (4 times in the same sequence).
It is not a floating pin problem, there is a proper pull-down resistor on the pin (1K + 4,7K Ohm to GND). The booting blink is "fine" because it replicates the onboard led, I don't know why it "blinks" 4 times during the restart.

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ERC-310 Tuya MCU dimmer
- [ ] Tasmota binary firmware version number used: 11.0 and 11.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
13:20:22.294 CMD: Backlog Template; Module; GPIO 255
13:20:22.366 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
13:20:22.576 RSL: RESULT = {"Module":{"54":"Tuya MCU"}}
13:20:22.829 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"2272":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"2304":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"288":"Led1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
13:20:24.381 RSL: STATE = {"Time":"2022-05-14T13:20:24","Uptime":"0T00:05:11","UptimeSec":311,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"VillaTorreZisa","BSSId":"E8:DF:70:9D:6F:7B","Channel":6,"Mode":"11n","RSSI":92,"Signal":-54,"LinkCount":1,"Downtime":"0T00:00:05"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
13:22:17.404 CMD: Status 0
13:22:17.409 RSL: STATUS = {"Status":{"Module":54,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_434E14","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
13:22:17.414 RSL: STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:07:04","StartupUTC":"2022-05-14T12:15:13","Sleep":50,"CfgHolder":4617,"BootCount":30,"BCResetTime":"2022-05-14T11:10:08","SaveCount":279,"SaveAddress":"F6000"}}
13:22:17.417 RSL: STATUS2 = {"StatusFWR":{"Version":"11.1.0(tasmota)","BuildDateTime":"2022-04-13T06:40:42","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"372/699"}}
13:22:17.421 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["VillaTorreZisa",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A190000000000","00000080","00006000","00004000"]}}
13:22:17.432 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":626,"Free":376,"Heap":26,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","04000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,56,62","Sensors":"1,2,3,4,5,6"}}
13:22:17.438 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-434E14-3604","IPAddress":"192.168.178.51","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.1","DNSServer2":"0.0.0.0","Mac":"2C:F4:32:43:4E:14","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
13:22:17.442 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_434E14","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
13:22:17.448 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-05-14T12:22:17","Local":"2022-05-14T13:22:17","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"05:09","Sunset":"20:23"}}
13:22:17.453 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-05-14T13:22:17"}}
13:22:17.458 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-05-14T13:22:17","Uptime":"0T00:07:04","UptimeSec":424,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"VillaTorreZisa","BSSId":"E8:DF:70:9D:6F:7B","Channel":6,"Mode":"11n","RSSI":96,"Signal":-52,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

To have a 3,3V output when selecting Relay (as when I select LED).

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-05-14:
This is not an Issue with Tasmota and should have been started as a Discussion

Please close and re-open as a Discussion giving detailed example of how you are connecting your relay in terms of hardware. An ESP pin is not enough to directly drive an electrimechanial relay.

Smandurlo, 2022-05-14:
It is indeed a Tasmota problem, because I am using an external power supply to drive the relay (it is a 12V relay), I used a tester (as relay there is no output and as LED there is 3,3V output as it should be) and finally because when I use the LED option, the relay works as expected (with those flickering I don't understand while restarting).

barbudor, 2022-05-14:
Your configuration above doesn't include any pin defined as Relay
You didn't provided any logs in WebLog 3 or WebLog 4 to show what happens when you try to power on/off such relay

Please fill correctly the template so we can advise.

It may be possible that when TuyaMCU is enabled, all POWER commands go to the TuyaMCU and an local relay may not be possible. But we need you to first do your homework and fll the Issue template.

Smandurlo, 2022-05-14:
this is the weblog 4 after restart 1 and LED configuration:
00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at F6, Count 295
00:00:00.056 QPC: Count 1
00:00:00.058 CFG: CR 372/699, Busy 0
00:00:00.062 ROT: Mode 1
00:00:00.108 SNS: Hardware Serial
00:00:00.108 TYA: Request MCU configuration at 9600 bps
00:00:00.109 SRC: Restart
00:00:00.123 TYA: Send "55aa0006000501010001000d"
00:00:00.145 Project tasmota - Tasmota Version 11.1.0(tasmota)-2_7_4_9(2022-04-13T06:40:42)
00:00:00.147 {"TuyaReceived":{"Data":"55AA03070005010100010011","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
00:00:00.148 TYA: fnId=11 is set for dpId=1
00:00:00.149 TYA: RX Relay-1 --> MCU State: Off Current State:Off
00:00:00.156 TYA: Set WiFi LED 2 (-1)
00:00:00.165 TYA: Send "55aa000300010205"
00:00:00.174 {"TuyaReceived":{"Data":"55AA0303000005","Cmnd":3}}
00:00:00.174 TYA: RX WiFi LED set ACK
00:00:00.224 WIF: Checking connection...
00:00:00.225 WIF: Attempting connection...
00:00:00.562 WIF: Connecting to AP1 VillaTorreZisa Channel 6 BSSId E8:DF:70:9D:6F:7B in mode 11n as tasmota-434E14-3604...
00:00:02.427 WIF: Checking connection...
00:00:02.428 WIF: Attempting connection...
00:00:03.426 WIF: Checking connection...
00:00:03.427 WIF: Connected
00:00:03.679 HTP: Web server active on tasmota-434E14-3604 with IP address 192.168.178.51
00:00:04.402 WIF: Sending Gratuitous ARP
00:00:04.403 NTP: Sync time...
00:00:04.466 RTC: UTC 2022-05-14T12:45:24, DST 2022-03-27T02:00:00, STD 2022-10-30T03:00:00
13:45:24.000 RTC: Synced by NTP
13:45:24.002 TYA: Set WiFi LED 3 (0)
13:45:24.011 TYA: Send "55aa000300010306"
13:45:24.019 {"TuyaReceived":{"Data":"55AA0303000005","Cmnd":3}}
13:45:24.020 TYA: RX WiFi LED set ACK
13:45:24.222 RSL: INFO1 = {"Info1":{"Module":"Tuya MCU","Version":"11.1.0(tasmota)","FallbackTopic":"cmnd/DVES_434E14_fb/","GroupTopic":"cmnd/tasmotas/"}}
13:45:24.224 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-434E14-3604","IPAddress":"192.168.178.51"}}
13:45:24.226 RSL: INFO3 = {"Info3":{"RestartReason":"Software/System restart","BootCount":34}}
13:45:24.228 RSL: RESULT = {"POWER":"OFF"}
13:45:24.230 RSL: POWER = OFF
13:45:24.238 TYA: Send "55aa0001000000"
13:45:24.292 {"TuyaReceived":{"Data":"55AA0301002A7B2270223A22617A743464707065666C656967766230222C2276223A22312E302E30222C226D223A307D2A","Cmnd":1,"CmndData":"7B2270223A22617A743464707065666C656967766230222C2276223A22312E302E30222C226D223A307D"}}
13:45:24.293 TYA: MCU Product ID: {"p":"azt4dppefleigvb0","v":"1.0.0","m":0}
13:45:24.302 TYA: Send "55aa0002000001"
13:45:24.309 {"TuyaReceived":{"Data":"55AA0302000004","Cmnd":2}}
13:45:24.311 TYA: RX MCU configuration Mode=0
13:45:24.312 TYA: Read MCU state
13:45:24.320 TYA: Send "55aa0008000007"
13:45:24.334 {"TuyaReceived":{"Data":"55AA03070005010100010011","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
13:45:24.335 TYA: fnId=11 is set for dpId=1
13:45:24.337 TYA: RX Relay-1 --> MCU State: Off Current State:Off
13:45:24.350 {"TuyaReceived":{"Data":"55AA0307000802020004000003E602","Cmnd":7,"CmndData":"02020004000003E6","DpType2Id2":998,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000003E6"}}}
13:45:24.351 TYA: fnId=0 is set for dpId=2
13:45:24.353 TYA: RX value 998 from dpId 2 
13:45:24.366 {"TuyaReceived":{"Data":"55AA03070008690200040000000080","Cmnd":7,"CmndData":"6902000400000000","DpType2Id105":0,"105":{"DpId":105,"DpIdType":2,"DpIdData":"00000000"}}}
13:45:24.367 TYA: fnId=0 is set for dpId=105
13:45:24.369 TYA: RX value 0 from dpId 105 
13:45:26.412 QPC: Reset
13:45:29.396 RSL: STATE = {"Time":"2022-05-14T13:45:29","Uptime":"0T00:00:11","UptimeSec":11,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"VillaTorreZisa","BSSId":"E8:DF:70:9D:6F:7B","Channel":6,"Mode":"11n","RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:05"}}

Smandurlo, 2022-05-14:
Configuration log with relay option:

13:47:43.962 CMD: Backlog Template; Module; GPIO 255
13:47:44.022 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
13:47:44.230 RSL: RESULT = {"Module":{"54":"Tuya MCU"}}
13:47:44.484 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"2272":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"2304":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"224":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

and weblog 4 with relay option after restart 1

00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at F8, Count 301
00:00:00.056 QPC: Count 1
00:00:00.058 CFG: CR 372/699, Busy 0
00:00:00.062 ROT: Mode 1
00:00:00.108 SNS: Hardware Serial
00:00:00.108 TYA: Request MCU configuration at 9600 bps
00:00:00.109 SRC: Restart
00:00:00.123 TYA: Send "55aa0006000501010001000d"
00:00:00.144 Project tasmota - Tasmota Version 11.1.0(tasmota)-2_7_4_9(2022-04-13T06:40:42)
00:00:00.147 {"TuyaReceived":{"Data":"55AA03070005010100010011","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
00:00:00.148 TYA: fnId=11 is set for dpId=1
00:00:00.149 TYA: RX Relay-1 --> MCU State: Off Current State:Off
00:00:00.156 TYA: Set WiFi LED 2 (-1)
00:00:00.165 TYA: Send "55aa000300010205"
00:00:00.174 {"TuyaReceived":{"Data":"55AA0303000005","Cmnd":3}}
00:00:00.174 TYA: RX WiFi LED set ACK
00:00:00.225 WIF: Checking connection...
00:00:00.225 WIF: Attempting connection...
00:00:00.562 WIF: Connecting to AP1 VillaTorreZisa Channel 6 BSSId E8:DF:70:9D:6F:7B in mode 11n as tasmota-434E14-3604...
00:00:02.425 WIF: Checking connection...
00:00:02.426 WIF: Connected
00:00:02.679 HTP: Web server active on tasmota-434E14-3604 with IP address 192.168.178.51
00:00:03.406 WIF: Sending Gratuitous ARP
00:00:03.407 NTP: Sync time...
00:00:03.464 RTC: UTC 2022-05-14T12:49:19, DST 2022-03-27T02:00:00, STD 2022-10-30T03:00:00
13:49:19.000 RTC: Synced by NTP
13:49:19.002 TYA: Set WiFi LED 3 (0)
13:49:19.011 TYA: Send "55aa000300010306"
13:49:19.019 {"TuyaReceived":{"Data":"55AA0303000005","Cmnd":3}}
13:49:19.020 TYA: RX WiFi LED set ACK
13:49:19.223 RSL: INFO1 = {"Info1":{"Module":"Tuya MCU","Version":"11.1.0(tasmota)","FallbackTopic":"cmnd/DVES_434E14_fb/","GroupTopic":"cmnd/tasmotas/"}}
13:49:19.225 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-434E14-3604","IPAddress":"192.168.178.51"}}
13:49:19.227 RSL: INFO3 = {"Info3":{"RestartReason":"Software/System restart","BootCount":37}}
13:49:19.229 RSL: RESULT = {"POWER1":"OFF"}
13:49:19.231 RSL: POWER1 = OFF
13:49:19.232 RSL: RESULT = {"POWER2":"OFF"}
13:49:19.234 RSL: POWER2 = OFF
13:49:19.242 TYA: Send "55aa0001000000"
13:49:19.295 {"TuyaReceived":{"Data":"55AA0301002A7B2270223A22617A743464707065666C656967766230222C2276223A22312E302E30222C226D223A307D2A","Cmnd":1,"CmndData":"7B2270223A22617A743464707065666C656967766230222C2276223A22312E302E30222C226D223A307D"}}
13:49:19.297 TYA: MCU Product ID: {"p":"azt4dppefleigvb0","v":"1.0.0","m":0}
13:49:19.305 TYA: Send "55aa0002000001"
13:49:19.313 {"TuyaReceived":{"Data":"55AA0302000004","Cmnd":2}}
13:49:19.315 TYA: RX MCU configuration Mode=0
13:49:19.316 TYA: Read MCU state
13:49:19.324 TYA: Send "55aa0008000007"
13:49:19.338 {"TuyaReceived":{"Data":"55AA03070005010100010011","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
13:49:19.340 TYA: fnId=11 is set for dpId=1
13:49:19.341 TYA: RX Relay-1 --> MCU State: Off Current State:Off
13:49:19.354 {"TuyaReceived":{"Data":"55AA0307000802020004000003E602","Cmnd":7,"CmndData":"02020004000003E6","DpType2Id2":998,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000003E6"}}}
13:49:19.356 TYA: fnId=0 is set for dpId=2
13:49:19.357 TYA: RX value 998 from dpId 2 
13:49:19.370 {"TuyaReceived":{"Data":"55AA03070008690200040000000080","Cmnd":7,"CmndData":"6902000400000000","DpType2Id105":0,"105":{"DpId":105,"DpIdType":2,"DpIdData":"00000000"}}}
13:49:19.372 TYA: fnId=0 is set for dpId=105
13:49:19.374 TYA: RX value 0 from dpId 105 
13:49:22.436 QPC: Reset

sfromis, 2022-05-14:
When you get a device with the affliction of a TuyaMCU, you do indeed not get to have other pins act as they would without the device being managed by the other MCU, with the ESP chip merely being a "wifi modem". What you try to do is not supported by Tasmota, which is no issue, as in Tasmota not doing what it should.

I think that someone had success in a similar situation configuring the intended relay pin as a PWM, and using it with a 100% duty cycle.

Smandurlo, 2022-05-14:
but the gpios are not connected with the Tuya MCU, they are not even wired to the dimmer board. I am doing an external connection.

Smandurlo, 2022-05-14:
> When you get a device with the affliction of a TuyaMCU, you do indeed not get to have other pins act as they would without the device being managed by the other MCU, with the ESP chip merely being a [...]

I tried this option and it actually works, but it doesn't make any sense:
1) the internal dimmer is not woking anymore from the tasmota console;
2) the Tuya MCU doesn't respond anymore to the heartbeat.

In this situation, you can just select "Generic 18" template and setup your gpio as relay. It will work.

I wish to use the Tuya MCU (I need the ON/OFF from the wireless momentary switches linked to the tuya dimmer) and an external relay. If I use the PWM option, I loose the wireless momentary switches triggers.

I really don't get what is the software problem that doesn't make the relay work while Tuya MCU template is selected :-(

sfromis, 2022-05-14:
> but the gpios are not connected with the Tuya MCU

Yes. That's why I made a suggestion for what might work. When configuring a TuayMCU, pin config as a relay will not work. I'd not expect light control with Dimmer to work either. What I meant was as a "basic mode" PWM, using [SetOption15](https://tasmota.github.io/docs/Commands/#setoption15) and the PWM command. And yes, this is no "full" solution, just a possible workaround.

There is no software problem here, it is just that you'd like the software do something it is not designed for.