arendst/Tasmota #15627: I2C does not reckognize SCD30 device after power up

clanganke, 2022-05-14:
### PROBLEM DESCRIPTION
The problem occurs with versions from Tasmota v10.1.0 on, including the current development v11.1.0.3, and did not occurr from v9.2.0 (first version being used here) up to and including v10.0.0:
With an ESP8266 with I2C attached 7segment display (address 0x70) and I2C attached SCD30 sensor (address 0x61), the sensor is not being detected (early enough..?) after a power up, even though the i2cscan command reports both devices right afterwards:
```
  21:25:43.873 MQT: stat/mydevice/RESULT = {"I2CScan":"Device(s) found at 0x61 0x70"}
```
Without a power cut, like with a restart and after e.g. reflashing Tasmota, the sensor is being reckognized on startup

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 mini
- [x] Tasmota binary firmware version number used: 11.1.0.3, 11.1.0, 11.0.0, 10.1.0 have got the error, versions before do not
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: esptool.py of platformio
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
21:40:49.254 CMD: Backlog Template; Module; GPIO 255
21:40:49.335 MQT: stat/mydevice/RESULT = {"NAME":"mydevice","GPIO":[0,0,0,0,640,608,0,0,0,64,0,0,0,0],"FLAG":0,"BASE":18}
21:40:49.555 MQT: stat/mydevice/RESULT = {"Module":{"0":"mydevice"}}
21:40:49.817 MQT: stat/mydevice/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"640":"I2C SDA"},"GPIO5":{"608":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"64":"Button_n1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
  -none-

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
21:51:11.437 MQT: stat/mydevice/STATUS = {"Status":{"Module":0,"DeviceName":"mydevice","FriendlyName":["mydevice"],"Topic":"mydevice","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":8,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
21:51:11.484 MQT: stat/mydevice/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T00:03:06","StartupUTC":"2022-05-14T20:48:05","Sleep":50,"CfgHolder":4617,"BootCount":23,"BCResetTime":"2020-12-29T14:07:50","SaveCount":187,"SaveAddress":"F7000"}}
21:51:11.528 MQT: stat/mydevice/STATUS2 = {"StatusFWR":{"Version":"11.1.0.3(tasmota)","BuildDateTime":"2022.05.14 22:23:53","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"386/699"}}
21:51:11.558 MQT: stat/mydevice/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MySSID",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000800B","2805C80001000600003C5A0A000000000000","00800080","00206000","00044000"]}}
21:51:11.595 MQT: stat/mydevice/STATUS4 = {"StatusMEM":{"ProgramSize":631,"Free":372,"Heap":18,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16400B","FlashFrequency":40,"FlashMode":3,"Features":["00000407","002A87C6","041813A1","000000C2","00400004","00000001","00000028","00000000","00000000"],"Drivers":"1,2,4,5,10,13,20,21,24","Sensors":"2,5,6,31,42,65"}}
21:51:11.638 MQT: stat/mydevice/STATUS5 = {"StatusNET":{"Hostname":"mydevice","IPAddress":"192.168.1.15","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"84:CC:A8:B0:ED:CA","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
21:51:11.672 MQT: stat/mydevice/STATUS6 = {"StatusMQT":{"MqttHost":"ha","MqttPort":8883,"MqttClientMask":"mydevice","MqttClient":"mydevice","MqttUser":"mydeviceuser","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:51:11.702 MQT: stat/mydevice/STATUS7 = {"StatusTIM":{"UTC":"2022-05-14T20:51:11","Local":"2022-05-14T21:51:11","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00"}}
21:51:11.728 MQT: stat/mydevice/STATUS10 = {"StatusSNS":{"Time":"2022-05-14T21:51:11","SCD30":{"CarbonDioxide":841,"eCO2":833,"Temperature":27.6,"Humidity":38.4,"DewPoint":12.2},"TempUnit":"C"}}
21:51:11.755 MQT: stat/mydevice/STATUS11 = {"StatusSTS":{"Time":"2022-05-14T21:51:11","Uptime":"0T00:03:06","UptimeSec":186,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"MySSID","BSSId":"2C:91:AB:96:AA:84","Channel":5,"Mode":"11n","RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:05"}}}
21:51:12.533 WIF: Prüfe Verbindung...

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: aus Flash geladen von F9, Anzahl 185
00:00:00.056 QPC: Count 1
00:00:00.059 ROT: Mode 1
00:00:00.061 I2C: SevenSeg gefunden bei 0x70
00:00:00.065 SRC: Restart
00:00:00.067 Projekt tasmota - mydevice Version 11.1.0.3(tasmota)-2_7_4_9(2022.05.14 22:23:53)
00:00:00.081 SRC: Rule
00:00:00.082 CMD: Grp 0, Cmd 'POWER', Idx 1, Len 1, Pld 1, Data '1'
00:00:00.085 RSL: RESULT = {"POWER":"ON"}
00:00:00.086 RSL: POWER = ON
00:00:00.087 SRC: Rule
00:00:00.088 CMD: Grp 0, Cmd 'DISPLAYMODE', Idx 1, Len 1, Pld 3, Data '3'
00:00:00.093 RSL: RESULT = {"DisplayMode":3}
00:00:00.096 SRC: Rule
00:00:00.097 CMD: Grp 0, Cmd 'DISPLAYDIMMER', Idx 1, Len 4, Pld 6, Data '6.00'
00:00:00.100 RSL: RESULT = {"DisplayDimmer":6}
00:00:00.102 SRC: Rule
00:00:00.103 CMD: Grp 0, Cmd 'DISPLAYBLINKRATE', Idx 1, Len 1, Pld 0, Data '0'
00:00:00.106 RSL: RESULT = {"DisplayBlinkrate":0}
00:00:00.220 WIF: Prüfe Verbindung...
00:00:00.221 WIF: Verbindungsversuch...
00:00:00.558 WIF: verbinden mit AP1 MySSID Channel 5 BSSId 2C:91:AB:96:AA:84 in Modus 11n als mydevice...
00:00:02.484 WIF: Prüfe Verbindung...
00:00:02.485 WIF: verbunden
00:00:02.687 HTP: Web-Server aktiv bei mydevice mit IP-Adresse 192.168.1.15
00:00:02.989 HTP: Konsolen
00:00:03.383 WIF: Sending Gratuitous ARP
00:00:03.385 NTP: Sync time...
00:00:03.413 RTC: UTC 2022-05-14T20:45:48, DST 2022-03-27T02:00:00, STD 2022-10-30T03:00:00
21:54:48.001 RTC: Synced by NTP
21:54:48.271 MQT: Verbindungsversuch...
21:54:50.064 MQT: TLS connected in 1621 ms, max ThunkStack used 4500
21:54:50.066 MQT: Server fingerprint: 69 35 83 D2 76 0F E9 73 22 A0 DD 65 22 AC FE EF E6 23 E1 6E
21:54:50.067 MQT: verbunden
21:54:50.121 MQT: tele/mydevice/LWT = Online (beibehalten)
21:54:50.133 MQT: cmnd/mydevice/POWER = 
21:54:50.135 MQT: abonniere cmnd/mydevice/#
21:54:50.143 MQT: abonniere cmnd/tasmotas/#
21:54:50.150 MQT: abonniere cmnd/mydevice_fb/#
21:54:50.212 MQT: tele/mydevice/INFO1 = {"Info1":{"Module":"mydevice","Version":"11.1.0.3(tasmota)","FallbackTopic":"cmnd/mydevice_fb/","GroupTopic":"cmnd/tasmotas/"}}
21:54:50.231 MQT: tele/mydevice/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"mydevice","IPAddress":"192.168.1.15"}}
21:54:50.247 MQT: tele/mydevice/INFO3 = {"Info3":{"RestartReason":"External System","BootCount":22}}
21:54:50.264 MQT: stat/mydevice/RESULT = {"POWER":"ON"}
21:54:50.274 MQT: stat/mydevice/POWER = ON
21:54:50.286 HTP: Konsole
21:54:52.435 QPC: Reset
21:54:54.381 APP: Anzahl Startvorgänge 22
```

After a restart the SCD30 sensor is being reckognized:
```
...
00:00:00.102 CMD: Grp 0, Cmd 'DISPLAYBLINKRATE', Idx 1, Len 1, Pld 0, Data '0'
00:00:00.105 RSL: RESULT = {"DisplayBlinkrate":0}
00:00:00.124 I2C: SCD30 gefunden bei 0x61
00:00:00.125 SCD: FW v3.66
00:00:00.181 WIF: Prüfe Verbindung...
00:00:00.182 WIF: Verbindungsversuch...
00:00:00.518 WIF: verbinden mit AP1 MySSID Channel 5 BSSId 2C:91:AB:96:AA:84 in Modus 11n als mydevice...
...
```

### TO REPRODUCE
* cut power
* reconnect power
* SCD30 sensor *is not* being reckognized at addres 0x61 on startup

### EXPECTED BEHAVIOUR
* sensor is to be reckongized as with versions up to v10.0.0

### SCREENSHOTS
none

### ADDITIONAL CONTEXT
not req.

sfromis, 2022-05-14:
Duplicate of https://github.com/arendst/Tasmota/issues/15438

clanganke, 2022-05-14:
Hmm searched for I2C there only, I bet ... maybe it is at least of value to have pinpointed the first version with the error.

closed 2022-05-15 by ascillato2

arnowelzel, 2022-08-13:
I still experience this problem with a D1 mini and Tasmota 12.0.1.1