arendst/Tasmota #15636: Rule which uses `publish2` action is being truncated based on (slight) variations or results in a crash on trigger

kquinsland, 2022-05-15:
### PROBLEM DESCRIPTION

A formerly working rule (9.4+ ) results in a crash/reboot when triggered. While drafting this issue, I found some interesting behavior around incomplete rule payloads.

The rule [publishes a supplementary Home Assistant auto configuration payload](https://karlquinsland.com/treatlife-ds03-tasmota-autoconfig-with-homeassistant/).

In all cases, the rule is under 511 chars which - apparently - was an issue (see https://github.com/arendst/Tasmota/issues/8533).

Here is the **original** rule that _was_ working up until very recently. I only noticed the issue because Home Assistant marked the device as "unavailable" as of 3 days ago. My _current_ home assistant instance has been running for a little over 6 days now with 0 restarts.

Thinking that the crash might be caused by some out of memory condition, I stared trying to optimize the mqtt payload and discovered something _interesting_: When i make _very subtle_ changes to the `pct_cmd_tpl` field in the MQTT payload, I observe tasmota either not fully storing the rule or storing the rule completely (but the rule execution still causes a crash).

I will provide my test results at the end.

---

### REQUESTED INFORMATION

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): [TreatLife DS03](https://templates.blakadder.com/treatlife_DS03.html)
- [X] Tasmota binary firmware version number used: Multiple. This issue has been written with the device in question using 11.1-LITE version.
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: authentic FTDI
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
11:53:05.558 CMD: Backlog Template; Module; GPIO 255
11:53:05.577 MQT: stat/bed_room_ceiling_fan/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
11:53:05.808 MQT: stat/bed_room_ceiling_fan/RESULT = {"Module":{"54":"Tuya MCU"}}
11:53:06.063 MQT: stat/bed_room_ceiling_fan/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"2272":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"2304":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
11:53:26.447 CMD: Backlog Rule1; Rule2; Rule3
11:53:26.467 MQT: stat/bed_room_ceiling_fan/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":344,"Free":167,"Rules":"on TuyaReceived#Data=55AA03070005030400010016 do publish2 stat/%topic%/speed 25 endon on TuyaReceived#Data=55AA03070005030400010117 do publish2 stat/%topic%/speed 50 endon on TuyaReceived#Data=55AA03070005030400010218 do publish2 stat/%topic%/speed 75 endon on TuyaReceived#Data=55AA03070005030400010319 do publish2 stat/%topic%/speed 100 endon"}}
11:53:26.697 MQT: stat/bed_room_ceiling_fan/RESULT = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
11:53:26.948 MQT: stat/bed_room_ceiling_fan/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
```
- [X] Provide the output of this command: `Status 0`:
```lua
11:53:47.176 CMD: status0
11:53:47.219 MQT: stat/bed_room_ceiling_fan/STATUS0 = {"Status":{"Module":54,"DeviceName":"Bedroom Ceiling Fan","FriendlyName":["Fan","Light"],"Topic":"bed_room_ceiling_fan","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0},"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Exception","Uptime":"0T00:26:09","StartupUTC":"2022-05-15T18:27:38","Sleep":50,"CfgHolder":4617,"BootCount":5587,"BCResetTime":"2021-02-10T03:39:38","SaveCount":9139,"SaveAddress":"F6000"},"StatusFWR":{"Version":"11.1.0(lite)","BuildDateTime":"2022-04-13T06:40:31","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"482/699"},"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["myWiFiNameHere",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000088","0000E000","00004000"]},"StatusMEM":{"ProgramSize":507,"Free":496,"Heap":27,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F000783","04108001","00000006","010001C0","00000000","00004004","00000000","04000020"],"Drivers":"1,2,3,4,9,10,12,16,20,21,37,62","Sensors":"3"},"StatusNET":{"Hostname":"bed-room-ceiling-fan-2580","IPAddress":"192.168.1.190","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.2","DNSServer2":"0.0.0.0","Mac":"DE;AD:BE:A2:8A:14","Webserver":2,"HTTP_API":1,"WifiConfig":2,"WifiPower":17.0},"StatusMQT":{"MqttHost":"mqtt.internal.lan","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A28A14","MqttUser":"tasmota","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4},"StatusTIM":{"UTC":"2022-05-15T18:53:47","Local":"2022-05-15T11:53:47","StartDST":"2022-04-10T02:00:00","EndDST":"2022-11-05T02:00:00","Timezone":99,"Sunrise":"21:08","Sunset":"12:24"},"StatusSNS":{"Time":"2022-05-15T11:53:47"},"StatusSTS":{"Time":"2022-05-15T11:53:47","Uptime":"0T00:26:09","UptimeSec":1569,"Vcc":3.458,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"myWiFiNameHere","BSSId":"DE:AD:BE:EF:F0:0B","Channel":11,"Mode":"11n","RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:04"}},"StatusSTK":{"Exception":28,"Reason":"Exception","EPC":["4023a0d2","00000000","00000000"],"EXCVADDR":"00000078","DEPC":"00000000","CallChain":["4023b285","4023a85e","4023ac93","4023a85e","4023a794","4023b2c0","40239ffd","40272a65","4023a85e","4023aa48","4023a040","40101062","4023a0c4","4023150a","4026b814","4023185d","4026b814","4023191c","40100da3","40202d43","4010102b","40100da3","4026b814","4021d319","402318b5","4023b195","4023a800","4023a794","4026b9f5","4026b9f5","4026b8f6"]}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
I don't have UART access to the device right now. The device reboots and does not emit much to the console, unfortunately. 

After the device reboots, I can get the RestartReason from the console. It typically looks like this:

11:55:48.473 MQT: tele/bed_room_ceiling_fan/INFO3 = {"Info3":{"RestartReason":{"Exception":28,"Reason":"Exception","EPC":["4023a0d2","00000000","00000000"],"EXCVADDR":"00000078","DEPC":"00000000","CallChain":["4023b285","4023a85e","4023ac93","4023a85e","4023a794","4023b2c0","40239ffd","40272a65","4023a85e","4023aa48","4023a040","40101062","4023a0c4","4023150a","4026b814","4023185d","4026b814","4023191c","40100da3","40202d43","4010102b","40100da3","4026b814","4021d319","402318b5","4026b9f5","4026b9f5","4026b8f6","4026b8f6","4026b814","4022085c"]},"BootCount":5588}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_ Are recorded at the bottom of this post

### EXPECTED BEHAVIOUR

I would expect the JSON document to be published to the MQTT topic.

### SCREENSHOTS

N/A 

### ADDITIONAL CONTEXT

[... 245 lines cut ...]

```
{"uniq_id":"DEADBEEF8A14","~":"bed_room_ceiling_fan/POWER1","cmd_t":"cmnd/~","pl_off":"OFF","pl_on":"ON","stat_t":"stat/~","pct_cmd_t":"cmnd/bed_room_ceiling_fan/tuyasend4","pct_cmd_tpl":"{123f v<=25},0{lse},3{ndif}","pct_stat_t":"stat/bed_room_ceiling_fan/speed","dev":{"cns":[["mac","DEADBEEF8A14"]]}}
```

The value for `pct_cmd_tpl` is mangled :(.

**crash:**

```
[... cut ...]
I have a rule that is meant to configure the fan component in HA for all 4 speeds but this rule seems to be causing a crash when it is executed. It did not used to do this and I am uncertain as to the cause. I suspect memory exhaustion but I don't know how to confirm this.

The second issue:

When attempting to minify the autodiscovery payload by playing with the `pct_cmd_t` field of the JSON document, I found that some variations would cause the rule to be truncated somewhere around 125 characters early!

barbudor, 2022-05-17:
Hi @kquinsland 
Thanks for your extensively detailled report.
I never though I would say such a thing but, may be a little too much as I have difficultis to follow the full explanations.

I've focused on the rule problem. As I don't have HA, I took your 1st rule and modify it slighly like below :
```haskel
Rule2 on event#test do publish2 debug/homeassistant/fan/%macaddr%/config {\"uniq_id\":\"A28A14\",\"~\":\"%topic%/POWER1\",\"cmd_t\":\"cmnd/~\",\"pl_off\":\"OFF\",\"pl_on\":\"ON\",\"stat_t\":\"stat/~\",\"avty_t\":\"tele/%topic%/LWT\",\"pl_avail\":\"Online\",\"pl_not_avail\":\"Offline\",\"pct_cmd_t\":\"cmnd/%topic%/tuyasend4\",\"pct_cmd_tpl\":\"{%set v=value|int%}{%if v<=25%}3,0{%elif v<=50%}3,1{%elif v<=75%}3,2{%elif
Rule2 +v<=100%}3,3{%endif%}\",\"pct_stat_t\":\"stat/%topic%/speed\",\"dev\":{\"cns\":[[\"mac\",\"%macaddr%\"]]}} endon
```
Which provides those logs:
```
21:52:29.766 MQT: stat/nodemcu/RULE = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
21:52:38.048 CMD: Rule2 on event#test do publish2 debug/homeassistant/fan/%macaddr%/config {\"uniq_id\":\"A28A14\",\"~\":\"%topic%/POWER1\",\"cmd_t\":\"cmnd/~\",\"pl_off\":\"OFF\",\"pl_on\":\"ON\",\"stat_t\":\"stat/~\",\"avty_t\":\"tele/%topic%/LWT\",\"pl_avail\":\"Online\",\"pl_not_avail\":\"Offline\",\"pct_cmd_t\":\"cmnd/%topic%/tuyasend4\",\"pct_cmd_tpl\":\"{%set v=value|int%}{%if v<=25%}3,0{%elif v<=50%}3,1{%elif v<=75%}3,2{%elif
21:52:38.098 RUL: Stored uncompressed, would compress from 413 to 286 (-31%)
21:52:38.104 MQT: stat/nodemcu/RULE = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":413,"Free":98,"Rules":"on event#test do publish2 debug/homeassistant/fan/%macaddr%/config {\\\"uniq_id\\\":\\\"A28A14\\\",\\\"~\\\":\\\"%topic%/POWER1\\\",\\\"cmd_t\\\":\\\"cmnd/~\\\",\\\"pl_off\\\":\\\"OFF\\\",\\\"pl_on\\\":\\\"ON\\\",\\\"stat_t\\\":\\\"stat/~\\\",\\\"avty_t\\\":\\\"tele/%topic%/LWT\\\",\\\"pl_avail\\\":\\\"Online\\\",\\\"pl_not_avail\\\":\\\"Offline\\\",\\\"pct_cmd_t\\\":\\\"cmnd/%topic%/tuyasend4\\\",\\\"pct_cmd_tpl\\\":\\\"{%set v=value|int%}{%if v<=25%}3,0{%elif v<=50%}3,1{%elif v<=75%}3,2{%elif"}}
21:52:47.295 CMD: Rule2 +v<=100%}3,3{%endif%}\",\"pct_stat_t\":\"stat/%topic%/speed\",\"dev\":{\"cns\":[[\"mac\",\"%macaddr%\"]]}} endon
21:52:47.329 RUL: Compressed from 525 to 366 (-30%)
21:52:47.332 RUL: Rule2 on event#test do publish2 debug/homeassistant/fan/%macaddr%/config {\"uniq_id\":\"A28A14\",\"~\":\"%topic%/POWER1\",\"cmd_t\":\"cmnd/~\",\"pl_off\":\"OFF\",\"pl_on\":\"ON\",\"stat_t\":\"stat/~\",\"avty_t\":\"tele/%topic%/LWT\",\"pl_avail\":\"Online\",\"pl_not_avail\":\"Offline\",\"pct_cmd_t\":\"cmnd/%topic%/tuyasend4\",\"pct_cmd_tpl\":\"{%set v=value|int%}{%if v<=25%}3,0{%elif v<=50%}3,1{%elif v<=75%}3,2{%elif
21:52:47.363 RUL: Rule2 +v<=100%}3,3{%endif%}\",\"pct_stat_t\":\"stat/%topic%/speed\",\"dev\":{\"cns\":[[\"mac\",\"%macaddr%\"]]}} endon
21:52:47.380 MQT: stat/nodemcu/RULE = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":525,"Free":143,"Rules":"on event#test do publish2 debug/homeassistant/fan/%macaddr%/config {\\\"uniq_id\\\":\\\"A28A14\\\",\\\"~\\\":\\\"%topic%/POWER1\\\",\\\"cmd_t\\\":\\\"cmnd/~\\\",\\\"pl_off\\\":\\\"OFF\\\",\\\"pl_on\\\":\\\"ON\\\",\\\"stat_t\\\":\\\"stat/~\\\",\\\"avty_t\\\":\\\"tele/%topic%/LWT\\\",\\\"pl_avail\\\":\\\"Online\\\",\\\"pl_not_avail\\\":\\\"Offline\\\",\\\"pct_cmd_t\\\":\\\"cmnd/%topic%/tuyasend4\\\",\\\"pct_cmd_tpl\\\":\\\"{%set v=value|int%}{%if v<=25%}3,0{%elif v<=50%}3,1{%elif v<=75%}3,2{%elif v<=100%}3,3{%endif%}\\\",\\\"pct_stat_t\\\":\\\"stat/%topic%/speed\\\",\\\"dev\\\":{\\\"cns\\\":[[\\\"mac\\\",\\\"%macaddr%\\\"]]}} endon"}}
```
As you can see the rule seems intact
Now switching to the serial, I execute the rule by `event test`: I then immediatly get an Exception 28 and the ESP reboots

I will investigate further what could be the cause

kquinsland, 2022-05-17:
>  I never though I would say such a thing but, may be a little too much as I have difficultis to follow the full explanations.

ha! "that's too much detail" might be a a "i never thought i'd hear..." as well :).

> I've focused on the rule problem. As I don't have HA, I took your 1st rule and modify it slighly like below :

That's completely fair. The HA information is something that I mentioned for background and context as to _why_ I am using the rule / trigger in the first place. I probably could have done a better job of minimizing that as the **core** issue here is that the rule does _seem_ to be stored correctly but something about it's execution causes a crash.

Perhaps inflating / hydrating the payload with the true values of `%macaddr%` and `%topic%` uses too much memory?

>  I execute the rule by `event test`

I didn't know this was a thing. Neat. I will for sure keep this in mind for any future dev/test of rules.

> I will investigate further what could be the cause

If you need any additional information from me, I am happy to help.  I don't have a compile environment for Tasmota set up, but I could probably do so if there's some custom code you need me to run _on my devices_.  

It sounds like you were able to hit the same crash and, apparently, [Exception code 28 is memory related](https://github.com/esp8266/Arduino/blob/master/doc/exception_causes.rst)  (`LoadProhibitedCause`).

Several months ago (around the time Tasmota 9.2 was latest stable) [when I was developing this rule, I hit a few "unexplained" crashes while developing the payload ](https://karlquinsland.com/treatlife-ds03-tasmota-autoconfig-with-homeassistant/#side-note-rules-crash) that I eventually worked around by compacting the payload. This is why I _suspect_ that hydrating the payload with the variables exhausts memory but I don't know enough about the code base to trace it down through the provided stack trace(s).

barbudor, 2022-05-18:
I already tried to replace %macaddress% with fixed values. Also splitting the rules in 2 part but that doesn't change a thing
So that seems to suggest a size problem but I haven't had the opportunity to go further due to a very busy week at work. Probaby not before the weekend

closed 2022-06-17 by github-actions[bot]

barbudor, 2022-06-17:
bump
the problem is still there AFAIK and I haven't had time to look further