arendst/Tasmota #15656: Restarts

mantielero, 2022-05-18:
### PROBLEM DESCRIPTION
My ESP32 gets restarted many times a day.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): ESP32-D0WD-V3
- [X] Tasmota binary firmware version number used: 11.1.0.3
  - [ ] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used: platformio
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
22:46:42.271 SRC: WebConsole from 192.168.0.207
22:46:42.276 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 26, Pld -99, Data 'Template; Module; GPIO 255'
22:46:42.307 SRC: Backlog
22:46:42.310 CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 0, Pld -99, Data ''
22:46:42.318 RSL: RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
22:46:42.514 SRC: Backlog
22:46:42.518 CMD: Grp 0, Cmd 'MODULE', Idx 1, Len 0, Pld -99, Data ''
22:46:42.522 RSL: RESULT = {"Module":{"1":"ESP32-DevKit"}}
22:46:42.764 SRC: Backlog
22:46:42.768 CMD: Grp 0, Cmd 'GPIO', Idx 1, Len 3, Pld 255, Data '255'
22:46:42.787 RSL: RESULT = {"GPIO0":{"0":"Ninguno"},"GPIO1":{"0":"Ninguno"},"GPIO2":{"0":"Ninguno"},"GPIO3":{"0":"Ninguno"},"GPIO4":{"0":"Ninguno"},"GPIO5":{"0":"Ninguno"},"GPIO6":{"0":"Ninguno"},"GPIO7":{"0":"Ninguno"},"GPIO8":{"0":"Ninguno"},"GPIO9":{"0":"Ninguno"},"GPIO10":{"0":"Ninguno"},"GPIO11":{"0":"Ninguno"},"GPIO12":{"0":"Ninguno"},"GPIO13":{"0":"Ninguno"},"GPIO14":{"0":"Ninguno"},"GPIO15":{"0":"Ninguno"},"GPIO16":{"3584":"Zigbee Rx"},"GPIO17":{"3552":"Zigbee Tx"},"GPIO18":{"0":"Ninguno"},"GPIO19":{"0":"Ninguno"},"GPIO20":{"0":"Ninguno"},"GPIO21":{"0":"Ninguno"},"GPIO22":{"0":"Ninguno"},"GPIO23":{"0":"Ninguno"},"GPIO24":{"0":"Ninguno"},"GPIO25":{"0":"Ninguno"},"GPIO26":{"0":"Ninguno"},"GPIO27":{"0":"Ninguno"},"GPIO32":{"0":"Ninguno"},"GPIO33":{"0":"Ninguno"},"GPIO34":{"0":"Ninguno"},"GPIO35":{"0":"Ninguno"},"GPIO36":{"0":"Ninguno"},"GPIO37":{"0":"Ninguno"},"GPIO38":{"0":"Ninguno"},"GPIO39":{"0":"Ninguno"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
22:48:15.275 SRC: WebConsole from 192.168.0.207
22:48:15.279 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 19, Pld -99, Data 'Rule1; Rule2; Rule3'
22:48:15.308 SRC: Backlog
22:48:15.311 CMD: Grp 0, Cmd 'RULE', Idx 1, Len 0, Pld -99, Data ''
22:48:15.317 RUL: Rule1 ON ZbReceived#Interior#Temperature DO Var1 %value% ENDON      ON ZbReceived#Interior#Humidity DO Var2 %value% ENDON      ON ZbReceived#Interior#Pressure DO Var3 %value% ENDON        ON ZbReceived#Exterior#Temperature DO Var4 %value% ENDON      ON ZbReceived#Exterior#Humidity DO Var5 %value% ENDON       ON ZbReceived#Exterior#Pressure DO Var6 %value% ENDON   ON ZbReceived#Puerta#Contact DO Var7 %value% ENDON   ON ZbReceived#Puerta DO WebQuery
22:48:15.361 RUL: Rule1 +https://script.google.com/macros/s/XXXXXXXXXXXXX/exec?s=Puerta&open=%var7% GET ENDON     ON ZBReceived#Interior DO WebQuery https://script.google.com/macros/s/XXXXXXXXXXXXX/exec?s=Tint&t=%var1%&h=%var2%&p=%var3% GET ENDON    ON ZBReceived#Exterior DO WebQuery
22:48:15.397 RUL: Rule1 +https://script.google.com/macros/s/XXXXXXXXXXXXX/exec?s=Text&t=%var4%&h=%var5%&p=%var6% GET ENDON     ON Time#Initialized DO WebQuery https://script.google.com/macros/s/XXXXXXXXXXXXX/exec?s=logger&val=NTP GET ENDON
22:48:15.428 RSL: RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":1164,"Free":120,"Rules":"ON ZbReceived#Interior#Temperature DO Var1 %value% ENDON      ON ZbReceived#Interior#Humidity DO Var2 %value% ENDON      ON ZbReceived#Interior#Pressure DO Var3 %value% ENDON        ON ZbReceived#Exterior#Temperature DO Var4 %value% ENDON      ON ZbReceived#Exterior#Humidity DO Var5 %value% ENDON       ON ZbReceived#Exterior#Pressure DO Var6 %value% ENDON   ON ZbReceived#Puerta#Contact DO Var7 %value% ENDON   ON ZbReceived#Puerta DO WebQuery https://script.google.com/macros/s/XXXXXXXXXXXXX/exec?s=Puerta&open=%var7% GET ENDON     ON ZBReceived#Interior DO WebQuery https://script.google.com/macros/s/XXXXXXXXXXXXX/exec?s=Tint&t=%var1%&h=%var2%&p=%var3% GET ENDON    ON ZBReceived#Exterior DO WebQuery https://script.google.com/macros/s/XXXXXXXXXXXXX/exec?s=Text&t=%var4%&h=%var5%&p=%var6% GET ENDON     ON Time#Initialized DO WebQuery https://script.google.com/macros/s/XXXXXXXXXXXXX/exec?s=logger&val=NTP GET ENDON"}}
22:48:15.607 SRC: Backlog
22:48:15.612 CMD: Grp 0, Cmd 'RULE', Idx 2, Len 0, Pld -99, Data ''
22:48:15.619 RSL: RESULT = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":169,"Free":342,"Rules":"ON System#Boot DO WebQuery https://script.google.com/macros/s/XXXXXXXXXXXXX/exec?s=logger&val=BOOT GET ENDON"}}
22:48:15.834 SRC: Backlog
22:48:15.838 CMD: Grp 0, Cmd 'RULE', Idx 3, Len 0, Pld -99, Data ''
22:48:15.845 RSL: RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
22:53:25.878 SRC: WebConsole from 192.168.0.207
22:53:25.882 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
22:53:25.889 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_D6D9DC","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
22:53:25.964 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32solo1.bin","RestartReason":"Software reset CPU","Uptime":"0T00:05:51","StartupUTC":"2022-05-18T20:47:34","Sleep":50,"CfgHolder":4617,"BootCount":483,"BCResetTime":"2022-04-24T17:13:40","SaveCount":527}}
22:53:26.036 RSL: STATUS2 = {"StatusFWR":{"Version":"11.1.0.3(tasmota)","BuildDateTime":"2022-05-18T21:26:02","Core":"2_0_3","SDK":"v4.4.1","CpuFrequency":80,"Hardware":"ESP32-D0WD-V3 rev.3","CR":"570/699"}}
22:53:26.080 RSL: STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["inalambrika",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C80001000600003C5A0A190000000000","00000080","00006002","00004000"]}}
22:53:26.152 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1405,"Free":1474,"Heap":151,"StackLowMark":0,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164054","FlashFrequency":40,"FlashMode":3,"Features":["0000040A","9F8A07CE","0051A001","B7F7BFCF","05DA9B00","20360C47","400840D2","20200000","0434480D"],"Drivers":"1,2,3,4,5,8,9,10,11,14,16,17,20,21,23,24,29,34,38,50,52,59,62,82,86","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,127"}}
22:53:26.247 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-D6D9DC-6620","IPAddress":"192.168.0.22","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.0.1","DNSServer2":"0.0.0.0","Mac":"E8:31:CD:D6:D9:DC","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"0.0.0.0","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00"},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
22:53:26.325 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-05-18T20:53:26","Local":"2022-05-18T22:53:26","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+02:00","Sunrise":"06:04","Sunset":"21:28"}}
22:53:26.376 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-05-18T22:53:26","ESP32":{"Temperature":53.3},"TempUnit":"C"}}
[... cut ...]
### TO REPRODUCE
_Steps to reproduce the behavior:_
Not easy. I seen happening when I run `WebQuery https://xxxx`, but in above's log, that is not happening.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Not crashing.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
N/A

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
N/A

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2022-06-21:
Do not run a solo1 image on a dual core processor.

Change your ``otaurl``.

Jason2866, 2022-06-21:
Good catch! Zigbee and webquery is probably to much to handle for one core

mantielero, 2022-06-21:
I don't know how to do that. Could you point me where should I configure to use both cores?

Jason2866, 2022-06-21:
Change the OtaUrl to by doing this command `OtaUrl https://github.com/tasmota/install/raw/main/firmware/unofficial/tasmota32-zigbee.bin` and than `Upgrade 1`

mantielero, 2022-06-21:
Just out of curiosity, why do I have to update the OTA URL? I compiled a custom image myself with customized settings.

I thought that I would have to customize some particular setting in the way I compiled the image.

Jason2866, 2022-06-21:
So you configured something wrong in your custom setup. The OTA Url i linked is a verified working build for ESP32 with ZNP Zigbee stack.
We can not provide support or solve issues for a custom compiled firmware.

mantielero, 2022-06-23:
Flashing that image:
```
$ esptool.py --port /dev/serial/by-id/usb-1a86_USB_Single_Serial_539E009512-if00 write_flash -fm dout 0x0 tasmota32-zigbee.bin
```

produces all the time:
```
...
invalid header: 0x07000810␍␊
invalid header: 0x07000810␍␊
ets Jul 29 2019 12:21:46␍␊
␍␊
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)␍␊
invalid header: 0x07000810␍␊
invalid header: 0x07000810␍␊
invalid header: 0x07000810␍␊
invalid header: 0x07000810␍␊
invalid header: 0x07000810␍␊
invalid header: 0x07000810␍␊
invalid header: 0x07000810␍␊
ets Jul 29 2019 12:21:46␍␊
␍␊
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)␍␊
invalid header: 0x07000810␍␊
invalid header: 0x07000810␍␊
invalid header: 0x07000810␍␊
invalid header: 0x07000810␍␊
invalid header: 0x07000810␍␊
invalid header: 0x07000810␍␊
invalid header: 0x07000810␍␊
ets Jul 29 2019 12:21:46␍␊
␍␊
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)␍␊
invalid header: 0x07000810␍␊
invalid header: 0x07000810␍␊
invalid header: 0x07000810␍␊
invalid header: 0x07000810␍␊
invalid header: 0x07000810␍␊
invalid header: 0x07000810␍␊
invalid header: 0x07000810␍␊
ets Jul 29 2019 12:21:46␍␊
␍␊
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)␍␊
invalid header: 0x0700081
...
```

sfromis, 2022-06-23:
No, the URL suggested was meant for **OTA**. For wired flashing the way you try to do it, you need to use the `.factory` image (if you do not prefer a more complex flashing process with 4 files at different addresses).
`tasmota32-zigbee.factory.bin`

Jason2866, 2022-06-23:
Build config is here https://github.com/Jason2866/Tasmota-build/blob/special_builder/tasmota/user_config_override.h

I have no restarts with this build. Can you swap the Esp32 board and check your power supply.

mantielero, 2022-06-23:
I don't have a spare ESP32 (I will buy a new one). Checking now with a different power supply.

On the other hand, what does this mean?
```json
"Reason":"InstructionFetchError",
"EPC":"00000001",
"EXCVADDR":"ffffffe0",
"CallChain":["fffffffe"]
```

mantielero, 2022-06-23:
With another power supply the issue continues. As before, it is random, but tends to occur after using webquery. I use it connected to three sensors: two for temperature and one for door open/close sensor.

For instance, I got the "door opened" notification at 12:33pm in my mobile (so the webquery got executed) an inmediately after the board got restarted (I got the notification for that in my mobile too).

After that restart I think that the zigbee CC-2530 stops working:
```
12:33:18.065 RSL: RESULT = {"WebQuery":"Done"}
12:33:19.119 QPC: Reset
12:33:20.805 ZIG: {"ZbZNPReceived":"44810000060060080101008B005FAC5700000718120A0000100060081D"}
12:33:21.132 APP: Boot Count 523
12:33:21.143 RSL: STATE = {"Time":"2022-06-23T12:33:21","Uptime":"0T00:00:15","UptimeSec":15,"Heap":164,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":51,"MqttCount":0,"Berry":{"HeapUsed":2,"Objects":28},"Wifi":{"AP":1,"SSId":"inalambrika","BSSId":"08:62:66:3B:21:B8","Channel":9,"Mode":"11n","RSSI":100,"Signal":-18,"LinkCount":1,"Downtime":"0T00:00:04"}}
12:33:21.201 RSL: SENSOR = {"Time":"2022-06-23T12:33:21","ESP32":{"Temperature":43.3},"TempUnit":"C"}
12:33:21.722 CFG: Saved, Count 575, Bytes 4096
12:33:23.059 ZIG: rebooting ZNP device
12:33:23.225 ZIG: ZbZNPSent 0xFF x 255
12:33:23.227 ZIG: ZbZNPSent 410000
12:33:28.210 ZIG: timeout, goto label 99
12:33:28.214 RSL: RESULT = {"ZbState":{"Status":99,"Message":"Abort"}}
12:33:28.242 ZIG: Abort
12:33:28.243 ZIG: Stopping (99)
```

If I go to `Zigbee Map` I get `Zigbee not started`

barbudor, 2022-06-23:
Is it stable without WebQuery?

mantielero, 2022-06-25:
Yes it is. It has been running for about 4 hours without failure (normally it fails within an hour). Zigbee is working fine too. 

Something that puzzles me is that the issue appears sometimes after calling WebQuery. But the WebQuery command itself seems to work fine (given that I get the notification in my Google Sheet).

@[arendst](https://github.com/arendst), just out of curiosity. What in the information provided makes you think I was running a `solo1`image? I checked how I compiled my custom image before and I run:
```
$ pio run -e tasmota32
```
not `pio run -e tasmota32solo1`.

arendst, 2022-06-25:
As noted in the comment your OtaUrl is set to "OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32solo1.bin"

Jason2866, 2022-06-26:
Probably, you "flashed over" the build. The settings from previous version are left intact.
It is general a good idea to erase whole flash when flashing a different build and avoiding unwanted side effects from previous versions. This can be the problem you encounter!

mantielero, 2022-06-26:
How can I do that?

mantielero, 2022-06-26:
On the other hand, I am testing with Berry rather than WebQuery. It looks like it is behaving much more stable with what I am trying. No crash so far.

Jason2866, 2022-06-26:
See https://tasmota.github.io/docs/Esptool/

Jason2866, 2022-06-26:
Avoiding WebQuery is always a good idea!

sfromis, 2022-06-26:
And using Ping is better than WebQuery too. Part of the story here is that http is not well-supported by Tasmota, it is only at a rudimentary level.

mantielero, 2022-06-26:
I have replaced Rules+WebQuery with Berry language. So far it looks much better (I am really amazed). I will come back to you and let you know if the issues have gone finally.

closed 2022-06-28 by Jason2866

klsc18, 2023-01-15:
> I have replaced Rules+WebQuery with Berry language. So far it looks much better (I am really amazed). I will come back to you and let you know if the issues have gone finally.
@mantielero 
Hello mantielero, I have the same problems as you had at the beginning. Can you post your berry scripts for https so others like me can see working examples? Would be great.
thanks

mantielero, 2023-08-01:
I am sorry I missed this post.

What I do in order to read from Google Sheet, looks like:
```berry
cl = webclient()
cl.begin("https://sheets.googleapis.com/v4/spreadsheets/xxxxx")
cl.add_header("Authorization","Bearer xxxxxx")
#cl.add_header("Content-Type","application/json")
cl.GET()
print(cl.get_string())
```
and in order to POST:
```berry
cl = webclient()
cl.begin("https://sheets.googleapis.com/v4/spreadsheets/xxxxxx")
cl.add_header("Authorization","Bearer xxxxxxxx)
cl.add_header("Content-Type","application/json")
cl.POST("{\"values\": [[12345,6789]]}")
```